evcc-io/evcc #15847: Sonnenanteil sinkt stetig

phL9r, 2024-09-01:
### Describe the bug

Hallo!

Ich betreibe eine PV Anlage. Hinter dieser hängen noch Balkonkraftwerke. Derzeit ist es so, dass der Sonnenanteil beim Laden meines E-Fahrzeuges stetig sinkt, obwohl kein Netzbezug stattfindet. Ich konnte es mittlerweile darauf eingrenzen, dass z. B. zum jetzigen Zeitpunkt etwa 1 kW neben der Haupt-PV produziert wird. Dieses erkennt evcc richtigerweise und nimmt diesen Zusatzüberschuss auch zum Laden, betrachtet diesen jedoch nicht als Sonnenanteil, sondern als Netzbezug. Ich habe zur Verdeutlichung noch Screenshots angehängt. Ein Netzbezug in der Übersicht wird nicht angezeigt, lediglich am Ladepunkt wird Co2-Verbrauch ermittelt und der Sonnenanteil sinkt stetig.

### Steps to reproduce

1. PV Anlage
2. Balkonkraftwerk
3. Ladung eines E-Fahrzeuges

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: §§§

interval: 30s # control cycle interval

sponsortoken: §§§§

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: 192.168.0.21
  port: 502
  usage: grid
  modbus: tcpip
  timeout: 15s
  name: grid1
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: 192.168.0.21
  port: 502
  usage: pv
  modbus: tcpip
  timeout: 15s
  name: pv2
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: 192.168.0.21
  port: 502
  usage: battery
  modbus: tcpip
  timeout: 15s
  name: battery3

chargers:
- type: template
  template: easee
  user: §§§§§
  password: §§§§§
[... 207 lines cut ...]
Sep 01 13:36:56 pi evcc[2950646]: [lp-1  ] DEBUG 2024/09/01 13:36:56 vehicle range: 226km
Sep 01 13:36:56 pi evcc[2950646]: [lp-1  ] DEBUG 2024/09/01 13:36:56 pv charge current: 9.27A = 9A + 0.271A (-187W @ 3p)
Sep 01 13:37:11 pi evcc[2950646]: [site  ] DEBUG 2024/09/01 13:37:11 ----
Sep 01 13:37:11 pi evcc[2950646]: [lp-1  ] DEBUG 2024/09/01 13:37:11 charge power: 6336W
Sep 01 13:37:11 pi evcc[2950646]: [lp-1  ] DEBUG 2024/09/01 13:37:11 charge currents: [8.88 9.07 8.79]A
Sep 01 13:37:11 pi evcc[2950646]: [site  ] DEBUG 2024/09/01 13:37:11 pv power: 5863W
Sep 01 13:37:11 pi evcc[2950646]: [site  ] DEBUG 2024/09/01 13:37:11 battery soc: 100%
Sep 01 13:37:11 pi evcc[2950646]: [site  ] DEBUG 2024/09/01 13:37:11 battery power: -4W
Sep 01 13:37:11 pi evcc[2950646]: [site  ] DEBUG 2024/09/01 13:37:11 grid meter: -457W
Sep 01 13:37:11 pi evcc[2950646]: [site  ] DEBUG 2024/09/01 13:37:11 grid currents: [1.76 -2.25 -1.96]A
Sep 01 13:37:11 pi evcc[2950646]: [site  ] DEBUG 2024/09/01 13:37:11 site power: -161W
Sep 01 13:37:11 pi evcc[2950646]: [lp-1  ] DEBUG 2024/09/01 13:37:11 detected active phases: 3p
Sep 01 13:37:11 pi evcc[2950646]: [lp-1  ] DEBUG 2024/09/01 13:37:11 charge total import: 924.858kWh
Sep 01 13:37:11 pi evcc[2950646]: [lp-1  ] DEBUG 2024/09/01 13:37:11 charger status: C
Sep 01 13:37:11 pi evcc[2950646]: [lp-1  ] DEBUG 2024/09/01 13:37:11 soc estimated: 56.00% (vehicle: 56.00%)
Sep 01 13:37:11 pi evcc[2950646]: [lp-1  ] DEBUG 2024/09/01 13:37:11 vehicle soc: 56%
```

### What type of operating system are you running?

Linux

### Version

0.130.6

MichaelSteg1986, 2024-09-02:
Hallo,
bei mir war's ähnlich. Mein Sohn hat bei den Wechselrichtern das Netzwerk ausgesteckt, Netzzähler noch erreichbar. Sonnenanteil wurde dann beim Laden immer weniger, obwohl eingespeist würde.
Bei mir läuft die V130.3

naltatis, 2024-09-02:
@phL9r Wenn ich das richtig verstehe hast du noch einen zweiten Energieproduzenten (die Balkon-PV) von der evcc aber nichts weiß, richtig? Man sieht ja im Screenshot und auch im Log, dass die Energiebilanz nicht aufgeht. Es wird weniger produziert als Verbrauch+Einspeisung. Wenn die Statistiken korrekt sein sollen, braucht evcc auch die Daten der Balkon-PV. Ich vermute, deine Anlage hat da keine Schnittstelle für, sonst hättest du das schon eingebunden, richtig?

Wenn du gar keine PV konfigurierst, geht evcc in einen Modus, die Daten allein aufgrund des Netzzählers hergeleitet werden. In diesem Fall sollte der %-Sonneanteil korrekt ermittelt werden. Dieser Mischzustand (evcc kennt nur Teile der Produktion) ist in der Tat etwas tricky. Ich schau mal, ob wir da eine non-hacky Lösung in der Berechnung für finde.

phL9r, 2024-09-02:
@naltatis

Ja so in etwa ist es korrekt.
Die Balkon PV hängt quasi einfach mit im Hauskreis und drückt teils den Hausverbrauch auf 0 bzw. darunter und speist auch mit ein. Der Huawei Power Sensor erkennt diesen Zusatz an verfügbarer Energie auch richtig, zählt diesen jedoch nicht als Erzeugung aus der Haupt PV mit hinzu. evcc erkennt, dass mehr Energie bereitsteht als die PV selbst produziert, wie man auf dem Screenshot sehen kann. Insofern ist genügend Energie vorhanden für die Ladung. evcc benutzt diese auch. Ich kann aber während der Ladung erkennen, dass der Sonnenanteil stetig sinkt, obwohl kein Netzbezug stattfindet. In evcc selbst sieht man auch, dass der aus der Balkon PV stammende Ladeanteil mit x Gramm co2 bemessen wird. Insofern geht evcc scheinbar irgendwie trotzdem von Netzbezug aus. Ich hoffe ich konnte das verständlich erklären!

premultiply, 2024-09-02:
Der "richtige" Ansatz wäre hier einfach die Messdaten der Kleinanlage auch mit zu erfassen. Notfalls kleiner SDM120-Zwischenzähler oder Shelly & Co.
Dann stimmen auch alle Daten.

phL9r, 2024-09-02:
Ich habe die Balkon PV an einem Fritzdect angeschlossen aber so wie ich das lese kann man diese nur als schaltbaren 'Charger' anlegen, nicht aber für die Leistungsmessung, richtig?
Problem ist auch, dass wir 2 Wohneinheiten haben, an jeder hängt jeweils eine Balkon-PV, da wir diese bereits vor der großen Anlage in Betrieb hatten. Beide hängen jeweils an einer schaltbaren DECT-Steckdose von AVM.

mdkeil, 2024-09-02:
> Ich habe die Balkon PV an einem Fritzdect angeschlossen aber so wie ich das lese kann man diese nur als schaltbaren 'Charger' anlegen, nicht aber für die Leistungsmessung, richtig?

geht auch als reine Messung..
https://docs.evcc.io/docs/devices/meters#avm-fritzdect

phL9r, 2024-09-02:
Dank des Tipps von mdkeil habe ich jetzt ein "workaround" zustande bekommen. Beide FritzDect sind jetzt als PV mit eingebunden. Jetzt wird zwar nachts der selbst gebaute Speicher als Solarleistung angezeigt, da er auch über das FritzDect mit einspeist (hatte ich vorher nicht erwähnt), aber ich warte mal den morgigen Tag ab, was er daraus macht. Sieht aber so schon ganz richtig aus! 

[image]

naltatis, 2024-09-02:
@phL9r ok das klingt doch gut.

Ich hab mir die Sonnen-Anteil-Berechnung noch mal angeschaut. Das funktioniert, wie erwartet. Grundproblem ist hier, wie vorher schon beschrieben, dass der Ladepunkt mehr Leistung nutzt, als das System selbstproduzierte Energie zur Verfügung hat (bzw. kennt). Hat man keine PV konfiguriert gibt es einen Fallback und wir treffen eine Produktionsannahme rein auf Basis des Netzzählers. Eine Mischform zu implementieren, die unplausible Summen noch mal in die eine oder andere Richtung anpasst ist nicht trivial. Daher mach ich hier mal zu.

closed 2024-09-02 by naltatis

jmtatsch, 2024-09-08:
Ich hatte genau das gleiche Problem. Seitdem das Balkonkraftwerk sauber gemessen wird funktioniert das Überschussladen wenigstens im mehrphasigen Betrieb sauber.

@naltatis Warum regelt ihr nicht einfach nur auf den Netzzähler? Der ist ja wohl der relevanteste Messwert oder gibt es gute Gründe die dagegen sprechen?

naltatis, 2024-09-08:
> Warum regelt ihr nicht einfach nur auf den Netzzähler? 

Genau das machen wir für die Regelung.

Für die Preisberechnung reicht das aber nicht aus. Hier spielt Produktion, Speicherleistung und Hausverbrauch auch eine Rolle.

jmtatsch, 2024-09-08:
Für die Preisberechnung verstehe ich das.

Wenn wirklich nur auf den Netzzähler geregelt wird verstehe ich nicht warum ich bislang immer Probleme (auch mit 3 phasigem) Überschussladen laden hatte siehe https://github.com/evcc-io/evcc/discussions/15329

Seit dem das Balkonkraftwerk gezählt ist funktionierte es gestern erstmals sauber 3 phasig.

[image]

Mit 1-phasig hatte ich heute wieder Netzbezug trotz korrekter Zählung aber das Thema scheint ja bekannt https://github.com/evcc-io/evcc/issues/15252#issuecomment-2336731396