evcc-io/evcc #15850: Evbox Elvi: set meterValues to empty without meter

premultiply, 2024-09-01:
chargers:
- type: template
  template: elvi 
  timeout: 2m  
  meter: false  
  name: wallbox5

_Originally posted by @WilMa5301 in https://github.com/evcc-io/evcc/discussions/14138#discussioncomment-10460498_

andig, 2024-09-01:
Verstehe ich nicht. Das elvi Template in master hat weder das eine noch das andere Setting. Was ist hier das Problem? https://github.com/evcc-io/evcc/pull/15838 Wurde nicht gemerged.

premultiply, 2024-09-01:
Ist nur ein Merker für mich.

Die Box gibt offensichtlich auch ohne verbautes Meter in der Autodetection der Measurands vor diese liefern zu können.
Dabei kommt dann aber am Ende nur fest Timeout oder 0 raus und die Schätzung von evcc wird nicht genutzt.

Hier brauchen wir also noch eine Lösung um bewusst keine Autodetection durchzuführen aber auch gar keine Measurands zu abonnieren.

andig, 2024-09-01:
Das wäre dann wieder das alte getconfiguration: false :(

andig, 2024-09-01:
Ist auf jeden Fall kein Bug 😀

premultiply, 2024-09-01:
> Das wäre dann wieder das alte getconfiguration: false :(

Das war was anders.

> Oder alternativ metervalues auch leer setzbar machen.

Sowas in der Richtung.

andig, 2024-09-07:
Gelöst:

    metervalues: Current.Offered

closed 2024-09-07 by andig