evcc-io/evcc #15851: Make battery-supported charging current configurable

cmock, 2024-09-01:
**Is your feature request related to a problem? Please describe.**
When using battery-supported charging, currently a "Min + PV" approach is taken.
This leads to very slow charging, which with some cars is inefficient. 
It also only delivers about 16kWh overnight (12 hours), some users might have the battery capacity, inverter sizing and need for way more charge to be delivered overnight.

**Describe the solution you'd like**
I would like to be able to configure the minimum current and number of phases for this feature.

**Describe alternatives you've considered**
I can't think of any good alternatives.

**Additional context**
The optimal current for this feature will depend on the specifics of the installation plus the use case of the user, so I think configurability is the best solution.

kscholty, 2024-09-05:
What I do in that case is to set the values at the Loadpoint.
You can configure number of phases, min current and max current using the GUI and then use "Fast" charging.
Sure this will not allow to use additional current from PV, but since I use it mostly at night that's not a problem.

andig, 2024-09-06:
The general issue with batteries it their non-linear behaviour. I think we need a general overhaul of how to best utilise and customise battery usage.

cmock, 2024-09-06:
@kscholty yes, I could do that, but that wouldn't be automatic, especially not when it comes to ending the battery discharge. It also requires changing (and afterwards changing back) multiple settings.

@andig what dou you mean by "non-linear"?

closed 2024-09-18 by github-actions[bot]