evcc-io/evcc #15856: OCPP doesn't start: cannot create charger type 'ocpp': ChangeConfiguration failed: NotSupported

andifanny, 2024-09-02:
### Describe the bug

After update to version 0.130.6 nightly evcc is not starting anymore due to FATAL error as shown in log.
After downgrade to 0.130.3 evcc starts normal

### Steps to reproduce

1.     update to 0.130.6 nightly
2.     start evcc
...

### Configuration details

```yaml
loadpoints:
  - title: Garage # display name for UI
    charger: AXXX # charger
    mode: pv 
    vehicle: ev4 

chargers:
  - name: AXXX
    type: template
    template: ocpp
    timeout: 2m # optional
```

### Log details

```text
[main ] INFO 2024/09/02 08:15:00 evcc 0.130.6 (095fa42b)
[main ] INFO 2024/09/02 08:15:01 using config file: /evcccfg/evcc.yaml
[db ] INFO 2024/09/02 08:15:01 using sqlite database: /root/.evcc/evcc.db
[main ] INFO 2024/09/02 08:15:01 listening at :7070
[ocpp-1] DEBUG 2024/09/02 08:15:01 waiting for chargepoint: 5m0s
[ocpp ] INFO 2024/09/02 08:15:06 charge point connected, registering: AXXX
[ocpp-1] DEBUG 2024/09/02 08:15:07 failed configuring availability: Rejected
[main ] FATAL 2024/09/02 08:15:37 cannot create charger 'AXXX': cannot create charger type 'template': cannot create charger type 'ocpp': ChangeConfiguration failed: NotSupported
[main ] FATAL 2024/09/02 08:15:37 will attempt restart in: 15m0s
```

### What type of operating system are you running?

Docker container

### Version

0.130.6 nightly

andig, 2024-09-02:
Was für ein Gerät ist das denn?

andig, 2024-09-02:
@premultiply hier braucht es einen Fallbackmechanismus für Availability/erste StatusNotification?

premultiply, 2024-09-02:
Fixed by #15856

andifanny, 2024-09-02:
> Was für ein Gerät ist das denn?

Sungrow AC011E-01

andig, 2024-09-02:
@premultiply hier fehlt der Fix?

premultiply, 2024-09-02:
Fixed by f5547ba0e79ff875ebae607542fd2583a11492f7

closed 2024-09-02 by premultiply