evcc-io/evcc #15864: OCCP: Wallbox Pulsar MAX in 0.130.x - charger disable: set charging profile: Rejected

Delorien84, 2024-09-02:
### Describe the bug

Hello, I have Wallbox pulsar max that works without any issue in version 0.129.0. Currently I tested version 0.130.7, and it report error: `charger disable: set charging profile: Rejected`.

When I leave previous configuration (without `remotestart: **true**`), the charger wont start charging, and complains about 
`set charging profile: Rejected`. The icon, shows, that I need to authenticate the session.

When set `remotestart: **true**`, the charger still reports `charging profile: Rejected` and start charging, even I have setup limit to 80% and car is already charged to that limit. Clicking on Turn off charging has no effect and car is still charging.

First log is when `remotestart: false`. Second is with enabled `remotestart`.

### Steps to reproduce

Upgrade to 0.130.7 and connect cable to the car.

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 15s # control cycle interval

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  cache: error
  db: error
  ocpp: trace

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
  - name: charger
    type: template
    template: shelly-pro-3em
    usage: charge
    host: 192.168.1.164 # IP-Adresse oder Hostname
  - name: grid
    type: custom
    power: # power (W)
      source: mqtt
      topic: solar/grid/power
    energy: # optional energy (kWh)
      source: mqtt
      topic: solar/grid/energy
    currents: # optional currents (A)
      - source: mqtt
        topic: solar/grid/current/l1
      - source: mqtt
        topic: solar/grid/current/l2
      - source: mqtt
        topic: solar/grid/current/l3
  - name: pv
    type: custom 
    power:
      source: mqtt
      topic: solar/pv/power
    energy:
      source: mqtt 
      topic: solar/pv/energy 
  - name: battery
    type: custom
    capacity: 11.1
    power:
      source: mqtt
      topic: solar/battery/power
    soc:
      source: mqtt
      topic: solar/battery/soc

# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for documentation see https://docs.evcc.io/docs/devices/chargers
chargers:
  - type: template
    template: pulsarplus
    stationid: wallbox
    name: wallbox
    #remotestart: true

# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for documentation see https://docs.evcc.io/docs/devices/vehicles
vehicles:
  - name: car1
    type: renault
    title: Zoe
    capacity: 52 # kWh
    user: xxx # user
    password: xxx # password
    vin: xxx 
    onIdentify: # set defaults when vehicle is identified
      mode: pv # enable PV-charging when vehicle is identified
      #minSoc: 20 # immediately charge to 0% regardless of mode unless "off" (disabled)
      #targetSoc: 80 # limit charge to 100%
      mincurrent: 9

# site describes the EVU connection, PV and home battery
site:
  title: Home # display name for UI
  meters:
    grid: grid # grid meter
    pv:
    - pv # list of pv inverters/ meters
    battery:
    - battery # list of battery meters
  residualPower: 100 # additional household usage margin
  #prioritySoc: 60 # give home battery priority up to this soc (empty to disable)
  #bufferSoc: 80 # continue charging on battery above soc (0 to disable)
  #bufferStartSoc: 95 # start charging on battery above soc (0 to disable)
  maxGridSupplyWhileBatteryCharging: 0 # ignore battery charging if AC consumption is above this value
  #smartCostLimit: 0 # set cost limit for automatic charging in PV mode

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
  - title: Garage # display name for UI
    charger: wallbox # charger
    meter: charger # charge meter
    mode: "pv" # set default charge mode, use "off" to disable by default if charger is publicly available
    phases: 3 # electrical connection (normal charger: default 3 for 3 phase, 1p3p charger: 0 for "auto" or 1/3 for fixed phases)
    minCurrent: 6 # minimum charge current (default 6A)
    maxCurrent: 16 # maximum charge current (default 16A)

    # remaining settings are experts-only and best left at default values
    priority: 0 # relative priority for concurrent charging in PV mode with multiple loadpoints (higher values have higher priority)
    soc:
      # polling defines usage of the vehicle APIs
      # Modifying the default settings it NOT recommended. It MAY deplete your vehicle's battery
      # or lead to vehicle manufacturer banning you from API use. USE AT YOUR OWN RISK.
      poll:
        # poll mode defines under which condition the vehicle API is called:
        #   charging: update vehicle ONLY when charging (this is the recommended default)
        #   connected: update vehicle when connected (not only charging), interval defines how often
        #   always: always update vehicle regardless of connection state, interval defines how often (only supported for single vehicle)
        mode: charging
        # poll interval defines how often the vehicle API may be polled if NOT charging
        interval: 60m
      estimate: true # set false to disable interpolating between api updates (not recommended)
    enable: # pv mode enable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 3m # threshold must be exceeded for this long
      threshold: 0 # maximum import power (W)
    guardDuration: 5m # switch charger contactor not more often than this (default 5m)

# tariffs are the fixed or variable tariffs
tariffs:
  currency: CZK # three letter ISO-4217 currency code (default EUR)
  grid:
    # either static grid price (or price zones)
    type: fixed
    price: 3.498462 # CZK/kWh
    zones:
      - days: Mo-So
        hours: 19-21
        price: 1.42079 # CZK/kWh
      - days: Mo-So
        hours: 0-6
        price: 1.42079 # EUR/kWh
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.0 # EUR/kWh
  co2:
    type: 
    uri: https://api.electricitymap.org/v3
    token: xxxxx
    zone: CZ

# mqtt message broker
mqtt:
  broker: localhost:1883
  topic: evcc # root topic for publishing, set empty to disable
  user: xxx
  password: xxx
```

### Log details

```text
See attached files
```

### What type of operating system are you running?

Linux

### Version

0.300.7

Delorien84, 2024-09-02:
[evcc-20240902-155524-trace-remotestart_false.log](https://github.com/user-attachments/files/16838762/evcc-20240902-155524-trace-remotestart_false.log)
[evcc-20240902-160534-trace-remotestart_true.log](https://github.com/user-attachments/files/16838763/evcc-20240902-160534-trace-remotestart_true.log)

mfuchs1984, 2024-09-02:
There are a lot of reports and discussions about the Pulsar chargers. They reject the charging profiles sent by evcc although they are valid. Seen e.g. 
https://github.com/evcc-io/evcc/issues/15581#issuecomment-2308356630
https://github.com/evcc-io/evcc/discussions/15537
https://github.com/evcc-io/evcc/issues/15581#issuecomment-2323259887

andig, 2024-09-02:
See https://github.com/evcc-io/evcc/issues/15581

closed 2024-09-02 by andig