evcc-io/evcc #15868: Add "Fast Green" Charging Mode

Stueflo, 2024-09-02:
**Is your feature request related to a problem? Please describe.**
My feature is not related to a problem but could be an enhancement. I regularly come across situations where I would like to charge my vehicle with the maximum power possible without mains electricity (i.e. only with Solar + battery). This is e.g. the case in the following situation:
- the battery has a high SoC and
- the car is not fully charged but will be disconnected soon and 
- it is foreseeable that PV energy delivered during the absence of the car will exceed the capacity of the battery.

**Describe the solution you'd like**
A solution could be an additional charging mode "Fast Green". In this charging mode, the vehicle would be charged with the entire PV power available (if any) plus the maximum battery power that can be provided (limited by either the maximum  discharging power of the battery or the maximum power of the inverter), reduced by the consumption of the house. 

**Describe alternatives you've considered**
Currently I see the following options:

- Choose "Min + Solar" and manually adjust the minimum power of the loadpoint
- Choose "Fast" and manually adjust the maximum power of the loadpoint
.

**Additional context**
Most likely several parameters would have to be configured / considered:

- the maximum discharging speed of the battery which determines the maximum power that can be delivered by the battery
- the maximum power that can be delivered by the inverter (solar + battery) because above this threshold the inverter will take energy from the grid
- Maybe an "Opposite Solar Surplus Charging" approach could be used alternatively. The system could detect the power taken from the grid (if any) and reduce the charging speed accordingly. In this case some power would be taken from the grid (until the system determined the ideal power level), but less parameters would have to be considered.

VolkerK62, 2024-09-02:
seem to be part of https://github.com/evcc-io/evcc/discussions/3530

andig, 2024-09-02:
Thank you, we would indeed consider this part of the PV mode redesign.

/cc @naltatis

closed 2024-09-02 by andig