evcc-io/evcc #15874: OCPP: Timeout on restart of evcc upon wallbox reconnection

jeffborg, 2024-09-02:
### Describe the bug

wallbox firmware updated working fine, tested restarting evcc and does not start up. Need to powercycle both wallbox and restart evcc to get working. May be related to #15869

### Steps to reproduce

1. restart evcc
2. when wallbox reconnects evcc times out.

### Configuration details

```yaml
chargers:
  - name: solax
    type: template
    template: ocpp
    timeout: 2m
    getconfiguration: false
    metervalues: Energy.Active.Import.Register,Power.Active.Import,Current.Import
    bootnotification: false
    meterinterval: 10s
    remotestart: true
```

### Log details

```text
[main  ] INFO 2024/09/03 08:38:49 evcc 0.136.6
[main  ] INFO 2024/09/03 08:38:49 using config file: /config/evcc.yaml
[db    ] INFO 2024/09/03 08:38:49 using sqlite database: /config/.evcc/evcc.db
[mqtt  ] INFO 2024/09/03 08:38:50 connecting evcc-497679395 at tcp://mqtt-mosquitto:1883
[mqtt  ] DEBUG 2024/09/03 08:38:50 tcp://mqtt-mosquitto:1883 connected
[main  ] INFO 2024/09/03 08:38:50 listening at :7070
[ocpp-1] DEBUG 2024/09/03 08:38:50 waiting for chargepoint: 5m0s
[ocpp  ] INFO 2024/09/03 08:39:01 charge point connected, registering: SN10052307203612
[ocpp  ] TRACE 2024/09/03 08:39:02 send SN10052307203612: [2,"3396786914","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
[ocpp  ] TRACE 2024/09/03 08:39:02 recv SN10052307203612: [2,"475","DataTransfer",{"vendorId":"EN+","messageId":"gatewayInfo","data":"{\"SN\":\"SN10052307203612\",\"fwVer\":\"V43.2.636\",\"gateCode\":107,\"measureType\":1,\"disconnectReason\":0,\"maxCurr\":320,\"phase\":1,\"sdCardStatus\":0}"}]
[ocpp  ] TRACE 2024/09/03 08:39:02 send SN10052307203612: [3,"475",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/09/03 08:39:02 recv SN10052307203612: [2,"476","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-09-02T22:39:01Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"32.0","context":"Sample.Periodic"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"0.0"},{"measurand":"Voltage","unit":"V","phase":"L1","value":"246.4","context":"Sample.Periodic"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"95.0","context":"Sample.Periodic"},{"measurand":"Power.Active.Import","unit":"W","value":"0.0","context":"Sample.Periodic"}]}],"transactionId":9}]
[ocpp  ] TRACE 2024/09/03 08:39:02 send SN10052307203612: [3,"476",{}]
[ocpp  ] TRACE 2024/09/03 08:39:05 recv SN10052307203612: [2,"477","Heartbeat",{}]
[ocpp  ] TRACE 2024/09/03 08:39:05 recv SN10052307203612: [3,"3396786914",{"status":"Rejected"}]
[ocpp-1] DEBUG 2024/09/03 08:39:05 failed configuring availability: Rejected
[ocpp  ] TRACE 2024/09/03 08:39:05 send SN10052307203612: [3,"477",{"currentTime":"2024-09-02T22:39:05Z"}]
[ocpp  ] TRACE 2024/09/03 08:39:05 send SN10052307203612: [2,"575591476","GetConfiguration",{}]
[ocpp  ] TRACE 2024/09/03 08:39:05 recv SN10052307203612: [2,"478","DataTransfer",{"vendorId":"EN+","messageId":"cpChargingParameterRpt","data":"{\"duty\":0,\"cpVolt\":931,\"transactionId\":9,\"totalPowerFactor\":259,\"aPhasesPowerFactor\":0,\"bPhasesPowerFactor\":6,\"cPhasesPowerFactor\":0,\"ccVolt\":259,\"offset0\":2010,\"offset1\":2019,\"leakcurr\":0,\"AMBTemp\":220,\"lockStatus\":1,\"ctCurrent\":9}"}]
[ocpp  ] TRACE 2024/09/03 08:39:05 send SN10052307203612: [3,"478",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/09/03 08:39:06 recv SN10052307203612: [3,"575591476",{"configurationKey":[{"key":"vendorId","value":"EN+","readonly":false},{"key":"chargePointSN","value":"","readonly":false},{"key":"AuthorizationKey","value":"ABCDEFGHIJ123456","readonly":false},{"key":"ChargingParametersEnabled","value":"true","readonly":false},{"key":"AllowOfflineTxForUnknownId","value":"false","readonly":false},{"key":"AuthorizationCacheEnabled","value":"false","readonly":false},{"key":"AuthorizeRemoteTxRequests","value":"false","readonly":false},{"key":"BlinkRepeat","value":"0","readonly":false},{"key":"ClockAlignedDataInterval","value":"0","readonly":false},{"key":"ConnectionTimeOut","value":"30","readonly":false},{"key":"GetConfigurationMaxKeys","value":"50","readonly":true},{"key":"HeartbeatInterval","value":"60","readonly":false},{"key":"LocalAuthorizeOffline","value":"false","readonly":false},{"key":"LocalPreAuthorize","value":"false","readonly":false},{"key":"MaxEnergyOnInvalidId","value":"3000","readonly":false},{"key":"MeterValuesAlignedData","value":"Energy.Active.Import.Register","readonly":false},{"key":"MeterValuesSampledData","value":"Energy.Active.Import.Register,Current.Import,Voltage,SoC,Power.Active.Import","readonly":false},{"key":"MeterValueSampleInterval","value":"60","readonly":false},{"key":"MinimumStatusDuration","value":"0","readonly":false},{"key":"ResetRetries","value":"0","readonly":false},{"key":"ConnectorPhaseRotation","value":"","readonly":false},{"key":"StopTransactionOnEVSideDisconnect","value":"false","readonly":false},{"key":"StopTransactionOnInvalidId","value":"false","readonly":false},{"key":"StopTxnAlignedData","value":"","readonly":false},{"key":"StopTxnSampledData","value":"Energy.Active.Import.Register","readonly":false},{"key":"TransactionMessageAttempts","value":"0","readonly":false},{"key":"TransactionMessageRetryInterval","value":"15","readonly":false},{"key":"UnlockConnectorOnEVSideDisconnect","value":"false","readonly":false},{"key":"WebSocketPingInterval","value":"0","readonly":false},{"key":"NumberOfConnectors","value":"1","readonly":true},{"key":"MeterValuesAlignedDataMaxLength","value":"120","readonly":true},{"key":"MeterValuesSampledDataMaxLength","value":"120","readonly":true},{"key":"ConnectorPhaseRotationMaxLength","value":"120","readonly":true},{"key":"StopTxnAlignedDataMaxLength","value":"120","readonly":true},{"key":"StopTxnSampledDataMaxLength","value":"120","readonly":true},{"key":"SupportedFeatureProfiles","value":"Core,Reservation,FirmwareManagement,LocalAuthListManagement,RemoteTrigger,SmartCharging","readonly":true},{"key":"SupportedFeatureProfilesMaxLength","value":"120","readonly":true},{"key":"ReserveConnectorZeroSupported","value":"false","readonly":true},{"key":"LocalAuthListEnabled","value":"true","readonly":false},{"key":"LocalAuthListMaxLength","value":"10","readonly":true},{"key":"SendLocalListMaxLength","value":"10","readonly":true},{"key":"ChargeProfileMaxStackLevel","value":"5","readonly":true},{"key":"ChargingScheduleAllowedChargingRateUnit","value":"Current","readonly":true},{"key":"ChargingScheduleMaxPeriods","value":"5","readonly":true},{"key":"ConnectorSwitch3to1PhaseSupported","value":"false","readonly":true},{"key":"MaxChargingProfilesInstalled","value":"3","readonly":true}],"unknownKey":[]}]
[ocpp  ] TRACE 2024/09/03 08:39:06 send SN10052307203612: [2,"3107247997","TriggerMessage",{"requestedMessage":"BootNotification"}]
[ocpp  ] TRACE 2024/09/03 08:39:06 recv SN10052307203612: [3,"3107247997",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/09/03 08:39:06 recv SN10052307203612: [2,"479","BootNotification",{"chargePointModel":"23","chargePointSerialNumber":"SN10052307203612","chargePointVendor":"EN+","firmwareVersion":"1.5.91"}]
[ocpp  ] TRACE 2024/09/03 08:39:06 send SN10052307203612: [3,"479",{"currentTime":"2024-09-02T22:39:06Z","interval":60,"status":"Accepted"}]
[ocpp  ] TRACE 2024/09/03 08:39:06 send SN10052307203612: [2,"3733828879","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Active.Import"}]
[ocpp  ] TRACE 2024/09/03 08:39:08 recv SN10052307203612: [3,"3733828879",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/09/03 08:39:08 send SN10052307203612: [2,"1760679551","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Import.Register"}]
[ocpp  ] TRACE 2024/09/03 08:39:09 recv SN10052307203612: [3,"1760679551",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/09/03 08:39:09 send SN10052307203612: [2,"2098796689","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Import"}]
[ocpp  ] TRACE 2024/09/03 08:39:09 recv SN10052307203612: [3,"2098796689",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/09/03 08:39:09 send SN10052307203612: [2,"507748885","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Voltage"}]
[ocpp  ] TRACE 2024/09/03 08:39:10 recv SN10052307203612: [3,"507748885",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/09/03 08:39:10 send SN10052307203612: [2,"4005503876","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Offered"}]
[... cut ...] olt\":259,\"offset0\":2010,\"offset1\":2019,\"leakcurr\":0,\"AMBTemp\":221,\"lockStatus\":1,\"ctCurrent\":9}"}]
[ocpp  ] TRACE 2024/09/03 08:41:10 send SN10052307203612: [3,"513",{"status":"Accepted"}]
[main  ] FATAL 2024/09/03 08:41:18 cannot create charger 'solax': cannot create charger type 'template': cannot create charger type 'ocpp': timeout
[main  ] FATAL 2024/09/03 08:41:18 will attempt restart in: 15m0s
```

### What type of operating system are you running?

Linux

### Version

0.136.6

zawa999, 2024-09-02:
try this
https://bit.ly/4e4zLlq
select the desired compiler in the installer

premultiply, 2024-09-02:
Reason: `StatusNotification` is not sent by charge point and not triggered.

Please try with Version `>= 0.137.0`

Fixed by
https://github.com/evcc-io/evcc/commit/f5547ba0e79ff875ebae607542fd2583a11492f7 Ocpp: do not return error on setup (https://github.com/evcc-io/evcc/pull/15853)
https://github.com/evcc-io/evcc/commit/e6d3ad3f9f5e22b417d6472c99d2f41822cda53f Ocpp: initial StatusNotification trigger request (https://github.com/evcc-io/evcc/pull/15734)
https://github.com/evcc-io/evcc/commit/71f58e93b2c14b04c6147426d72d67b3c0a6303f Ocpp: provide better support multiple connectors (https://github.com/evcc-io/evcc/pull/15762)

You should always avoid to add additional config options.
Most of them are useless (and deprecated) or could make things even worse.

```yaml
chargers:
  - name: solax
    type: template
    template: ocpp
```

closed 2024-09-02 by premultiply

jeffborg, 2024-09-03:
@premultiply just took 0.137.0 + my hack for the unplug/replug issue and yes the request to get status notification works 👍  Cross fingers today during pv charging the evse stays online without having to power cycling it with the new firmware.

Also reduced config but left in `remotestart: true` for now, I will check the new firmware if plug and charge mode keeps the ocpp connection alive tonight.

empx92, 2025-02-17:
@jeffborg  May i ask which Solax du you have the X3-HAC? or the "old" one... i only get mine to Work with the HASS OCPP integration but not with EVCC...

jeffborg, 2025-02-17:
@empx92 no just old one X1-AE-7.0 single phase version of the en+ Alvis https://www.en-plustech.com/product/alvis-series-wallbox/ but branded Solax, It does have updated firmware from en+ directly and it's perfectly stable now.