evcc-io/evcc #15878: Tariffs ENTSO-E: no data for resolution: PT60M

Skysurfer-14, 2024-09-03:
### Describe the bug

evcc no longer starts after update to version 0.130.7 because the dynamic electricity tariff from ENTSO-E cannot be called up.

### Steps to reproduce

1. Update to version 0.130.7
2. Use the tariff ENTSO-E like this:
```
# eprimo Taris 21,63 ct/kWh (brutto) + EPEX
  grid:
    type: entsoe # Entso-E european market data
    domain: BZN|DE-LU # https://transparency.entsoe.eu/content/static_content/Static%20content/web%20api/Guide.html#_areas
    securitytoken: bd8fe94d-add4-4e3a-88f9-70415d481155
    charges: 0.1817647 # Nettopreis: 0.2163 / 1.19
    tax: 0.19
```
3. Start evcc
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: a1cc34b439cef856606b780ce9a2a07d4e91f0b52384bee708d3f6661a201b01

interval: 10s # control cycle interval

sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJleHAiOjE4MDEwNDc2MDAsImlhdCI6MTcwNjQzOTYwMCwiaXNzIjoiZXZjYy5pbyIsInN1YiI6IlNreXN1cmZlci0xNCIsInNyYyI6ImdoIn0.cELipokACv2xMT6pMAE3bHQ7s9Mus3MHS0vAbWEnSfY

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

# mqtt message broker
mqtt:
  broker: 192.168.178.132:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  user: mqtt-explorer
  password: CEL243243

modbusproxy:
- port: 5200
  uri: 192.168.178.191:502

meters:
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.191  
  port: 502  
  usage: grid  
  modbus: tcpip  
  timeout: 5s  
  name: grid1
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.191  
  port: 502  
  usage: pv  
  modbus: tcpip  
  timeout: 5s  
  name: pv2
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.191  
  port: 502  
  usage: battery  
  modbus: tcpip  
  timeout: 5s  
  name: battery3

chargers:
- type: template
  template: go-e-v3 
  host: 192.168.178.188  
  name: wallbox5

vehicles:
- type: template
  template: ovms 
  title: e-up!  
  user: Skysurfer  
  password: Wqa12v1ktk-3JqWb8_p9  
  vehicleid: WVWZZZAAZMD913760  
  capacity: 32.3 
  phases: 2
  maxcurrent: 16
  name: ev4
#  mode: pv

loadpoints:
- title: Garage
  charger: wallbox5
  vehicle: ev4
  mode: pv
#  phases: 3
#  mincurrent: 6
#  maxcurrent: 16

site:
  title: Ladesteuerung
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

tariffs:
# Veganstrom
#  grid:
#    type: fixed
#    price: 0.3247

# ostrom Tarif 20,09 ct/kWh (brutto) + EPEX 
#  grid:
#    type: entsoe # Entso-E european market data
#    domain: BZN|DE-LU # https://transparency.entsoe.eu/content/static_content/Static%20content/web%20api/Guide.html#_areas
#    securitytoken: bd8fe94d-add4-4e3a-88f9-70415d481155
#    charges: 0.1688235 # Nettopreis: 0.2009 / 1.19
#    tax: 0.19
    
# eprimo Taris 21,63 ct/kWh (brutto) + EPEX
  grid:
    type: entsoe # Entso-E european market data
    domain: BZN|DE-LU # https://transparency.entsoe.eu/content/static_content/Static%20content/web%20api/Guide.html#_areas
    securitytoken: bd8fe94d-add4-4e3a-88f9-70415d481155
    charges: 0.1817647 # Nettopreis: 0.2163 / 1.19
    tax: 0.19
    
  feedin:
    type: fixed
    price: 0.082
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/09/03 09:15:01 evcc 0.130.7
[main  ] INFO 2024/09/03 09:15:01 using config file: /config/evcc.yaml
[db    ] INFO 2024/09/03 09:15:01 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2024/09/03 09:15:02 connecting evcc-1918748725 at tcp://192.168.178.132:1883
[mqtt  ] DEBUG 2024/09/03 09:15:02 tcp://192.168.178.132:1883 connected
[main  ] INFO 2024/09/03 09:15:02 listening at :7070
[proxy-5200] DEBUG 2024/09/03 09:15:02 modbus proxy for 192.168.178.191:502 listening at :5200
[entsoe] ERROR 2024/09/03 09:15:19 no data for resolution: PT60M
[main  ] FATAL 2024/09/03 09:15:19 cannot create tariff type 'entsoe': no data for resolution: PT60M
[main  ] FATAL 2024/09/03 09:15:19 will attempt restart in: 15m0s[lp-1  ] DEBUG 2024/09/03 08:35:17 !! session: chargeRater.chargedEnergy=0.6 - chargedAtStartup=0.0
```

### What type of operating system are you running?

HomeAssistant Add-on

mdkeil, 2024-09-03:
Ggfs solange auf Frauenhofer ISE ausweichen
https://docs.evcc.io/docs/devices/tariffs#fraunhofer-ise

Läuft es mit älteren Versionen?

mdkeil, 2024-09-03:
Ich würde meiden, dass es sich um temporäre Störungen beim Anbieter handelt.

closed 2024-09-03 by andig

sdebruyn, 2025-09-29:
I now have the same since updating to v0.208.1