evcc-io/evcc #15881: SMA Sunny Boy Storage Template Request

wittmannchris, 2024-09-03:
Sunny Boy Storage 2.5 - new template 

For Battery Loading:
40236 ->2289
40793 -> 2500
40795 -> 2500
40797 -> 0
40799 -> 2650
40801 -> 0

For 0-Usage Loading:

40236 ->2424
40793 -> 0
40795 -> 0
40797 -> 0
40799 -> 0
40801 -> 0

Normal : 

40236 ->2424
40793 -> 0
40795 -> 2500
40797 -> 0
40799 -> 2650
40801 -> 0

All6 Registers must be written within 60 seconds and than cyclical (also all 6 together).
I made a 2 sek break between the first 3 and the second three Registers, not to overload the modbus (somehwere has someone written this ;) )

Flash is not getting damaged as marked as a "ANlagensteuerobjekt" as well details in the documentation
[image]

Again.... 40149 and 40151 is not working anymore for loading !

_Originally posted by @wittmannchris in https://github.com/evcc-io/evcc/issues/15294#issuecomment-2278588953_

wittmannchris, 2024-09-03:
[Modbus - Parameter und Messwerte.pdf](https://github.com/user-attachments/files/16846373/Modbus.-.Parameter.und.Messwerte.pdf)

Page 64

[SMA_Modbus-TI-de-15.pdf](https://github.com/user-attachments/files/16846378/SMA_Modbus-TI-de-15.pdf)

andig, 2024-09-14:
Es ist unklar, warum es das braucht. Spricht etwas gegen `sma-sbs-modbus`? Geht `sma-si-modbus` auch nicht?

wittmannchris, 2024-09-14:
Nein geht weder noch. 

Der SBS 2.5 hat sein eigenes Ding wie beschrieben und nur dies funktioniert anscheinend noch. Doku und Register habe ich ja angehängt . Gerne mehr per pn bzw. kann auch testen.

andig, 2024-09-14:
Das SBS Template sollte alle Varianten abdecken. Hat SMA da die SW geändert? Oder ist nur den 2.5 "anders"? Sind die Register oben Sunspec? Falls ja- wie heissen die?

wittmannchris, 2024-09-14:
Ja seit dem neuen Update den Sunny home managers werden die bisherigen Register ( aus dem SI Template) beim SBS 2.5 übergeordnet verwendet.
Zur direkten Modbusansteuerung sind dann nur noch die Register wie ich beschrieben habe möglich. 
Hatte hier auch viel Kontakt mit SMA als Feldtester .
Der SBS 1.5-2.5 ist da auch komplett anders als der SBS 3.7-5.0 , für welche das SBS Template da ist. 

Laden: oberster Punkt 
0 usage war die batteriesperre gemeint bei Schnellladen und normal ist normal 😉

andig, 2024-09-14:
> Ja seit dem neuen Update den Sunny home managers werden die bisherigen Register ( aus dem SI Template) beim SBS 2.5 übergeordnet verwendet.

Ich verstehe leider nur Bahnhof. Was bedeutet das? Warum geht das SI Template dann nicht?

> Der SBS 1.5-2.5 ist da auch komplett anders als der SBS 3.7-5.0 , für welche das SBS Template da ist.

Ah

> 0 usage war die batteriesperre gemeint 

Ah

> So wie ich das sehe aus dem pdf sind die 6 Register kein sunspec

I see. Also neues Template.

wittmannchris, 2024-09-14:
Danke 👍😊

andig, 2024-09-14:
Achso: power/energy/soc aus dem SI Template wie gehabt?

wittmannchris, 2024-09-14:
Schau ich mir morgen mal an. 
Muss los 🍺

TheTrueRandom, 2024-09-15:
Gerade mal die modbus register manuell ausprobiert, alle 3 modi (laden, normal und 0-usage) funktionieren.
Beim Laden, geht er aber nicht über 623W.

wittmannchris, 2024-09-15:
Mein  Netzladen UI in der Batterieansicht ist weg?

andig, 2024-09-15:
> Beim Laden geht er aber nicht über 623W.

Das scheint mir bekanntes Problem zu sein- die Leistung wird wohl extrem langsam hoch gefahren. Falls sich dafür ein Register finden sollte können wir das gerne auf nehmen.

/cc @premultiply

wittmannchris, 2024-09-15:
> > Beim Laden geht er aber nicht über 623W.
> 
> Das scheint mir bekanntes Problem zu sein- die Leistung wird wohl extrem langsam hoch gefahren. Falls sich dafür ein Register finden sollte können wir das gerne auf nehmen.
> 
> /cc @premultiply

Kann ich gerade nicht beobachten

closed 2024-09-15 by andig

TheTrueRandom, 2024-09-15:
Hab viel rumprobiert, geht nicht über 623W beim laden. Hier mit modbus-cli als beispiel (30775 liest aktuelle power)
```
➜  cat loading.sh
modbus -s 3 192.168.10.180:502 40236/I=2289
modbus -s 3 192.168.10.180:502 40793/I=2500
modbus -s 3 192.168.10.180:502 40795/I=2500
modbus -s 3 192.168.10.180:502 40797/I=0
modbus -s 3 192.168.10.180:502 40799/I=2650
modbus -s 3 192.168.10.180:502 40801/I=0                                                                                                     
➜  modbus -s 3 192.168.10.180:502 30775/i 
Parsed 0 registers definitions from 1 files
30775: 2497 0x9c1
➜  ./loading.sh 
Parsed 0 registers definitions from 1 files
Parsed 0 registers definitions from 1 files
Parsed 0 registers definitions from 1 files
Parsed 0 registers definitions from 1 files
Parsed 0 registers definitions from 1 files
Parsed 0 registers definitions from 1 files
➜  modbus -s 3 192.168.10.180:502 30775/i
Parsed 0 registers definitions from 1 files
30775: -623 -0x26f

```

fabian440, 2024-09-15:
Hallo zusammen,

ich würde das neue Template gerne ausprobieren. Verstehe ich richtig, dass es dann im nächsten Release enthalten ist und ich dann als template name einfach nur auf "sma-sbs-15-25-modbus" umstellen müsste? Sorry für die doof Frage und vielen Dank vorab

andig, 2024-09-15:
Es wäre toll, die gleiche Frage nicht bei jedem issue/pr gestellt zu bekommen

premultiply, 2024-11-17:
@wittmannchris Akzeptiert der SBS2.5 hier auch statt 2500 W bzw. 2560 W größere Werte wie z. B. 100000 W ?

wittmannchris, 2024-11-18:
> @wittmannchris Akzeptiert der SBS2.5 hier auch statt 2500 W bzw. 2560 W größere Werte wie z. B. 100000 W ?
> 
> Das wäre relevant um die Templates angleichen zu können und die Benutzerkonfiguration zu vereinfachen.

Hab ich gerade so reproduzieren können.

Der SBS 2.5 hat sowohl 5000 als auch 10000 angenommen und auf max.Wert 2504 W geregelt.

Ich musste halt zum aufwecken aus dem Standby einmalig an die 40151 die 802 (active) senden.
Anders funktioniert anscheinend kein BWR, weder SBS 2.5 - 3.7 oder der Hybride.

premultiply, 2024-11-18:
Ok, danke für den Test. Dann kann ich die Templates angleichen.

Mein SBS wacht aus dem Tiefschlaf auf. Man muss nur offensichtlich die richtige Kombi senden.

wittmannchris, 2024-11-18:
> Ok, danke für den Test. Dann kann ich die Templates angleichen.
> 
> Mein SBS wacht aus dem Tiefschlaf auf. Man muss nur offensichtlich die richtige Kombi senden.

Wie machst du das ? ich sende die 6 register, jeweils drei gleichzeitig, die anderen 3 verzögert.

Es funktioniert auch ne weile ohne die 40151 .... bis er mal einige Stunden "schläft"....

andig, 2024-11-18:
Kann ich hier helfen? Betrifft das "aufwachen" nur "charge" oder auch "hold"? Letzteres würde ich nicht erwarten- denn dafür müsste der WR ja vorher "wach" sein?