evcc-io/evcc #15884: EEBUS - Plug & Charge - `evcc charger` returns : Identifier = <none>

gramss, 2024-09-03:
### Describe the bug

Ich würde gerne dieses Feature verwenden:
https://docs.evcc.io/docs/features/vehicle#erkennung-via-plug--charge

Mein Setup:
- evcc 0.130.6 auf Mac mit `brew service start`
- PMCC SW-Vers 3030
- PLC Verbindung
- EEBUS Energy Manager verbunden (evcc)
- Auto verbunden, Profile gesetzt (lädt ohne evcc daher nicht automatisch)
- Laden Starten/Stoppen funktioniert über evcc

Leider wird mir immer ein "Gastfahrzeug" bei EVCC angezeigt, wenn ich das Fahrzeug anstecke.

Mein Verständis ist, dass ich den identifier in der `evcc.yaml` auf Basis der Anleitung (siehe Link oben) angeben / herausfinden muss.

Leider funktioniert das bei mir nicht, siehe logs.

`Identifier:      <none>`

### Steps to reproduce

1. evcc is running as `brew service start evcc`
2. `$ evcc charger` -> Error: Websocket already in use (not surprising)
3. shut down `evcc` -> `brew service stop evcc`
4. make sure PMCC is available on network
5. make sure car is connected
6. car does not charge because of Profile (25% min + local + no direct charge) and SoC >25%
7. run `evcc charger`

Receive:

```
Power:           0W
Current L1..L3:  0A 0A 0A
.Soc:            not available
Charge status:   B
Enabled:         false
Charged:         0.0kWh
Mix/Max Current: 2.0/16.0A
Identifier:      <none>
```

Question:
Does the car need to charge during this process? (temporally set the charge profile to e.g 100% ..?) -> This info is missing in the documentation.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 8a5591514266a12f732e2b3e30d90f130e204e0832f09b360764a3cb53faffe5

interval: 30s # control cycle interval

chargers:
- type: template
  template: pmcc 
  ski: f...fae  
  name: wallbox1
[... 180 lines cut ...]
Charged:         0.0kWh
Mix/Max Current: 2.0/16.0A
Identifier:      <none>
[eebus ] DEBUG 2024/09/01 17:52:37 Recv: d:_i:19667_PorscheEVSE-0007XXX5:LoadControl to LoadControl result 543 45 ResultData 0
[eebus ] DEBUG 2024/09/01 17:52:37 Recv: d:_i:19667_PorscheEVSE-0007XXX5:LoadControl to LoadControl notify 545 LoadControlLimitListData
[eebus ] DEBUG 2024/09/01 17:52:37 ski f...fae event cem-opev-DataUpdateLimit
[eebus ] DEBUG 2024/09/01 17:52:37 ski f...fae event cem-oscev-DataUpdateLimit
[eebus ] DEBUG 2024/09/01 17:52:37 Recv: d:_i:19667_PorscheEVSE-0007XXX5:Measurement to Measurement notify 547 MeasurementListData
[eebus ] DEBUG 2024/09/01 17:52:37 ski f...fae event cem-evcem-DataUpdateCurrentPerPhase
[eebus ] DEBUG 2024/09/01 17:52:37 ski f...fae event cem-evcem-DataUpdatePowerPerPhase
[eebus ] DEBUG 2024/09/01 17:52:37 ski f...fae event cem-evcem-DataUpdateEnergyCharged
[eebus ] DEBUG 2024/09/01 17:52:37 ski f...fae event cem-evsoc-DataUpdateStateOfCharge
[eebus ] DEBUG 2024/09/01 17:52:38 mdns: stop announcement
[eebus ] DEBUG 2024/09/01 17:52:38 ski f...fae disconnected
[eebus ] ERROR 2024/09/01 17:52:38 websocket server error: http: Server closed
```
```

### What type of operating system are you running?

Linux

### Version

0.130.6

DerAndereAndi, 2024-09-03:
No, the car only needs to be connected.

' evcc  charger` only works with EEBUS if evcc is NOT running at the same time!

gramss, 2024-09-03:
ok.

I just tested with SoC 53% and min charge 60% (forcing the car to load).
EVCC off and send a `evcc charger`.

Finally got an identification MAC-Address. (which also works detecting the car in EVCC now! 👍 )

retested with min-charge-profile back to 25%. and have the same issue as before.

The Charge Status was different this time `Charge Status: C`

Logs:
```
➜  ~ evcc charger
[main  ] INFO 2024/09/03 13:09:19 evcc 0.130.6
[main  ] INFO 2024/09/03 13:09:19 using config file: /Users/f/evcc.yaml
[db    ] INFO 2024/09/03 13:09:19 using sqlite database: /Users/f/.evcc/evcc.db
[eebus ] INFO 2024/09/03 13:09:19 Local SKI:  552a790b8bc8a9e32e68155a8a845c2d8d3ea88c
[eebus ] DEBUG 2024/09/03 13:09:19 starting websocket server on :4712
[eebus ] DEBUG 2024/09/03 13:09:19 mdns: announce
[eebus ] DEBUG 2024/09/03 13:09:19 mdns: using zeroconf
[eebus ] DEBUG 2024/09/03 13:09:19 mdns: start search
[eebus ] DEBUG 2024/09/03 13:09:46 incoming connection request from f...fae
[eebus ] DEBUG 2024/09/03 13:09:46 ski f...fae connected
[eebus ] DEBUG 2024/09/03 13:09:46 Send:  read 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/09/03 13:09:46 Recv: d:_i:19667_PorscheEVSE-0007XXX5:NodeManagement to NodeManagement read 8951 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/09/03 13:09:46 Send: d:_i:19667_PorscheEVSE-0007XXX5 reply 2 8951 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/09/03 13:09:46 Recv: d:_i:19667_PorscheEVSE-0007XXX5:NodeManagement to NodeManagement reply 8956 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/09/03 13:09:46 Send: d:_i:19667_PorscheEVSE-0007XXX5 call 3 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/09/03 13:09:46 Send: d:_i:19667_PorscheEVSE-0007XXX5 read 4 NodeManagementUseCaseData
[eebus ] DEBUG 2024/09/03 13:09:46 cs-lpc: 0 DeviceDiagnosis Server found
[eebus ] DEBUG 2024/09/03 13:09:46 cs-lpp: 0 DeviceDiagnosis Server found
[eebus ] DEBUG 2024/09/03 13:09:46 Send: d:_i:19667_PorscheEVSE-0007XXX5 call 5 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/09/03 13:09:46 Send: d:_i:19667_PorscheEVSE-0007XXX5 call 6 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/09/03 13:09:46 Send: d:_i:19667_PorscheEVSE-0007XXX5 call 7 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/09/03 13:09:46 Send: d:_i:19667_PorscheEVSE-0007XXX5 read 10 MeasurementDescriptionListData
[eebus ] DEBUG 2024/09/03 13:09:46 Send: d:_i:19667_PorscheEVSE-0007XXX5 read 11 ElectricalConnectionDescriptionListData
[eebus ] DEBUG 2024/09/03 13:09:46 operation is not supported on function measurementConstraintsListData
[eebus ] DEBUG 2024/09/03 13:09:46 Send: d:_i:19667_PorscheEVSE-0007XXX5 call 8 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/09/03 13:09:46 Send: d:_i:19667_PorscheEVSE-0007XXX5 read 12 ElectricalConnectionParameterDescriptionListData
[eebus ] DEBUG 2024/09/03 13:09:46 Send: d:_i:19667_PorscheEVSE-0007XXX5 read 13 DeviceClassificationManufacturerData
[eebus ] DEBUG 2024/09/03 13:09:46 Send: d:_i:19667_PorscheEVSE-0007XXX5 read 9 DeviceClassificationManufacturerData
[eebus ] DEBUG 2024/09/03 13:09:46 Send: d:_i:19667_PorscheEVSE-0007XXX5 call 15 NodeManagementBindingRequestCall
[eebus ] DEBUG 2024/09/03 13:09:46 Send: d:_i:19667_PorscheEVSE-0007XXX5 read 16 DeviceDiagnosisStateData
[eebus ] DEBUG 2024/09/03 13:09:46 Send: d:_i:19667_PorscheEVSE-0007XXX5 call 14 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/09/03 13:09:46 Send: d:_i:19667_PorscheEVSE-0007XXX5 read 17 LoadControlLimitDescriptionListData
[eebus ] DEBUG 2024/09/03 13:09:46 Send: d:_i:19667_PorscheEVSE-0007XXX5 call 18 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/09/03 13:09:46 ski f...fae event cem-evsecc-EvseConnected
[eebus ] DEBUG 2024/09/03 13:09:46 operation is not supported on function loadControlLimitConstraintsListData
[eebus ] DEBUG 2024/09/03 13:09:46 Send: d:_i:19667_PorscheEVSE-0007XXX5 read 19 MeasurementDescriptionListData
[eebus ] DEBUG 2024/09/03 13:09:46 Send: d:_i:19667_PorscheEVSE-0007XXX5 read 20 DeviceConfigurationKeyValueDescriptionListData
[eebus ] DEBUG 2024/09/03 13:09:46 operation is not supported on function measurementConstraintsListData
[eebus ] DEBUG 2024/09/03 13:09:46 Send: d:_i:19667_PorscheEVSE-0007XXX5 call 21 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/09/03 13:09:46 Send: d:_i:19667_PorscheEVSE-0007XXX5 read 22 DeviceDiagnosisStateData
[eebus ] DEBUG 2024/09/03 13:09:46 Send: d:_i:19667_PorscheEVSE-0007XXX5 call 23 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/09/03 13:09:46 Send: d:_i:19667_PorscheEVSE-0007XXX5 read 24 ElectricalConnectionParameterDescriptionListData
[eebus ] DEBUG 2024/09/03 13:09:46 Send: d:_i:19667_PorscheEVSE-0007XXX5 read 25 ElectricalConnectionPermittedValueSetListData
[eebus ] DEBUG 2024/09/03 13:09:46 Send: d:_i:19667_PorscheEVSE-0007XXX5 call 26 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/09/03 13:09:46 Send: d:_i:19667_PorscheEVSE-0007XXX5 read 27 IdentificationListData
[eebus ] DEBUG 2024/09/03 13:09:46 ski f...fae event cem-evcc-EvConnected
[eebus ] DEBUG 2024/09/03 13:09:46 Recv: d:_i:19667_PorscheEVSE-0007XXX5:NodeManagement to NodeManagement read 8961 NodeManagementDestinationListData
[eebus ] DEBUG 2024/09/03 13:09:46 Send: d:_i:19667_PorscheEVSE-0007XXX5 reply 28 8961 NodeManagementDestinationListData
[... cut ...]
Mix/Max Current: 2.0/16.0A
Identifier:      0X:XX:XX:XX:XX:XD
[eebus ] DEBUG 2024/09/03 13:09:49 mdns: stop announcement
[eebus ] DEBUG 2024/09/03 13:09:49 ski f...fae disconnected
[eebus ] ERROR 2024/09/03 13:09:49 websocket server error: http: Server closed
```

Logs without charge_min-profile > SoC:

```
Power:           0W
Current L1..L3:  0A 0A 0A
.Soc:            not available
Charge status:   B
Enabled:         false
Charged:         0.0kWh
Mix/Max Current: 2.0/16.0A
Identifier:      <none>
```

closed 2024-09-03 by gramss

gramss, 2024-09-03:
Ah cool! Thanks for the explanation!

I am glad that this works now.. :)
I just want to get the (more or less) current SoC without the expensive P-Connect Service..

Next is trying out the evcc Testflight iOS App and generally seeing if evcc keeps the last known SoC in memory while the car is disconnected.

### Off topic:
Any idea how to get the SoC from the Apple Maps App on iOS..? This CarPlay integration (EV route planning with initial current SoC, unsure if updated during navigation..) works still without the Connection Services. 
Might have something to do with evcc if the evcc app could access this information somehow.. But obv. nothing to do with EEBUS.

DerAndereAndi, 2024-09-03:
The off-topic comment isn't clear to me and what you mean.

There is no way to get the Apple Maps provided data, also no way to access the source of that data (which is a Porsche service).

Also you don't need this anyway: evcc gets the SoC data with our setup from the PMCC. Just make sure the Plug & Charge is enabled in the EV.

gramss, 2024-09-03:
Thank you for that answer. That triggered me to write down my use cases and the potential options to fulfill all of them.
I think it is not worth to investigate any more options than what evcc with EEBUS already covers in my daily life.

----

The rest is just FYI..:

What are my **use cases**:
- checking SoC in the office -> SoC from home - normal drive (10%) is enough -> maybe go home eariler regarding dynamic pricing = Covered with evcc and P&C and your EEBUS implementation (awesome!)
- during holiday / larger travel / camping -> the SoC from home is no use to me -> need SoC for exact planning with ABRP (I did 20k km this half year..) -> Not covered through evcc
- during extended weekend stays at family -> not always a PMCC is handy with evcc on site.. I will try to cover this in 50% of the time with an additional local setup, PMCC exists there as well..

_Getting the Apple Maps Data - Options:_
- Looking into Apple Maps APIs (such things can be used in own apps or even Pythonista/Scriptable)
- Jailbreaking - finding the data - No high interest/feasability
- Data is in the iCloud (update cycle/initiator still unknown, also on iOS) -> Mac has also access to this data -> also boils down to finding (hidden) APIs

_Porsche Services_ -> Without Subscription, no way -> just pay them, if SoC while on the move is so important

_Getting the data from the source_ (the car directly while not on the PMCC with EVCC), _Options_:

- OBD Dongle -> some app needed -> only works while near car / car on -> could work for rooftent-camping use-case..
- Tapping the CAN interface of the car -> unsure which CAN protocol is used by Taycan (normal CAN or CAN-FB -> encryption? ->nope) -> would need to rewire car -> low liklyhood & could drain 12V Battery which is no fun as you might already know... Projects in that direction have no indication of support for my car
  - http://api.openvehicles.com
  - comma.ai
  
  
As I write this list down.. I think I should just be happy that I can cover 85% of the time that I want to access the SoC and should not overthink the rest.