evcc-io/evcc #15886: Elli Pro: Charger out of sync

CarG70, 2024-09-03:
### Describe the bug

Meine Elli Pro (ohne externen Zähler) lief mit dem Eebus-Profil bisher sehr stabil. Nach Update auf 0.130.7 bekomme ich nun die folgende Fehlermeldung: "Charge currents: not available". Ab 13:04:18 im Log-File. Nach dem Update hatte ich sowohl den Raspberry als auch die Wallbox rebootet.
[yaml.txt](https://github.com/user-attachments/files/16847918/yaml.txt)
[evcc-20240903-132111-debug.log](https://github.com/user-attachments/files/16847922/evcc-20240903-132111-debug.log)

### Steps to reproduce

1. Auto mit Wallbox verbinden
2. Ladevorgang starten
3. Fehlermeldung entsprechend Aktualisierungintervall
...

### Configuration details

```yaml
als Datei angefügt
```

### Log details

```text
als Datei angefügt
```

### What type of operating system are you running?

Linux

### Version

0.130.7

DerAndereAndi, 2024-09-03:
Das Logfile zeigt dass 2 evcc Instanzen laufen. Es zeigt keinen der beschriebenen Fehler. Bitte sicherstellen dass evcc nur 1 Mal läuft.

CarG70, 2024-09-03:
wie bekommt man das denn 2x zum Laufen und wie kann ich wieder abstellen?

DerAndereAndi, 2024-09-03:
Man kann evcc nicht 2 Mal laufen haben. Das geht bei EEBUS generell nicht.

Um an die Logs (EEBUS mit Trace !) zu kommen, bitte entweder die UI benutzen oder in der Dokumentation nachlesen wie man die aus dem System bekommt.

CarG70, 2024-09-03:
Danke für die Info. Nach Neustart von evcc ist die Meldung nun weg. Was auch immer das nun wieder war.

closed 2024-09-03 by CarG70