evcc-io/evcc #15890: OCPP: CP got into disconnected state even though evse is clearly has reconnected successfully

jeffborg, 2024-09-03:
### Describe the bug

Somehow evcc thinks the evse is disconnected after there were some internal network stability issues see logs. The evse did reconnect but somehow the disconnect handler got called AFTER the connected handler and connect boolean inside cp.go was set to false

I have attached the logs for the time covering the period in question showing the disconnect handler getting called multiple times after the connect handler was called see timestamp 3:12:36 for the ordering being out of order.

mqtt etc.. perfectly recovered on it's own, but ocpp thinks the evse is connected even though it's still sending messages to it and getting responses from it. I have also attached some logs later at 8:30am showing same state.

Knowing this I got it working by knocking the wifi connection of the evse off so it reconnected and hence connected was set to true here via connected method https://github.com/evcc-io/evcc/blob/master/charger/ocpp/cp.go#L22

I did force a charge and it evcc also appears to be sending `SetChargingProfile` twice see 3rd part of logs, I'm not sure if intended or not, I won't restart evcc yet in case you want me to do something else with the current instance of evcc.

_PS: the firmware update from en+ support in china for my evse has seemed to fix **all** network stability issues._

I've had to attach the logs as comments for parts 2 and 3

### Steps to reproduce

1. left evcc running for 22 hours
2. evse was also perfectly running (Thanks to the firmware update for it 👍 )
3. had some minor network issues at 3am
4. could not charge do anything at 8:30am

To get working I simply used the wifi controller to disconnect the evse so it reconnected, I didn't restart evcc nor the evse

### Configuration details

```yaml
chargers:
  - name: solax
    type: template
    template: ocpp
    remotestart: true
```

### Log details

```text
See comments below
```

### What type of operating system are you running?

Linux

### Version

0.130.7

jeffborg, 2024-09-03:
main network stability issues log
```
[lp-1  ] DEBUG 2024/09/04 03:07:25 charge power: 0W
[lp-1  ] DEBUG 2024/09/04 03:07:25 charge currents: [0 0 0]A
[circuit-main] DEBUG 2024/09/04 03:07:25 power: 855.37W
[circuit-main] DEBUG 2024/09/04 03:07:25 current: 0A
[site  ] ERROR 2024/09/04 03:07:25 pv 1 power: outdated
[site  ] DEBUG 2024/09/04 03:07:25 pv power: 0W
[site  ] DEBUG 2024/09/04 03:07:25 grid meter: 855W
[site  ] DEBUG 2024/09/04 03:07:25 site power: 855W
[lp-1  ] DEBUG 2024/09/04 03:07:25 charge voltages: [247 0 0]V
[lp-1  ] DEBUG 2024/09/04 03:07:25 detected connected phases: 1p
[lp-1  ] DEBUG 2024/09/04 03:07:26 charge total import: 18.099kWh
[lp-1  ] DEBUG 2024/09/04 03:07:26 charger status: B
[lp-1  ] DEBUG 2024/09/04 03:07:26 pv charge current: 0A = 0A + -3.72A (855W @ 1p)
[ocpp  ] TRACE 2024/09/04 03:07:26 recv SN10052307203612: [2,"19423","DataTransfer",{"vendorId":"EN+","messageId":"cpChargingParameterRpt","data":"{\"duty\":0,\"cpVolt\":931,\"transactionId\":2,\"totalPowerFactor\":256,\"aPhasesPowerFactor\":0,\"bPhasesPowerFactor\":6,\"cPhasesPowerFactor\":0,\"ccVolt\":256,\"offset0\":2010,\"offset1\":2019,\"leakcurr\":0,\"AMBTemp\":83,\"lockStatus\":1,\"ctCurrent\":9}"}]
[ocpp  ] TRACE 2024/09/04 03:07:26 send SN10052307203612: [3,"19423",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/09/04 03:07:36 send SN10052307203612: [2,"3987456050","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2024/09/04 03:07:36 recv SN10052307203612: [2,"19424","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-09-03T17:07:33Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"32.0","context":"Sample.Periodic"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"0.0"},{"measurand":"Voltage","unit":"V","phase":"L1","value":"246.9","context":"Sample.Periodic"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"18099.0","context":"Sample.Periodic"},{"measurand":"Power.Active.Import","unit":"W","value":"0.0","context":"Sample.Periodic"}]}],"transactionId":2}]
[ocpp  ] TRACE 2024/09/04 03:07:36 send SN10052307203612: [3,"19424",{}]
[ocpp  ] TRACE 2024/09/04 03:07:37 recv SN10052307203612: [3,"3987456050",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/09/04 03:07:37 recv SN10052307203612: [2,"19425","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-09-03T17:07:36Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"32.0","context":"Trigger"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"0.0"},{"measurand":"Voltage","unit":"V","phase":"L1","value":"246.9","context":"Trigger"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"18099.0","context":"Trigger"},{"measurand":"Power.Active.Import","unit":"W","value":"0.0","context":"Trigger"}]}],"transactionId":2}]
[ocpp  ] TRACE 2024/09/04 03:07:37 send SN10052307203612: [3,"19425",{}]
[ocpp  ] TRACE 2024/09/04 03:07:38 recv SN10052307203612: [2,"19426","DataTransfer",{"vendorId":"EN+","messageId":"cpChargingParameterRpt","data":"{\"duty\":0,\"cpVolt\":931,\"transactionId\":2,\"totalPowerFactor\":256,\"aPhasesPowerFactor\":0,\"bPhasesPowerFactor\":6,\"cPhasesPowerFactor\":0,\"ccVolt\":256,\"offset0\":2010,\"offset1\":2019,\"leakcurr\":0,\"AMBTemp\":83,\"lockStatus\":1,\"ctCurrent\":9}"}]
[ocpp  ] TRACE 2024/09/04 03:07:38 send SN10052307203612: [3,"19426",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/09/04 03:07:38 recv SN10052307203612: [2,"19427","DataTransfer",{"vendorId":"EN+","messageId":"cpChargingParameterRpt","data":"{\"duty\":0,\"cpVolt\":931,\"transactionId\":2,\"totalPowerFactor\":256,\"aPhasesPowerFactor\":0,\"bPhasesPowerFactor\":6,\"cPhasesPowerFactor\":0,\"ccVolt\":256,\"offset0\":2010,\"offset1\":2019,\"leakcurr\":0,\"AMBTemp\":83,\"lockStatus\":1,\"ctCurrent\":9}"}]
[ocpp  ] TRACE 2024/09/04 03:07:38 send SN10052307203612: [3,"19427",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/09/04 03:07:44 recv SN10052307203612: [2,"19428","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-09-03T17:07:43Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"32.0","context":"Sample.Periodic"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"0.0"},{"measurand":"Voltage","unit":"V","phase":"L1","value":"247.1","context":"Sample.Periodic"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"18099.0","context":"Sample.Periodic"},{"measurand":"Power.Active.Import","unit":"W","value":"0.0","context":"Sample.Periodic"}]}],"transactionId":2}]
[ocpp  ] TRACE 2024/09/04 03:07:44 send SN10052307203612: [3,"19428",{}]
[ocpp  ] TRACE 2024/09/04 03:07:45 recv SN10052307203612: [2,"19429","DataTransfer",{"vendorId":"EN+","messageId":"cpChargingParameterRpt","data":"{\"duty\":0,\"cpVolt\":931,\"transactionId\":2,\"totalPowerFactor\":256,\"aPhasesPowerFactor\":0,\"bPhasesPowerFactor\":6,\"cPhasesPowerFactor\":0,\"ccVolt\":256,\"offset0\":2010,\"offset1\":2019,\"leakcurr\":0,\"AMBTemp\":83,\"lockStatus\":1,\"ctCurrent\":9}"}]
[site  ] DEBUG 2024/09/04 03:07:52 ----
[lp-1  ] DEBUG 2024/09/04 03:07:52 charge power: 0W
[lp-1  ] DEBUG 2024/09/04 03:07:52 charge currents: [0 0 0]A
[circuit-main] DEBUG 2024/09/04 03:07:52 power: 892.78W
[circuit-main] DEBUG 2024/09/04 03:07:52 current: 0A
[site  ] ERROR 2024/09/04 03:07:53 pv 1 power: outdated
[site  ] DEBUG 2024/09/04 03:07:53 pv power: 0W
[site  ] DEBUG 2024/09/04 03:07:53 grid meter: 893W
[site  ] DEBUG 2024/09/04 03:07:53 site power: 893W
[... cut ...]
[ocpp  ] TRACE 2024/09/04 09:09:47 send SN10052307203612: [3,"26176",{}]
[ocpp  ] TRACE 2024/09/04 09:09:48 recv SN10052307203612: [2,"26177","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-09-03T23:09:47Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"32.0","context":"Sample.Periodic"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"0.0"},{"measurand":"Voltage","unit":"V","phase":"L1","value":"248.0","context":"Sample.Periodic"},{"measurand":"Energy.Active.Import.Regist
```

andig, 2024-09-04:
Might be a race condition in the ocpp library. Probably needs upstream issue. Is this easily reproducible?

jeffborg, 2024-09-04:
@andig not really, home k8s cluster had a "moment" at 3am this morning, I should have got the logs from traefik proxy which sits in front of evcc as well for the ocpp connection but they are gone now. If it happens again happy to gather this as well.

andig, 2024-09-04:
Could you prune all the mqtt stuff for readability?

andig, 2024-09-04:
Thank you. Opened upstream ticket to get more insights.

andig, 2024-09-05:
See update in related issue. If this happens more than once we'd need to add additional logging and again full log. Otherwise we could close?

jeffborg, 2024-09-05:
@andig ok to close, if happens more often I will provide the ocpp layer debug logs

One thing I noticed in your bug report is ChargePointDisconnected handler does not use the lock, this may have triggered the out of order calling inside evcc.

```
// NewChargePoint implements ocpp16.ChargePointConnectionHandler
func (cs *CS) NewChargePoint(chargePoint ocpp16.ChargePointConnection) {
	cs.mu.Lock()
	defer cs.mu.Unlock()

	// check for configured charge point
	cp, ok := cs.cps[chargePoint.ID()]
	if ok {
		cs.log.DEBUG.Printf("charge point connected: %s", chargePoint.ID())
		cp.connect(true)
	}
}

// ChargePointDisconnected implements ocpp16.ChargePointConnectionHandler
func (cs *CS) ChargePointDisconnected(chargePoint ocpp16.ChargePointConnection) {
	cs.log.DEBUG.Printf("charge point disconnected: %s", chargePoint.ID())

	if cp, err := cs.ChargepointByID(chargePoint.ID()); err == nil {
		cp.connect(false)
	}
}
```

andig, 2024-09-05:
Yes- but it only calls functions that use locking.