evcc-io/evcc #15906: Tronity: time between API queries

sebnaf, 2024-09-05:
### Describe the bug

Looking at my trace log, I have found out that time between queries should be 15mins, however, there seems to be hickups from time to time not updating from guest vehicle to my Tesla.

Also, looking at https://docs.evcc.io/docs/reference/configuration/vehicles there is `cache` mentioned, which i want to use with `cache: 5m` with Tronity (due to the instant connection of Tesla <-> Tronity). 

However, setting up

```
vehicles:
  - type: template
    template: tronity 
    title: EM-IL 55E
    clientid: 96a
    clientsecret: 8b73
    vin: LRW
    capacity: 72  
    name: tesla
    cache: 5m
```

Results in
`[main  ] FATAL 2024/09/05 11:27:12 cannot create vehicle 'tesla': cannot create vehicle type 'template': invalid key: cache`

### Steps to reproduce

1. use cache: 5 m with tronity template
2.
3.
...

### Configuration details

```yaml
vehicles:
  - type: template
    template: tronity 
    title: EM-IL 55E
    clientid: 96a
    clientsecret: 8b73
    vin: LRW
    capacity: 72  
    name: tesla
    cache: 5m
```
```

### Log details

```text
[tronity] TRACE 2024/09/04 22:00:45 POST https://api.tronity.tech/authentication
[tronity] TRACE 2024/09/04 22:00:45 GET https://api.tronity.tech/tronity/vehicles/668aedd2d9d3250012584ce4/last_record
[tronity] TRACE 2024/09/04 22:00:45 {"odometer":35204,"range":99.7,"level":22.45,"charging":"Disconnected","plugged":false,"chargerPower":0,"chargeRemainingTime":null,"latitude":48.124067,"longitude":7.844362,"status":null,"timestamp":1725480026000,"lastUpdate":1725480026000}
[tronity] TRACE 2024/09/04 22:01:45 GET https://api.tronity.tech/tronity/vehicles/668aedd2d9d3250012584ce4/last_record
[tronity] TRACE 2024/09/04 22:01:45 {"odometer":35204,"range":99.7,"level":22.45,"charging":"Disconnected","plugged":false,"chargerPower":0,"chargeRemainingTime":null,"latitude":48.124067,"longitude":7.844362,"status":null,"timestamp":1725480058000,"lastUpdate":1725480058000}
[tronity] TRACE 2024/09/04 22:02:45 GET https://api.tronity.tech/tronity/vehicles/668aedd2d9d3250012584ce4/last_record
[tronity] TRACE 2024/09/04 22:02:45 {"odometer":35204,"range":99.7,"level":22.45,"charging":"Disconnected","plugged":false,"chargerPower":0,"chargeRemainingTime":null,"latitude":48.124067,"longitude":7.844362,"status":null,"timestamp":1725480058000,"lastUpdate":1725480058000}
[tronity] TRACE 2024/09/04 22:03:45 GET https://api.tronity.tech/tronity/vehicles/668aedd2d9d3250012584ce4/last_record
[tronity] TRACE 2024/09/04 22:03:45 {"odometer":35204,"range":99.7,"level":22.45,"charging":"Disconnected","plugged":false,"chargerPower":0,"chargeRemainingTime":null,"latitude":48.124067,"longitude":7.844362,"status":null,"timestamp":1725480058000,"lastUpdate":1725480058000}
[tronity] TRACE 2024/09/04 22:04:45 GET https://api.tronity.tech/tronity/vehicles/668aedd2d9d3250012584ce4/last_record
[tronity] TRACE 2024/09/04 22:04:45 {"odometer":35204,"range":99.7,"level":22.45,"charging":"Disconnected","plugged":false,"chargerPower":0,"chargeRemainingTime":null,"latitude":48.124067,"longitude":7.844362,"status":null,"timestamp":1725480058000,"lastUpdate":1725480058000}
[tronity] TRACE 2024/09/04 22:05:45 GET https://api.tronity.tech/tronity/vehicles/668aedd2d9d3250012584ce4/last_record
[tronity] TRACE 2024/09/04 22:05:45 {"odometer":35204,"range":99.7,"level":22.45,"charging":"Disconnected","plugged":false,"chargerPower":0,"chargeRemainingTime":null,"latitude":48.124067,"longitude":7.844362,"status":null,"timestamp":1725480058000,"lastUpdate":1725480058000}
[tronity] TRACE 2024/09/04 22:06:45 GET https://api.tronity.tech/tronity/vehicles/668aedd2d9d3250012584ce4/last_record
[tronity] TRACE 2024/09/04 22:06:45 {"odometer":35204,"range":99.7,"level":22.45,"charging":"Disconnected","plugged":false,"chargerPower":0,"chargeRemainingTime":null,"latitude":48.124067,"longitude":7.844362,"status":null,"timestamp":1725480058000,"lastUpdate":1725480058000}
[tronity] TRACE 2024/09/04 22:07:45 GET https://api.tronity.tech/tronity/vehicles/668aedd2d9d3250012584ce4/last_record
[tronity] TRACE 2024/09/04 22:07:45 {"odometer":35204,"range":99.7,"level":22.45,"charging":"Disconnected","plugged":false,"chargerPower":0,"chargeRemainingTime":null,"latitude":48.124067,"longitude":7.844362,"status":null,"timestamp":1725480058000,"lastUpdate":1725480058000}
[tronity] TRACE 2024/09/04 22:08:45 GET https://api.tronity.tech/tronity/vehicles/668aedd2d9d3250012584ce4/last_record
[tronity] TRACE 2024/09/04 22:08:45 {"odometer":35204,"range":99.7,"level":22.45,"charging":"Disconnected","plugged":false,"chargerPower":0,"chargeRemainingTime":null,"latitude":48.124067,"longitude":7.844362,"status":null,"timestamp":1725480058000,"lastUpdate":1725480058000}
[tronity] TRACE 2024/09/04 22:09:45 GET https://api.tronity.tech/tronity/vehicles/668aedd2d9d3250012584ce4/last_record
[tronity] TRACE 2024/09/04 22:09:45 {"odometer":35204,"range":99.7,"level":22.45,"charging":"Disconnected","plugged":false,"chargerPower":0,"chargeRemainingTime":null,"latitude":48.124067,"longitude":7.844362,"status":null,"timestamp":1725480058000,"lastUpdate":1725480058000}

No query for over 30 minutes

[tronity] TRACE 2024/09/04 22:47:05 GET https://api.tronity.tech/tronity/vehicles/668aedd2d9d3250012584ce4/last_record
[tronity] TRACE 2024/09/04 22:47:05 {"odometer":35204,"range":97.9,"level":22.014,"charging":"Disconnected","plugged":false,"chargerPower":0,"chargeRemainingTime":null,"latitude":48.124067,"longitude":7.844362,"status":null,"timestamp":1725482617000,"lastUpdate":1725482617000}

Selected Tesla in evcc manually - results in instant query

[tronity] TRACE 2024/09/04 23:01:05 POST https://api.tronity.tech/authentication
[tronity] TRACE 2024/09/04 23:01:05 GET https://api.tronity.tech/tronity/vehicles/668aedd2d9d3250012584ce4/last_record
[tronity] TRACE 2024/09/04 23:01:06 {"odometer":35204,"range":112.1,"level":25.191,"charging":"Charging","plugged":true,"chargerPower":11,"chargeRemainingTime":null,"latitude":48.124067,"longitude":7.844362,"status":null,"timestamp":1725483621000,"lastUpdate":1725483621000}
```

### What type of operating system are you running?

Docker container

### Version

0.130.7

sebnaf, 2024-09-05:
Now, at 13:44 the latest query was:

```
[tronity] TRACE 2024/09/05 11:42:37 GET https://api.tronity.tech/tronity/vehicles/668aedd2d9d3250012584ce4/last_record
[tronity] TRACE 2024/09/05 11:42:37 {"odometer":35204,"range":113.7,"level":25.56,"charging":"Disconnected","plugged":false,"chargerPower":0,"chargeRemainingTime":null,"latitude":48.124067,"longitude":7.844362,"status":null,"timestamp":1725527753000,"lastUpdate":1725527753000}
```

andig, 2024-09-05:
> Also, looking at https://docs.evcc.io/docs/reference/configuration/vehicles there is cache mentioned, which i want to use with cache: 5m with Tronity (due to the instant connection of Tesla <-> Tronity).

If cache is not exposed in the template, you'll need to use the plain config (`type: tronity`). See tronity.go for relevant parameters.

closed 2024-09-05 by andig