evcc-io/evcc #15908: UI: Warte auf autorisierung wird angezeigt, wenn kein Auto angesteckt ist

TobiasHuber1980, 2024-09-05:
### Describe the bug

1) wenn kein Auto verbunden ist, wird trotzdem "warte auf autorisierung" im UI angezeigt [#15873 Fix ChargerStatusReason not resetting](https://github.com/evcc-io/evcc/pull/15873)
2) nebenbei aufgefallen: config mit `evcc dump --cfg >~/config.txt` erstellt, in der ausgabe steht: 
```
Fehlermeldung: failed configuring loadpoints: failed configuring loadpoint: not found: loadpoints
```

### Steps to reproduce

1. evcc starten
2. **kein** Auto anstecken / nach dem abstecken
3. screenshot
[image]

...

### Configuration details

```yaml
[main  ] INFO 2024/09/05 12:15:20 evcc 0.130.7 (5489a15b)
[main  ] INFO 2024/09/05 12:15:20 using config file: /etc/evcc.yaml
[db    ] INFO 2024/09/05 12:15:20 using sqlite database: /home/pi/.evcc/evcc.db

Fehlermeldung:

    failed configuring loadpoints: failed configuring loadpoint: not found: loadpoints

Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug

# unique installation id
plant: *****

#interval: 30s # control cycle interval

sponsortoken: *****

##################################

site:
  - title: name
    meters:
      grid: grid
        
##################################     

meters:
  - name: grid
    type: template
    template: plexlog
    usage: grid
    host: 192.168.1.26
    port: 503

##################################

chargers:
- type: template
  template: keba-modbus 
  host: 192.168.1.21
  name: wallbox1
  modbus: tcpip
[... 169 lines cut ...]
[site ] DEBUG 2024/09/05 11:42:22 pv power: 70000W
[site ] DEBUG 2024/09/05 11:42:22 grid meter: -50346W
[site ] DEBUG 2024/09/05 11:42:22 site power: -50346W
[site ] DEBUG 2024/09/05 11:42:37 ----
[lp-1 ] DEBUG 2024/09/05 11:42:37 charge power: 0W
[lp-1 ] DEBUG 2024/09/05 11:42:37 charge currents: [0 0 0]A
[circuit-loadpoints] DEBUG 2024/09/05 11:42:37 power: 0W
[circuit-loadpoints] DEBUG 2024/09/05 11:42:37 current: 0A
[circuit-main] DEBUG 2024/09/05 11:42:37 power: -52431W
[circuit-main] DEBUG 2024/09/05 11:42:37 current: 0A
[site ] DEBUG 2024/09/05 11:42:37 pv power: 70050W
[site ] DEBUG 2024/09/05 11:42:37 grid meter: -52431W
[site ] DEBUG 2024/09/05 11:42:37 site power: -52431W
[site ] DEBUG 2024/09/05 11:42:52 ----
[lp-1 ] DEBUG 2024/09/05 11:42:52 charge power: 0W
[lp-1 ] DEBUG 2024/09/05 11:42:52 charge currents: [0 0 0]A
```

### What type of operating system are you running?

Linux

### Version

0.130.7 (5489a15b)

premultiply, 2024-09-05:
Da bräuchte es bitte erstmal ein Tracelog mit der Modbus-Kommunikation von dem betroffenen Charger in dem jeweiligen Zustand  z. B. auch einzeln mittels `evcc charger --log trace,db:error` sofern möglich.
- abgesteckt
- angeschlossen, wartend auf RFID-Karte
- angeschlossen, erfolgreich Authentififiziert
- ladend

Ref https://github.com/evcc-io/evcc/pull/15399/files#r1719649903

closed 2024-09-05 by andig

TobiasHuber1980, 2024-09-05:
Logs liefere ich am Montag, dann bin ich wieder vor Ort.

andig, 2024-09-05:
Danke- das wäre weiterhin hilfreich. Aktuell ist nur Pflaster aufgeklebt.

TobiasHuber1980, 2024-09-07:
[status_b_nach_authentifizierung.txt](https://github.com/user-attachments/files/16917609/status_b_nach_authentifizierung.txt)
[status_a_nach_authentifizierung.txt](https://github.com/user-attachments/files/16917610/status_a_nach_authentifizierung.txt)
[status_b_vor_authorisierung.txt](https://github.com/user-attachments/files/16917634/status_b_vor_authorisierung.txt)

hier die restlichen logs von status a/b mit vor/nach authorisireung.

premultiply, 2024-09-09:
Danke.
Was bedeutet "Status A nach Authentifizierung"? Wie sieht da das Szenario aus?
Ist das ein Interpreationsfehler seitens evcc oder was meinst du damit?

na-ma-ste, 2025-01-29:
Hallo Zusammen,

ich bin ziemlich neu bei evcc. Ich hab auch das Problem von oben mit dem folgenden Szenario:

a) Ich fahr mit dem Auto in die Garage
b) RFID Karte an die Keba, Keba Akzeptiert die Karte
c) Kabel ans Auto Stecken
d) Mit Evcc den Ladevorgang starten

Das war bisher immer mein Ablauf. Evtl. muss ich den "ändern" in

a) Ich fahr mit dem Auto in die Garage
b) Kabel ans Auto Stecken
c) RFID Karte an die Keba, Keba Akzeptiert die Karte
d) Mit Evcc den Ladevorgang starten

Und der Würde das Problem dann schon lösen?

na-ma-ste, 2025-01-29:
Ok, auch die neue Abfolge schafft keine Abhilfe

na-ma-ste, 2025-02-05:
Hallo @TobiasHuber1980 , hat sich das Problem bei dir gelöst?

TobiasHuber1980, 2025-02-05:
> Hallo @TobiasHuber1980 , hat sich das Problem bei dir gelöst?

Ich habe kein Problem mit den Keba wallboxen. Welcomecharge = true, Intervall 12 Sekunden. 
Erst Karte, kurz warten, dann Auto anstecken und wieder kurz warten bis das Relais in der Keba klickt.

na-ma-ste, 2025-02-05:
Danke. Welcomecharge = true hab ich jetzt aktiviert und teste. Ich meine, als ich evcc aufgesetzt hatte, lief es auch ohne welcomecharge richtig. Danke für die Rückmeldung