evcc-io/evcc #15915: Sungrow SHxxT series: Battery charging/discharging not available

wapman1979, 2024-09-05:
### Describe the bug

The new Sungrow SHxxT series has no more direct LAN Port at the inverter.
But WiNet-S2 has now modbus functionality.
evcc works in this environment (SH15T & SBR128) "out of the box" with the exception that the load on "Battery charging" and "Battery dischargin" is not shown.

Several other SH15T, SH20T or SH25T users are reporting similarly issues in Forums and here on github. 
I run evcc on a Home Assistent system and had the same issue with the mkaiser integertaion (as other users).
For mkaiser sungrow integration, I now could solve the issue by changing the yaml entries for:
- Battery charging
- Battery discharging

As described here: 
https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/discussions/310#discussioncomment-10562073

Sungrow also changed the "data_type" for "Battery current" to int16 insted of uint16.
Even in the available Sungrow datasheet this is still mentioned as uint16:
https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/commit/4e6b3bc41ebe14eb67a05114816bc19a2758e9f5

How can we implement these changes to evcc to see if this is also solving the issue on evcc?

### Steps to reproduce

The Issue "Battery charging/discharging not available" occurs continuously.
This was reproduced with different configuration and connectivity.
- Via LAN/WLAN
- Via Waveshare rs485 LAN Adapter
- Modbus Proxy on/of 

For the mkaiser sungrow integration it was possible to display the charging discharging load reproducible with some adoptions.

Reproducible in by now installed evcc versions.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: homeassistant.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

interval: 30s # control cycle interval

meters:
- name: my_grid1
  type: template
  template: sungrow-hybrid 
  usage: grid
  modbus: tcpip
  id: 1
  host: 192.168.39.33 # Hostname
  port: 502 # Port
#  timeout: 1m # optional for modbus: tcpip?

- name: my_pv1
  type: template
  template: sungrow-hybrid 
  usage: pv
  modbus: tcpip
  id: 1
  host: 192.168.39.33 # Hostname
  port: 502 # Port
#  timeout: 1m # optional for modbus: tcpip

- name: my_battery1
  type: template
  template: sungrow-hybrid 
  usage: battery
  modbus: tcpip
  id: 1
  host: 192.168.39.33 # Hostname
  port: 502 # Port
#  timeout: 1m # optional for modbus: tcpip
  capacity: 12.8 # Akkukapazität in kWh (optional)

chargers:
- type: template
  template: go-e-v3 
  host: 192.168.39.81  
  name: Haid9GoE

vehicles:
- type: template
  template: vw 
  title: ID.5 Michael  
  icon: car  
  user: login.mh@hoepfl.cc  
  password: bivjuj-xiqNig-dunmi6  
  vin: WVGZZZE2ZPP515720  
  capacity: 77  
  phases: 3  
  cache: 1m  
  mode: minpv  
  minCurrent: 6  
  maxCurrent: 16  
  timeout: 10s  
  name: ev4

loadpoints:
- title: Haid 9 Carport
  charger: Haid9GoE
  mode: pv

site:
  title: Haid am Bühl 9, Schorndorf
  meters:
    grid: my_grid1
    pv:
    - my_pv1
    battery:
    - my_battery1
  residualPower: 100 # additional household usage margin
#  maxGridSupplyWhileBatteryCharging: 0 # ignore battery charging if AC consumption is above this value

# tariffs are the fixed or variable tariffs
tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    # either static grid price (or price zones)
    type: fixed
    price: 0.34 # EUR/kWh
#    zones:
#     - days: Mon-Fri
#        hours: 2-5
#        price: 0.2 # EUR/kWh
#      - days: Sat,Sun
#        price: 0.15 # EUR/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.08 # EUR/kWh

# mqtt message broker
#mqtt:
  # broker: localhost:1883
  # topic: evcc # root topic for publishing, set empty to disable
  # user:
  # password:
```

### Log details

```text
[site  ] DEBUG 2024/09/05 23:26:04 ----
[lp-1  ] DEBUG 2024/09/05 23:26:04 charge power: 0W
[lp-1  ] DEBUG 2024/09/05 23:26:04 charge currents: [0 0 0]A
[site  ] DEBUG 2024/09/05 23:26:04 pv power: 0W
[site  ] DEBUG 2024/09/05 23:26:04 battery soc: 87%
[site  ] DEBUG 2024/09/05 23:26:04 battery power: 0W
[site  ] DEBUG 2024/09/05 23:26:04 grid meter: -0W
[site  ] DEBUG 2024/09/05 23:26:04 grid currents: [-0.492 0.0175 0.0836]A
[site  ] DEBUG 2024/09/05 23:26:04 site power: 100W
[lp-1  ] DEBUG 2024/09/05 23:26:04 !! session: chargeRater.chargedEnergy=3.8 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/09/05 23:26:04 charge total import: 7115.929kWh
[lp-1  ] DEBUG 2024/09/05 23:26:04 charger status: B
[lp-1  ] DEBUG 2024/09/05 23:26:04 pv charge current: min 6A > 0A (100W @ 1p, battery: false)
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.130.7

andig, 2024-09-06:
It's unclear what you're asking. Does SHT have different registers? Which? Happy to reopen when that information is available. Thank you!

closed 2024-09-06 by andig

wapman1979, 2024-09-06:
Thx for your quick reply, Andi.
Where can I check if evcc uses  int16 for "Battery current"?
This is regsiter 13020.

andig, 2024-09-06:
See https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/sungrow-hybrid.yaml

wapman1979, 2024-09-06:
ok. that is the case...

- source: modbus
      {{- include "modbus" . | indent 4 }}
      register:
        address: 13021 # Battery power
        type: input
        decode: int16

will try to figure out if the calculation is a difrent one and come back.

DarcDan, 2024-09-13:
@wapman1979 do you have any new information?

leonidnf, 2024-10-20:
Hi @wapman1979 ,
I have a similar setup SungrowSH15T with an SBR128 battery and the same issue with the battery charging/discharging status. I created a ticket and went further with debugging this issue. It seems to be that sungrow-hybrid.yaml template doesn't read the correct values for the 13000 address.
    - source: calc
      add:
      - source: modbus
        {{- include "modbus" . | indent 6 }}
        register:
          type: input
          address: 13000 # Battery running state
          decode: bool16
          bitmask: 1 # Charging
        scale: -1
      - source: modbus
        {{- include "modbus" . | indent 6 }}
        register:
          type: input
          address: 13000 # Battery running state
          decode: bool16
          bitmask: 2 # Discharging

A little bit more details about is in the issue I created here:
https://github.com/evcc-io/evcc/issues/16704

Have you solved this issue?

leonidnf, 2024-10-21:
A temporary solution is mentioned here:
https://github.com/evcc-io/evcc/issues/16704