evcc-io/evcc #15916: Unavailable Grünstrom API preventing EVCC startup

pandamouse84, 2024-09-05:
### Describe the bug

EVCC is not starting when configured Grünsteom API can‘t be reached. 

### Steps to reproduce

1. Disable internet access for EVCC host
2. Try to start EVCC

### Configuration details

```yaml
log: info
levels:
  cache: error

# unique installation id
plant: xxxx

interval: 30s # control cycle interval

meters:
  - type: template
    template: varta
    usage: grid
    host: 172.20.20.1
    port: 502
    name: gridVarta

chargers:
  - name: wallboxElli
    type: template
    template: ellipro
    ip: 172.20.20.3
    ski: xxxx

loadpoints:
  - title: Carport
    charger: wallboxElli

site:
  title: Mein Zuhause
  meters:
    grid: gridVarta
  residualPower: 100
```

### Log details

```text
[main  ] INFO 2024/09/06 01:07:20 listening at :7070
[gsi   ] ERROR 2024/09/06 01:08:20 Get "https://api.corrently.io/v2.0/gsi/prediction?zip=***": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[main  ] FATAL 2024/09/06 01:08:20 cannot create tariff type 'grünstromindex': Get "https://api.corrently.io/v2.0/gsi/prediction?zip=xxxxx": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[main  ] FATAL 2024/09/06 01:08:20 will attempt restart in: 15m0s
[eebus ] ERROR 2024/09/06 01:10:20 websocket server error: http: Server closed
Using config file: /config/custom_yaml/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/custom_yaml/evcc.yaml'
[main  ] INFO 2024/09/06 01:10:23 evcc 0.130.7
[main  ] INFO 2024/09/06 01:10:23 using config file: /config/custom_yaml/evcc.yaml
[db    ] INFO 2024/09/06 01:10:23 using sqlite database: /data/evcc.db
[eebus ] INFO 2024/09/06 01:10:23 Local SKI:  xxxx
[main  ] INFO 2024/09/06 01:10:23 listening at :7070
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.130.7

andig, 2024-09-06:
Works as expected. Devices need be available at startup. Only alternative: ignore tariffs. At a later stage we will try to restart services while evcc is running.

closed 2024-09-06 by andig

pandamouse84, 2024-09-06:
@andig: Kannst du mir dann einen Tipp geben wie ich EVCC aktuell gestartet bekomme?, scheinbar ist die Grünstrom API seit gestern offline. Da ich die config für Grünstrom schon in die DB migrieren habe, kann ich den Teil auch nicht mehr in der yaml aus-kommentieren ;D P.s. ich nutze das HA Addon.

andig, 2024-09-06:
Das ist doof 🤪. Mit sqlite in der config Tabelle löschen?

/cc @naltatis

naltatis, 2024-09-06:
Aber du solltest auch in Fehlermodus weiterhin Zugriff auf die Config UI haben. Da kannst du den Tarif löschen.

pandamouse84, 2024-09-06:
@naltatis: Leider nicht das UI wird nicht geladen.

Mir ist erst aufgefallen das der Grünstromindex ein Problem hatte nachdem ich Home Assistent wegen eines anderen Updates neu gestartet hatte.

naltatis, 2024-09-06:
Ok, das sollte nicht so sein. I'll check.

pandamouse84, 2024-09-09:
Hi @naltatis: Ist das Problem behoben oder müsste der Bug wieder geöffnet werden?