evcc-io/evcc #15926: Netzladen mit Sonnenbatterie funktioniert nicht

mabasger, 2024-09-06:
### Describe the bug

Netzladen ist konfiguriert. Das Event tritt ein. Es ist in UI und Logs ersichtlich, dass EVCC etwas versucht. Die Batterie reagiert jedoch nicht, es findet kein Laden statt.

### Steps to reproduce

1. Netzladen ist konfiguriert
2. Preisgrenze tritt ein
3. UI zeigt "Batterie laden günstige Netzenergie"
----> Logik durch EVCC hat offensichtlich gegriffen
4. "set batterie mode charge" in logs ersichtlich
5. Sonnenbatterie arbeitet aber völlig normal weiter

### Configuration details

```yaml
meters:
- type: template
  template: sonnenbatterie 
  usage: grid  
  host: 192.168.1.80  
  port: 8080  
  name: grid1
- type: template
  template: sonnenbatterie 
  usage: pv  
  host: 192.168.1.80  
  port: 8080  
  name: pv2
- type: template
  template: sonnenbatterie 
  usage: battery  
  host: 192.168.1.80  
  port: 8080  
  name: battery3
  token: ed8966d3-fd2c-467e-b529-undnocheinpaarzahlen
  capacity: 6
```

### Log details

```text
Sep 06 16:11:16 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:11:16 ----
Sep 06 16:11:16 raspberrypi evcc[20471]: [lp-1  ] DEBUG 2024/09/06 16:11:16 charge power: 0W
Sep 06 16:11:16 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:11:16 set battery mode: charge
Sep 06 16:11:16 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:11:16 pv power: 1968W
Sep 06 16:11:16 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:11:16 battery soc: 66%
Sep 06 16:11:16 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:11:16 battery power: -1569W
Sep 06 16:11:16 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:11:16 grid meter: -62W
Sep 06 16:11:16 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:11:16 battery has priority at soc 66% (< 80%)
Sep 06 16:11:16 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:11:16 site power: -62W
Sep 06 16:11:16 raspberrypi evcc[20471]: [lp-1  ] DEBUG 2024/09/06 16:11:16 charger status: A
Sep 06 16:11:46 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:11:46 ----
Sep 06 16:11:46 raspberrypi evcc[20471]: [lp-1  ] DEBUG 2024/09/06 16:11:46 charge power: 0W
Sep 06 16:11:46 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:11:46 pv power: 2073W
Sep 06 16:11:46 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:11:46 battery soc: 66%
Sep 06 16:11:46 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:11:46 battery power: -1668W
Sep 06 16:11:46 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:11:46 grid meter: 33W
Sep 06 16:11:46 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:11:46 battery has priority at soc 66% (< 80%)
Sep 06 16:11:46 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:11:46 site power: 33W
Sep 06 16:11:46 raspberrypi evcc[20471]: [lp-1  ] DEBUG 2024/09/06 16:11:46 charger status: A
Sep 06 16:11:56 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:11:56 set grid charge limit: 0.1
Sep 06 16:12:16 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:12:16 ----
Sep 06 16:12:16 raspberrypi evcc[20471]: [lp-1  ] DEBUG 2024/09/06 16:12:16 charge power: 0W
Sep 06 16:12:16 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:12:16 set battery mode: normal
Sep 06 16:12:16 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:12:16 pv power: 2149W
Sep 06 16:12:16 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:12:16 battery soc: 67%
Sep 06 16:12:16 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:12:16 battery power: -1686W
Sep 06 16:12:16 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:12:16 grid meter: -71W
Sep 06 16:12:16 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:12:16 battery has priority at soc 67% (< 80%)
Sep 06 16:12:16 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:12:16 site power: -71W
Sep 06 16:12:16 raspberrypi evcc[20471]: [lp-1  ] DEBUG 2024/09/06 16:12:16 charger status: A
Sep 06 16:12:29 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:12:29 set grid charge limit: 0.4
Sep 06 16:12:46 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:12:46 ----
Sep 06 16:12:46 raspberrypi evcc[20471]: [lp-1  ] DEBUG 2024/09/06 16:12:46 charge power: 0W
Sep 06 16:12:46 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:12:46 set battery mode: charge
Sep 06 16:12:46 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:12:46 pv power: 1937W
Sep 06 16:12:46 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:12:46 battery soc: 67%
Sep 06 16:12:46 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:12:46 battery power: -1562W
Sep 06 16:12:46 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:12:46 grid meter: -13W
Sep 06 16:12:46 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:12:46 battery has priority at soc 67% (< 80%)
Sep 06 16:12:46 raspberrypi evcc[20471]: [site  ] DEBUG 2024/09/06 16:12:46 site power: -13W
Sep 06 16:12:46 raspberrypi evcc[20471]: [lp-1  ] DEBUG 2024/09/06 16:12:46 charger status: A
```

### What type of operating system are you running?

Linux

closed 2024-09-06 by andig

mabasger, 2024-09-07:
Danke fürs nachgucken! Und unverschämterweise würde ich auch gleich noch mehr anfragen: eine Codeänderung. Copy & paste von 2 Zeilen. Das wäre sehr freundlich :-)

Mir ist unklar, warum Case 3 (Charge) nicht umgesetzt wurde, da das nötige Kommando in case 2 (hold) bereits vorhanden ist. Wahrscheinlich war das "damals" einfach unwichtig. Case 3 müsste den gleichen API-Call (/api/v2/setpoint/charge/) wie Case 2 enthalten. Nur anstelle von /charge/0 müsste /charge/3000 stehen, um eine Ladung mit den 3kW der Sonnenbatterie auszulösen. Siehe Code der Sonnenbatterie.yaml:

   - source: http
          **uri: http://{{ .host }}/api/v2/setpoint/charge/0**
          insecure: true
          **method: POST**
          headers:
          - content-type: application/json
          - Auth-Token: {{ .token }}
    - case: 3 # charge
      set:
        source: http
        **uri: http://{{ .host }}/api/v2/configurations** hier müsste uri: http://{{ .host }}/api/v2/setpoint/charge/3000 stehen
        insecure: true
        **method: PUT** das zu POST
        headers:
        - content-type: application/json
        - Auth-Token: {{ .token }}
        body: '{"EM_OperatingMode":"2"}'  # self consumption

Dokumentation von Sonnen:

Charge/Discharge battery
The discharging power of a storage system can be controlled by setting a setpoint in watts. The corresponding value of the setpoint is kept until the battery receives a new charging or discharging value. If VPP is active, the request will be rejected.
direction (required) string charge\discharge 
Setpoint direction: must be "charge" or "discharge".
watts (required) string <int32>
Amount of charge or discharge, in watts. Must be a positive number less than 1.000.000.000 (1000 MW).

Besonderer Return 403 If VPP is enabled for this battery, it's not possible to charge or discharge using this endpoint.

andig, 2024-09-07:
Gerne PR

rivengh, 2024-09-07:
> Gerne PR /cc @rivengh

Alles klar, mache ich gerne.