evcc-io/evcc #15934: Elli using EEBus logging error "charge currents: not available" after upgrade to version 0.130.7 from 0.130.6

ilion1, 2024-09-07:
### Describe the bug

During charging charge currents are received only from time to time. In between error message "charge currents: not available" is logged.
Full logs are stored as attachment because they cannot be uploaded in the form due to size:
eebus6.log: log using version 0.130.6 (charging without errors)
eebus7.log: log using version 0.130.7 (charging with errors)
[evcc--logs.zip](https://github.com/user-attachments/files/16917018/evcc--logs.zip)

### Steps to reproduce

1. run version 0.130.7
2. start charging
3. errors are logged
4. stop and run version 0.130.6
5. no errors are logged

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error
#  eebus: trace

# unique installation id
plant: xxxx

interval: 10s # control cycle interval

meters:
- type: template
  template: sonnenbatterie 
  usage: grid  
  host: 192.168.189.38  
  port: 8080  
  name: grid1
- type: template
  template: sonnenbatterie 
  usage: pv  
  host: 192.168.189.38  
  port: 8080  
  name: pv2
- type: template
  template: sonnenbatterie 
  usage: battery  
  host: 192.168.189.38  
  port: 8080  
  name: battery3

chargers:
- name: wallbox5
  type: eebus 
  ski: D9A2 03D9 E918 F950 EC97 E593 7251 E179 79A3 905E  
  meter: true
  chargedEnergy: false
  ip: 192.168.189.34

loadpoints:
- title: Elli Wallbox
  charger: wallbox5
  mode: pv

site:
  title: Zu Hause
  residualPower: 100
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

eebus:
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----
          MHcCAQEEIIBSMKp7Kl3HS98/cf+kvdAGyZx8VNQ+AASdw6/Ev+tMoAoGCCqGSM49
          28l9YxSYO3tXp3PgFFMa50KPrSQhD1ZKHQ==
          -----END EC PRIVATE KEY-----
      public: |
          -----BEGIN CERTIFICATE-----
          MIIBvDCCAWOgAwIBAgIRAzkyyrnQ7mwJT8P2XxF07ZUwCgYIKoZIzj0EAwIwPjEL
          W8tXWK9rBQ5mcO6GLC2qrA==
          -----END CERTIFICATE-----
```

### Log details

```text
full logs as attachments above!

[lp-1  ] DEBUG 2024/09/07 09:26:02 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/09/07 09:26:12 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/09/07 09:26:22 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/09/07 09:26:32 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/09/07 09:26:42 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/09/07 09:26:52 charge currents: [0 0 0]A
[lp-1  ] ERROR 2024/09/07 09:27:02 charge currents: not available
[lp-1  ] DEBUG 2024/09/07 09:27:12 charge currents: [7.23 7.2 7]A
[lp-1  ] ERROR 2024/09/07 09:27:22 charge currents: not available
[lp-1  ] ERROR 2024/09/07 09:27:32 charge currents: not available
[lp-1  ] ERROR 2024/09/07 09:27:42 charge currents: not available
[lp-1  ] ERROR 2024/09/07 09:27:52 charge currents: not available
[lp-1  ] ERROR 2024/09/07 09:28:02 charge currents: not available
[lp-1  ] ERROR 2024/09/07 09:28:12 charge currents: not available
[lp-1  ] ERROR 2024/09/07 09:28:22 charge currents: not available
[lp-1  ] ERROR 2024/09/07 09:28:32 charge currents: not available
[lp-1  ] ERROR 2024/09/07 09:28:42 charge currents: not available
[lp-1  ] ERROR 2024/09/07 09:28:52 charge currents: not available
[lp-1  ] ERROR 2024/09/07 09:29:02 charge currents: not available
[lp-1  ] ERROR 2024/09/07 09:29:12 charge currents: not available
[lp-1  ] ERROR 2024/09/07 09:29:22 charge currents: not available
[lp-1  ] ERROR 2024/09/07 09:29:32 charge currents: not available
[lp-1  ] ERROR 2024/09/07 09:29:43 charge currents: not available
[lp-1  ] ERROR 2024/09/07 09:29:53 charge currents: not available
[lp-1  ] ERROR 2024/09/07 09:30:02 charge currents: not available
[lp-1  ] ERROR 2024/09/07 09:30:04 charge currents: not available
[lp-1  ] DEBUG 2024/09/07 09:30:12 charge currents: [0.05 0 0]A
[lp-1  ] DEBUG 2024/09/07 09:30:22 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/09/07 09:30:22 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/09/07 09:30:32 charge currents: [0 0 0]A
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.130.7

andig, 2024-09-07:
Elli or Elli Pro?

ilion1, 2024-09-07:
Elli Pro with meter. I am testing updates from time to time but I am still running 0.128.4 using the ellipro template as the most stable version.

StefanSchoof, 2024-09-07:
I think https://github.com/evcc-io/evcc/pull/15778 makes the error existing also in previous version visible. So before old measurements from elli where used. Now evcc is giving an error when the measurement no longer are sent.

andig, 2024-09-07:
@DerAndereAndi ich habe das Gefühl, dass hier noch etwas nicht stimmt:

```
[eebus ] DEBUG 2024/09/07 09:27:11 ski d9a203d9e918f950ec97e5937251e17979a3905e event cem-evcem-DataUpdateCurrentPerPhase
[eebus ] TRACE 2024/09/07 09:27:11 Recv: d9a203d9e918f950ec97e5937251e17979a3905e {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_i:47859_Elli-Wallbox-2125A1G53O"},{"entity":[1,1]},{"feature":11}]},{"addressDestination":[{"device":"d:_n:EVCC_HEMS-EVCC-3639656234373032"},{"entity":[1]},{"feature":6}]},{"msgCounter":55283},{"cmdClassifier":"notify"}]},{"payload":[{"cmd":[[{"function":"measurementListData"},{"filter":[[{"cmdControl":[{"partial":[]}]}]]},{"measurementListData":[{"measurementData":[[{"measurementId":0},{"valueType":"value"},{"value":[{"number":723},{"scale":-2}]}],[{"measurementId":1},{"valueType":"value"},{"value":[{"number":72},{"scale":-1}]}],[{"measurementId":2},{"valueType":"value"},{"value":[{"number":7},{"scale":0}]}]]}]}]]}]}]}}]}
[eebus ] DEBUG 2024/09/07 09:27:11 Recv: d:_i:47859_Elli-Wallbox-2125A1G53O:Measurement to Measurement notify 55283 MeasurementListData
[eebus ] DEBUG 2024/09/07 09:27:11 ski d9a203d9e918f950ec97e5937251e17979a3905e event cem-evcem-DataUpdateCurrentPerPhase
[site  ] DEBUG 2024/09/07 09:27:12 ----
[lp-1  ] DEBUG 2024/09/07 09:27:12 charge power: 4929W
[lp-1  ] DEBUG 2024/09/07 09:27:12 charge currents: [7.23 7.2 7]A
...
[eebus ] DEBUG 2024/09/07 09:27:22 ski d9a203d9e918f950ec97e5937251e17979a3905e event cem-evcem-DataUpdateCurrentPerPhase
[eebus ] TRACE 2024/09/07 09:27:22 Recv: d9a203d9e918f950ec97e5937251e17979a3905e {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_i:47859_Elli-Wallbox-2125A1G53O"},{"entity":[1,1]},{"feature":11}]},{"addressDestination":[{"device":"d:_n:EVCC_HEMS-EVCC-3639656234373032"},{"entity":[1]},{"feature":6}]},{"msgCounter":55310},{"cmdClassifier":"notify"}]},{"payload":[{"cmd":[[{"function":"measurementListData"},{"filter":[[{"cmdControl":[{"partial":[]}]}]]},{"measurementListData":[{"measurementData":[[{"measurementId":0},{"valueType":"value"},{"value":[{"number":1301},{"scale":-2}]}],[{"measurementId":1},{"valueType":"value"},{"value":[{"number":1297},{"scale":-2}]}],[{"measurementId":2},{"valueType":"value"},{"value":[{"number":1301},{"scale":-2}]}]]}]}]]}]}]}}]}
[eebus ] DEBUG 2024/09/07 09:27:22 Recv: d:_i:47859_Elli-Wallbox-2125A1G53O:Measurement to Measurement notify 55310 MeasurementListData
[eebus ] DEBUG 2024/09/07 09:27:22 ski d9a203d9e918f950ec97e5937251e17979a3905e event cem-evcem-DataUpdateCurrentPerPhase
[lp-1  ] ERROR 2024/09/07 09:27:22 charge currents: not available
```

Die Elli schickt in beiden Fällen Updates soweit ich das verstehe- das letzte verarbeiten wir anscheinend aber nicht mehr?

DerAndereAndi, 2024-09-07:
Ja, bin schon seit gestern dran. Braucht Änderungen in EEBUS aber auch in evcc. Bin gerade am Testen, kann aber erst morgen einen PR machen.

andig, 2024-09-07:
Klasse, danke Dir!

sebnaf, 2024-09-07:
@andig Ich sehe ein ähnliches Verhalten nach dem Update 0.130.4 -> 0.130.7 ohne Konfigänderung mit meiner go-e:

`Elektronenpumpe: max charge current 5A: unexpected status: 500 (Internal Server Error)`

closed 2024-09-08 by andig