evcc-io/evcc #15936: UI: Warte auf Autorisierung wird angezeigt, obwohl Auto bereits lädt

An-Wi, 2024-09-07:
### Describe the bug

...und Autorisierung erfolgreich durchgeführt wurde.
(Wallbox: Dadapower)

[image]

### Steps to reproduce

1. Ladeplanung "Ankunft" gesetzt
2. SOC des EVs ist unter "Min Ladung %"
3. Fahrzeug an Wallbox anstecken
4. Autorisieren
5. Ladevorgang startet

### Configuration details

```yaml
[...]

chargers:
  - name: wallbox1
    type: template
    template: dadapower
    modbus: tcpip
    id: 1
    host: 192.168.115.11
    port: 502

[...]

loadpoints:
  - title: Einfahrt # display name for UI
    charger: wallbox1 # charger
    mode: pv   
    vehicle: ev1 
    phases: 0   

    priority: 0 
    soc:
      poll:
        mode: always    #charging
        interval: 60m
      estimate: true # set false to disable interpolating between api updates (not recommended)
    enable: # pv mode enable behavior
      delay: 2m         #1m # threshold must be exceeded for this long
      threshold: -444   
    disable: # pv mode disable behavior
      delay: 5m         # 3m # threshold must be exceeded for this long
      threshold: 999    
    guardDuration: 10m # 5m # switch charger contactor not more often than this (default 5m)
```

### Log details

```text
[site  ] DEBUG 2024/09/05 19:12:27 ----
[lp-1  ] DEBUG 2024/09/05 19:12:27 charge power: 0W
[lp-1  ] DEBUG 2024/09/05 19:12:27 charge currents: [0 0 0]A
[script] DEBUG 2024/09/05 19:12:27 /fupcService/pvCurrentPower.bat: 94
[site  ] DEBUG 2024/09/05 19:12:27 pv power: 94W
[site  ] DEBUG 2024/09/05 19:12:27 grid meter: 312W
[site  ] DEBUG 2024/09/05 19:12:27 grid currents: [0.432 0.567 1.48]A
[site  ] DEBUG 2024/09/05 19:12:27 site power: 312W
[lp-1  ] DEBUG 2024/09/05 19:12:27 charge total import: 6564.734kWh
[lp-1  ] DEBUG 2024/09/05 19:12:27 charger status: A
[lp-1  ] DEBUG 2024/09/05 19:12:27 next soc poll remaining time: 51m59s
[site  ] DEBUG 2024/09/05 19:12:47 ----
[lp-1  ] DEBUG 2024/09/05 19:12:47 charge power: 0W
[lp-1  ] DEBUG 2024/09/05 19:12:47 charge currents: [0 0 0]A
[script] DEBUG 2024/09/05 19:12:47 /fupcService/pvCurrentPower.bat: 84
[site  ] DEBUG 2024/09/05 19:12:47 pv power: 84W
[site  ] DEBUG 2024/09/05 19:12:47 grid meter: 310W
[site  ] DEBUG 2024/09/05 19:12:47 grid currents: [0.439 0.554 1.43]A
[site  ] DEBUG 2024/09/05 19:12:47 site power: 310W
[lp-1  ] DEBUG 2024/09/05 19:12:47 charge total import: 6564.734kWh
[lp-1  ] DEBUG 2024/09/05 19:12:47 charger status: B
[lp-1  ] INFO 2024/09/05 19:12:47 car connected
[lp-1  ] DEBUG 2024/09/05 19:12:47 charge total import: 6564.734kWh
[nissan] TRACE 2024/09/05 19:12:47 GET https://alliance-platform-caradapter-prod.apps.eu2.kamereon.io/car-adapter/v2/cars/***/battery-status
[nissan] TRACE 2024/09/05 19:12:48 {"id":"***","timestamp":"2024-09-05T17:12:38Z","batteryLevel":30,"batteryAutonomy":177,"batteryCapacity":87,"batteryAvailableEnergy":28,"plugStatus":1,"chargingStatus":0.4,"chargingRemainingTime":518}
[lp-1  ] DEBUG 2024/09/05 19:12:48 vehicle status: B
[lp-1  ] DEBUG 2024/09/05 19:12:48 vehicle soc: 30%
[lp-1  ] DEBUG 2024/09/05 19:12:48 vehicle range: 177km
[lp-1  ] DEBUG 2024/09/05 19:12:48 forced charging at vehicle soc 30% (< 40% min soc)
[lp-1  ] DEBUG 2024/09/05 19:12:48 max charge current: 16A
[lp-1  ] DEBUG 2024/09/05 19:12:48 charger enable
[lp-1  ] DEBUG 2024/09/05 19:12:48 wake-up timer: start
[site  ] DEBUG 2024/09/05 19:13:07 ----
[lp-1  ] DEBUG 2024/09/05 19:13:07 charge power: 10615W
[lp-1  ] DEBUG 2024/09/05 19:13:07 charge currents: [15.2 15.2 15.2]A
[script] DEBUG 2024/09/05 19:13:07 /fupcService/pvCurrentPower.bat: 88
[site  ] DEBUG 2024/09/05 19:13:07 pv power: 88W
[site  ] DEBUG 2024/09/05 19:13:07 grid meter: 11040W
[site  ] DEBUG 2024/09/05 19:13:07 grid currents: [15.5 15.3 16.2]A
[site  ] DEBUG 2024/09/05 19:13:07 site power: 11040W
[lp-1  ] DEBUG 2024/09/05 19:13:07 !! session: chargeRater.chargedEnergy=0.0 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/09/05 19:13:07 charge total import: 6564.760kWh
[lp-1  ] DEBUG 2024/09/05 19:13:07 charger status: C
[lp-1  ] INFO 2024/09/05 19:13:07 start charging ->
[lp-1  ] DEBUG 2024/09/05 19:13:07 wake-up timer: stop
[db    ] TRACE 2024/09/05 19:13:07 INSERT INTO `sessions` (`created`,`finished`,`loadpoint`,`identifier`,`vehicle`,`odometer`,`meter_start_kwh`,`meter_end_kwh`,`charged_kwh`,`charge_duration`,`solar_percentage`,`price`,`price_per_kwh`,`co2_per_kwh`) VALUES ("2024-09-05 19:13:07.47","0000-00-00 00:00:00","Einfahrt","","Ariya",NULL,6564.734,NULL,0,NULL,NULL,NULL,NULL,NULL) RETURNING `id` 1 <nil>
[lp-1  ] DEBUG 2024/09/05 19:13:07 charger vehicle id: P-543-Test                              
[lp-1  ] DEBUG 2024/09/05 19:13:07 soc estimated: 30.03% (vehicle: 30.00%)
[lp-1  ] DEBUG 2024/09/05 19:13:07 vehicle soc: 30%
[lp-1  ] DEBUG 2024/09/05 19:13:07 vehicle range: 177km
[lp-1  ] DEBUG 2024/09/05 19:13:07 forced charging at vehicle soc 30% (< 40% min soc)
[site  ] DEBUG 2024/09/05 19:13:27 ----
[lp-1  ] DEBUG 2024/09/05 19:13:27 charge power: 10618W
[lp-1  ] DEBUG 2024/09/05 19:13:27 charge currents: [15.2 15.2 15.2]A
[lp-1  ] DEBUG 2024/09/05 19:13:27 forced charging at vehicle soc 30% (< 40% min soc)
[script] DEBUG 2024/09/05 19:13:27 /fupcService/pvCurrentPower.bat: 87
[site  ] DEBUG 2024/09/05 19:13:27 pv power: 87W
[site  ] DEBUG 2024/09/05 19:13:27 grid meter: 11044W
[site  ] DEBUG 2024/09/05 19:13:27 grid currents: [15.5 15.3 16.3]A
[site  ] DEBUG 2024/09/05 19:13:27 site power: 11044W
[lp-1  ] DEBUG 2024/09/05 19:13:27 detected active phases: 3p
[lp-1  ] DEBUG 2024/09/05 19:13:27 !! session: chargeRater.chargedEnergy=0.1 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/09/05 19:13:27 charge total import: 6564.818kWh
[lp-1  ] DEBUG 2024/09/05 19:13:27 charger status: C
[lp-1  ] DEBUG 2024/09/05 19:13:27 soc estimated: 30.09% (vehicle: 30.00%)
[lp-1  ] DEBUG 2024/09/05 19:13:27 vehicle soc: 30%
[lp-1  ] DEBUG 2024/09/05 19:13:27 vehicle range: 177km
[lp-1  ] DEBUG 2024/09/05 19:13:27 forced charging at vehicle soc 30% (< 40% min soc)
[site  ] DEBUG 2024/09/05 19:13:47 ----
[lp-1  ] DEBUG 2024/09/05 19:13:47 charge power: 10624W
[lp-1  ] DEBUG 2024/09/05 19:13:47 charge currents: [15.2 15.2 15.2]A
[lp-1  ] DEBUG 2024/09/05 19:13:47 forced charging at vehicle soc 30% (< 40% min soc)
[script] DEBUG 2024/09/05 19:13:47 /fupcService/pvCurrentPower.bat: 87
[site  ] DEBUG 2024/09/05 19:13:47 pv power: 87W
[site  ] DEBUG 2024/09/05 19:13:47 grid meter: 11050W
[site  ] DEBUG 2024/09/05 19:13:47 grid currents: [15.5 15.3 16.3]A
[site  ] DEBUG 2024/09/05 19:13:47 site power: 11050W
[lp-1  ] DEBUG 2024/09/05 19:13:47 detected active phases: 3p
[lp-1  ] DEBUG 2024/09/05 19:13:47 !! session: chargeRater.chargedEnergy=0.1 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/09/05 19:13:47 charge total import: 6564.877kWh
[lp-1  ] DEBUG 2024/09/05 19:13:47 charger status: C
[lp-1  ] DEBUG 2024/09/05 19:13:47 soc estimated: 30.15% (vehicle: 30.00%)
[lp-1  ] DEBUG 2024/09/05 19:13:47 vehicle soc: 30%
[lp-1  ] DEBUG 2024/09/05 19:13:47 vehicle range: 177km
[lp-1  ] DEBUG 2024/09/05 19:13:47 forced charging at vehicle soc 30% (< 40% min soc)
[site  ] DEBUG 2024/09/05 19:14:07 ----
[lp-1  ] DEBUG 2024/09/05 19:14:07 charge power: 10587W
[lp-1  ] DEBUG 2024/09/05 19:14:07 charge currents: [15.1 15.1 15.2]A
[lp-1  ] DEBUG 2024/09/05 19:14:07 forced charging at vehicle soc 30% (< 40% min soc)
[script] DEBUG 2024/09/05 19:14:07 /fupcService/pvCurrentPower.bat: 83
[site  ] DEBUG 2024/09/05 19:14:07 pv power: 83W
[site  ] DEBUG 2024/09/05 19:14:07 grid meter: 11013W
[site  ] DEBUG 2024/09/05 19:14:07 grid currents: [15.4 15.3 16.3]A
[site  ] DEBUG 2024/09/05 19:14:07 site power: 11013W
[lp-1  ] DEBUG 2024/09/05 19:14:07 detected active phases: 3p
[lp-1  ] DEBUG 2024/09/05 19:14:07 !! session: chargeRater.chargedEnergy=0.2 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/09/05 19:14:07 charge total import: 6564.936kWh
[lp-1  ] DEBUG 2024/09/05 19:14:07 charger status: C
[lp-1  ] DEBUG 2024/09/05 19:14:07 soc estimated: 30.21% (vehicle: 30.00%)
[lp-1  ] DEBUG 2024/09/05 19:14:07 vehicle soc: 30%
[lp-1  ] DEBUG 2024/09/05 19:14:07 vehicle range: 177km
[lp-1  ] DEBUG 2024/09/05 19:14:07 forced charging at vehicle soc 30% (< 40% min soc)
[site  ] DEBUG 2024/09/05 19:14:27 ----
```

### What type of operating system are you running?

Windows

### Version

0.130.7

andig, 2024-09-07:
Im Master gehoben.

closed 2024-09-07 by andig

An-Wi, 2024-09-07:
ok, hatte soeben auch das Verhalten beim normalem Laden mit "Min+PV" (ohne "Min.Ladung").
Aber ich warte jetzt einfach mal auf das nächste Update. Danke für die schnelle Reaktion. ⚡

pkuehnel, 2024-09-20:
In welcher Version ist das korrigiert?

andig, 2024-09-20:
Guckst Du in die Release Notes ;)

pkuehnel, 2024-09-20:
Hab isch geguckt und nix gefunden, `15936` kommt nischt vor und `Aut` kommt nischt vor (nur bei Einführung des Feature in [0.130.3](https://github.com/evcc-io/evcc/releases/tag/0.130.3))

An-Wi, 2024-09-20:
Ich konnte kein fehlerhaftes Verhalten in der UI bzgl. „Warte auf Autorisierung“ mehr feststellen… etwa seit 0.130.9

pkuehnel, 2024-09-21:
Ok danke. Ich habe jetzt wieder die 0.127.3 installiert, die scheint mir stabil zu laufen, mit der 130.x laufe ich von einem Problem ins nächste. Charger rejected Ampere Änderungen usw, Auto beginnt nicht zu laden usw.