evcc-io/evcc #15937: Follow-up zu #15772: smart #1 Auto-Detection, SoC, km-Stand u.a. funktionieren bei mir nicht

cyberways, 2024-09-07:
### Describe the bug

Vielen Dank für den Fix der smart-hello Gen-1 API.
Habe ich direkt nach Erschienen der 0.130.7 noch von unterwegs remote auf der heimischen Anlage installiert und hatte erstmals ever in "evcc vehicle --log trace smart" auf der Shell keine Fehler mehr und gültige Werte - auch in der GUI Konfiguration beim Fahrzeug danach alles grün - bei abgestecktem Fahrzeug weil unterwegs.
Einen Tag später ist mir bereits aufgefallen, dass die Fahrzeugdaten des smart in der GUI Konfig. wieder rot und alle Werte 0 sind. Beim ersten Laden zurück zuhause dann wieder die gleichen Fehler wie vor dem Fix. Wie ich heute bei genauer Analyse feststelle auch wieder im "evcc vehicle --log trace smart" auf der Shell.

### Steps to reproduce

Zugangsdaten hat andig ja schon.

Eine Auto-Erkennung des smart scheint evcc gar nicht zu versuchen.
Im Logfile steht beim Start 

[site ] WARN 2024/09/07 10:32:27 vehicle 'smart' does not support automatic detection 
[site ] INFO 2024/09/07 10:32:27 vehicle 2: range ✓ finish ✗ status ✗ climate ✓ wakeup ✗ 

Status x ist wohl das vermeintliche Problem. Die json API liefert den Ladezustand aber in "chargerState", das von 0 auf 2 wechselt wenn das Auto lädt. Könnte man das bitte "nachrüsten"?

Ich hänge unten einmal das komplette Log Trace an bei abgestecktem Auto an und nochmal das json während des Ladens.
Vielen Dank!

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: ***

interval: 30s # control cycle interval

sponsortoken: ***

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: victron-energy 
  usage: grid  
  host: 192.168.24.11  
  port: 502  
  name: grid
  meterid: 276
- type: template
  template: victron-energy 
  usage: pv  
  host: 192.168.24.11  
  port: 502  
  name: pv
- type: template
  template: victron-energy 
  usage: battery  
  host: 192.168.24.11  
  port: 502  
  name: battery
  minsoc: 25
  capacity: 10

chargers:
- type: template
[... 301 lines cut ...]
Capacity:     62.0kWh
Range:        json: invalid use of ,string struct tag, trying to unmarshal "" into int: %!s(<nil>)
Odometer:     json: invalid use of ,string struct tag, trying to unmarshal "" into int: %!s(<nil>)
Climater:     json: invalid use of ,string struct tag, trying to unmarshal "" into int: %!s(<nil>)
Position:     json: invalid use of ,string struct tag, trying to unmarshal "" into int: %!s(<nil>)
OnIdentified: MaxCurrent:10
[eebus ] DEBUG 2024/09/07 10:23:36 mdns: stop announcement
[eebus ] ERROR 2024/09/07 10:23:36 websocket server error: http: Server closed

json Antwort während des Ladens: chargerState wechselt von 0 auf 2

[smart-hello] TRACE 2024/09/07 10:31:36 {"code":"1000","data":{"result":{"serviceResult":{"error":null,"operationResult":1},"sessionId":"PS048570000000029789636525951824"},"vehicleStatus":{"basicVehicleStatus":{"usageMode":"1","engineStatus":"engine_off","position":{"altitude":"","posCanBeTrusted":"","latitude":"","carLocatorStatUploadEn":"false","marsCoordinates":"","longitude":""},"carMode":"0","speed":"0.0","speedValidity":"false","direction":""},"notification":{"notifForEmgyCallStatus":"0"},"eg":{"enableRunning":"false","blocked":{"status":"0"},"panicStatus":"false"},"parkTime":{"status":"1725697061555"},"theftNotification":{"time":"1716550899","activated":"2"},"configuration":{"propulsionType":"4","fuelType":"4","vin":"HESXA*****"},"updateTime":"1725697890073","additionalVehicleStatus":{"maintenanceStatus":{"tyreTempWarningPassengerRear":"0","daysToService":"171","engineHrsToService":"500","odometer":"5232.000","brakeFluidLevelStatus":"3","tyreTempDriverRear":"23.000","tyreTempWarningPassenger":"0","tyreTempWarningDriverRear":"0","mainBatteryStatus":{"stateOfCharge":"1","chargeLevel":"92.4","energyLevel":"0","stateOfHealth":"0","powerLevel":"0","voltage":"12.225"},"tyreTempDriver":"23.000","tyreTempPassengerRear":"22.000","tyrePreWarningDriver":"0","distanceToService":"24768","tyrePreWarningPassengerRear":"0","tyreTempWarningDriver":"0","tyreStatusPassengerRear":"241.648","tyreStatusPassenger":"237.529","tyreStatusDriverRear":"241.648","serviceWarningStatus":"0","tyreStatusDriver":"247.140","tyreTempPassenger":"23.000","tyrePreWarningDriverRear":"0","tyrePreWarningPassenger":"0","washerFluidLevelStatus":"1"},"electricVehicleStatus":{"disChargeUAct":"0.0","disChargeSts":"0","wptFineAlignt":"0","chargeLidAcStatus":"2","distanceToEmptyOnBatteryOnly":"93","distanceToEmptyOnBattery100Soc":"429","chargeSts":"0","averPowerConsumption":"-86.3","chargerState":"2","timeToTargetDisCharged":"2047","distanceToEmptyOnBattery20Soc":"84","disChargeConnectStatus":"3","chargeLidDcAcStatus":"1","dcChargeSts":"0","ptReady":"0","chargeLevel":"22","statusOfChargerConnection":"3","dcDcActvd":"1","indPowerConsumption":"0.0","dcDcConnectStatus":"0","disChargeIAct":"0.0","dcChargeIAct":"-6.7","chargeUAct":"406.7","bookChargeSts":"0","chargeIAct":"5.900","timeToFullyCharged":"878"},"chargeHvSts":"1","drivingBehaviourStatus":{"gearAutoStatus":"0","gearManualStatus":"0","engineSpeed":"0.000"},"runningStatus":{"ahbc":"0","goodbye":"0","homeSafe":"0","cornrgLi":"0","frntFog":"0","stopLi":"0","tripMeter1":"4378.6","approach":"0","tripMeter2":"149.8","indFuelConsumption":"0","hiBeam":"0","engineCoolantLevelStatus":"3","fuelEnLevel":"0","loBeam":"0","posLiRe":"0","ltgShow":"0","welcome":"0","drl":"0","fuelLevelPct":"0","ahl":"0","fuelEnCns":"0","trunIndrLe":"0","trunIndrRi":"0","afs":"0","dbl":"0","avgSpeed":"24","posLiFrnt":"0","reverseLi":"0","hwl":"0","reFog":"0","flash":"0","allwl":"0","fuelEnCnsFild":"0"},"trailerStatus":{"trailerTurningLampSts":"0","trailerFogLampSts":"0","trailerBreakLampSts":"0","trailerReversingLampSts":"0","trailerPosLampSts":"0"},"climateStatus":{"drvHeatSts":"0","winPosDriver":"0","rrVentDetail":"0","rlVentSts":"0","passVentSts":"0","interiorTemp":"25.000","passVentDetail":"0","sunroofPos":"101","cdsClimateActive":"false","sunroofOpenStatus":"1","rrHeatingDetail":"0","winStatusPassenger":"2","fragActive":false,"winStatusDriver":"2","drvVentSts":"0","winStatusPassengerRear":"2","sunCurtainRearOpenStatus":"1","preClimateActive":false,"rlHeatingDetail":"0","winPosPassengerRear":"0","curtainPos":"0","rlVentDetail":"0","curtainOpenStatus":"1","climateOverHeatProActive":"true","rrVentSts":"0","rrHeatingSts":"0","winPosPassenger":"0","steerWhlHeatingSts":"2","drvVentDetail":"0","winPosDriverRear":"0","exteriorTemp":"21.500","rlHeatingSts":"0","winStatusDriverRear":"2","defrost":"false","drvHeatDetail":"2","passHeatingDetail":"2","airBlowerActive":"false","sunCurtainRearPos":"101","passHeatingSts":"0"},"drivingSafetyStatus":{"doorLockStatusDriverRear":"1","srsCrashStatus":"0","doorOpenStatusPassengerRear":"0","doorPosPassengerRear":"0","doorOpenStatusDriver":"0","seatBeltStatusPassenger":"false","doorPosDriver":"0","seatBeltStatusThPassengerRear":"false","electricParkBrakeStatus":"1","doorLockStatusDriver":"1","seatBeltStatusThDriverRear":"false","tankFlapStatus":"2","seatBeltStatusPassengerRear":"false","doorOpenStatusPassenger":"0","doorPosPassenger":"0","vehicleAlarm":{"alrmSt":"1","alrmTrgSrc":"7"},"doorPosDriverRear":"0","centralLockingStatus":"2","seatBeltStatusDriver":"false","doorLockStatusPassenger":"1","seatBeltStatusMidRear":"false","trunkLockStatus":"1","seatBeltStatusDriverRear":"false","engineHoodOpenStatus":"0","doorOpenStatusDriverRear":"0","doorLockStatusPassengerRear":"1","trunkOpenStatus":"0"},"pollutionStatus":{"interiorPM25":"11","interiorSecondPM25Level":"0","interiorPM25Level":"0","relHumSts":"80","exteriorPM25Level":"0"}},"temStatus":{"swVersion":null,"serialNumber":null,"powerSource":null,"networkAccessStatus":{"mobileNetwork":null,"simInfo":{"iccId":null,"imsi":null,"msisdn":null}},"mcuVersion":null,"mpuVersion":null,"backupBattery":{"stateOfCharge":null,"stateOfHealth":null,"voltage":null},"hwVersion":null,"powerMode":null,"healthStatus":null,"sleepCycleNextWakeupTime":null,"rvsEnable":"true","imei":null,"state":null,"connectivityStatus":null}}},"success":true,"hint":null,"httpStatus":"OK","sessionId":"f1002eb8985d5136c02a2ec6af49769c","message":"operation succeed"}
```

### What type of operating system are you running?

Linux

### Nightly build

- [ ] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.7

andig, 2024-09-07:
Scheint ja als würde Smart da ständig etwas ändern. Bitte bei Smart beschweren. Ich sehe nicht, dass wir das alle paar Tage wieder nachziehen wollen.

cyberways, 2024-09-07:
Verstehe ich irgendwie, ist aber nicht gut. Für eine Beschwerde bräuchte ich bitte Infos: Was ändern die denn und warum kommt es dann immer zu den gleichen Fehlermeldungen? Gibt es eine Gen-1 API und noch eine andere Version die funktioniert und bin ich der einzige Smart #1, der diese Probleme hat und bei allen anderen läuft es einfach?

andig, 2024-09-07:
Zuerst musst du wohl mal Smart fragen, ob und wann sie überhaupt ein extern nutzbares API anbieten wollen das sie dann auch mal dokumentieren. Da fängt der Ärger nämlich schon an- es gibt GAR NICHTS...

closed 2024-09-07 by andig

cyberways, 2024-09-07:
Für GAR NICHTS an Doku habt ihr aber eine richtig gute Anbindung gebaut, auch wenn sie bei mir nicht funktioniert.
Das Auto lädt übrigens gerade noch ca. 1,5h, wenn du mit den Zugangsdaten nochmal während des Ladens draufschauen wollen würdest. ;-) Vielleicht gibt es neues zu sehen?

andig, 2024-09-07:
>Für GAR NICHTS an Doku habt ihr aber eine richtig gute Anbindung gebaut,

Isso ;)

Beim Ladestatus musst Du selbst Hand anlegen und mit Hilfe von

    evcc charger --log trace

rausfinden, woran man disconnected/ connected/ charging fest machen kann. Vmtl ist das `chargerState`. Aber welche Werte sind was?

cyberways, 2024-09-07:
Das habe ich schon 2x geschrieben im Laufe unserer Tickets. chargerState = 0 ist abgesteckt, 2 ist ladend wie jetzt gerade und ich meine 3 ist ladebereit wartend wenn nicht genug PV überschuss von evcc oder der im Auto selbst eingestellte SoC erreicht ist und das Auto noch an der Wallbox hängt.

andig, 2024-09-07:
> und ich meine

Es braucht "connected". Ist er das jetzt oder nicht? "ich meine" lässt sich schlecht implementieren ;)

cyberways, 2024-09-07:
Das können wir gerne zusammen rausfinden wenn ihr es implementieren möchtet. Jetzt gerade lädt das Auto und ist "2". 
Ich hatte die "3" im Gedächtnis von vor einigen Tagen als das Auto nach dem Laden pausiert hat, müsste aber nochmal sicherstellen ob durch evcc wegen zu wenig Sonne oder wegen "voll".

cyberways, 2024-09-21:
@andig Du hast die smart-hello API trotz des verständlichen Unmuts über die "Unterstützung" seitens des Herstellers ja doch noch einmal gefixt und auch den Charge State eingebaut. Mega Danke dafür! 
Heute getestet mit der 0.130.11, funktioniert alles 100% inkl. Fahrzeugerkennung und SoC Anzeige – zum ersten Mal überhaupt seit ich das Auto habe. DANKE!

andig, 2024-09-21:
Gerne :)