evcc-io/evcc #15943: Charge Amps Halo: timeout

premultiply, 2024-09-07:
> Bitte mal mit dem aktuellen Nightly probieren. Dabei erst den evcc-Dienst gestoppt lassen, Box rebooten und vollständig hochfahren lassen und dann erst evcc hochfahren. Sicherheitshalber mal ocpp auf Tracelevel mitloggen lassen.

Hier sieht es leider nicht besser aus.
Version: `evcc version 0.130.7 (dab8dfae)`

<details>
  <summary>Config</summary>

```
chargers:
- name: wallbox4
  type: template
  template: ocpp
  connector: 1
  connecttimeout: 10m
  timeout: 10m
```

</details>

<details>
  <summary>Testsetup</summary>

- 2 unterschiedliche Halo-Modelle getestet: Ausführungen mit und ohne Schuko
- Fahrzeug nicht eingesteckt
- In der Box ist als OCPP die IP von EVCC eingetragen. Box läuft seit über einer Stunde, seit letztem Reboot. Dann erst EVCC gestartet
</details>

<details>
  <summary>Log: Erster Versuch</summary>

```
[ocpp-1] DEBUG 2024/09/07 07:29:25 waiting for chargepoint: 10m0s
[ocpp  ] INFO 2024/09/07 07:30:10 charge point connected, registering: 2403082631M
[ocpp  ] TRACE 2024/09/07 07:30:10 send 2403082631M: [2,"2672514354","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
[ocpp-1] DEBUG 2024/09/07 07:30:40 failed configuring availability: timeout
[ocpp  ] TRACE 2024/09/07 07:31:10 send 2403082631M: [2,"1173943367","GetConfiguration",{}]
[ocpp  ] DEBUG 2024/09/07 07:31:10 charge point disconnected: 2403082631M
[main  ] FATAL 2024/09/07 07:31:10 cannot create charger 'wallbox4': cannot create charger type 'template': cannot create charger type 'ocpp': timeout
[main  ] FATAL 2024/09/07 07:31:10 will attempt restart in: 15m0s
[ocpp  ] DEBUG 2024/09/07 07:31:13 charge point connected: 2403082631M
[ocpp  ] TRACE 2024/09/07 07:31:13 recv 2403082631M: [2, "671510120-1736422715-167515484", "BootNotification", {"chargePointModel":"HALO","chargePointSerialNumber":"2403082631M","chargePointVendor":"Charge Amps","firmwareVersion":"180","iccid":"0","meterType":"ATM90E32AS"}]
[ocpp  ] TRACE 2024/09/07 07:31:13 send 2403082631M: [3,"671510120-1736422715-167515484",{"currentTime":"2024-09-07T06:31:13Z","interval":60,"status":"Accepted"}]
[ocpp  ] TRACE 2024/09/07 07:31:14 recv 2403082631M: [2, "2031540435-1694113035-15976211", "StartTransaction", {"connectorId":1,"idTag":"00000000000000","meterStart":528564,"timestamp":"2024-09-05T15:38:49Z"}]
[ocpp  ] TRACE 2024/09/07 07:32:14 recv 2403082631M: [2, "2068488941-356120837-172445389", "StartTransaction", {"connectorId":1,"idTag":"00000000000000","meterStart":528564,"timestamp":"2024-09-05T15:38:49Z"}]
[ocpp  ] DEBUG 2024/09/07 07:33:44 charge point disconnected: 2403082631M
[ocpp  ] DEBUG 2024/09/07 07:34:20 charge point connected: 2403082631M
[ocpp  ] TRACE 2024/09/07 07:34:50 recv 2403082631M: [2, "181277323-1394981785-152828202", "StartTransaction", {"connectorId":1,"idTag":"00000000000000","meterStart":528564,"timestamp":"2024-09-05T15:38:49Z"}]
[ocpp  ] TRACE 2024/09/07 07:34:50 recv 2403082631M: [2, "2016303320-1825317966-12025150", "StopTransaction", {"idTag":"00000000000000","meterStop":534859,"timestamp":"2024-09-05T17:20:49Z","transactionId":-1,"reason":"PowerLoss"}]
[ocpp  ] TRACE 2024/09/07 07:34:50 send 2403082631M: [3,"2016303320-1825317966-12025150",{"idTagInfo":{"status":"Accepted"}}]
[ocpp  ] TRACE 2024/09/07 07:34:50 recv 2403082631M: [2, "895983091-948277659-606742936", "StatusNotification", {"connectorId":0,"errorCode":"NoError","status":"Available"}]
[ocpp  ] TRACE 2024/09/07 07:34:50 send 2403082631M: [3,"895983091-948277659-606742936",{}]
[ocpp  ] TRACE 2024/09/07 07:34:50 recv 2403082631M: [2, "1541277574-768373858-117390205", "StatusNotification", {"connectorId":1,"errorCode":"NoError","status":"Available"}]
[ocpp  ] TRACE 2024/09/07 07:34:50 send 2403082631M: [3,"1541277574-768373858-117390205",{}]
[ocpp  ] TRACE 2024/09/07 07:34:50 recv 2403082631M: [2, "725626001-282802793-1725834041", "StatusNotification", {"connectorId":2,"errorCode":"NoError","status":"Available"}]
[ocpp  ] TRACE 2024/09/07 07:34:50 send 2403082631M: [3,"725626001-282802793-1725834041",{}]
[ocpp  ] TRACE 2024/09/07 07:34:50 recv 2403082631M: [2, "1288956161-252932288-162012984", "StatusNotification", {"connectorId":0,"errorCode":"NoError","status":"Available"}]
[ocpp  ] TRACE 2024/09/07 07:34:50 send 2403082631M: [3,"1288956161-252932288-162012984",{}]
[ocpp  ] TRACE 2024/09/07 07:34:50 recv 2403082631M: [2, "1663822829-1255058936-10718463", "StatusNotification", {"connectorId":1,"errorCode":"NoError","status":"Available"}]
[ocpp  ] TRACE 2024/09/07 07:34:50 send 2403082631M: [3,"1663822829-1255058936-10718463",{}]
[ocpp  ] TRACE 2024/09/07 07:34:50 recv 2403082631M: [2, "1667158257-651599149-191625422", "StatusNotification", {"connectorId":2,"errorCode":"NoError","status":"Available"}]
[ocpp  ] TRACE 2024/09/07 07:34:50 send 2403082631M: [3,"1667158257-651599149-191625422",{}]
[ocpp  ] TRACE 2024/09/07 07:35:50 recv 2403082631M: [2, "1077768330-673479359-669834701", "Heartbeat", {}]
[ocpp  ] TRACE 2024/09/07 07:35:50 send 2403082631M: [3,"1077768330-673479359-669834701",{"currentTime":"2024-09-07T06:35:50Z"}]
[ocpp  ] TRACE 2024/09/07 07:36:50 recv 2403082631M: [2, "841865678-286462040-1025874218", "Heartbeat", {}]
[ocpp  ] TRACE 2024/09/07 07:36:50 send 2403082631M: [3,"841865678-286462040-1025874218",{"currentTime":"2024-09-07T06:36:50Z"}]
[ocpp  ] TRACE 2024/09/07 07:37:50 recv 2403082631M: [2, "2038588940-734826475-148776524", "Heartbeat", {}]
[ocpp  ] TRACE 2024/09/07 07:37:50 send 2403082631M: [3,"2038588940-734826475-148776524",{"currentTime":"2024-09-07T06:37:50Z"}]
[ocpp  ] TRACE 2024/09/07 07:38:50 recv 2403082631M: [2, "198383716-843156544-438875118", "Heartbeat", {}]
[ocpp  ] TRACE 2024/09/07 07:38:50 send 2403082631M: [3,"198383716-843156544-438875118",{"currentTime":"2024-09-07T06:38:50Z"}]
[... 108 lines cut ...]
[ocpp  ] TRACE 2024/09/07 10:10:10 recv 2207064354M: [3, "892096400", {"status":"Rejected"}]
[ocpp  ] TRACE 2024/09/07 10:10:10 send 2207064354M: [2,"3705225881","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Active.Import,Energy.Active.Import.Register,Current.Import,Voltage,Current.Offered"}]
[ocpp  ] TRACE 2024/09/07 10:10:10 recv 2207064354M: [3, "3705225881", {"status":"Accepted"}]
[ocpp  ] TRACE 2024/09/07 10:10:10 send 2207064354M: [2,"860024731","TriggerMessage",{"requestedMessage":"MeterValues"}]
[ocpp  ] TRACE 2024/09/07 10:10:10 recv 2207064354M: [3, "860024731", {"status":"Accepted"}]
[ocpp  ] TRACE 2024/09/07 10:10:11 recv 2207064354M: [2, "1793420242-2114067607-10141938", "MeterValues", {"connectorId":1,"transactionId":-1,"meterValue":[{"timestamp":"2024-09-07T09:10:09Z","sampledValue":[{"value":"586757","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"0.0","context":"Sample.Periodic","measurand":"Current.Offered","unit":"A"},{"value":"0.2","context":"Sample.Periodic","measurand":"Power.Active.Import","unit":"W"},{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Import","phase":"L1","unit":"A"},{"value":"242.00","context":"Sample.Periodic","measurand":"Voltage","phase":"L1","unit":"V"},{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Import","phase":"L2","unit":"A"},{"value":"240.31","context":"Sample.Periodic","measurand":"Voltage","phase":"L2","unit":"V"},{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Import","phase":"L3","unit":"A"},{"value":"240.58","context":"Sample.Periodic","measurand":"Voltage","phase":"L3","unit":"V"}]}]}]
[ocpp  ] TRACE 2024/09/07 10:10:11 send 2207064354M: [2,"4166563109","ChangeConfiguration",{"key":"MeterValueSampleInterval","value":"10"}]
[ocpp-1] WARN 2024/09/07 10:10:41 failed configuring MeterValueSampleInterval: timeout
[ocpp  ] TRACE 2024/09/07 10:11:10 recv 2207064354M: [2, "933238601-1913056354-641130695", "MeterValues", {"connectorId":1,"transactionId":-1,"meterValue":[{"timestamp":"2024-09-07T09:10:09Z","sampledValue":[{"value":"586757","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"0.0","context":"Sample.Periodic","measurand":"Current.Offered","unit":"A"},{"value":"0.2","context":"Sample.Periodic","measurand":"Power.Active.Import","unit":"W"},{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Import","phase":"L1","unit":"A"},{"value":"242.00","context":"Sample.Periodic","measurand":"Voltage","phase":"L1","unit":"V"},{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Import","phase":"L2","unit":"A"},{"value":"240 [... cut ...]
Habt Ihr einen Ansatz, woran das liegen könnte, oder muss ich mich da mal reinfuchsen, wie ich den Traffic in Klartext mitschneide, um da tiefer rein zu gucken?

Noch ein unqualifizierter Gedanke: Es gibt Boxen mit und ohne zusätzlicher Schuko-Dose. Gehe davon aus, dass die Schuko-Variante mit zusätzlichem Connector angesprochen werden kann?! Können die Probleme daraus resultieren, dass der Connector des Typ-2 Steckers ggf. eine andere Connector-ID bekommt?

_Originally posted by @Chillkroete1206 in https://github.com/evcc-io/evcc/discussions/3624#discussioncomment-10576145_

andig, 2024-09-07:
Bitte Fehlerbeschreibung ergänzen.

premultiply, 2024-09-07:
Hier liefert evcc als CS für manche eingehenden Nachrichten (z.B. StartTransaction, MeterValues, ...) in bestimmten Situationen keine zugehörige Confirmation.

Sobald bzw. solange diese ausbleibt stellt die Box sofort die Kommunikation ein.

Originally posted by @premultiply in https://github.com/evcc-io/evcc/discussions/3624#discussioncomment-10576218

andig, 2024-09-14:
Tentatively closing, since logfile is missing this might be identical to https://github.com/evcc-io/evcc/issues/15991

PR #15945 (2024-09-07): Ocpp: always send confirmation
Fixes #15943 https://github.com/evcc-io/evcc/discussions/3624#discussioncomment-10576145
closed 2024-09-14 by andig