evcc-io/evcc #15952: After upgrading from 0.130.6 to 0.130.7 no charging possible with go-e

sebnaf, 2024-09-07:
### Describe the bug

After upgrading to 0.130.7 without changing any configuration charging using go-e is not possible anymore, failing with

`[lp-1 ] ERROR 2024/09/07 16:29:37 max charge current 5A: unexpected status: 500 (Internal Server Error)`

### Steps to reproduce

1. update to 0.130.7
2.
3.
...

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local
  port: 7070

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

plant: 2d2xyzde6

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken:
sponsortoken: eyJhbxyz9i0

telemetry: true

# log settings
log: error
levels:
  site: info
  lp-1: info
  lp-2: info
  cache: trace
  db: warn
  vehicle: info
  tesla: info
  elektronenpumpe: info
  tronity: trace

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
  #    # rtu: true
  #    # readonly: true # use `deny` to raise modbus errors

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
  - name: grid
    type: movingaverage
    decay: 0.2
    meter:
      type: custom
      power:
        source: sunspec
        #model: sunspec
[... 223 lines cut ...]
      msg: Unknown vehicle, guest connected?
  services:
    - type: pushover
      app: aq5gxyz27w
      recipients:
        - uanxyz2c
```

### Log details

```text
-> https://pastes.dev/AQ7PNjCiso
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.7

sebnaf, 2024-09-07:
This can be narrowed down to

`[lp-1  ] DEBUG 2024/09/07 16:30:07 pv charge current: min 5A > 0A (263W @ 1p, battery: false)
[go-e  ] TRACE 2024/09/07 16:30:07 GET http://10.10.20.22/api/set?amp=5`

and

`[go-e  ] TRACE 2024/09/07 16:30:07 {"amp":"amp: failed: Value 5 exceeds range 6 to 32 (5)"}
[lp-1  ] ERROR 2024/09/07 16:30:07 max charge current 5A: unexpected status: 500 (Internal Server Error)`

However - why did this work prior 0.130.7 with the same settings?

VolkerK62, 2024-09-07:
do you have a mincurrent of 5A configured?
is it working with 6A?

andig, 2024-09-08:
5A geht halt nicht und kann egtl nie funktioniert haben….

closed 2024-09-08 by andig

sebnaf, 2024-09-08:
Sorry - genau die Config hat über Monate funktioniert. Deswegen widerspreche ich dem "eigentlich".
Ich habe es auch gerade nochmals mit der 0.130.6 erfolgreich getestet.

Möchtet ihr das Tracelog der 0.130.6 haben?

andig, 2024-09-08:
Brauchts nicht. Warum auch immer das jetzt anders sein mag- 5A geht einfach nicht.