evcc-io/evcc #15953: evcc token produces wrong misleading configuration entry

deadrabbit87, 2024-09-07:
### Describe the bug

generating a token for psa or tesla produces the following configuration entry: 
```
Add the following tokens to the vehicle config:

    type: peugeot
    tokens:
      access: xyz
      refresh: xyz
```

If this is done, evcc produces the error 
`FATAL 2024/09/07 16:03:26 cannot create vehicle '208e': cannot create vehicle type 'template': invalid key: tokens`

The correct entry is: 
```
    type: peugeot
    accesstoken: xyz
    refreshtoken: xyz
```

### Steps to reproduce

1. Generate token with `evcc token`
2. Copy configuration entry to `evcc.yaml`

### Configuration details

```yaml
see above
```

### Log details

```text
FATAL 2024/09/07 16:03:26 cannot create vehicle '208e': cannot create vehicle type 'template': invalid key: tokens
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.130.7

andig, 2024-09-08:
Deine Beschreibung ist fehlerhaft. Type und Template sind unterschiedliche Dinge.

closed 2024-09-08 by andig

andig, 2024-09-11:
Schau dir mal die Fehlermeldung an und vergleiche mit deiner Config

deadrabbit87, 2024-09-11:
Sorry, ich seh das Brett nicht. Für mich ist der config-Eintrag falsch, den `evcc token` generiert.