evcc-io/evcc #15957: Hausbatterie Huawei

jwedenig, 2024-09-07:
### Describe the bug

Hausbatterie von Huawei Sun 2000 wird entleert obwohl der "Verhindere" Switch aktiv ist

### Steps to reproduce

Hausbatterie von Huawei Sun 2000 wird entleert obwohl der "Verhindere" Switch aktiv ist

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error
  modbus: trace

# unique installation id
plant: e6d7cc4d527d7b88b0d1de8d656033f37cbd95ba9db08f7dbfed3xxx

interval: 30s # control cycle interval

sponsortoken: xxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.1.70  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.1.70  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- name: my_battery
  type: template
  template: huawei-dongle-powersensor
  usage: battery
  modbus: tcpip
  id: 1
  host: 192.168.1.70 # Hostname
  port: 502 # Port
  timeout: 15s # optional
  capacity: 10

chargers:
- type: template
  template: go-e-v3 
  host: 192.168.1.86  
  name: wallbox5

vehicles:
- type: template
  template: id 
[... 81 lines cut ...]
[id    ] TRACE 2024/09/07 12:15:59 _csrf=uSSu1hhl7Fiu8UwEgLzfGykGqmANSG1Sy-5kPoMh4D7VOP2E3ByetH5d2m-DkCk1t5HrLR9jhwJuf1V_-NhQC7cW0FjkD8zm&email=***&hmac=5b107417d7a110081d20e1fd8e847b6d161a08f2bd55d8688b854ef9bd401310&relayState=c976a7e8110b77cd24a3f177e65f048a1dae9d2e
[id    ] TRACE 2024/09/07 12:15:59 GET https://identity.vwgroup.io/signin-service/v1/a24fba63-34b3-4d43-b181-942111e6bda8@apps_vw-dilab_com/login/authenticate?relayState=c976a7e8110b77cd24a3f177e65f048a1dae9d2e
[id    ] TRACE 2024/09/07 12:16:00 <!DOCTYPE html>
    <html lang="en" xmlns="http://www.w3.org/1999/html">
<head>
    <meta charset="utf-8"/>
    <meta name="viewport"
          content="width=device-width, height=device-height, initial-scale=1, maximum-scale=1, user-scalable=no"/>
    <meta name="identitykit" content="loginAuthenticate"/>
    
<script type="text/javascript">(window.NREUM||(NREUM={})).init={privacy:{cookies_enabled:true},ajax:{deny_list:["bam.eu01.nr-data.net"]},distributed_tracing:{enabled:true}};(window.NREUM||(NREUM={})).loader_config={agentID:"535875755",accountID:"2547829",trustKey:"2217462",xpid:"VgMDVl5RARAEUlFQBQQOXlY=",licenseKey:"NRJS-ecd5929e8859410bbd9",applicationID:"460225982"};;/*! For license information please see nr-loader-spa-1.265.1.min.js.LICENSE.txt */
(()=>{var e,t,r={8122:(e,t,r)=>{"use strict";r.d(t,{a:()=>i});var n=r(944);function i(e,t){try{if(!e||"object"!=typeof e)return(0,n.R)(3);if(!t||"object"!=typeof t)return(0,n.R)(4);const r=Object.create(Object.getPrototypeOf(t),Object.getOwnPropertyDescriptors(t)),o=0===Object.keys(r).length?e:r;for(let a in o)if(void 0!==e[a])try{if(null===e[a]){r[a]=null;continue}Array.isArray(e[a])&&Array.isArray(t[a])?r[a]=Array.from(new Set([...e[a],...t[a]])):"object"==typeof e[a]&&"object"==typeof t[a]?r[a]=i(e[a],t[a]):r[a]=e[a]}catch(e){(0,n.R)(1,e)}return r}catch(e){(0,n.R)(2,e)}}},2555:(e,t,r)=>{"use strict";r.d(t,{Vp:()=>c,fn:()=>s,x1:()=>u});var n=r(384),i=r(8122);const o={beacon:n.NT.beacon,errorBeacon:n.NT.errorBeacon,licenseKey:void 0,applicationID:void 0,sa:void
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.130.7

mdkeil, 2024-09-07:
Dein Log ist leider nicht brauchbar.. nur 3s und nichts wo überhaupt mal geladen wird.

aus den normalen Log aus der Diskussion #15929 
`site  ] DEBUG 2024/09/06 17:20:38 set battery mode: hold `

Batterie hold wird gesetzt, wird aber ignoriert.. Bitte das aussagekräftiges Trace-Log oben einfügen.

jwedenig, 2024-09-07:
Hi, leider bekam ich eine Fehlermeldung beim Erstellen des Issues, zuviele Charakters oder so.
Hier das längere Log:
[evcc-20240907-121709-trace.log](https://github.com/user-attachments/files/16919493/evcc-20240907-121709-trace.log)

mdkeil, 2024-09-07:
Also der Ladevorgang beginnt bei 12:16:11, das Log endet bei 12:17:08 -- ich konnte da zumindest nicht feststellen, dass überhaupt der hold an die Batterie ging.. weil vermutlich zu kurz.. Meiner Ansicht nach wurde auch nicht mit "schnell" geladen, kann mich aber auch täuschen.

jwedenig, 2024-09-07:
ok, ich werde es morgen wieder versuchen und dann das Log posten, wie definiere ich die Länge des Logs?

mdkeil, 2024-09-07:
Also das Log wird geschrieben, sobald evcc gestartet ist.. da muss nichts definiert werden.. oder kopierst Du das immer über das UI? --das ist glaube nur ein Ausschnitt.. Da du Linux verwendest, sollte dir das gesamte Log zur Verfügung stehen.. wenn Du nicht genau weißt, was du davon kopieren sollst, einfach komplett hochladen aber zumindest grob die Zeit sagen, wann die Ladung gestartet wurde.. das verkürzt dann die Suche--

jwedenig, 2024-09-07:
ich machs hier:
[image]

jwedenig, 2024-09-07:
So, hier mal alles, ich werde es aber wie gesagt morgen mal versuchen und ein neues Log schicken bei Bedarf
[evcc.log](https://github.com/user-attachments/files/16919830/evcc.log)

Chris45671, 2024-09-07:
@jwedenig WIe DU ein vollständiges Log unter Linux erstellen kannst, findest Du auf der EVCC Seite unter Häufige Fragen.
https://docs.evcc.io/docs/faq#fehlersuche

jwedenig, 2024-09-07:
ja, danke, habs vor 3 Minuten geschickt

mucki12, 2024-09-08:
An welcher Stelle wird denn in einem der angehängten Logs auf "Schnell" umgeschaltet und anschließend mit 11kW geladen?

jwedenig, 2024-09-08:
Hier nochmals ein aktuelles Log von heute!
Habe ca. um 17 Uhr den Schnellmodus aktiviert, und warum auch immer heute funktioniert es.
Es kam zwar kurz die Batterie oben ins Bild für ca. 5 sec und seitdem nur Last vom Netz und Batterie gesperrt!
Freu mich, warum auch immer es jetzt funktioniert! Mal schauen, ob es so bleibt!
[evcc-1.log](https://github.com/user-attachments/files/16922730/evcc-1.log)

Chris45671, 2024-09-08:
Für einen Laien wie mich, sieht es doch ganz gut aus.

```
Sep 08 17:05:16 octopi evcc[15057]: [lp-1  ] DEBUG 2024/09/08 17:05:16 set charge mode: now
Sep 08 17:05:18 octopi evcc[15057]: [site  ] DEBUG 2024/09/08 17:05:18 set battery mode: hold
Sep 08 17:05:25 octopi evcc[15057]: [lp-1  ] DEBUG 2024/09/08 17:05:25 switched phases: 3p
Sep 08 17:05:26 octopi evcc[15057]: [lp-1  ] DEBUG 2024/09/08 17:05:26 max charge current: 16A
Sep 08 17:05:42 octopi evcc[15057]: [lp-1  ] INFO 2024/09/08 17:05:42 stop charging
Sep 08 17:06:03 octopi evcc[15057]: [site  ] DEBUG 2024/09/08 17:06:03 set battery mode: normal
Sep 08 17:06:12 octopi evcc[15057]: [lp-1  ] DEBUG 2024/09/08 17:06:12 !! session: chargeRater.chargedEnergy=0.4 - chargedAtStartup=0.0
```

Allerdings war der Ladevorgang mit unter 30sec sehr kurz, um hier wirklich zu sehen ob es über einen längeren Zeitraum funktioniert.

jwedenig, 2024-09-08:
Habe ca 30 min geladen und es hat funktioniert! Gestern nicht, warum auch immer

closed 2024-09-14 by andig