evcc-io/evcc #15971: EVCC startet nicht nach Upgrade von 0.130.7 auf 0.130.8

StevieC121176, 2024-09-08:
### Describe the bug

beim starten von EVCC kommt ein Fehler (Fatal), sh. Logfile 
`[main  ] FATAL 2024/09/08 17:18:17 cannot create meter 'db:8': cannot create meter type 'template': invalid key: minsoc`

geändert an der Konfiguration wurde nichts. Downgrade auf 0.130.7 läßt ivcc wieder starten

### Steps to reproduce

1. upgrade auf 0.130.8
2. start evcc

[evcc-20240908-171921-trace.log](https://github.com/user-attachments/files/16922724/evcc-20240908-171921-trace.log)

### Configuration details

```yaml
konfiguriert per EVCC UI
```

### Log details

```text
sh. logfile
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.8

andig, 2024-09-08:
> konfiguriert per EVCC UI

Was denn? Hast du zufällig einen Kostal?

andig, 2024-09-08:
Closed in https://github.com/evcc-io/evcc/commit/7bd213f0b17e55daa378e023ad7457073e6adc72

closed 2024-09-08 by andig

StevieC121176, 2024-09-08:
> > konfiguriert per EVCC UI
> 
> Was denn? Hast du zufällig einen Kostal?

Ja, 2 Kostal plenticore plus

StevieC121176, 2024-09-09:
kurze Rückmeldung.
Mit der aktuellen Nightly von heute morgen funktioniert es wieder. EVCC startet und auch die Batterieladung mit Netzbezug funktioniert.

JanSie85, 2024-09-10:
Bzw. wenn man den minsoc Eintrag aus der YAML löscht oder temporär ausklammert (via #) , startet evcc auch wieder normal und es funktioniert einwandfrei (PV Überschusslladen. Netzladung habe ich nicht probiert). Aber ich fand die Fehlermeldung wenigstens sehr eindeutig und man wusste sofort, wo man was beheben muss.

ulfinho, 2024-09-11:
Wie kann ich denn den minsoc ohne den Eintrag ändern, wenn ich den Standardwert von 20 zu niedrig finde?