evcc-io/evcc #15976: Cannot create "Meter:" with only x1 "currents" source ? Needed for the "circuits" load management

IamMattM, 2024-09-08:
### Describe the bug

I am trying to add a current clamp to my 1-phase mains grid.

However, it seems that the "currents" require three sources for a 3-phase grid that I do not have.

The error that I get:

```
/app # evcc meter
[main  ] INFO 2024/09/08 17:30:36 evcc 0.130.7
[main  ] INFO 2024/09/08 17:30:36 using config file: /etc/evcc.yaml
[db    ] INFO 2024/09/08 17:30:36 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2024/09/08 17:30:36 connecting evcc at tcp://192.168.1.180:1883
[mqtt  ] ERROR 2024/09/08 17:30:36 tcp://192.168.1.180:1883 connection lost: EOF
[main  ] FATAL 2024/09/08 17:30:36 cannot create meter 'myGRIDmeter': cannot create meter type 'custom': currents: need one per phase, total three
/app # 
```

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
Some extracts :

#################################################################################################################################################################
#################################################################################################################################################################
circuits: # Loadmanagement
- name: main
  title: main
  # maxCurrent: 60
  maxpower: 13800
  meter: myGRIDmeter
##################################################################################################################################################################
##################################################################################################################################################################

###################################################################################################################################################################
###################################################################################################################################################################
# Meters (current measurement devices) is a list of devices in the house that can measure power and energy consumption, PV generation, or house battery usage
meters:

  - name: myGRIDmeter
    type: custom
    power:
        source: mqtt
        topic: glow/4417935260B4/SENSOR/electricitymeter
        jq: .electricitymeter.power.value
        timeout: 30s # don't accept values older than timeout
        scale: 1000

    currents:
        source: mqtt
        topic: zigbee2mqtt/garage_CT_clamps
        jq: .current_a # Phase 1
        timeout: 30s # don't accept values older than timeout
        scale: 1.0

        # source: mqtt
        # topic: zigbee2mqtt/garage_CT_clamps
        # jq: .current_b # Phase 2
        # timeout: 30s # don't accept values older than timeout
        # scale: 1.0

        # source: mqtt
        # topic: zigbee2mqtt/garage_CT_clamps
        # jq: .current # Phase 3
        # timeout: 30s # don't accept values older than timeout
        # scale: 1.0
```

### Log details

```text
[main ] INFO 2024/09/08 17:30:06 evcc 0.130.7
[main ] INFO 2024/09/08 17:30:06 using config file: /etc/evcc.yaml
[db ] INFO 2024/09/08 17:30:06 using sqlite database: /root/.evcc/evcc.db
[mqtt ] INFO 2024/09/08 17:30:06 connecting evcc at tcp://192.168.1.180:1883
[mqtt ] DEBUG 2024/09/08 17:30:06 tcp://192.168.1.180:1883 connected
[main ] INFO 2024/09/08 17:30:06 listening at :7070
[main ] FATAL 2024/09/08 17:30:06 cannot create meter 'myGRIDmeter': cannot create meter type 'custom': currents: need one per phase, total three
[main ] FATAL 2024/09/08 17:30:06 will attempt restart in: 15m0s
[main ] INFO 2024/09/08 17:30:06 new version available: 0.130.8
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.7

IamMattM, 2024-09-08:
I have found a work-around by simply repeating the call for the same phase-1 current measure.

Seems to now work.

```
circuits: # Loadmanagement
- name: main
  title: main
  maxCurrent: 60
  maxpower: 13800
  meter: myGRIDmeter
```

```

meters:

  - name: myGRIDmeter
    type: custom
    power:
        source: mqtt
        topic: glow/4417935260B4/SENSOR/electricitymeter
        jq: .electricitymeter.power.value
        timeout: 30s # don't accept values older than timeout
        scale: 1000

    currents:
        - source: mqtt
          topic: zigbee2mqtt/garage_CT_clamps
          jq: .current_a # Phase 1
          timeout: 30s # don't accept values older than timeout
          scale: 1.0

        - source: mqtt
          topic: zigbee2mqtt/garage_CT_clamps
          jq: .current_a # Phase 1 (no Phase 2 on Grid)
          timeout: 30s # don't accept values older than timeout
          scale: 1.0

        - source: mqtt
          topic: zigbee2mqtt/garage_CT_clamps
          jq: .current_a # Phase 1 (no Phase 3 on Grid)
          timeout: 30s # don't accept values older than timeout
          scale: 1.0

```

IamMattM, 2024-09-08:
I get "evcc meter"
```
myGRIDmeter
-----------
Power:          877W
Current L1..L3: 4.97A 4.97A 4.97A
```

closed 2024-09-08 by IamMattM

VolkerK62, 2024-09-08:
Alternative:

```
currents:
        - source: mqtt
          topic: zigbee2mqtt/garage_CT_clamps
          jq: .current_a # Phase 1
          timeout: 30s # don't accept values older than timeout
          scale: 1.0

        - source: const
          value: 0

        - source: const
          value: 0
```