evcc-io/evcc #15991: Nach Update von 0.130.7 auf 0.130.8 Fehler am Charger EvBox Elvi

DaEffendi, 2024-09-09:
### Describe the bug

Hallo,
nach Update von 0.130.7 auf 0.130.8 Fehler am Charger EvBox Elvi. 
Ein anderer User hat das gleiche Problem #15984 
Ein downgrade auf 0.130.7 behebt das Problem.
Dankeschön!

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
chargers:
- type: template
  template: elvi 
  stationid: EVB-P2XXXXX
  connector: 1  
  idtag: 'XXXXXXX'
  connecttimeout: 5m  
  timeout: 2m  
  meter: false  
  name: my_charger1
```

### Log details

```text
sudo evcc -l trace charger
[main  ] INFO 2024/09/09 15:08:31 evcc 0.130.8
[main  ] INFO 2024/09/09 15:08:31 using config file: /etc/evcc.yaml
[db    ] INFO 2024/09/09 15:08:31 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2024/09/09 15:08:31 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/09/09 15:08:31 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/09/09 15:08:31 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/09/09 15:08:31 SELECT * FROM `settings` 4 <nil>
[db    ] TRACE 2024/09/09 15:08:34 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/09/09 15:08:34 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/09/09 15:08:34 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/09/09 15:08:34 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/09/09 15:08:34 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/09/09 15:08:34 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/09/09 15:08:34 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/09/09 15:08:34 SELECT * FROM `configs` WHERE `configs`.`class` = 1 0 <nil>
[EVB-P22214203-1] DEBUG 2024/09/09 15:08:34 waiting for chargepoint: 5m0s
[ocpp  ] DEBUG 2024/09/09 15:08:38 charge point connected: EVB-P2XXXXXX
[ocpp  ] TRACE 2024/09/09 15:08:38 send EVB-P2XXXXXXXX: [2,"315626860","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
[ocpp  ] TRACE 2024/09/09 15:08:38 recv EVB-P2XXXXXX: [2,"2e71afd5-df11-4674-b7e8-40cf7de8cfda","Heartbeat",{}]
[ocpp  ] TRACE 2024/09/09 15:08:38 send EVB-P2XXXXXXX: [3,"2e71afd5-df11-4674-b7e8-40cf7de8cfda",{"currentTime":"2024-09-09T13:08:38Z"}]
[EVB-P22214203-1] DEBUG 2024/09/09 15:09:08 failed configuring availability: timeout
[ocpp  ] TRACE 2024/09/09 15:09:37 recv EVB-P2XXXXXX: [2,"2e71b012-df11-4674-b7e8-40cfb19b9f3f","Heartbeat",{}]
[ocpp  ] TRACE 2024/09/09 15:09:37 send EVB-P2XXXXXX: [3,"2e71b012-df11-4674-b7e8-40cfb19b9f3f",{"currentTime":"2024-09-09T13:09:37Z"}]
[ocpp  ] TRACE 2024/09/09 15:09:38 send EVB-P2XXXXX: [2,"1386310307","GetConfiguration",{}]
[main  ] FATAL 2024/09/09 15:09:38 cannot create charger 'my_charger1': cannot create charger type 'template': cannot create charger type 'ocpp': timeout
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.130.8

andig, 2024-09-09:
Jetzt muss ich mich wundern. Wärst Du so nett und würdest noch ein 0.130.7er Log daneben legen?

DaEffendi, 2024-09-09:
Log von 0.130.7

sudo evcc -l trace charger
[main  ] INFO 2024/09/09 16:30:21 evcc 0.130.7
[main  ] INFO 2024/09/09 16:30:21 using config file: /etc/evcc.yaml
[db    ] INFO 2024/09/09 16:30:21 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2024/09/09 16:30:21 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/09/09 16:30:21 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/09/09 16:30:21 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/09/09 16:30:21 SELECT * FROM `settings` 4 <nil>
[db    ] TRACE 2024/09/09 16:30:24 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/09/09 16:30:24 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/09/09 16:30:24 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/09/09 16:30:24 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/09/09 16:30:24 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/09/09 16:30:24 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/09/09 16:30:24 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/09/09 16:30:24 SELECT * FROM `configs` WHERE `configs`.`class` = 1 0 <nil>
[EVB-P2XXXXX-1] DEBUG 2024/09/09 16:30:24 waiting for chargepoint: 5m0s
[ocpp  ] DEBUG 2024/09/09 16:30:25 charge point connected: EVB-P2XXXXXX
[ocpp  ] TRACE 2024/09/09 16:30:25 send EVB-PXXXXX: [2,"345140724","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
[ocpp  ] TRACE 2024/09/09 16:30:25 recv EVB-P2XXXX: [2,"2e71c300-df11-4674-b7e8-40cf823ee41a","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-09-09T14:30:15Z","sampledValue":[{"value":"0.0","context":"Trigger","measurand":"Current.Import","phase":"L1","unit":"A"},{"value":"0.0","context":"Trigger","measurand":"Current.Import","phase":"L2","unit":"A"},{"value":"0.0","context":"Trigger","measurand":"Current.Import","phase":"L3","unit":"A"},{"value":"90.0","context":"Trigger","measurand":"Current.Offered","unit":"A"},{"value":"0","context":"Trigger","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"0","context":"Trigger","measurand":"Power.Active.Import","unit":"W"},{"value":"32","context":"Trigger","measurand":"Temperature","location":"Body","unit":"Celsius"},{"value":"0","context":"Trigger","measurand":"Voltage","phase":"L1","unit":"V"},{"value":"0","context":"Trigger","measurand":"Voltage","phase":"L2","unit":"V"},{"value":"0","context":"Trigger","measurand":"Voltage","phase":"L3","unit":"V"},{"value":"0.00","context":"Trigger","measurand":"Frequency"}]}]}]
[ocpp  ] TRACE 2024/09/09 16:30:25 recv EVB-P2XXXXX: [2,"2e71c300-df11-4674-b7e8-40cfba70231c","DataTransfer",{"vendorId":"EV-BOX","messageId":"evbStatusNotification","data":"1,Available,NoError,No error.,0,Green,0,{0,8017,100},{0,0},{A,11984,12060,61568},0,0,320,2024-09-09T14:30:15Z,0,3,{0,0,0,0,0,0,0,0,0},240,0,0,97,0,270,900,0"}]
[ocpp  ] TRACE 2024/09/09 16:30:25 send EVB-P2XXXXXX: [3,"2e71c300-df11-4674-b7e8-40cfba70231c",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/09/09 16:30:39 recv EVB-P2XXXXXX: [2,"2e71c30f-df11-4674-b7e8-40cfe3db6673","Heartbeat",{}]
[ocpp  ] TRACE 2024/09/09 16:30:39 send EVB-P2XXXXXX: [3,"2e71c30f-df11-4674-b7e8-40cfe3db6673",{"currentTime":"2024-09-09T14:30:39Z"}]
[EVB-P2XXXXXXX] DEBUG 2024/09/09 16:30:55 failed configuring availability: timeout
[ocpp  ] TRACE 2024/09/09 16:31:25 send EVB-P2XXXXXX: [2,"3320360888","GetConfiguration",{}]
[ocpp  ] TRACE 2024/09/09 16:31:26 recv EVB-P2XXXXXXX: [3,"3320360888",{"configurationKey":[{"key":"AllowOfflineTxForUnknownId","readonly":false,"value":"false"},{"key":"AuthorizationCacheEnabled","readonly":false,"value":"true"},{"key":"AuthorizationKey","readonly":false,"value":"FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"},{"key":"AuthorizeRemoteTxRequests","readonly":false,"value":"true"},{"key":"ChargeProfileMaxStackLevel","readonly":true,"value":"6"},{"key":"ChargingScheduleAllowedChargingRateUnit","readonly":true,"value":"Current,Power"},{"key":"ChargingScheduleMaxPeriods","readonly":true,"value":"25"},{"key":"ClockAlignedDataInterval","readonly":false,"value":"0"},{"key":"ConnectionTimeOut","readonly":false,"value":"60"},{"key":"ConnectorPhaseRotation","readonly":false,"value":"0.Unknown,1.RST"},{"key":"ConnectorPhaseRotationMaxLength","readonly":true,"value":"21"},{"key":"ConnectorSwitch3to1PhaseSupported","readonly":true,"value":"false"},{"key":"DefaultChargingDelay","readonly":false,"value":"0"},{"key":"GetConfigurationMaxKeys","readonly":true,"value":"1"},{"key":"HeartbeatInterval","readonly":false,"value":"60"},{"key":"LightIntensity","readonly":false,"value":"25"},{"key":"LocalAuthListEnabled","readonly":false,"value":"true"},{"key":"LocalAuthListMaxLength","readonly":true,"value":"500"},{"key":"LocalAuthorizeOffline","readonly":false,"value":"true"},{"key":"LocalPreAuthorize","readonly":false,"value":"true"},{"key":"MaxChargingProfilesInstalled","readonly":true,"value":"66"},{"key":"MessageTimeout","readonly":false,"value":"60"},{"key":"MeterValuesAlignedData","readonly": [... cut ...]
[ocpp  ] TRACE 2024/09/09 16:31:31 send EVB-P2XXXXX: [2,"547046351","GetCompositeSchedule",{"connectorId":1,"duration":60}]
[ocpp  ] TRACE 2024/09/09 16:31:32 recv EVB-P2XXXXX: [3,"547046351",{"status":"Accepted","connectorId":1,"scheduleStart":"2024-09-09T14:31:31Z","chargingSchedule":{"duration":60,"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":0.0,"numberPhases":3}],"minChargingRate":0.0}}]
Power:          0W
Energy:         not available
Current L1..L3: 0A 0A 0A
Voltage L1..L3: not available
Charge status:  A
Enabled:        false
Identifier:     <none>

andig, 2024-09-09:
Die Änderung ist, dass 

    [2,"345140724","ChangeAvailability",{"connectorId":0,"type":"Operative"}]

in .7 nicht auf Rückgabe geprüft wird.

andig, 2024-09-09:
Jetzt bin ich ernsthaft verwirrt- denn auch die 0.130.7 prüft auf Ergebnis:

```go
func (cs *CS) ChangeAvailabilityRequest(id string, connector int, availabilityType core.AvailabilityType) error {
	rc := make(chan error, 1)
	err := cs.ChangeAvailability(id, func(request *core.ChangeAvailabilityConfirmation, err error) {
		if err == nil && request != nil && request.Status != core.AvailabilityStatusAccepted {
			err = errors.New(string(request.Status))
		}
		rc <- err
	}, connector, availabilityType)
	return Wait(err, rc, cs.timeout)
}
```

Jetzt bin ich ratlos.

/cc @premultiply hast Du eine Idee?

premultiply, 2024-09-09:
Es trifft nach dem Verbindungsaufbau keine `StatusNotification` innerhalb des Timeouts ein.

Das lange, vergebliche Warten auf die Rückmeldung von `ChangeAvailability` ist hier unnötig bis kontraproduktiv, da uns das Ergebnis ohnehin nicht interessiert. Dies ist ja nur eine Methode um manche Boxen überhaupt zur Zusammenarbeit zu bewegen und zweitens in einigen Fällen implizit eine StatusNotification anzutriggern.
Hier wird aber nur unnötig Zeit mit Warten verplempert statt einfach weiterzumachen und nach `getConfiguration `gleich eine `StatusNotification ` explizit anzutriggern.

Wir haben hier mehrere Möglichkeiten eine elementare `StatusNotification` zu erhalten:

1. Die Box schickt sie unaufgefordert direkt nach Verbindungsaufbau für alle Connectoren (so sollte es eigentlich streng nach Standard sein!).
2. `ChangeAvailability` triggert sie indirekt für alle Connectoren (`connectorId: 0`)
3. `TriggerMessage` triggert sie explizit (falls Trigger unterstützt werden (nach getConfiguration) für alle Connectoren
4. Theoretisch könnte man 2.+3. auch nochmal pro Connector machen (machen wir derzeit noch nicht, 2. würde HIER soweiso nicht bringen)
5. Der Watchdog führt 3. nochmal stumpf nach `Timeout/2` aus. Das scheint hier aber irgendwie nicht (mehr) zu funktionieren?

Letztlich muss nur eine `StatusNotification` innerhalb von Timeout nach dem Websockt-Verbindungsaufbau eintreffen.

Lange Rede, kurzer Sinn: Als allererstes muss hier mal das Warten auf die Antwort von `ChangeAvailability` weg.

andig, 2024-09-09:
>Das lange, vergebliche Warten auf die Rückmeldung von ChangeAvailability ist hier unnötig bis kontraproduktiv, da uns das Ergebnis ohnehin nicht interessiert.

Siehe https://github.com/evcc-io/evcc/issues/15991#issuecomment-2338571511. Das war in der alten Version auch so. Was hat sich also geändert?

premultiply, 2024-09-09:
Die Stelle im Code ist mehr oder weniger "Zufall".
Dort ist gerade der Timeout abgelaufen weil auf dem Channel bislang keine `StatusNotification` eingegangen ist.
Der Läuft ja im Hintergrund mit.

andig, 2024-09-09:
>Der Läuft ja im Hintergrund mit.

Wo?

premultiply, 2024-09-09:
Hier:
https://github.com/evcc-io/evcc/blob/4e9fb2f3ceb19f732aec3eee6dde057b8b04b0ee/charger/ocpp/connector.go#L133-L149

andig, 2024-09-09:
Verstehe ich nicht. Was läuft da im Hintergrund? `Initialized()` wird am Ende der Initialisierung aufgerufen. Zu dem Zeitpunkt ist `GetConfiguration` lange fertig. Davon sehe ich im Log nichts.

benkuly, 2024-09-10:
Haben das selbe Problem mit der ABB Terra AC seit dem neuesten update.

e-fahrer-jochen, 2024-09-10:
gleiches Problem bei meiner ELVI V3
nach dem downgrade startet die Version 0.130.7 bei mir erst nach einem reboot vom Rechner.

coderph0x, 2024-09-11:
Ich werfe ein, dass ich zwei Elvis habe, die sowohl mit .7 als auch mit .8 einwandfrei funktionieren.

musbach, 2024-09-12:
@coderph0x Auf welche Firmware-Version läuft deine EV-Box?

coderph0x, 2024-09-12:
In der App sehe ich P0424B0424v0.220706_W6.0...

Müsste also 424v0 sein.

musbach, 2024-09-12:
Ich bin auf 423v0.

pdonovan, 2024-09-13:
Strangely, I see the same `failed configuring availability: timeout` message with my ZJ Beny chargepoint, but only when I upgraded to 0.130.9 -  0.130.8 actually works fine with that CP.

musbach, 2024-09-13:
Hier noch mal die Trace-Ausschnitte mit den Versionen 0.130.7 und 0.130.9. Mit der 0.130.7 funktioniert es. Ich hoffe, dass es hilft:

```
$ sudo evcc -l trace charger
[main  ] INFO 2024/09/13 08:59:09 evcc 0.130.7
[main  ] INFO 2024/09/13 08:59:09 using config file: /home/satellit/evcc.yaml
[db    ] INFO 2024/09/13 08:59:09 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2024/09/13 08:59:09 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/09/13 08:59:09 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/09/13 08:59:09 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/09/13 08:59:09 SELECT * FROM `settings` 0 <nil>
[db    ] TRACE 2024/09/13 08:59:10 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/09/13 08:59:10 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/09/13 08:59:10 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/09/13 08:59:10 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/09/13 08:59:10 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/09/13 08:59:10 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/09/13 08:59:10 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/09/13 08:59:10 SELECT * FROM `configs` WHERE `configs`.`class` = 1 0 <nil>
[EVB-Pxyz-1] DEBUG 2024/09/13 08:59:10 waiting for chargepoint: 5m0s
[ocpp  ] DEBUG 2024/09/13 08:59:10 charge point connected: EVB-Pxyz
[ocpp  ] TRACE 2024/09/13 08:59:10 send EVB-Pxyz: [2,"229274085","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
[ocpp  ] TRACE 2024/09/13 08:59:22 recv EVB-Pxyz: [2,"2e769f4a-df11-4675-b7f9-40efb8cab50c","Heartbeat",{}]
[ocpp  ] TRACE 2024/09/13 08:59:22 send EVB-Pxyz: [3,"2e769f4a-df11-4675-b7f9-40efb8cab50c",{"currentTime":"2024-09-13T06:59:22Z"}]
[ocpp  ] TRACE 2024/09/13 08:59:32 recv EVB-Pxyz: [2,"2e769f53-df11-4675-b7f9-40ef21203748","MeterValues",{"connectorId":1,"transactionId":3,"meterValue":[{"timestamp":"2024-09-13T06:59:31Z","sampledValue":[{"value":"0.0","measurand":"Current.Import","phase":"L1","unit":"A"},{"value":"0.0","measurand":"Current.Import","phase":"L2","unit":"A"},{"value":"0.0","measurand":"Current.Import","phase":"L3","unit":"A"},{"value":"0.0","measurand":"Current.Offered","unit":"A"},{"value":"1079157","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"0","measurand":"Power.Active.Import","unit":"W"},{"value":"28","measurand":"Temperature","location":"Body","unit":"Celsius"},{"value":"234","measurand":"Voltage","phase":"L1","unit":"V"},{"value":"235","measurand":"Voltage","phase":"L2","unit":"V"},{"value":"235","measurand":"Voltage","phase":"L3","unit":"V"},{"value":"50.04","measurand":"Frequency"}]}]}]
[ocpp  ] TRACE 2024/09/13 08:59:33 recv EVB-Pxyz: [2,"2e769f54-df11-4675-b7f9-40efc0aadb49","DataTransfer",{"vendorId":"EV-BOX","messageId":"evbStatusNotification","data":"1,SuspendedEVSE,NoError,No error.,0,Yellow,1,{20,8008,0},{1079157,0},{B,11888,8764,61584},0,0,280,2024-09-13T06:59:31Z,3,42,{234,235,235,0,0,0,0,0,0},170,1,1,118,0,200,0,5004"}]
[ocpp  ] TRACE 2024/09/13 08:59:33 send EVB-Pxyz: [3,"2e769f54-df11-4675-b7f9-40efc0aadb49",{"status":"Accepted"}]
[EVB-Pxyz-1] DEBUG 2024/09/13 08:59:40 failed configuring availability: timeout
[ocpp  ] TRACE 2024/09/13 09:00:10 send EVB-Pxyz: [2,"1671759230","GetConfiguration",{}]
[ocpp  ] TRACE 2024/09/13 09:00:11 recv EVB-Pxyz: [3,"1671759230",{"configurationKey":[{"key":"AllowOfflineTxForUnknownId","readonly":false,"value":"false"},{"key":"AuthorizationCacheEnabled","readonly":false,"value":"true"},{"key":"AuthorizationKey","readonly":false,"value":"FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"},{"key":"AuthorizeRemoteTxRequests","readonly":false,"value":"true"},{"key":"ChargeProfileMaxStackLevel","readonly":true,"value":"6"},{"key":"ChargingScheduleAllowedChargingRateUnit","readonly":true,"value":"Current,Power"},{"key":"ChargingScheduleMaxPeriods","readonly":true,"value":"25"},{"key":"ClockAlignedDataInterval","readonly":false,"value":"0"},{"key":"ConnectionTimeOut","readonly":false,"value":"60"},{"key":"ConnectorPhaseRotation","readonly":false,"value":"0.Unknown,1.RST"},{"key":"ConnectorPhaseRotationMaxLength","readonly":true,"value":"21"},{"key":"ConnectorSwitch3to1PhaseSupported","readonly":true,"value":"false"},{"key":"DefaultChargingDelay","readonly":false,"value":"0"},{"key":"GetConfigurationMaxKeys","readonly":true,"value":"1"},{"key":"HeartbeatInterval","readonly":false,"value":"60"},{"key":"LightIntensity","readonly":false,"value":"50"},{"key":"LocalAuthListEnabled","readonly":false,"value":"true"},{"key":"LocalAuthListMaxLength","readonly":true,"value":"500"},{"key":"LocalAuthorizeOffline","readonly":false,"value":"true"},{"key":"LocalPreAuthorize","readonly":false,"value":"true"},{"key":"MaxChargingProfilesInstalled","readonly":true,"value":"66"},{"key":"MessageTimeout","readonly":false,"value":"60"},{"key":"MeterValuesAlignedData","readonly":false,"value":"Energy.Active.Import.Register"},{"key":"MeterValuesAlignedDat [... cut ...]
[EVB-Pxyz-1] DEBUG 2024/09/13 09:02:47 failed configuring availability: timeout
[ocpp  ] TRACE 2024/09/13 09:03:17 recv EVB-Pxyz: [2,"2e76a035-df11-4675-b7f9-40ef45576b85","Heartbeat",{}]
[ocpp  ] TRACE 2024/09/13 09:03:17 send EVB-Pxyz: [3,"2e76a035-df11-4675-b7f9-40ef45576b85",{"currentTime":"2024-09-13T07:03:17Z"}]
[ocpp  ] TRACE 2024/09/13 09:03:17 send EVB-Pxyz: [2,"4194758043","GetConfiguration",{}]
[main  ] FATAL 2024/09/13 09:03:17 cannot create charger 'wallbox6': cannot create charger type 'template': cannot create charger type 'ocpp': timeout
$
```

smiley1803, 2024-09-13:
Habe ebenfalls das gleiche Problem mit ABB Terra AC und 0.130.9 - mit 0.130.8 lief es bei mir noch

e-fahrer-jochen, 2024-09-13:
bin mal die Versionen zurück gegangen

diese Version startet nicht: 0.130.7+1725589553

diese Version startet 0.130.7+1725503150

andig, 2024-09-13:
Kannst du sagen, welche Commits das sind? Steht im Log!

e-fahrer-jochen, 2024-09-14:
keine Ahnung wo ich das finden kann?
Die Buchstaben und Zahlen hinter der Version in Klammern ändern sich nach jedem Test.

Hab mal main auf trace gestellt. Eventuell hilft das.

<details><summary>evcc 0.130.7 startet nicht</summary>
<p>

```
Sep 14 10:39:14 raspberrypi5 systemd[1]: Stopping evcc.service - evcc...
Sep 14 10:39:14 raspberrypi5 systemd[1]: evcc.service: Main process exited, code=exited, status=1/FAILURE
Sep 14 10:39:14 raspberrypi5 systemd[1]: evcc.service: Failed with result 'exit-code'.
Sep 14 10:39:14 raspberrypi5 systemd[1]: Stopped evcc.service - evcc.
Sep 14 10:39:20 raspberrypi5 systemd[1]: Started evcc.service - evcc.
Sep 14 10:39:20 raspberrypi5 evcc[29182]: [main  ] INFO 2024/09/14 10:39:20 evcc 0.130.7 (c35eaca3)
Sep 14 10:39:20 raspberrypi5 evcc[29182]: [main  ] INFO 2024/09/14 10:39:20 using config file: /etc/evcc.yaml
Sep 14 10:39:20 raspberrypi5 evcc[29182]: [main  ] INFO 2024/09/14 10:39:20 listening at :7070
Sep 14 10:39:24 raspberrypi5 evcc[29182]: [main  ] FATAL 2024/09/14 10:39:24 cannot create charger 'evbox': cannot create charger type 'template': cannot create charger type 'ocpp': decoding failed due to the following error(s):
Sep 14 10:39:24 raspberrypi5 evcc[29182]: error decoding 'Timeout': time: invalid duration "<no value>"
Sep 14 10:39:24 raspberrypi5 evcc[29182]: [main  ] FATAL 2024/09/14 10:39:24 will attempt restart in: 15m0s
Sep 14 10:39:24 raspberrypi5 evcc[29182]: [main  ] TRACE 2024/09/14 10:39:24 GET https://api.github.com/repos/evcc-io/evcc/releases/latest
Sep 14 10:39:24 raspberrypi5 evcc[29182]: [main  ] TRACE 2024/09/14 10:39:24 {"url":"https://api.github.com/repos/evcc-io/evcc/releases/174616705","assets_url":"https://api.github.com/repos/evcc-io/evcc/releases/174616705/assets","upload_url":"https://uploads.github.com/repos/evcc-io/evcc/releases/174616705/assets{?name,label}","html_url":"https://github.com/evcc-io/evcc/releases/tag/0.130.9","id":174616705,"author":{"login":"andig","id":184815,"node_id":"MDQ6VXNlcjE4NDgxNQ==","avatar_url":"https://avatars.githubusercontent.com/u/184815?v=4","gravatar_id":"","url":"https://api.github.com/users/andig","html_url":"https://github.com/andig","followers_url":"https://api.github.com/users/andig/followers","following_url":"https://api.github.com/users/andig/following{/other_user}","gists_url":"https://api.github.com/users/andig/gists{/gist_id}","starred_url":"https://api.github.com/users/andig/starred{/owner}{/repo}","subscriptions_url":"https://api.github.com/users/andig/subscriptions","organizations_url":"https://api.github.com/users/andig/orgs","repos_url":"https://api.github.com/users/andig/repos","events_url":"https://api.github.com/users/andig/events{/privacy}","received_events_url":"https://api.github.com/users/andig/received_events","type":"User","site_admin":false},"node_id":"RE_kwDODX4bUs4KaHCB","tag_name":"0.130.9","target_commitish":"master","name":"0.130.9","draft":false,"prerelease":false,"created_at":"2024-09-11T18:11:47Z","published_at":"2024-09-11T18:28:54Z","assets":[{"url":"https://api.github.com/repos/evcc-io/evcc/releases/assets/192020353","id":192020353,"node_id":"RA_kwDODX4bUs4Lcf-B","name":"checksums.txt","label":"","uploader":{"login":"andig","id":184815,"node_id":"MDQ6VXNlcjE4NDgxNQ==","avatar_url":"https://avatars.githubusercontent.com/u/184815?v=4","gravatar_id":"","url":"https://api.github.com/users/andig","html_url":"https://github.com/andig","followers_url":"https://api.github.com/users/andig/followers","following_url":"https://api.github.com/users/andig/following{/other_user}","gists_url":"https://api.github.com/users/andig/gists{/gist_id}","starred_url":"https://api.github.com/users/andig/starred{/owner}{/repo}","subscriptions_url":"https://api.github.com/users/andig/subscriptions","organizations_url":"https://api.github.com/users/andig/orgs","repos_url":"https://api.github.com/users/andig/repos","events_url":"https://api.github.com/users/andig/events{/privacy}","received_events_url":"https://api.github.com/users/andig/received_events","type":"User","site_admin":false},"content_type":"text/plain; charset=utf-8","state":"uploaded","size":754,"download_count":2,"created_at":"2024-09-11T18:40:18Z","updated_at":"2024-09-11T18:40:18Z","browser_download_url":"https://github.com/evcc-io/evcc/releases/download/0.130.9/checksums.txt"},{"url":"https://api.github.com/repos/evcc-io/evcc/releases/assets/192020339","id":192020339,"node_id":"RA_kwDODX4bUs4Lcf9z","name":"evcc_0.130.9_amd64.deb","label":"","uploader":{"login":"andig","id":184815,"node_id":"MDQ6VXNlcjE4NDgxNQ==","avatar_url":"https://avatars.githubusercontent.com/u/184815?v=4","gravatar_id":"","url":"https://api.github.com/users/andig","html_url":"https://github.com/andig","followers_url":"https://api.github.com/users/andig/followers","following_url":"https://api.github.com/users/andig/following{/other_user}","gists_url":"https://api.github.com/users/andig/gists{/gist_id}","starred_url":"https://api.github.com/users/andig/starred{/owner}{/repo}","subscriptions_url":"https://api.github.com/users/andig/subscriptions","organizations_url":"https://api.github.com/users/andig/orgs","repos_url":"https://api.github.com/users/andig/repos","events_url":"https://api.github.com/users/andig/events{/privacy}","received_events_url":"https://api.github.com/users/andig/received_events","type":"User","site_admin":false},"content_type":"application/vnd.debian.binary-package","state":"uploaded","size":24664136,"download_count":7,"created_ [... cut ...]
Sep 14 10:52:38 raspberrypi5 evcc[29656]: [ocpp  ] TRACE 2024/09/14 10:52:38 recv EVB-P22338027: [2,"2e780b56-df11-4666-b72a-40eb4b1c57bc","DataTransfer",{"vendorId":"EV-BOX","messageId":"evbStatusNotification","data":"1,Available,NoError,No error.,0,Off,0,{0,4752,100},{863225,0},{A,11872,12028,61568},0,0,300,2024-09-14T08:52:38Z,0,96,{237,238,237,0,0,0,0,0,0},200,1,2,113,0,210,900,5005"}]
Sep 14 10:52:38 raspberrypi5 evcc[29656]: [ocpp  ] TRACE 2024/09/14 10:52:38 send EVB-P22338027: [3,"2e780b56-df11-4666-b72a-40eb4b1c57bc",{"status":"Accepted"}]

``` 

</p>
</details>

andig, 2024-09-14:
Das sind:

```
evcc 0.130.7 (c35eaca3)
evcc 0.130.7 (9ca0bd48)
```

andig, 2024-09-14:
Ich habe den Fehler gefunden. Ein Timeout in `ChangeAvailability` führt dazu, dass der folgende `GetConfiguration` Request in einen Timeout läuft. Nach erster Einschätzung scheint das ein Problem in der OCPP Library zu sein das irgendwie durch die evcc Änderungen getriggert wird, ohne dass die konkrete Ursache erkennbar wäre.

e-fahrer-jochen, 2024-09-14:
@andig 
Ist das schon die neue Version?

apt list --upgradable
Auflistung… Fertig
evcc/bookworm **0.130.9+1726280751** arm64 [aktualisierbar von: 0.130.7+1725503150]

[image]

würde dann gleich testen.

andig, 2024-09-14:
>Ist das schon die neue Version?

Bitte nicht jedesmal nachfragen. Wenn ein PR nicht gemerged ist kann er auch nicht im Build sein. Danke.

Wäre jemand in der Lage, https://github.com/evcc-io/evcc/pull/16100 zu compilieren und testen? Der eigens gebaute Testfall zeigt den Fehler noch, ich bin aber nicht mehr sicher, ob das noch real ist oder nur den Testbedingungen zuzuschreiben. Evtl. könnte das Problem also schon gelöst sein.

emkadoc, 2024-09-14:
@andig Ich habe es gebaut und "real" getestet -> [main  ] INFO 2024/09/14 13:26:35 evcc 0.130.9 (518ba81c)
Meine beiden ABB Terra ACs verbinden sich und evcc startet somit bei mir zumindest wieder korrekt. Vielen Dank! :satisfied:

andig, 2024-09-14:
Zur Erläuterung: in https://github.com/evcc-io/evcc/pull/16100/files#diff-a3867ee833cfe8f6b556ca94f3a9356f5eb08ae4a9f093552529aa74dd11455cL35 haben wir 1m Timeout gesetzt. In https://github.com/evcc-io/evcc/pull/16100/files#diff-c4ed558802ad437478e41f32be5fe1f2cf43f3360749e2b4a9b7f7375953d5edL18 brechen wir jedoch mit einem _anderen_ Timeout schon nach 30s ab. Die Ocpp Library wartet jedoch intern weiter auf ihren Timeout. Der schlägt dann beim nächsten Request- nämlich `GetConfiguration` zu.

Die Lösung besteht jetzt darin, den evcc-spezifischen Timeout zu eliminieren und alle Warteschritte der Library zu überlassen. Dafür wird der Timeout der Library wieder auf den Standardwert (welchen auch evcc) verwendet verkürzt.

closed 2024-09-14 by andig

musbach, 2024-09-14:
Vielen Dank! Mit der 0.130.10 läuft die EVBox wieder!