evcc-io/evcc #15994: Battery Charging Optimization: Flexible Control of SOC and Charging Power

Loony2392, 2024-09-09:
Describe the solution you'd like

Deutsch:

Ich würde mir wünschen, dass man beim Netzladen der Batterie einen minimalen und maximalen SOC-Wert angeben kann, um genau festzulegen, wann der Ladevorgang starten oder stoppen soll. Darüber hinaus wäre es toll, die Ladeleistung flexibel anpassen zu können, indem man auch hier minimale und maximale Werte definiert. So könnte das Laden optimal an die jeweiligen Anforderungen angepasst werden.

Englisch:

I would like the ability to specify minimum and maximum SOC values when charging the battery from the grid, to precisely determine when the charging process should start or stop. Additionally, it would be great to adjust the charging power flexibly by defining minimum and maximum values, allowing the charging process to be optimally tailored to specific needs.

Describe alternatives you've considered

Deutsch:

Eine Alternative wäre, den Ladevorgang manuell zu überwachen und bei Erreichen des gewünschten Ladezustands einzugreifen. Dies ist jedoch eher unpraktisch und aufwendig. Auch die Nutzung von externen Tools oder Geräten habe ich in Erwägung gezogen, aber eine direkt integrierte Lösung wäre sicherlich viel benutzerfreundlicher und eleganter.

Englisch:

An alternative would be to monitor the charging process manually and intervene when the desired state of charge is reached. However, this is rather impractical and time-consuming. I've also considered using external tools or devices, but a directly integrated solution would certainly be much more user-friendly and elegant.

Additional context

Deutsch:

Mit einer solchen Funktion könnte der Ladevorgang deutlich effizienter und flexibler gestaltet werden. Besonders in Situationen, in denen der Ladevorgang nach bestimmten Zeitplänen oder Energieverfügbarkeiten erfolgen soll, wäre die Möglichkeit, sowohl den Ladezustand als auch die Ladeleistung zu regulieren, ein großer Vorteil.

Englisch:

With such a feature, the charging process could be made significantly more efficient and flexible. Especially in situations where charging should occur according to specific schedules or energy availability, the ability to regulate both the state of charge and the charging power would be a major advantage.

andig, 2024-09-09:
Danke für die ausführliche Beschreibung- noch hat es aber nicht "klick" gemacht...

> So könnte das Laden optimal an die jeweiligen Anforderungen angepasst werden.

Welche wären das bzw. was ist der Use Case?

> Mit einer solchen Funktion könnte der Ladevorgang deutlich effizienter und flexibler gestaltet werden. Besonders in Situationen, in denen der Ladevorgang nach bestimmten Zeitplänen oder [...]

Genau das würde ich als Nachteil sehen- da dann detaillierte Anforderungen an Konfigurierbarkeit von Zeitplänen die nächste Folge wären?

Generell möchten wir a) leicht verständliche Use Cases ausliefern und dabei b) die individuelle (externe) Konfigurierbarkeit des Produktes über Schnittstellen hoch halten.

core-tinkerer, 2024-09-09:
Hi, bin gerade hierauf gestoßen nachdem ich hier war: https://github.com/evcc-io/evcc/discussions/15557

Ich sehe hier 2 in 1:

Als User würde ich die Funktion Netzladen auf einen konfigurierbaren max SoC setzen wollen. Dies soll ermöglichen, dass ich zu Zeiten günstigen Stroms einen konfigurierbarten SoC erreiche aber noch "Luft" habe um PV aufzunehmen. Z.B. im Winter mit i.d.R. günstigerem Nacht-Strom netzdienlich auf z.B. 80% bis täglich 06:00 Uhr laden und die wenige Wintersonne noch einspeichern zu können. Teils fast vergleichbar mit den Ladeplänen für eAuto. 

Weiterhin möchte ich die Ladeleistung der Batterie auf einen konfigurierbaren Wert setzen. Hier ist aus meiner ehrlichen persönlichen Sicht der Grund, dass die Fronius Gen24 nur mit 200-500W freiwillig aus dem Netz laden. Die benötigen eine explizite Ladeleistung in % um mehr zu machen. Statt einer Sonderlocke für den Gen24 (vermutlich ein extra Modbus Reg 40366 = 10000 mitzugeben) könnte ein konfigurierbarer Wert eine aufwändigere aber generische Lösung sein. Sollten andere WR auch so was machen könnte man das generisch über die template yaml als advanced mit einbauen.

[Fronius Gen24, BYD HVM, 2x Keba P30]

Loony2392, 2024-09-09:
Hallo,

wie @core-tinkerer bereits gut erklärt hat, wäre es äußerst hilfreich, den SOC-Wert (State of Charge) innerhalb von EVCC flexibel begrenzen zu können. So könnte man die Batterie bei günstigen Strompreisen bis zu einem definierten Wert laden und dennoch Kapazität für PV-Leistung freihalten.

Aktuell beträgt die Ladeleistung meines Fronius Gen24 beim Netzbezug lediglich 100 bis 200 W. Das ist bei einem Speichervolumen von über 10 kWh kaum effizient. Eine Erweiterung von EVCC, die es ermöglicht, gezielt mit höherer Leistung (z.B. 1-2 kW) zu laden, wäre ideal – insbesondere, wenn der Strompreis für die nächsten Stunden günstig ist.

Diese Funktion würde es ermöglichen, den Speicher präzise zu steuern und gleichzeitig die Vorteile des dynamischen Stromtarifs voll auszuschöpfen. Dadurch könnte man nicht nur Kosten sparen, sondern auch die Eigenverbrauchsquote maximieren, indem man bei günstigen Preisen lädt und bei Bedarf weiterhin PV-Leistung aufnehmen kann. 

Eine solche Erweiterung würde EVCC noch flexibler und effizienter machen, insbesondere im Hinblick auf die wirtschaftliche Nutzung von Stromspeichern. 😊

andig, 2024-09-09:
> Eine Erweiterung von EVCC, die es ermöglicht, gezielt mit höherer Leistung (z.B. 1-2 kW) zu laden, wäre ideal – insbesondere, wenn der Strompreis für die nächsten Stunden günstig ist.

Bitte neues Thema. Dafür müsste dann auch klar sein, wie das funktionieren soll wenn die aktuelle Implementierung das nicht her gibt.

andig, 2024-09-09:
> wie @core-tinkerer bereits gut erklärt hat, wäre es äußerst hilfreich, den SOC-Wert (State of Charge) innerhalb von EVCC flexibel begrenzen zu können. So könnte man die Batterie bei günstigen Strompreisen bis zu einem definierten Wert laden und dennoch Kapazität für PV-Leistung freihalten.

Wäre die Lösung da nicht eher eine Ladeplanung?

Loony2392, 2024-09-09:
> Wäre die Lösung da nicht eher eine Ladeplanung?

ja, eine Ladeplanung wäre hier wahrscheinlich die bessere Lösung. 

Mein Vorschlag wäre:
Man könnte den Ladeprozess basierend auf dem PV-Forecast, dem aktuellen SOC und den Strompreisen dynamisch steuern. Dabei würden Ladezeitfenster automatisch berechnet, und der SOC wird so begrenzt, dass genug Kapazität für PV-Leistung übrig bleibt. 

Oder wie würdest du dies aufbauen wollen ?

andig, 2024-09-09:
Genau wie Ladeplanung heute bis Limit. Forecast wäre dann ein separates Thema. Gibts Erfahrungswerte, ob/dass die günstigen Slots tatsächlich nachts liegen?

Loony2392, 2024-09-09:
Aus der Vergangenheit kann ich sagen das im Winter die Slots meistens Nachts lagen zumindest bei mir mit Tibber 0-4 Uhr meistens.

Im Sommer aktuell eher Mittag und teilweise Nachts wobei das auch sehr stark Wetter Abhängig ist.

Ich könnte dir aber wenn du genaue Werte brauchst auch einen Report aus der Datenbank besorgen.

core-tinkerer, 2024-09-09:
Wegs Preise is das hier nützlich: https://energy-charts.info/charts/price_heatmaps/chart.htm?l=de&c=DE
x = Tag, y= Zeit. Im Frühjahr ist mehr blau 0-6h, im Sommer in der Mitte 12-15h (viel PV und Wind). Rot (teuer) ist morgends und Abends der Übergang von der i.d.R. nächtlichen Grundlast bis die Erneuerbaren richtig anfangen - die hochphase der schnellen aber teuren Gaskraftwerke....

core-tinkerer, 2024-09-09:
Stimme @andig zu dass das aufgetrennt werden muss. IMHO:

1) "Fix" für WR wie Gen24 damit diese überhaupt laden (i.e. 100% default). Vielleicht gibt's noch mehr. Den threads nach könnte das sein.
2) Konfigurierbares Leistungs-Limit für Netzladen der Hausbatterie (für die die das unterstützen)
3) Ladeplanung Hausbatterie
4) More sophisticated ;o): PV-Forecast mit einbinden. Aber das ist imho ein dickeres Brett. Da muss auch mehr Konfig überall rein wie z.B. mehrere PV-Flächen mit je Anlagen-kWp, Anlagen-Standort, Anlagen-Ausrichtung und -Neigung damit z.B. https://forecast.solar/ (evtl mit api key) angebunden werden kann. Und einiges an Hirnschmalz wie man das in einen Regel-Algo packt der dann noch das Auto berücksichtigt...

[Kein Algorithmus kann mehr PV Panele ersetzen ;o) ]

andig, 2024-09-20:
> Aktuell beträgt die Ladeleistung meines Fronius Gen24 beim Netzbezug lediglich 100 bis 200 W. Das ist bei einem Speichervolumen von über 10 kWh kaum effizient. 

Das ist per https://github.com/evcc-io/evcc/pull/16209 behoben. Ladeplanung schauen wir uns an und Forecast ist nicht ganz trivial. Akutes Problem erstmal gelöst 👍🏻

closed 2024-09-20 by andig

schnekrehuesli, 2024-10-05:
Gibt es zur Ladeplanung bereits weitere Überlegungen?
Das könnte m.E. relativ simpel sein, wie im Screenshop in [#16008](https://github.com/evcc-io/evcc/issues/16008) angedeutet: also Netzladung bis xx % SOC. Das wäre, wie schon hier erwähnt, vor allem im Winter sinnvoll.
Mein Fronius Gen24 lädt bei evcc 0.130.13 mit maximaler Leistung in die Batterie. Eine Begrenzung der Ladeleistung wäre nice-to-have, um die Batterie zu schonen aber nicht oberste Priorität.

andig, 2024-10-05:
Dieses Issue ist erledigt. Gerne neues für Deinen Use Case.

schnekrehuesli, 2024-10-05:
d.h. Ihr implementiert die Begrenzung der Netzladung bis xx % SOC bereits? Der Use Case ist hier ja schon ziemlich gut beschrieben.
Oder soll ich nochmal einen Issue damit aufmachen?
Sorry für die Nachfrage, bin neu hier, finde  evcc ziemlich genial.

andig, 2024-10-06:
Der Punkt war eher, dass wir die Ladeplanung auch für den Heimakku ermöglichen wollen. Noch gibts das aber nicht. Eine Soc-Begrenzung kannst Du bis dahin extern über die APIs umsetzen indem Du das Ladefeature einfach abschaltest wenn Dir der Akku hinreichend voll ist.