evcc-io/evcc #16000: "Lock Battery" during fast charging not working since 0.130.0 - Sungrow Hybrid Inverter

andyhielscher, 2024-09-09:
### Describe the bug

Enabling the setting "Verhindere Entladung im Schnell-Modus und bei geplantem Laden." does not lock the battery but it appears an error in the log:
[site ] ERROR 2024/09/09 18:45:02 battery mode: write tcp [ip EVCC]:55410->[ip sungrow SH inverter]:502: write: broken pipe

### Steps to reproduce

1. Clean install of evcc
2. use Sungrow SH inverter
3. Start fast charging the car
4. switching the described setting causes error message

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local
  port: 7070

log: debug

interval: 30s

levels:
  cache: error

plant: xxx

sponsortoken: xxx

meters:
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.1.14
  port: 502
  timeout: 10s
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.1.14
  port: 502
  timeout: 10s
  usage: pv
  modbus: tcpip
  name: pv1
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.1.14
  port: 502
  timeout: 10s
  usage: battery
  modbus: tcpip
  name: battery1
  capacity: 8.64

chargers:
- type: template
  template: go-e-v3
  host: 192.168.1.16
  name: wallbox1

loadpoints:
- title: Go-E Garage
  charger: wallbox1
  mode: pv
  enable:
    delay: 10m
  disable:
    delay: 2m

site:
  title: Zuhause
  meters:
    grid: grid1
    pv:
    - pv1
    battery:
    - battery1
  residualPower: 10

tariffs:
  currency: EUR
  grid:
    type: tibber
    token: xxx
  feedin:
    type: fixed
    price: 0.08
```

### Log details

```text
evcc meter --log trace,db:error --battery-mode hold --battery-mode-wait 1m battery1
[main  ] INFO 2024/09/09 18:38:47 evcc 0.130.8
[main  ] INFO 2024/09/09 18:38:47 using config file: /etc/evcc.yaml
[modbus] TRACE 2024/09/09 18:38:48 modbus: send 00 01 00 00 00 06 01 06 32 f9 00 02
[modbus] TRACE 2024/09/09 18:38:48 modbus: recv 00 01 00 00 00 06 01 06 32 f9 00 02
[modbus] TRACE 2024/09/09 18:38:48 modbus: send 00 01 00 00 00 06 01 06 32 fa 00 cc
[modbus] TRACE 2024/09/09 18:38:48 modbus: recv 00 01 00 00 00 06 01 06 32 fa 00 cc
[main  ] INFO 2024/09/09 18:38:48 waiting for: 1m0s
[modbus] TRACE 2024/09/09 18:39:48 modbus: closing connection due to idle timeout: 1m0.000584391s

evcc meter --log trace,db:error --battery-mode normal --battery-mode-wait 1m battery1
[main  ] INFO 2024/09/09 18:39:57 evcc 0.130.8
[main  ] INFO 2024/09/09 18:39:58 using config file: /etc/evcc.yaml
[modbus] TRACE 2024/09/09 18:39:58 modbus: send 00 01 00 00 00 06 01 06 32 f9 00 00
[modbus] TRACE 2024/09/09 18:39:58 modbus: recv 00 01 00 00 00 06 01 06 32 f9 00 00
[modbus] TRACE 2024/09/09 18:39:58 modbus: send 00 01 00 00 00 06 01 06 32 fa 00 cc
[modbus] TRACE 2024/09/09 18:39:58 modbus: recv 00 01 00 00 00 06 01 06 32 fa 00 cc
[main  ] INFO 2024/09/09 18:39:58 waiting for: 1m0s
[modbus] TRACE 2024/09/09 18:40:58 modbus: closing connection due to idle timeout: 1m0.000598163s
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.130.8

andig, 2024-09-09:
Hast Du das nicht schonmal gepostet?

andig, 2024-09-09:
> [site ] ERROR 2024/09/09 18:45:02 battery mode: write tcp [ip EVCC]:55410->[ip sungrow SH inverter]:502: write: broken pipe

Ich sehe keinen Fehler im Log?

DarcDan, 2024-09-10:
Also ich habe ebenfalls dieses Verhalten @andyhielscher setzt du auch den SH15T mit SBRXXX ein?

andyhielscher, 2024-09-10:
@andig Nein, ich hatte diesen Fehler noch nicht gepostet, das war jemand anders. Bitte schließe das Issue noch nicht, ich werde heute Nachmittag versuchen, bei dem Error auch mit loglevel: trace vielleicht mehr Informationen herauszubekommen. Dann werde ich das hier hinzufügen. 

@DarcDan Ich setze einen SH5RT mit SBR096 ein. Jemand anders hatte das Problem mit SHxx und BYD Batterie.

Dunkelgraf, 2024-09-10:
Das Verhalten tritt bei mir auch auf. Ich setze eine. Sungrow SH10RT mit SBR096 ein.

andig, 2024-09-10:
"ich auch" hilft nicht. Hier fehlt weiter ein Logfile.

DarcDan, 2024-09-10:
Verstehe ich @andig ist nur nicht immer direkt auf die schnelle alles zusammen zu tragen. Unser Auto ist eben zu Hause an gekommen und ich habe es zum Laden angesteckt, diesmal einfach nur PV Überschussladen Batterie auf 100% in der Konfig, trotzdem entnimmt er der Batterie den Saft.

Anbei mal Screenshots und Logs, wenn mehr benötigt wird muss ich das Szenario noch mal umbauen. Ich vermute aber das es hier um das gleiche Problem geht das er die Batterie nicht sperrt. Wenn ich mich irre kann ich gerne noch mal ein neues Thema draus machen

[image]
[image]

site ] DEBUG 2024/09/10 12:55:40 ----
[lp-1 ] DEBUG 2024/09/10 12:55:40 charge power: 4018W
[lp-1 ] DEBUG 2024/09/10 12:55:40 charge currents: [6.17 6 5.91]A
[site ] DEBUG 2024/09/10 12:55:40 pv power: 1145W
[site ] DEBUG 2024/09/10 12:55:40 battery soc: 97%
[site ] DEBUG 2024/09/10 12:55:40 battery power: 0W
[site ] DEBUG 2024/09/10 12:55:40 grid meter: -0W
[site ] DEBUG 2024/09/10 12:55:40 grid currents: [0.41 -0.0819 -0.531]A
[site ] DEBUG 2024/09/10 12:55:40 site power: 0W
[lp-1 ] DEBUG 2024/09/10 12:55:40 detected active phases: 3p
[lp-1 ] DEBUG 2024/09/10 12:55:40 !! session: chargeRater.chargedEnergy=0.0 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/09/10 12:55:40 charge total import: 76.660kWh
[lp-1 ] DEBUG 2024/09/10 12:55:40 charger status: C
[lp-1 ] DEBUG 2024/09/10 12:55:40 soc estimated: 81.00% (vehicle: 81.00%)
[lp-1 ] DEBUG 2024/09/10 12:55:40 vehicle soc: 81%
[lp-1 ] DEBUG 2024/09/10 12:55:40 vehicle range: 220km
[lp-1 ] DEBUG 2024/09/10 12:55:40 pv charge current: 6A = 6A + -0A (0W @ 3p)
[site ] DEBUG 2024/09/10 12:55:55 ----
[lp-1 ] DEBUG 2024/09/10 12:55:55 charge power: 4018W
[lp-1 ] DEBUG 2024/09/10 12:55:55 charge currents: [6.17 6 5.91]A
[site ] DEBUG 2024/09/10 12:55:55 pv power: 1145W
[site ] DEBUG 2024/09/10 12:55:55 battery soc: 97%
[site ] DEBUG 2024/09/10 12:55:55 battery power: 0W
[site ] DEBUG 2024/09/10 12:55:55 grid meter: -0W
[site ] DEBUG 2024/09/10 12:55:55 grid currents: [0.41 -0.0819 -0.531]A
[site ] DEBUG 2024/09/10 12:55:55 site power: 0W
[lp-1 ] DEBUG 2024/09/10 12:55:55 detected active phases: 3p
[lp-1 ] DEBUG 2024/09/10 12:55:55 !! session: chargeRater.chargedEnergy=0.0 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/09/10 12:55:55 charge total import: 76.660kWh
[lp-1 ] DEBUG 2024/09/10 12:55:55 charger status: C
[lp-1 ] DEBUG 2024/09/10 12:55:55 soc estimated: 81.00% (vehicle: 81.00%)
[lp-1 ] DEBUG 2024/09/10 12:55:55 vehicle soc: 81%
[lp-1 ] DEBUG 2024/09/10 12:55:55 vehicle range: 220km
[lp-1 ] DEBUG 2024/09/10 12:55:55 pv charge current: 6A = 6A + -0A (0W @ 3p)

premultiply, 2024-09-10:
> trotzdem entnimmt er der Batterie den Saft.

> [site ] DEBUG 2024/09/10 12:55:55 battery power: 0W

Irgendwie nicht nachvollziehbar.
Irgendwas scheint an den Meteringdaten auch nicht ganz zu stimmen.

Darüber hinaus wirkt die Batteriesperre ja nur im Modus "Sofort", nicht bei PV.

DarcDan, 2024-09-10:
OK, das scheint ja dann noch einmal ein anderes Problem zu sein. Soll ich dazu noch mal ein eigenes Thema öffnen? Ich könnte mir gut vorstellen das es sich um ein Problem handelt den WiNet-S2 Dongle auszulesen? Da bin ich aber überhaupt nicht im Thema.

Dunkelgraf, 2024-09-10:
> "ich auch" hilft nicht. Hier fehlt weiter ein Logfile.

Hier ein Auszug aus dem Log:
[modbus] TRACE 2024/09/10 11:06:58 modbus: send 00 2b 00 00 00 06 01 06 32 f9 00 02
[site  ] ERROR 2024/09/10 11:06:58 battery mode: write tcp 172.18.0.2:45926->192.168.20.168:502: write: broken pipe
[modbus] TRACE 2024/09/10 11:06:58 modbus: send 36 d8 00 00 00 06 01 04 13 98 00 02
[modbus] TRACE 2024/09/10 11:06:58 modbus: recv 36 d8 00 00 00 07 01 04 04 10 bb 00 00
[modbus] TRACE 2024/09/10 11:06:58 modbus: send 1b c0 00 00 00 06 01 04 32 ca 00 02
[modbus] TRACE 2024/09/10 11:06:58 modbus: recv 1b c0 00 00 00 07 01 04 04 48 1a 00 01
[site  ] DEBUG 2024/09/10 11:06:58 pv power: 4283W
[modbus] TRACE 2024/09/10 11:06:58 modbus: send 1b d5 00 00 00 06 01 04 32 dd 00 01
[modbus] TRACE 2024/09/10 11:06:58 modbus: recv 1b d5 00 00 00 05 01 04 02 10 4e
[modbus] TRACE 2024/09/10 11:06:58 modbus: send 1b d4 00 00 00 06 01 04 32 c8 00 01
[modbus] TRACE 2024/09/10 11:06:58 modbus: recv 1b d4 00 00 00 05 01 04 02 00 2d
[modbus] TRACE 2024/09/10 11:06:58 modbus: send 1b c1 00 00 00 06 01 04 32 c8 00 01
[modbus] TRACE 2024/09/10 11:06:58 modbus: recv 1b c1 00 00 00 05 01 04 02 00 2d
[modbus] TRACE 2024/09/10 11:06:58 modbus: send 1b c0 00 00 00 06 01 04 32 e2 00 02
[modbus] TRACE 2024/09/10 11:06:58 modbus: recv 1b c0 00 00 00 07 01 04 04 43 99 00 00
[modbus] TRACE 2024/09/10 11:06:58 modbus: send 1b c0 00 00 00 06 01 04 32 de 00 01
[modbus] TRACE 2024/09/10 11:06:58 modbus: recv 1b c0 00 00 00 05 01 04 02 00 d2
[site  ] DEBUG 2024/09/10 11:06:58 battery soc: 21%
[site  ] DEBUG 2024/09/10 11:06:58 battery power: 4174W
[modbus] TRACE 2024/09/10 11:06:58 modbus: send 1b c3 00 00 00 06 01 04 32 d1 00 02
[modbus] TRACE 2024/09/10 11:06:58 modbus: recv 1b c3 00 00 00 07 01 04 04 f2 e0 ff ff
[site  ] DEBUG 2024/09/10 11:06:58 grid meter: 3360W
[modbus] TRACE 2024/09/10 11:06:58 modbus: send 1b c0 00 00 00 06 01 04 15 e2 00 02
[modbus] TRACE 2024/09/10 11:06:58 modbus: recv 1b c0 00 00 00 07 01 04 04 06 30 00 00
[modbus] TRACE 2024/09/10 11:06:58 modbus: send 1b 8b 00 00 00 06 01 04 13 9a 00 01
[modbus] TRACE 2024/09/10 11:06:58 modbus: recv 1b 8b 00 00 00 05 01 04 02 09 27
[modbus] TRACE 2024/09/10 11:06:58 modbus: send 1b 8e 00 00 00 06 01 04 15 e4 00 02
[modbus] TRACE 2024/09/10 11:06:58 modbus: recv 1b 8e 00 00 00 07 01 04 04 01 be 00 00
[modbus] TRACE 2024/09/10 11:06:58 modbus: send 1b 5d 00 00 00 06 01 04 13 9b 00 01
[modbus] TRACE 2024/09/10 11:06:58 modbus: recv 1b 5d 00 00 00 05 01 04 02 09 16
[modbus] TRACE 2024/09/10 11:06:58 modbus: send 1b 8d 00 00 00 06 01 04 15 e6 00 02
[modbus] TRACE 2024/09/10 11:06:58 modbus: recv 1b 8d 00 00 00 07 01 04 04 01 e7 00 00
[modbus] TRACE 2024/09/10 11:06:58 modbus: send 1b 55 00 00 00 06 01 04 13 9c 00 01
[modbus] TRACE 2024/09/10 11:06:58 modbus: recv 1b 55 00 00 00 05 01 04 02 09 31
[site  ] DEBUG 2024/09/10 11:06:58 grid currents: [6.76 1.92 2.07]A
[modbus] TRACE 2024/09/10 11:06:58 modbus: send 1b c0 00 00 00 06 01 04 32 ec 00 02
[modbus] TRACE 2024/09/10 11:06:58 modbus: recv 1b c0 00 00 00 07 01 04 04 56 4f 00 00
[site  ] DEBUG 2024/09/10 11:06:58 site power: 7734W
[lp-2  ] DEBUG 2024/09/10 11:06:58 charge total import: 274.636kWh
[lp-2  ] DEBUG 2024/09/10 11:06:58 charger status: A
[site  ] DEBUG 2024/09/10 11:07:28 ----
[lp-1  ] DEBUG 2024/09/10 11:07:28 charge power: 10552W
[lp-1  ] DEBUG 2024/09/10 11:07:28 charge currents: [15.2 15.1 15.2]A
[lp-2  ] DEBUG 2024/09/10 11:07:28 charge power: 0W
[lp-2  ] DEBUG 2024/09/10 11:07:28 charge currents: [0 0 0]A
[site  ] WARN 2024/09/10 11:07:28 planner: outdated
[modbus] TRACE 2024/09/10 11:07:28 modbus: send 00 2c 00 00 00 06 01 06 32 f9 00 02
[site  ] ERROR 2024/09/10 11:07:28 battery mode: write tcp 172.18.0.2:37732->192.168.20.168:502: write: broken pipe
[modbus] TRACE 2024/09/10 11:07:28 modbus: send 36 d9 00 00 00 06 01 04 13 98 00 02
[modbus] TRACE 2024/09/10 11:07:28 modbus: recv 36 d9 00 00 00 07 01 04 04 1a 70 00 00
[modbus] TRACE 2024/09/10 11:07:28 modbus: send 1b c1 00 00 00 06 01 04 32 ca 00 02
[modbus] TRACE 2024/09/10 11:07:28 modbus: recv 1b c1 00 00 00 07 01 04 04 48 1b 00 01
[site  ] DEBUG 2024/09/10 11:07:28 pv power: 6768W

Die Umschaltung hat nach ca. einer Stunde laden funktioniert, nach Beendigung des Ladevorgangs gab es beim freigeben der Batterie zum gleichen Problem. Ich kann die kompletten Logs hier anhängen wenn das gewünscht ist.
Der Sungrow WR ist über den zweiten Anschluß angebunden, nicht über das WiNet-Dongle.

andig, 2024-09-11:
> Der Fehler tritt an der gleichen Stelle auf, laut TRACE wird per Modbus 00 02 00 00 00 06 01 06 32 f9 00 02 gesendet und danach kommt die ERROR-message write: broken pipe. Dies geschieht dann, wenn ich den Schalter "Verhindere Entladung im Schnell-Modus und bei geplantem Laden." aktiviere.

Es wäre doch prima, das Problem ein wenig einzugrenzen. Dann müssen auch keine Screeshots gepostet werden. Das lässt sich ganz einfach hinsichtlich Ursache testen:

    evcc meter --log trace,db:error --battery-mode hold --battery-wait 1m

Und _falls_ der Befehl dazu führt, dass der WR die Verbindung trennt stellen sich ein paar Fragen:
- ist das schlimm?
- was sagt der Herstellersupport dazu?
- welche Register müssten stattdessen genutzt werden?
- was war vor 0.130.0 anders (Logfile)?

andyhielscher, 2024-09-11:
Ich habe den Befehl ` evcc meter --log trace,db:error --battery-mode hold --battery-mode-wait 1m` ausgeführt und folgende Logfile erhalten:
[main  ] INFO 2024/09/11 13:33:37 evcc 0.130.8
[main  ] INFO 2024/09/11 13:33:37 using config file: /etc/evcc.yaml
[modbus] TRACE 2024/09/11 13:33:37 modbus: send 00 01 00 00 00 06 01 06 32 f9 00 02
[modbus] TRACE 2024/09/11 13:33:37 modbus: recv 00 01 00 00 00 06 01 06 32 f9 00 02
[modbus] TRACE 2024/09/11 13:33:37 modbus: send 00 01 00 00 00 06 01 06 32 fa 00 cc
[modbus] TRACE 2024/09/11 13:33:37 modbus: recv 00 01 00 00 00 06 01 06 32 fa 00 cc
[main  ] INFO 2024/09/11 13:33:37 waiting for: 1m0s
[modbus] TRACE 2024/09/11 13:34:37 modbus: closing connection due to idle timeout: 1m0.000132066s

Mir fällt auf, dass das ein anderes Register ist. Hier wird  00 **01** 00 00 00 06 01 06 32 f9 00 02 benutzt und bei der Fehlermeldung im Log 00 **02** 00 00 00 06 01 06 32 f9 00 02. Ist das ggf die Ursache?

VG

andig, 2024-09-11:
Das ist bei Modbus TCP die transaction id. Ich wundere mich eher, wieso die hier konstant 01 bleiben kann. Aber das tut vmtl. nix zur Sache.

Dunkelgraf, 2024-09-11:
> * was war vor 0.130.0 anders (Logfile)?

Ich habe mal ein Testsystem mit 0.128.4 aufgesetzt und die Bereiche "site", "modbus" und "lp-2" für einen Ladevorgang mitgeloggt. Das Sperren der Batterie funktioniert in dieser Version.

Anbei das anomysierte Logfile
[evcc-20240911-Session-trace.log](https://github.com/user-attachments/files/16967042/evcc-20240911-Session-trace.log)

Falls mehr benötigt wird, bitte Bescheid geben.

andig, 2024-09-11:
@Dunkelgraf  Wäre super wenn Du das analog https://github.com/evcc-io/evcc/issues/16000#issuecomment-2342867413 machen könntest

andig, 2024-09-11:
> Das ist bei Modbus TCP die transaction id. Ich wundere mich eher, wieso die hier konstant 01 bleiben kann. Aber das tut vmtl. nix zur Sache.

Das ist in der Tat ein super versteckter Fehler bei Modbus TCP, mittels https://github.com/evcc-io/evcc/commit/a44ba0d20d0b2a9b704a7f800290d6333ea43ea2 behoben.

Dunkelgraf, 2024-09-11:
> @Dunkelgraf Wäre super wenn Du das analog [#16000 (comment)](https://github.com/evcc-io/evcc/issues/16000#issuecomment-2342867413) machen könntest

bitte sehr

root@evcc-test:~# evcc meter --log trace,db:error --battery-mode hold --battery-mode-wait 1m
[main  ] INFO 2024/09/11 18:32:42 evcc 0.128.4
[main  ] INFO 2024/09/11 18:32:42 using config file: /etc/evcc.yaml
[main  ] INFO 2024/09/11 18:32:42 waiting for: 1m0s
[main  ] INFO 2024/09/11 18:33:42 waiting for: 1m0s
[modbus] TRACE 2024/09/11 18:34:42 modbus: send 00 01 00 00 00 06 01 06 32 f9 00 02
[modbus] TRACE 2024/09/11 18:34:42 modbus: recv 00 01 00 00 00 06 01 06 32 f9 00 02
[modbus] TRACE 2024/09/11 18:34:42 modbus: send 00 02 00 00 00 06 01 06 32 fa 00 cc
[modbus] TRACE 2024/09/11 18:34:42 modbus: recv 00 02 00 00 00 06 01 06 32 fa 00 cc
[main  ] INFO 2024/09/11 18:34:42 waiting for: 1m0s
[modbus] TRACE 2024/09/11 18:35:42 modbus: closing connection due to idle timeout: 1m0.000784459s

andyhielscher, 2024-09-12:
Ich habe die Modbus-Anbindung von meinem Sungrow-Hybrid-Wechselrichter jetzt umgestellt und steuere das Gerät ab jetzt über den Winet-S Stick. Damit funktioniert alles. Es scheint also irgendwie am Wechselrichter zu liegen. Kannst du gleiches feststellen @Dunkelgraf?

Dunkelgraf, 2024-09-12:
Das Steuern funktioniert über den WiNet-S-Dongle, allerdings bekomme ich darüber keinen "Battery Power" Wert.

andig, 2024-09-14:
> allerdings bekomme ich darüber keinen "Battery Power" Wert.

Hier gehts nicht Batterieleistung. Ich schließe das Issue mal tentatively, ggf. durch ModbusTCP Fix gelöst.

Tornado1907, 2024-09-18:
Hallo, ich habe das gleiche Problem mit der broken pipe Meldung und der nicht funktionierenden Batteriesperre. Sungrow Hybrid SH10RT mit Sungrow Batterie. Modbud TCP über den integrierten LAN Port. EVCC Version 0.130.11. was muss ich tun, um das Trace Log zu anonymisieren, nur die IPs ersetzen?

andig, 2024-09-18:
> Ich schließe das Issue mal tentatively, ggf. durch ModbusTCP Fix gelöst.

Aktuelle Version getestet?

jks1981, 2024-09-18:
Ich hatte das Problem mit der 0.130.10 nicht mehr, seit der 0.130.11 aber wieder.

andig, 2024-09-18:
Es braucht einen Reproducer mit

    evcc meter

Irgendwo mitten im Log lässt sich fürs Debugging leider nicht verwenden.

Tornado1907, 2024-09-18:
Bei mir tritt der Fehler sowohl mit der .10 als auch mit der .11 auf.

Dunkelgraf, 2024-09-19:
> Es braucht einen Reproducer mit
> 
> ```
> evcc meter
> ```
Brauchst Du diese Ausgabe oder ein "evcc meter" während des Ladevorgangs?
root@xxx:~# evcc meter --log trace,db:error --battery-mode hold --battery-mode-wait 1m
[main  ] INFO 2024/09/19 12:13:11 evcc 0.130.11
[main  ] INFO 2024/09/19 12:13:11 using config file: /etc/evcc.yaml
[mqtt  ] INFO 2024/09/19 12:13:11 connecting evcc-1717554873 at tcp://evcc.xxx.xxx:1883
[mqtt  ] DEBUG 2024/09/19 12:13:11 tcp://evcc.xxx.xxx:1883 connected
[main  ] INFO 2024/09/19 12:13:11 waiting for: 1m0s
[main  ] INFO 2024/09/19 12:14:11 waiting for: 1m0s
[modbus] TRACE 2024/09/19 12:15:11 modbus: send 00 01 00 00 00 06 01 06 32 f9 00 02
[modbus] TRACE 2024/09/19 12:15:11 modbus: recv 00 01 00 00 00 06 01 06 32 f9 00 02
[modbus] TRACE 2024/09/19 12:15:11 modbus: send 00 02 00 00 00 06 01 06 32 fa 00 cc
[modbus] TRACE 2024/09/19 12:15:11 modbus: recv 00 02 00 00 00 06 01 06 32 fa 00 cc
[main  ] INFO 2024/09/19 12:15:11 waiting for: 1m0s
[modbus] TRACE 2024/09/19 12:16:11 modbus: closing connection due to idle timeout: 1m0.000160074s

Eine Sache ist mir dabei aufgefallen: evcc meter ... von der Kommandozeile läuft sauber durch, aus dem laufenden Programm kommt "broken pipe". Nach vielen Versuchen (ca. eine Stunde) wird die Batterie irgendwann gesperrt.

jks1981, 2024-09-30:
Wird an dem Fehler noch gearbeitet? Was genau wird gebraucht? Der Fehler ist bei mir immer noch vorhanden (Version 0.130.12).

ebloody1, 2024-10-03:
Guten Abend zusammen. Ich habe auch den gleichen Fehler wie jks1981. SH8.0RT, SBR096, Sungrow Energy Meter
Ich weiss nicht mehr ab welcher Version der Fehler aufgetreten ist, aber ich glaube vor der 0.130.x war noch alles in Ordnung. Kann ich mit irgendwelchen weiteren Information weiterhelfen? 

evcc meter --log trace,db:error --battery-mode hold --battery-mode-wait 1m
[main  ] INFO 2024/10/03 22:15:27 evcc 0.130.12
[main  ] INFO 2024/10/03 22:15:27 using config file: /etc/evcc.yaml
[main  ] INFO 2024/10/03 22:15:27 waiting for: 1m0s
[main  ] INFO 2024/10/03 22:16:27 waiting for: 1m0s
[modbus] TRACE 2024/10/03 22:17:27 modbus: send 00 01 00 00 00 06 01 06 32 f9 00 02
[modbus] TRACE 2024/10/03 22:17:27 modbus: recv 00 01 00 00 00 06 01 06 32 f9 00 02
[modbus] TRACE 2024/10/03 22:17:27 modbus: send 00 02 00 00 00 06 01 06 32 fa 00 cc
[modbus] TRACE 2024/10/03 22:17:27 modbus: recv 00 02 00 00 00 06 01 06 32 fa 00 cc
[main  ] INFO 2024/10/03 22:17:27 waiting for: 1m0s
[modbus] TRACE 2024/10/03 22:18:27 modbus: closing connection due to idle timeout: 1m0.000694447s

andig, 2024-10-04:
Das Logfile zeit ja erstmal keinen Fehler- Battery Mode hold geht. Kein broken pipe. Wird denn auch die Batterie dadurch gesperrt? Falls parallel noch evcc läuft muss da die Batteriesteuerung deaktiviert sein damit Du das hier parallel im cli testen kannst.

DiDu0815, 2024-10-04:
Will nur insofern beitragen indem ich bestätige, dass der Fehler broken pipe ab 0.130.0 auch bei mir (praktisch permanent) auftrat (SH10RT + Axitec Axistorage Li SV2, verbunden mit evcc über den LAN-Anschluss des Wechselrichters). Nach Downgrade heute auf 0.129.0  - vor allem um den SOH der Hausbatterie zu schonen - ist der Fehler wieder verschwunden, d.h. das Sperren/Freigeben der Batterie bei/nach Schnellladen funktioniert nun wieder wenn der entsprechende Schalter gesetzt ist. Während der Nutzung von 0.130.x hatte ich auch ein Update der [Wechselrichter-Firmware](https://github.com/sungrow-firmware/firmware/tree/main/Hybrid%20Inverter/SH10RT) auf die aktuelle Version vollzogen. Bezgl. des Problems machte das jedoch keinen Unterschied. Ich werde erstmal auf Version 0.129.0 verweilen (müssen) und derweil diesen Thread verfolgen.

andig, 2024-10-04:
Am `sungrow-hybrid` Template gab es zwischen 0.129.0 und heute keine relevante Änderung, auf jeden Fall nicht bei der Batteriesteuerung. An Template/Registern liegt es also nicht. 

Damit wäre die These, dass es am `broken pipe` Netzwerkfehler liegt. Leider haben wir **null** Idee wo der her kommen könnte und damit auch keine Lösung.

DiDu0815, 2024-10-04:
Habe selbst null Ahnung von Go, aber vielleicht jemand anderes [Golang - What is Broken Pipe Error? Tcp Http Connections and Pools](https://medium.com/trendyol-tech/golang-what-is-broken-pipe-error-tcp-http-connections-and-pools-3988b79f28e5)

andig, 2024-10-08:
@DiDu0815 könntest Du mal bitte den Battery Mode mit

    evcc meter --log trace,db:error

in der alten und neuen Version setzen? Ich würde gerne- im direkten Vergleich- sehen, welche Unterschiede es da gibt. Irgendeinen Grund muss es ja haben. evcc dabei bitte deaktivieren.

Tornado1907, 2024-10-08:
Hallo zusammen,  ich glaube nicht, das wir dem Problem mit den evcc -meter eingrenzen zu können. Die BrokenPipe Condition kommt aus dem Netzwerkstack und passiert wahrscheinlich wie im Artikel, den DiDu0815 gepostet hat, wenn die Serververbindung (in diesem Fall ModBus TCP) einseitig terminiert wurde. Bei mir tritt der Fehler auch direkt nach einem Neutstart von evcc nicht auf, dann aber später beim deaktivieren der Batteriesperre. Irgendetwas in der Art und weise wie die TCP Connections gehandelt werden, hat sich zwischen 0.129.x und 1.30.x geändert. Ist leider schwierig einzugrenzen, da wir hier ein Problem haben, was erst nach einer gewissen Zeit entsteht.

Tornado1907, 2024-10-08:
Leider hat Sungrow da nichts in petto. Aber irgendetwas muss sich auf Seite evcc geändert haben, das es sonst mit der älteren Version nicht gehen würde. Vielleicht ist es auch eine shared Komponente, die mit eingebunden wird. wir können ja mal testen, zwischen welchen Builds das Problem zuerst auftritt und dann schauen, was sich dazwischen geändert hat. 
Ansonsten würde nur der Holzhammer helfen, und bei jeder Modbusabfrage die Verbindung neu aufzubauen bzw. sie nach der Abfrage wieder zu beenden.

andig, 2024-10-08:
Bei evcc hat sich geändert, dass sich jetzt mehrere Geräte eine physikalische Verbindung sauber teilen und wir dadurch auch ohne race conditions parallel drauf zugreifen können. Weggefallen dadurch ist ein 100ms Delay zwischen Zugriffen unterschiedlicher Slave IDs. Das sollte bei TCP aber auch völlig unerheblich sein (bei Serial hat sich dadurch ein Bug in der Implementierung gezeigt den wir auch gelöst haben). Was Du probieren könntest wäre, ob ein `delay: 100ms` oder sowas den WR vllt. so glücklich machen kann, dass er auf den Verbindungsabbruch verzichtet.

closed 2024-10-12 by andig

KarlKnapp, 2024-10-12:
Wurde das Problem im Code gelöst oder muss etwas in der config geändert werden bzwl `delay` ?

andig, 2024-10-12:
Es gilt weiter https://github.com/evcc-io/evcc/issues/16000#issuecomment-2399711177

Dunkelgraf, 2024-10-12:
ich möchte das morgen gerne testen, weiß aber nicht wo ich den delay einstellen kann.

clemenslermen, 2024-10-12:
Geht mir genauso. Gleiches setup und hätte morgen bei 18cent bei Tibber gute Möglichkeit es zu testen. Wo kann ich den Delay einstellen?

KarlKnapp, 2024-10-13:
mir erschließt sich leider nicht warum das hier geschlossen wurde, wenn nicht mal klar ist wie man es testen kann

clemenslermen, 2024-10-13:
> mir erschließt sich leider nicht warum das hier geschlossen wurde, wenn nicht mal klar ist wie man den workaround testen kann

Ich kann es schon verstehen, weil es ja offensichtlich ein Sungrow Fehler ist. Wäre allerdings über einen kleinen Hint froh, wo der Delay gesetzt werden muss. In der Doku sehe ich nur delays for loadpoints und die beigen mich an der Stelle ja nicht weiter

HunZ83, 2024-10-13:
> Will nur insofern beitragen indem ich bestätige, dass der Fehler broken pipe ab 0.130.0 auch bei mir (praktisch permanent) auftrat (SH10RT + Axitec Axistorage Li SV2, verbunden mit evcc über den [...]

Ha ich dasselbe Problem habe, wollte ich als Zwischenlösung eben auch einmal auf 0.129.0 downgraden. Da ich evcc als Addon im Home Assistant laufen habe, finde ich keine Möglichkeit, dies zu tun. Kannst Du mir einen Tipp geben, wie ich zurück zur 0.129.0 komme?
Danke!

DiDu0815, 2024-10-13:
@HunZ83 Ich betreibe evcc nativ in einer DietPi Linux VM auf Synology NAS, prinzipiell so wie es [hier](https://docs.evcc.io/docs/installation/linux) beschrieben ist. Für diese Konstellation wäre das Installationskommando für genau 0.129.0 dann `sudo apt install evcc=0.129.0`. Auf diese Weise kann man schnell beliebige, existierende Versionen installieren. 
Optional kann man ungewollte automatische Updates weg von dieser Version dann noch mittels `sudo apt-mark hold evcc` verhindern. Diese Upgrade-Sperre könnte mittels `sudo apt-mark unhold evcc` später auch wieder entfernt werden.
Zu anderen Konstellationen bis ich nicht aussagefähig.

lmfgpeace, 2024-10-14:
@andyhielscher 
@andig 

Der Fehler kommt bei mir im Schnelllademodus nicht.

SH10.0RT + SBR096 & go-e Wallbox.

Ich habe gestern erstmalig evcc (Docker Container auf Synology NAS) installiert.

matschex, 2024-10-20:
Hallo,
Also bei mir geht es nicht. Ich konnte auch kein Delay eintragen sondern nur ein timeout. Oder wo muss ich das Delay eintragen?

chris-e-codes, 2024-10-20:
> Ich kann es schon verstehen, weil es ja offensichtlich ein Sungrow Fehler ist. Wäre allerdings über einen kleinen Hint froh, wo der Delay gesetzt werden muss. In der Doku sehe ich nur delays for [...]

So klar ist das für mich nicht. 129er Version geht, ab 130 kaputt. Wieso ist es dann ein Sungrow Fehler?

andig, 2024-10-20:
Spielt keine Rolle. Solange es niemand testet gibts auch für uns nix zu tun.

matschex, 2024-10-20:
Wie ich schrieb, zwei Posts davor getestet mit timeout, da kein Delay Parameter beim WR oder Batterie verfügbar ist. Was wird noch benötigt?

chris-e-codes, 2024-10-20:
Moin ich habe mal versucht die 0.129.0 gegen die 0.130.13 zu testen und die wichtigsten logs zu ziehen:

0.129.0:
```
[lp-1  ] DEBUG 2024/10/20 17:05:03 charge power: 0W
[lp-1  ] DEBUG 2024/10/20 17:05:03 charge currents: [0.085 0.027 0.059]A
[modbus] TRACE 2024/10/20 17:05:03 modbus: send 00 29 00 00 00 06 01 04 13 98 00 02
[modbus] TRACE 2024/10/20 17:05:03 modbus: send 00 2a 00 00 00 06 01 04 13 98 00 02
[modbus] TRACE 2024/10/20 17:05:03 modbus: recv 00 2a 00 00 00 07 01 04 04 01 d2 00 00
[modbus] TRACE 2024/10/20 17:05:03 modbus: send 00 2b 00 00 00 06 01 04 32 ca 00 02
[modbus] TRACE 2024/10/20 17:05:03 modbus: recv 00 2b 00 00 00 07 01 04 04 a2 94 00 01
[site  ] DEBUG 2024/10/20 17:05:03 pv power: 466W
[modbus] TRACE 2024/10/20 17:05:03 modbus: send 00 2c 00 00 00 06 01 04 32 dd 00 01
[modbus] TRACE 2024/10/20 17:05:03 modbus: recv 00 2c 00 00 00 05 01 04 02 01 3e
[modbus] TRACE 2024/10/20 17:05:03 modbus: send 00 2d 00 00 00 06 01 04 32 c8 00 01
[modbus] TRACE 2024/10/20 17:05:03 modbus: recv 00 2d 00 00 00 05 01 04 02 00 0d
[modbus] TRACE 2024/10/20 17:05:03 modbus: send 00 2e 00 00 00 06 01 04 32 c8 00 01
[modbus] TRACE 2024/10/20 17:05:03 modbus: recv 00 2e 00 00 00 05 01 04 02 00 0d
[modbus] TRACE 2024/10/20 17:05:03 modbus: send 00 2f 00 00 00 06 01 04 32 e2 00 02
[modbus] TRACE 2024/10/20 17:05:03 modbus: recv 00 2f 00 00 00 07 01 04 04 6b 3d 00 00
[modbus] TRACE 2024/10/20 17:05:03 modbus: send 00 30 00 00 00 06 01 04 32 de 00 01
[modbus] TRACE 2024/10/20 17:05:03 modbus: recv 00 30 00 00 00 05 01 04 02 00 3f
[site  ] DEBUG 2024/10/20 17:05:03 battery soc: 6%
[site  ] DEBUG 2024/10/20 17:05:03 battery power: 318W
[modbus] TRACE 2024/10/20 17:05:03 modbus: send 00 31 00 00 00 06 01 04 32 d1 00 02
[modbus] TRACE 2024/10/20 17:05:03 modbus: recv 00 31 00 00 00 07 01 04 04 00 09 00 00
[site  ] DEBUG 2024/10/20 17:05:03 grid meter: -9W
[modbus] TRACE 2024/10/20 17:05:03 modbus: send 00 32 00 00 00 06 01 04 32 ec 00 02
[modbus] TRACE 2024/10/20 17:05:03 modbus: recv 00 32 00 00 00 07 01 04 04 54 f1 00 00
[site  ] DEBUG 2024/10/20 17:05:03 site power: 459W
[lp-1  ] DEBUG 2024/10/20 17:05:18 set charge mode: now
[site  ] DEBUG 2024/10/20 17:05:18 ----
[lp-1  ] DEBUG 2024/10/20 17:05:19 charge power: 0W
[lp-1  ] DEBUG 2024/10/20 17:05:19 charge currents: [0.085 0.027 0.059]A
[modbus] TRACE 2024/10/20 17:05:19 modbus: send 00 33 00 00 00 06 01 04 13 98 00 02
[modbus] TRACE 2024/10/20 17:05:19 modbus: recv 00 33 00 00 00 07 01 04 04 01 d1 00 00
[modbus] TRACE 2024/10/20 17:05:19 modbus: send 00 34 00 00 00 06 01 04 32 ca 00 02
[modbus] TRACE 2024/10/20 17:05:19 modbus: recv 00 34 00 00 00 07 01 04 04 a2 94 00 01
[site  ] DEBUG 2024/10/20 17:05:19 pv power: 465W
[modbus] TRACE 2024/10/20 17:05:19 modbus: send 00 35 00 00 00 06 01 04 32 dd 00 01
[modbus] TRACE 2024/10/20 17:05:19 modbus: recv 00 35 00 00 00 05 01 04 02 00 00
[modbus] TRACE 2024/10/20 17:05:19 modbus: send 00 36 00 00 00 06 01 04 32 c8 00 01
[modbus] TRACE 2024/10/20 17:05:19 modbus: recv 00 36 00 00 00 05 01 04 02 00 29
[modbus] TRACE 2024/10/20 17:05:19 modbus: send 00 37 00 00 00 06 01 04 32 c8 00 01
[modbus] TRACE 2024/10/20 17:05:19 modbus: recv 00 37 00 00 00 05 01 04 02 00 29
[modbus] TRACE 2024/10/20 17:05:19 modbus: send 00 38 00 00 00 06 01 04 32 e2 00 02
[modbus] TRACE 2024/10/20 17:05:19 modbus: recv 00 38 00 00 00 07 01 04 04 6b 3d 00 00
[modbus] TRACE 2024/10/20 17:05:19 modbus: send 00 39 00 00 00 06 01 04 32 de 00 01
[modbus] TRACE 2024/10/20 17:05:19 modbus: recv 00 39 00 00 00 05 01 04 02 00 3f
[site  ] DEBUG 2024/10/20 17:05:19 battery soc: 6%
[site  ] DEBUG 2024/10/20 17:05:19 battery power: 0W
[modbus] TRACE 2024/10/20 17:05:19 modbus: send 00 3a 00 00 00 06 01 04 32 d1 00 02
[modbus] TRACE 2024/10/20 17:05:19 modbus: recv 00 3a 00 00 00 07 01 04 04 ff e5 ff ff
[site  ] DEBUG 2024/10/20 17:05:19 grid meter: 27W
[modbus] TRACE 2024/10/20 17:05:19 modbus: send 00 3b 00 00 00 06 01 04 32 ec 00 02
[modbus] TRACE 2024/10/20 17:05:19 modbus: recv 00 3b 00 00 00 07 01 04 04 54 f1 00 00
[site  ] DEBUG 2024/10/20 17:05:19 site power: 177W
[lp-1  ] DEBUG 2024/10/20 17:05:19 charger status: B
[lp-1  ] DEBUG 2024/10/20 17:05:19 next soc poll remaining time: 3m14s
[lp-1  ] DEBUG 2024/10/20 17:05:19 max charge current: 16A
[lp-1  ] DEBUG 2024/10/20 17:05:20 charger enable
[lp-1  ] DEBUG 2024/10/20 17:05:20 wake-up timer: start
[site  ] DEBUG 2024/10/20 17:05:33 ----
[lp-1  ] DEBUG 2024/10/20 17:05:33 charge power: 0W
[lp-1  ] DEBUG 2024/10/20 17:05:33 charge currents: [0.085 0.027 0.059]A
[modbus] TRACE 2024/10/20 17:05:33 modbus: send 00 3c 00 00 00 06 01 04 13 98 00 02
[modbus] TRACE 2024/10/20 17:05:33 modbus: recv 00 3c 00 00 00 07 01 04 04 01 d2 00 00
[modbus] TRACE 2024/10/20 17:05:33 modbus: send 00 3d 00 00 00 06 01 04 32 ca 00 02
[modbus] TRACE 2024/10/20 17:05:33 modbus: recv 00 3d 00 00 00 07 01 04 04 a2 94 00 01
[site  ] DEBUG 2024/10/20 17:05:33 pv power: 466W
[... 319 lines cut ...]
[site  ] DEBUG 2024/10/20 17:10:57 grid meter: 7311W
[modbus] TRACE 2024/10/20 17:10:57 modbus: send 00 68 00 00 00 06 01 04 15 e2 00 02
[modbus] TRACE 2024/10/20 17:10:57 modbus: recv 00 68 00 00 00 07 01 04 04 08 d4 00 00
[modbus] TRACE 2024/10/20 17:10:57 modbus: send 00 69 00 00 00 06 01 04 13 9a 00 01
[modbus] TRACE 2024/10/20 17:10:57 modbus: recv 00 69 00 00 00 05 01 04 02 09 38
[modbus] TRACE 2024/10/20 17:10:57 modbus: send 00 6a 00 00 00 06 01 04 15 e4 00 02
[... cut ...]
[modbus] TRACE 2024/10/20 17:10:57 modbus: recv 00 6e 00 00 00 07 01 04 04 54 f4 00 00
[site  ] DEBUG 2024/10/20 17:10:57 site power: 11546W
[lp-1  ] DEBUG 2024/10/20 17:11:26 set charge mode: pv
```

In der 130.13 reagiert der WR oder ?evcc?auf den Modbus "00 5f 00 00 00 06 01 06 32 f9 00 02" sofort mit broken pipe

In der 129.0 wird der Modbus "00 59 00 00 00 06 01 06 32 f9 00 02" mit "00 59 00 00 00 06 01 06 32 f9 00 02" beantwortet.

Kann ich irgendwie noch detaillierter testen? Evtl. tcpdump im docker richtung wr ziehen?

Dunkelgraf, 2024-10-20:
> Spielt keine Rolle. Solange es niemand testet gibts auch für uns nix zu tun.

Solange ich nicht weiß wo ich den Delay einbauen muss, kann ich das nicht testen. Das habe ich bereits geschrieben.

lmfgpeace, 2024-10-20:
> > > mir erschließt sich leider nicht warum das hier geschlossen wurde, wenn nicht mal klar ist wie man den workaround testen kann
> 
> > 
> 
> > Ich kann es schon verstehen, weil es ja offensichtlich ein Sungrow Fehler ist. Wäre allerdings über einen kleinen Hint froh, wo der Delay gesetzt werden muss. In der Doku sehe ich nur delays for loadpoints und die beigen mich an der Stelle ja nicht weiter
> 
> 
> 
> So klar ist das für mich nicht. 129er Version geht, ab 130 kaputt. Wieso ist es dann ein Sungrow Fehler?
> 
> Klar kann eine Anpassung bei evcc im Modbus Stack jetzt einen Fehler von Sungrow aufdecken. 
> 
> Schade dass es einfach zu ist. Seit 130 wird jetzt beim Plan oder schnell laden immer die Batterie geplättet und Batterie aus Netzladen läuft ebenfalls auf einen Fehler 😭

Nochmal:
Bei mir kommt es zu keinem Fehler mit dem SH10.0RT und dem SBR096 und der go-e Gemini flex 22kW (v3).

Tornado1907, 2024-10-20:
> Nochmal: Bei mir kommt es zu keinem Fehler mit dem SH10.0RT und dem SBR096 und der go-e Gemini flex 22kW (v3).

@lmfgpeace 
Nur kurz zur Eingrenzung:
-  du nutzt den LAN Port des Wechselrichters und nicht den WiConnect Dongle
- Du benutzt nicht den Modbus Proxy, oder doch?

Tornado1907, 2024-10-20:
@andig 

wo müssen wir das Delay eintragen, um zu testen, ob das hilft?

Der Fehler tritt übrigens auch auf, wenn man die Batterie zwangsladen will. Startet man evcc neu, geht das sofort, evcc packt den WR in den “Promiscous” Mode aber beim Ende des Ladevorgangs dann wieder “Broken Pipe” was dann bedeutet, das die Batterie nicht mehr genutzt wird. Ich weiß mittlerweile, wie ich den WR wieder manuell in den gewünschten Betriebsmodus bringe…

lmfgpeace, 2024-10-20:
> 
> 
> > Nochmal: Bei mir kommt es zu keinem Fehler mit dem SH10.0RT und dem SBR096 und der go-e Gemini flex 22kW (v3).
> 
> 
> 
> @lmfgpeace 
> 
> Nur kurz zur Eingrenzung:
> 
> -  du nutzt den LAN Port des Wechselrichters und nicht den WiConnect Dongle
> 
> - Du benutzt nicht den Modbus Proxy, oder doch?
> 
> 
> 
> 
> 
> 

Yee. Lan Port des WR verbunden mit dem Router. 

Dann via modbus tcp:

```

meters:
  - name: Sungrow_SH10.0RT_grid
    type: template
    template: sungrow-hybrid
    usage: grid

    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.178.105 # Hostname
    port: 502 # Port
    timeout: 10s # optional
  - name: Sungrow_SH10.0RT_pv
    type: template
    template: sungrow-hybrid
    usage: pv

    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.178.105 # Hostname
    port: 502 # Port
    timeout: 10s # optional
  - name: Sungrow_SH10.0RT_battery
    type: template
    template: sungrow-hybrid
    usage: battery

    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.178.105 # 

```

andig, 2024-10-20:
Ist die FW relevant?

lmfgpeace, 2024-10-20:
> Ist die FW relevant?

FW vom WR, Batterie und Kommunikationsmodul ist bei mir up to date.

[image]

[image]

chris-e-codes, 2024-10-21:
Die Logs von mir waren ein nachstellen mehr nicht :) schon gekürzt auf 1 Minute.

Klar könnte man ein Ticket bei Sungrow aufmachen. Aber was beschreiben wir denen als Issue? Haben wir ne Richtung wie es durch die Änderung der Version kommt?

andig, 2024-10-21:
> Die Logs von mir waren ein nachstellen mehr nicht

Kannst Du das auf Knopfdruck? Dann sollte es mit `evcc meter` reproduzierbar sein!

> Klar könnte man ein Ticket bei Sungrow aufmachen. Aber was beschreiben wir denen als Issue?

"WR bricht TCP Verbindung nach Setzen Batteriemodus ab". Das kann man- wenn auf Knopfdruck reproduzierbar- noch mit Wireshark ergänzen. evcc Version spielt keine Rolle. Irgendwas im WR- und sei es nur anderes (völlig legales) Timing (von evcc)- muss das Verhalten auslösen.

chris-e-codes, 2024-10-21:
Wenn das Auto Abends da ist ja. Dank Docker geht der Wechsel sehr schnell.
Ich ziehe mal nen Wireshark und führe es mit meter aus. Melde mich

[43 more replies]