evcc-io/evcc #16001: Ocpp status field validation genericerror causes segfault?

scotthernandez, 2024-09-09:
### Describe the bug

[main  ] INFO 2024/09/08 16:09:11 evcc 0.130.8
[main  ] INFO 2024/09/08 16:09:11 using config file: /config/evcc.yaml
[mqtt  ] INFO 2024/09/08 16:09:11 connecting evcc-922132188 at tcp://192.168.1.5:1883
[mqtt  ] DEBUG 2024/09/08 16:09:11 tcp://192.168.1.5:1883 connected
[mqtt  ] DEBUG 2024/09/08 16:09:11 tcp://192.168.1.5:1883 subscribe solar_assistant/total/battery_power/state
[40252204784191-1] DEBUG 2024/09/08 16:09:11 waiting for chargepoint: 5m0s
[ocpp  ] DEBUG 2024/09/08 16:09:41 charge point connected: 40252204784191
[40252204784191-1] WARN 2024/09/08 16:10:12 meter timeout
[40252204784191-1] DEBUG 2024/09/08 16:10:12 failed configuring WebSocketPingInterval: ChangeConfiguration failed: NotSupported
[lp-1  ] WARN 2024/09/08 16:10:27 deprecated: mincurrent setting is ignored, please remove
[lp-1  ] WARN 2024/09/08 16:10:27 deprecated: maxcurrent setting is ignored, please remove
[lp-1  ] WARN 2024/09/08 16:10:27 deprecated: phases setting is ignored, please remove
[site  ] WARN 2024/09/08 16:10:27 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site  ] INFO 2024/09/08 16:10:28 site config:
[site  ] INFO 2024/09/08 16:10:28   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/09/08 16:10:28     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/09/08 16:10:28     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/09/08 16:10:28     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2024/09/08 16:10:28   vehicles:
[site  ] INFO 2024/09/08 16:10:28     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/09/08 16:10:28 loadpoint 1:
[lp-1  ] INFO 2024/09/08 16:10:28   mode:        pv
[lp-1  ] INFO 2024/09/08 16:10:28   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/09/08 16:10:28   meters:      charge ✗
[site  ] DEBUG 2024/09/08 16:10:28 set buffer soc: 100
[site  ] DEBUG 2024/09/08 16:10:28 set buffer start soc: 0
[site  ] DEBUG 2024/09/08 16:10:28 set priority soc: 50
[lp-1  ] DEBUG 2024/09/08 16:10:28 phase timer inactive
[lp-1  ] DEBUG 2024/09/08 16:10:28 pv timer inactive
[ocpp  ] ERROR 2024/09/08 16:10:28 ocpp message (2011687811): GenericError - Key: 'CallResult.Payload.Status' Error:Field validation for 'Status' failed on the 'compositeScheduleStatus' tag ([3,"2011687811",{"status":"NotImplemented"}])
[site  ] DEBUG 2024/09/08 16:10:58 ----
[lp-1  ] DEBUG 2024/09/08 16:10:58 charge power: 0W
[site  ] DEBUG 2024/09/08 16:10:58 pv power: 6121W
[site  ] DEBUG 2024/09/08 16:10:58 battery soc: 100%
[site  ] DEBUG 2024/09/08 16:10:58 battery power: -73W
[site  ] DEBUG 2024/09/08 16:10:58 grid meter: -3523W
[site  ] DEBUG 2024/09/08 16:10:58 site power: -3596W
[lp-1  ] DEBUG 2024/09/08 16:10:58 charger status: A
[lp-1  ] INFO 2024/09/08 16:10:58 car disconnected
[lp-1  ] DEBUG 2024/09/08 16:10:58 set session soc limit: 0
[lp-1  ] DEBUG 2024/09/08 16:10:58 set session energy limit: 0
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x20 pc=0x18637e0]

goroutine 284 [running]:
github.com/evcc-io/evcc/charger/ocpp.(*CS).GetCompositeScheduleRequest.func1(0xc001b80f50?, {0x3877c40, 0xc0015cfc20})
	github.com/evcc-io/evcc/charger/ocpp/cs_core.go:80 +0xc0
github.com/lorenzodonini/ocpp-go/ocpp1%2e6.(*centralSystem).GetCompositeSchedule.func1({0x0?, 0x0?}, {0x3877c40?, 0xc0015cfc20?})
	github.com/lorenzodonini/ocpp-go@v0.18.0/ocpp1.6/central_system.go:335 +0x54
created by github.com/lorenzodonini/ocpp-go/ocpp1%2e6.(*centralSystem).handleCanceledRequest in goroutine 39
	github.com/lorenzodonini/ocpp-go@v0.18.0/ocpp1.6/central_system.go:565 +0xb8

### Steps to reproduce

Not sure, other than ocpp charger

### Configuration details

```yaml
chargers:%0A%20%20-%20name%3A%20vista-garage%0A%20%20%20%20type%3A%20template%0A%20%20%20%20template%3A%20ocpp%0A%20%20%20%20idtag%3A%201%0A%20%20%20%20stationid%3A%2040252204784191%0A%0A%23%20vehicle%20definitions%0A%23%20vehicles:%0A%23%20%20-%20n
```

### Log details

```text
Lmk if more details are needed
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

I did not check the nightly build and have no easy way to do that.

andig, 2024-09-09:
Sehr lustig:

    [3,"2011687811",{"status":"NotImplemented"}]

closed 2024-09-09 by andig

scotthernandez, 2024-09-09:
Thanks for the fix and quick turnaround ;)