evcc-io/evcc #16006: Cachetime for BMW ignored while connected to station

endurance1968, 2024-09-09:
### Describe the bug

Error message out of volume quota since cache time seems to be ignored

### Steps to reproduce

1. connect vehicle
2.start charging
you get error messages like
[lp-1 ] WARN 2024/09/09 19:46:42 vehicle soc: 403: Out of call volume quota. Quota will be replenished in 00:13:18. (unexpected status: 403 (Forbidden)) (ignored by estimator)
...

### Configuration details

```yaml
vehicle cache time is set to 20min (set in DB not via configfile)
```

### Log details

```text
[lp-1 ] WARN 2024/09/09 19:45:42 vehicle soc: 403: Out of call volume quota. Quota will be replenished in 00:14:18. (unexpected status: 403 (Forbidden)) (ignored by estimator)
[lp-1 ] DEBUG 2024/09/09 19:45:42 soc estimated: 55.45% (vehicle: 52.00%)
[lp-1 ] DEBUG 2024/09/09 19:45:42 vehicle soc: 55%
**[lp-1 ] ERROR 2024/09/09 19:45:42 vehicle soc limit: 403: Out of call volume quota. Quota will be replenished in 00:14:18. (unexpected status: 403 (Forbidden))
[lp-1 ] ERROR 2024/09/09 19:45:42 vehicle range: 403: Out of call volume quota. Quota will be replenished in 00:14:18. (unexpected status: 403 (Forbidden))**
[site ] DEBUG 2024/09/09 19:46:12 ----
[lp-1 ] DEBUG 2024/09/09 19:46:12 charge power: 3607W
[lp-1 ] DEBUG 2024/09/09 19:46:12 charge currents: [15.3 0.038 0.076]A
[lp-1 ] DEBUG 2024/09/09 19:46:12 detected active phases: 1p
[lp-1 ] DEBUG 2024/09/09 19:46:12 !! session: chargeRater.chargedEnergy=3.5 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/09/09 19:46:12 charge total import: 48.465kWh
[lp-1 ] DEBUG 2024/09/09 19:46:12 charger status: C
[lp-1 ] WARN 2024/09/09 19:46:12 vehicle soc: 403: Out of call volume quota. Quota will be replenished in 00:13:48. (unexpected status: 403 (Forbidden)) (ignored by estimator)
[lp-1 ] DEBUG 2024/09/09 19:46:12 soc estimated: 55.59% (vehicle: 52.00%)
[lp-1 ] DEBUG 2024/09/09 19:46:12 vehicle soc: 56%
**[lp-1 ] ERROR 2024/09/09 19:46:12 vehicle soc limit: 403: Out of call volume quota. Quota will be replenished in 00:13:48. (unexpected status: 403 (Forbidden))
[lp-1 ] ERROR 2024/09/09 19:46:12 vehicle range: 403: Out of call volume quota. Quota will be replenished in 00:13:48. (unexpected status: 403 (Forbidden))**
[site ] DEBUG 2024/09/09 19:46:42 ----
[lp-1 ] DEBUG 2024/09/09 19:46:42 charge power: 3603W
[lp-1 ] DEBUG 2024/09/09 19:46:42 charge currents: [15.3 0.038 0.076]A
[lp-1 ] DEBUG 2024/09/09 19:46:42 detected active phases: 1p
[lp-1 ] DEBUG 2024/09/09 19:46:42 !! session: chargeRater.chargedEnergy=3.5 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/09/09 19:46:42 charge total import: 48.495kWh
[lp-1 ] DEBUG 2024/09/09 19:46:42 charger status: C
[lp-1 ] WARN 2024/09/09 19:46:42 vehicle soc: 403: Out of call volume quota. Quota will be replenished in 00:13:18. (unexpected status: 403 (Forbidden)) (ignored by estimator)
[lp-1 ] DEBUG 2024/09/09 19:46:42 soc estimated: 55.74% (vehicle: 52.00%)
[lp-1 ] DEBUG 2024/09/09 19:46:42 vehicle soc: 56%
[lp-1 ] ERROR 2024/09/09 19:46:42 vehicle soc limit: 403: Out of call volume quota. Quota will be replenished in 00:13:18. (unexpected status: 403 (Forbidden))
[lp-1 ] ERROR 2024/09/09 19:46:42 vehicle range: 403: Out of call volume quota. Quota will be replenished in 00:13:18. (unexpected status: 403 (Forbidden))
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.7

andig, 2024-09-09:
> Error message out of volume quota since cache time seems to be ignored

Dafür brauchts ein trace log für BMW. Hier ist das nicht erkennbar.

closed 2024-09-09 by andig

endurance1968, 2024-09-10:
Was mit tracelog für BMW gemeint?
Ich dachte 
[lp-1 ] ERROR 2024/09/09 19:46:42 vehicle soc limit: 403: Out of call volume quota. Quota will be replenished in 00:13:18. (unexpected status: 403 (Forbidden))
[lp-1 ] ERROR 2024/09/09 19:46:42 vehicle range: 403: Out of call volume quota. Quota will be replenished in 00:13:18. (unexpected status: 403 (Forbidden))
kommt durch den BMW - ja da steht lp (load point) aber der Fehler kommt nur wenn der BMW angesteckt wird. Was soll ich da noch mehr an logs liefern?

andig, 2024-09-10:
Bmw auf log level trace.

endurance1968, 2024-09-10:
Vehicles habe ich via GUI eingerichtet - In der GUI sehe ich keine Log-settings. Muss ich das auf Konfigfile umstellen oder übersehe ich etwas? 
Ist wohl erstmal besser alles auf konfigfile zu setzen.

andig, 2024-09-10:
Guckst Du Doku: https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen. Oder ganz einfach: im Hamburgermenü!

endurance1968, 2024-09-10:
https://docs.evcc.io/docs/reference/configuration/log#levels
hatte ich tatsächlich übersehen - probiere ich aus. Danke.

endurance1968, 2024-09-10:
Sodele here we are
```
root@lx-evcc:~# journalctl -u evcc -f
Sep 10 17:00:13 lx-evcc evcc[38961]: [bmw   ] TRACE 2024/09/10 17:00:13 client_id=31c357a0-7a1d-4590-aa99-33b97244d048&code_challenge=MMfBJXS6w8TKLecD54w9FMNnGXkKqBIwviWaD2C4xME&code_challenge_method=S256&grant_type=authorization_code&nonce=login_nonce&password=***&redirect_uri=com.bmw.connected%3A%2F%2Foauth&response_type=code&scope=openid+profile+email+offline_access+smacc+vehicle_data+perseus+dlm+svds+cesim+vsapi+remote_services+fupo+authenticate_user&state=cEG9eLAIi6Nv-aaCAniziE_B6FPoobva3qr5gukilYw&username=***
```
Sep 10 17:00:13 lx-evcc evcc[38961]: --
Sep 10 17:00:13 lx-evcc evcc[38961]: {"redirect_to" : "redirect_uri=com.bmw.connected://oauth?client_id=31c357a0-7a1d-4590-aa99-33b97244d048&response_type=code&scope=openid profile email offline_access smacc vehicle_data perseus dlm svds cesim vsapi remote_services fupo authenticate_user&state=cEG9eLAIi6Nv-aaCAniziE_B6FPoobva3qr5gukilYw&authorization=d7GiMW2Iu-UvMQhcMTVTunpM1TY.*AAJTSQACMDIAAlNLABxKYWFSQm95a01vaTd5TUVtcXQxTGZWS3VLVXM9AAR0eXBlAANDVFMAAlMxAAIwMQ..*"}
Sep 10 17:00:13 lx-evcc evcc[38961]: [bmw   ] TRACE 2024/09/10 17:00:13 POST https://customer.bmwgroup.com/gcdm/oauth/authenticate
Sep 10 17:00:13 lx-evcc evcc[38961]: [bmw   ] TRACE 2024/09/10 17:00:13 authorization=d7GiMW2Iu-UvMQhcMTVTunpM1TY.%2AAAJTSQACMDIAAlNLABxKYWFSQm95a01vaTd5TUVtcXQxTGZWS3VLVXM9AAR0eXBlAANDVFMAAlMxAAIwMQ..%2A&client_id=31c357a0-7a1d-4590-aa99-33b97244d048&code_challenge=MMfBJXS6w8TKLecD54w9FMNnGXkKqBIwviWaD2C4xME&code_challenge_method=S256&nonce=login_nonce&redirect_uri=com.bmw.connected%3A%2F%2Foauth&response_type=code&scope=openid+profile+email+offline_access+smacc+vehicle_data+perseus+dlm+svds+cesim+vsapi+remote_services+fupo+authenticate_user&state=cEG9eLAIi6Nv-aaCAniziE_B6FPoobva3qr5gukilYw
Sep 10 17:00:13 lx-evcc evcc[38961]: [bmw   ] TRACE 2024/09/10 17:00:13 POST https://customer.bmwgroup.com/gcdm/oauth/token
Sep 10 17:00:13 lx-evcc evcc[38961]: [bmw   ] TRACE 2024/09/10 17:00:13 code=3kw8sPuoHK1yofH_LOsL0Ek1KJU&code_verifier=kPabXLGkyydsikUxrwTnqoF2PMyIPcQfzfQSgSZaTPY&grant_type=authorization_code&redirect_uri=com.bmw.connected%3A%2F%2Foauth
Sep 10 17:00:13 lx-evcc evcc[38961]: --
Sep 10 17:00:13 lx-evcc evcc[38961]: {"gcid":"22a7d43b-71d5-4077-89c0-bfcf7af2405b","token_type":"Bearer","access_token":"uyDoRNO_M_8ikdlMC-LwqcbECDM","refresh_token":"mlIrXM1j49mfG797Oh170sewNOM","scope":"smacc vehicle_data perseus dlm svds openid profile vsapi remote_services authenticate_user cesim offline_access email fupo","expires_in":3599,"id_token":"eyJ0eXAiOiJKV1QiLCJraWQiOiJiMDZEZXIvYndra04reFNHMVFKQ0puQ1djK0k9IiwiYWxnIjoiUlMyNTYifQ.eyJhdF9oYXNoIjoiRlo4by1vNVZrWFRyd3N3b3dmR0F0dyIsInN1YiI6IjIyYTdkNDNiLTcxZDUtNDA3Ny04OWMwLWJmY2Y3YWYyNDA1YiIsImF1ZGl0VHJhY2tpbmdJZCI6ImY3ZGNiYjQxLTQxMDYtNDZmYy1iOGNkLWY4OWUyYzcyZTEwMi0yNTA0MTczMzkiLCJnY2lkIjoiMjJhN2Q0M2ItNzFkNS00MDc3LTg5YzAtYmZjZjdhZjI0MDViIiwiaXNzIjoiaHR0cHM6Ly9jdXN0b21lci5ibXdncm91cC5jb20vZ2NkbS9vYXV0aC9vcGVuaWQiLCJ0b2tlbk5hbWUiOiJpZF90b2tlbiIsImFjciI6IjAiLCJhenAiOiIzMWMzNTdhMC03YTFkLTQ1OTAtYWE5OS0zM2I5NzI0NGQwNDgiLCJhdXRoX3RpbWUiOjE3MjU5ODc2MTMsImV4cCI6MTcyNjA3NDAxMywiaWF0IjoxNzI1OTg3NjEzLCJlbWFpbCI6Ik9sYWYua3JhdXNlQG9rLWVkdi5kZSIsImVtYWlsX3ZlcmlmaWVkIjoiTUFJTF9BQ1RJVkUiLCJzdWJuYW1lIjoiMjJhN2Q0M2ItNzFkNS00MDc3LTg5YzAtYmZjZjdhZjI0MDViIiwiaG9tZV9tYXJrZXQiOiJERSIsImdpdmVuX25hbWUiOiJPbGFmIiwibm9uY2UiOiJsb2dpbl9ub25jZSIsImF1ZCI6IjMxYzM1N2EwLTdhMWQtNDU5MC1hYTk5LTMzYjk3MjQ0ZDA0OCIsImNfaGFzaCI6Iks1RFp3a3FaRk9jRU5yUjhqci1lTWciLCJzX2hhc2giOiJwcXpwa0pfS09mQ2htTTg4dFVLcExRIiwibG9naW5fbmFtZSI6Ik9sYWYua3JhdXNlQG9rLWVkdi5kZSIsInJlYWxtIjoiL2N1c3RvbWVyIiwic2FsdXRhdGlvbiI6IlNBTF9NUiIsInRva2VuVHlwZSI6IkpXVFRva2VuIiwiZmFtaWx5X25hbWUiOiJLcmF1c2UifQ.FTzsBlFZ_FV7-xd59xGw3eFEwc4bKO0VdlG-lf95iF76ZfFYN9IokCnmUfLP8bD8fFlJT3_GtrEZZBbkStCReYnc_LNLrmXPixtMZDhFnSB7UBQaEmD2q4jH8vl0YB9r0PqxRLgqrLQHTpT0EFx9pxsRnv_GD5bsBZIOs_Laj6QFcKA8N3CBRMa5djO-ks3MzbNCP4O3J13-nz6ocaE8nRv452l0W9MFD-652MMI7KM4blzxtmBNuIuorAuvzDypGlbbA-nOx68h9Cojhkhg6Q6bjxg5d3FqfFtYu4FUrH7jWuRDGFd7lAik_fK6reNtgYgpOJSHJXBe8MoWKhxn8w"}
Sep 10 17:00:13 lx-evcc evcc[38961]: [bmw   ] TRACE 2024/09/10 17:00:13 GET https://cocoapi.bmwgroup.com/eadrax-vcs/v4/vehicles?apptimezone=120&appDateTime=1725987613255
Sep 10 17:00:13 lx-evcc evcc[38961]: [bmw   ] TRACE 2024/09/10 17:00:13 [{"vin":"***","mappingInfo":{"isAssociated":null,"isLmmEnabled":false,"mappingStatus":"CONFIRMED","isPrimaryUser":true},"appVehicleType":"CONNECTED","attributes":{"lastFetched":"2024-09-10T17:00:13.560Z","model":"i3","year":2014,"color":4285228633,"brand":"BMW_I","driveTrain":"ELECTRIC","headUnitType":"NBT","headUnitRaw":"NBT","hmiVersion":"ID4","softwareVersionCurrent":{"puStep":{"month":7,"year":20},"iStep":510,"seriesCluster":"I001"},"softwareVersionExFactory":{"puStep":{"month":3,"year":14},"iStep":502,"seriesCluster":"I001"},"bodyType":"I01","countryOfOrigin":"DE","driverGuideInfo":{"androidAppScheme":"com.bmwgroup.driversguide.row","iosAppScheme":"bmwdriversguide:///open","androidStoreUrl":"https://play.google.com/store/apps/details?id=com.bmwgroup.driversguide.row","iosStoreUrl":"https://apps.apple.com/de/app/id714042749?mt=8"}}}]
Sep 10 17:00:14 lx-evcc evcc[38961]: [lp-1  ] DEBUG 2024/09/10 17:00:14 charge total import: 61.634kWh
Sep 10 17:00:14 lx-evcc evcc[38961]: [site  ] WARN 2024/09/10 17:00:14 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/sit [... cut ...]
Sep 10 17:01:45 lx-evcc evcc[38961]: [cache ] TRACE 2024/09/10 17:01:45 lp-1/planProjectedEnd: 0001-01-01 00:00:00 +0000 UTC
Sep 10 17:01:45 lx-evcc evcc[38961]: [cache ] TRACE 2024/09/10 17:01:45 lp-1/planOverrun: 0s
Sep 10 17:01:45 lx-evcc evcc[38961]: [cache ] TRACE 2024/09/10 17:01:45 greenShareHome: 1
Sep 10 17:01:45 lx-evcc evcc[38961]: [cache ] TRACE 2024/09/10 17:01:45 greenShareLoadpoints: 1
Sep 10 17:01:45 lx-evcc evcc[38961]: [cache ] TRACE 2024/09/10 17:01:45 tariffPriceHome: 0
Sep 10 17:01:45 lx-evcc evcc[38961]: [cache ] TRACE 2024/09/10 17:01:45 tariffPriceLoadpoints: 0

endurance1968, 2024-09-11:
I switch off charging last night -> no errors. After switching on the charging (PV+MIN) this morning. I got: (filtered on BMW only)
``
	Zeile 164: Sep 11 04:11:15 lx-evcc evcc[38961]: [bmw   ] TRACE 2024/09/11 04:11:15 GET https://cocoapi.bmwgroup.com/eadrax-vcs/v4/vehicles/state?apptimezone=120&appDateTime=1726027875491
	Zeile 170: Sep 11 04:11:16 lx-evcc evcc[38961]: [bmw   ] TRACE 2024/09/11 04:11:16 {"state":{"isLeftSteering":true,"lastFetched":"2024-09-11T04:11:15.766Z","lastUpdatedAt":"2024-09-11T03:57:18Z","isLscSupported":true,"range":118,"doorsState":{"combinedSecurityState":"SECURED","leftFront":"CLOSED","leftRear":"CLOSED","rightFront":"CLOSED","rightRear":"CLOSED","combinedState":"CLOSED","hood":"CLOSED","trunk":"CLOSED"},"windowsState":{"leftFront":"CLOSED","rightFront":"CLOSED","combinedState":"CLOSED"},"currentMileage":132785,"climateTimers":[{"isWeeklyTimer":false,"timerAction":"DEACTIVATE","timerWeekDays":[],"departureTime":{"hour":7,"minute":0}},{"isWeeklyTimer":true,"timerAction":"DEACTIVATE","timerWeekDays":["MONDAY"],"departureTime":{"hour":7,"minute":0}},{"isWeeklyTimer":true,"timerAction":"DEACTIVATE","timerWeekDays":["MONDAY"],"departureTime":{"hour":7,"minute":0}}],"requiredServices":[],"checkControlMessages":[],"chargingProfile":{"chargingControlType":"WEEKLY_PLANNER","reductionOfChargeCurrent":{"start":{"hour":6,"minute":0},"end":{"hour":12,"minute":0}},"chargingMode":"IMMEDIATE_CHARGING","chargingPreference":"CHARGING_WINDOW","departureTimes":[{"id":1,"timeStamp":{"hour":12,"minute":0},"action":"DEACTIVATE","timerWeekDays":[]},{"id":2,"timeStamp":{"hour":10,"minute":0},"action":"DEACTIVATE","timerWeekDays":["SUNDAY"]},{"id":3,"timeStamp":{"hour":8,"minute":0},"action":"DEACTIVATE","timerWeekDays":["WEDNESDAY"]},{"id":4,"action":"DEACTIVATE","timerWeekDays":[]}],"climatisationOn":false,"chargingSettings":{"targetSoc":100,"idcc":"NO_ACTION","hospitality":"NO_ACTION"}},"electricChargingState":{"chargingLevelPercent":91,"range":118,"isChargerConnected":true,"chargingStatus":"NOT_CHARGING","chargingTarget":100},"combustionFuelLevel":{},"driverPreferences":{"lscPrivacyMode":"OFF"},"securityOverviewMode":null,"vehicleSoftwareVersion":{"puStep":{"month":7,"year":20},"iStep":{"seriesCluster":"I001","year":20,"month":7,"iStep":510}}},"capabilities":{"remoteServices":{"inCarCamera":{"id":"inCarCamera","sta ...
	Zeile 170: Sep 11 04:11:16 lx-evcc evcc[38961]: [bmw   ] TRACE 2024/09/11 04:11:16 {"state":{"isLeftSteering":true,"lastFetched":"2024-09-11T04:11:15.766Z","lastUpdatedAt":"2024-09-11T03:57:18Z","isLscSupported":true,"range":118,"doorsState":{"combinedSecurityState":"SECURED","leftFront":"CLOSED","leftRear":"CLOSED","rightFront":"CLOSED","rightRear":"CLOSED","combinedState":"CLOSED","hood":"CLOSED","trunk":"CLOSED"},"windowsState":{"leftFront":"CLOSED","rightFront":"CLOSED","combinedState":"CLOSED"},"currentMileage":132785,"climateTimers":[{"isWeeklyTimer":false,"timerAction":"DEACTIVATE","timerWeekDays":[],"departureTime":{"hour":7,"minute":0}},{"isWeeklyTimer":true,"timerAction":"DEACTIVATE","timerWeekDays":["MONDAY"],"departureTime":{"hour":7,"minute":0}},{"isWeeklyTimer":true,"timerAction":"DEACTIVATE","timerWeekDays":["MONDAY"],"departureTime":{"hour":7,"minute":0}}],"requiredServices":[],"checkControlMessages":[],"chargingProfile":{"chargingControlType":"WEEKLY_PLANNER","reductionOfChargeCurrent":{"start":{"hour":6,"minute":0},"end":{"hour":12,"minute":0}},"chargingMode":"IMMEDIATE_CHARGING","chargingPreference":"CHARGING_WINDOW","departureTimes":[{"id":1,"timeStamp":{"hour":12,"minute":0},"action":"DEACTIVATE","timerWeekDays":[]},{"id":2,"timeStamp":{"hour":10,"minute":0},"action":"DEACTIVATE","timerWeekDays":["SUNDAY"]},{"id":3,"timeStamp":{"hour":8,"minute":0},"action":"DEACTIVATE","timerWeekDays":["WEDNESDAY"]},{"id":4,"action":"DEACTIVATE","timerWeekDays":[]}],"climatisationOn":false,"chargingSettings":{"targetSoc":100,"idcc":"NO_ACTION","hospitality":"NO_ACTION"}},"electricChargingState":{"chargingLevelPercent":91,"range":118,"isChargerConnected":true,"chargingStatus":"NOT_CHARGING","chargingTarget":100},"combustionFuelLevel":{},"driverPreferences":{"lscPrivacyMode":"OFF"},"securityOverviewMode":null,"vehicleSoftwareVersion":{"puStep":{"month":7,"year":20},"iStep":{"seriesCluster":"I001","year":20,"month":7,"iStep":510}}},"capabilities":{"remoteServices":{"inCarCamera":{"id":"inCarCamera","sta ...
	Zeile 171: Sep 11 04:11:16 lx-evcc evcc[38961]: [bmw   ] TRACE 2024/09/11 04:11:16 GET https://cocoapi.bmwgroup.com/eadrax-vcs/v4/vehicles/state?apptimezone=120&appDateTime=1726027876089
	Zeile 172: Sep 11 04:11:16 lx-evcc evcc[38961]: [bmw   ] TRACE 2024/09/11 04:11:16 {"state":{"isLeftSteering":true,"lastFetched":"2024-09-11T04:11:16.127Z","lastUpdatedAt":"2024-09-11T03:57:18Z","isLscSupported":true,"range":118,"doorsState":{"combinedSecurityState":"SECURED","leftFront":"CLOSED","leftRear":"CLOSED","rightFront":"CLOSED","rightRear":"CLOSED","combinedState":"CLOSED","hood":"CLOSED","trunk":"CLOSED"},"windowsState":{"leftFront":"CLOSED","rightFront":"CLOSED","combinedState":"CLOSED"},"currentMileage":132785,"climateTimers":[{"isWeeklyTimer":false,"timerAction":"DEACTIVATE","timerWeekDays":[],"departureTime":{"hour":7,"minute":0}},{"isWeeklyTimer":true,"timerAction":"DEACTIVATE","timerWeekDays":["MONDAY"],"departureTime":{"hour":7,"minute":0}},{"isWeeklyTimer":true,"timerAction":"DEACTIVATE","timerWeekDays":["MONDAY"],"departureTime":{"hour":7,"minute":0}}],"requiredSe [... cut ...] r":20},"iStep":{"seriesCluster":"I001","year":20,"month":7,"iStep":510}}},"capabilities":{"remoteServices":{"inCarCamera ...
	Zeile 570: Sep 11 04:13:25 lx-evcc evcc[38961]: [bmw   ] TRACE 2024/09/11 04:13:25 GET https://cocoapi.bmwgroup.com/eadrax-vcs/v4/vehicles/state?apptimezone=120&appDateTime=1726028005451
	Zeile 574: Sep 11 04:13:25 lx-evcc evcc[38961]: [bmw   ] TRACE 2024/09/11 04:13:25 { "statusCode": 403, "message": "Out of call volume quota. Quota will be replenished in 00:46:35." }
``

andig, 2024-09-11:
These do indeed look like new requests. Without context in the log (site/loadpoint) or config impossible to say whats going on. As usual: need complete issues...

endurance1968, 2024-09-12:
Moin, 
ich hab gestern mal alles auf config.yaml umgestellt, sprich das experimentell GUI Feature eleminiert.
jetzt läuft es ohne Fehler. Es scheint als gäbe es da noch ein Problem die cachetime aus der DB zu lesen oder korrekt von der GUI abzuspeichern. oder in der Kombi korrekt zu initialisieren oder was weiss ich. 
Ohne dass ich es jetzt noch verifizieren kann
[image]
ich meine man kann hier nur Zahlen ohne "m" eingeben. Wenn dann also 20 und nicht 20m in der DB landet und beim auslesen dann was schiefgeht... aber das ist Spekulation.
Ich bleibe jetzt beim yaml - ist für mich auch schneller anpassbar als über GUI.