evcc-io/evcc #16008: Fronius Gen24 Ladeleistung variabel einstellen

Loony2392, 2024-09-09:
Hallo zusammen,

wie bereits in #15994 erwähnt, ist die Netzladeleistung des Fronius Gen24 aktuell auf lediglich 100-200 W begrenzt. Es wäre hilfreich, im Reiter "Netzladen" eine zusätzliche Option hinzuzufügen, bei der man die gewünschte Ladeleistung festlegen kann, um den Fronius entsprechend effizienter beim Netzladen der Batterie zu steuern.

man könnte dies z.b auch in den Prozentschritten machen (in Abhängigkeit der WR Leistung)

[image]

Möglicher Weise wäre diese Funktion auch für andere Wechselrichter interessant.

[1683535878986-9958f969-6c1c-444e-a777-6d1959234873-gen24_primo_symo_inverter_register_map_int-amp-sf_storage.xlsx](https://github.com/user-attachments/files/16937017/1683535878986-9958f969-6c1c-444e-a777-6d1959234873-gen24_primo_symo_inverter_register_map_int-amp-sf_storage.xlsx)

andig, 2024-09-10:
Warum ist hier etwas einzustellen? Hier scheint eher laden mit Maximalleistung relevant um den günstigsten Slot nutzen zu können?

Welches Register ist für maximale Leistung notwendig?

TomF79, 2024-09-10:
> Warum ist hier etwas einzustellen? Hier scheint eher laden mit Maximalleistung relevant um den günstigsten Slot nutzen zu können?
> 
> Welches Register ist für maximale Leistung notwendig? Konntest du das lokal testen?

Sinnvoll wäre wenn es zumindest eine Auswahl geben würde mit wieviel Watt man laden möchte, so wie auch im Wechselrichter im Batteriemanagement die Regel dazu angelegt werden kann (Mindestladeleistung). Standardmässig mit 200W ist sinnfrei.

andig, 2024-09-10:
> Standardmässig mit 200W ist sinnfrei.

Check. Wir möchten das nicht für alle konfigurierbar machen (wer das braucht kann es extern umsetzen). Ein sinnvoller Defaultwert wäre aber notwendig.

> Register Prüfe ich.

Prima- schau bitte auch mal, ob es reicht, das 1x zu setzen oder ob wirklich evcc das tun muss...

TomF79, 2024-09-10:
> > Standardmässig mit 200W ist sinnfrei.
> 
> Check. Wir möchten das nicht für alle konfigurierbar machen (wer das braucht kann es extern umsetzen). Ein sinnvoller Defaultwert wäre aber notwendig.
> 

Ich habe bis jetzt durschnittlich und die meiste Zeit mit 2500W geladen, somit ist mein leerer 7,68kWh Akku in 3 Stunden voll, was gut in den günstigen Stunden umzusetzen ist und auch die Leistung im gesamten Haus in einen vernünftigen Rahmen belässt.

panzenpeda, 2024-09-10:
ich würde auch maximale Leistung bevorzugen, um die (ohnehin schon weniger werdenden) günstigen stunden zu nutzen.
spitze wäre aber natürlich eine einstellmöglichkeit

WolfImBusch, 2024-09-10:
Hallo,
super interessant, zumal ich die identischen Komponenten verwende und ab dem 1.1.2025 zu Tibber wechsle ... 😄

Auf GitHub gibt es ein Projekt mit Namen "Batcontrol" als separate Lösung. Ich bin leider kein besonders guter Programmierer aber vielleicht kann jemand anhand des Codes dort ableiten, welches Register und wie oft dieses zu setzen ist ableiten... 
Viele Grüße 
Alex

JauntyJosef, 2024-09-12:
Zum erzwungenen Netzladen schreibe ich in das Modbusregister 40355_OutWRte einem negativen Wert. Wenn ich zB -3000 (% mal 100) hineinschreibe, wird der Akku mit mindestens 30% von dem Wert, der in dem Register 40345_WChaMAx steht, geladen. Was an PV Leistung fehlt, kommt vom Netz.
**Ein positiver Wert in dem 40355 Register, gibt die maximale Entladeleistung an. Ein negativer Wert die minimale Ladeleistung.**
In dem Register 40348_StorCtl_Mod habe ich 3 stehen.

Loony2392, 2024-09-12:
40355 kann aber eigentlich nicht sein, nach der Doku von Fronius ist das für den Status

[image]

Wenn dann müsste es das register 40357 sein

[image]

JauntyJosef, 2024-09-12:
Mein Fehler: du musst meine angegebenen Register um 1 erhöhen, damit diese zu denen in der Doku passen (ich hab diese nur schnell aus dem Modbus Adapter im ioBroker entnommen … da muss man com Register 1 abziehen). 
OutWRte negativ schreiben passt aber. 

Wenn du in InWRte einen negativen Wert schreibst, dann erzwingst du, wenn ich mich recht erinnere, eine Entladung des Speichers. Müsste ich aber in der Modbus Doku noch einmal nachsehen.

JauntyJosef, 2024-09-13:
Auszug aus der Fronius Modus Doku Seite 34/35:

> **Beispiel 5: Laden im Bereich von 50% bis 75% der nominalen Leistung**

> Dieses Verhalten kann durch Limitierung der maximalen Ladeleistung auf 75% und Limitierung der maximalen Entladeleistung auf -50% erreicht werden => resultiert in Fenster [1650 W, 2475 W]
> - InWRte = 75% (setze Ladelimit auf 75% von WchaMax)
> - OutWRte = -50% (setze Entladelimit auf -50% von WchaMax)
> - StorCtl_Mod = 3 (schalte beide Grenzwerte aktiv, Bit-Muster: 11)
> - Der Batteriestatus in Fronius Solar.web wechselt zu „Erzwungene Nachladung“
> 
> **Beispiel 6: Entladen mit 50% der nominalen Leistung**

> Dieses Verhalten kann durch Limitierung der maximalen Ladeleistung auf -50% und Limitierung der maximalen Entladeleistung auf 50% erreicht werden => resultiert in Fenster [-1650 W, -1650 W]
> - InWRte = -50% (setze Ladelimit auf -50% von WchaMax)
> - OutWRte = 50% (setze Entladelimit auf 50% von WchaMax)
> - StorCtl_Mod = 3 (schalte beide Grenzwerte aktiv, Bit-Muster: 11)
> 
> **Beispiel 7: Laden mit 50% bis 100% der nominalen Leistung**
> Dieses Verhalten kann durch Limitierung der maximalen Entladeleistung auf -50% erreicht werden => resultiert in Fenster [1650 W, 3300 W]
> - OutWRte = -50% (setze Entladelimit auf -50% von WchaMax)
> - StorCtl_Mod = 2 (schaltet Entladegrenzwert aktiv, Bit-Muster: 10)
> - InWRte ist in diesem Fall nicht relevant
> - Der Batteriestatus in Fronius Solar.web wechselt zu „Erzwungene Nachladung“

andig, 2024-09-14:
> Zum erzwungenen Netzladen schreibe ich in das Modbusregister 40355_OutWRte einem negativen Wert. Wenn ich zB -3000 (% mal 100) hineinschreibe, wird der Akku mit mindestens 30% von dem Wert, der in dem Register 40345_WChaMAx steht, geladen. Was an PV Leistung fehlt, kommt vom Netz.

Laut Template wird `OutWRte` bereits mit -100% geschrieben. Die Frage ist also, warum das nicht zu maximaler Ladeleistung führt? Kann es sein, dass `WChaMAx` hier auf einem falschen Wert steht? Warum ist dieser falsch gesetzt?

Aus Sicht evcc scheint hier nichts falsch zu sein, daher erstmal tentatively geschlossen. Eine konfigurierbare Ladeleistung ist nicht geplant.

panzenpeda, 2024-09-14:
das wars also?
wir laden den akku also mit ~100Watt??
keine "besserung" in Aussicht?
schade!

andig, 2024-09-14:
Da es Probleme zu geben scheint, ein Update zu lesen und verstehen oder einen inhaltlichen Beitrag zu liefern Kommentar versteckt.

JauntyJosef, 2024-09-16:
> > Zum erzwungenen Netzladen schreibe ich in das Modbusregister 40355_OutWRte einem negativen Wert. Wenn ich zB
> Laut Template wird `OutWRte` bereits mit -100% geschrieben. Die Frage ist also, warum das nicht zu maximaler Ladeleistung führt? Kann es sein, dass `WChaMAx` hier auf einem falschen Wert steht? Warum ist dieser falsch gesetzt?
> 
> Aus Sicht evcc scheint hier nichts falsch zu sein, daher erstmal tentatively geschlossen. Eine konfigurierbare Ladeleistung ist nicht geplant.

@andig 
Darin könnte aber das Problem liegen. Kannst du das überprüfen, ob hier tatsächlich -100% geschrieben werden. Ich denke es wird "-100" in OutWRte geschrieben, aber das sind eben nur 1% und nicht 100%. Dazu müsste in das register "-10000" geschrieben werden. Wenn ich über evcc das Netzladen starten, steht dann jedenfalls -100 in dem Register ... und das sind eben 1%.

Loony2392, 2024-09-16:
> > > Zum erzwungenen Netzladen schreibe ich in das Modbusregister 40355_OutWRte einem negativen Wert. Wenn ich zB
> > > Laut Template wird `OutWRte` bereits mit -100% geschrieben. Die Frage ist also, warum das nicht zu maximaler Ladeleistung führt? Kann es sein, dass `WChaMAx` hier auf einem falschen Wert steht? Warum ist dieser falsch gesetzt?
> > 
> > 
> > Aus Sicht evcc scheint hier nichts falsch zu sein, daher erstmal tentatively geschlossen. Eine konfigurierbare Ladeleistung ist nicht geplant.
> 
> @andig Darin könnte aber das Problem liegen. Kannst du das überprüfen, ob hier tatsächlich -100% geschrieben werden. Ich denke es wird "-100" in `OutWRte` geschrieben, aber das sind eben nur 1% und nicht 100%. Dazu müsste in das register "-10000" geschrieben werden. Wenn ich über evcc das Netzladen starten, steht dann jedenfalls -100 in dem Register ... und das sind eben 1%. Der Scale Faktor steht in `InOutWRte_SF` und ist -2.
> 
> Edit: Und beim Ausschalten des Netzladens, wird glaube ich "100" geschrieben ... sollte "10000" sein. Da aber evcc beim Deaktivieren der Netzladung `StorCTLMod` mit 0 schreibt, hat das keine Auswirkung.

Das könnte daran liegen.

Wenn ich das laden der Batterie mal durch hochsetzen der Kosten erzwinge sagt HA das OutWRte steht auf -1% steht wenn ich das ganze Stoppe (Batterie gesperrt laut EVCC) steht der wert auf 0. #

[image]

[image]

andig, 2024-09-16:
Sowas steht im Logfile- da könnten wir einfach schauen!

Loony2392, 2024-09-16:
> Sowas steht im Logfile- da könnten wir einfach schauen!

Anbei das Modbus / SunSpec log

[evcc-20240916-120452-trace.log](https://github.com/user-attachments/files/17012529/evcc-20240916-120452-trace.log)

andig, 2024-09-16:
Ohje, ich sehe gerade das Todo im Code bei den Schreibbefehlen:

    // TODO scale factors

Ich hab kein Sunspecgerät. Wer mir Remote Zugang über VPN/SSH geben könnte bitte bei info@evcc.io melden. Vielen Dank!

JauntyJosef, 2024-09-17:
Was würdet du denn genau benötigen?

WolfImBusch, 2024-09-17:
@JauntyJosef Ich habe gestern für @andig einen VPN Zugang eingerichtet. Ich glaube er war noch nicht "hier" - aber das kommt bestimmt noch .. :-) Bin nur unschlüssig, ob ich evcc beenden muss, damit er debuggen kann 

Zur Info: ich habe die Komponenten Fronius Smartmeter TS 65A-3, Fronius Wechselrichter Fronius Symo Gen24 Plus sowie BYD Battery-Box Premium HVS. 

Steuerung erfolgt über Modbus und **nicht** über das Solar-API von Fronius - ich glaube, das war die Voraussetzung, dass das Laden der Hausbatterie über das Netz überhaupt klappt. Im Fronius habe ich "Batterieladung aus dem öffentlichen Netz zulassen" aktiviert sowie natürlich die Modus-Steuerung.

evcc hat gestern kurz angefangen zu laden (mit ca. 50W) - aber das Laden kurz danach beendet. Mal sehen, was andig rausfindet ...

Viele Grüße!

JauntyJosef, 2024-09-17:
Sehr gut, danke für die Info. Da bin ich gespannt.

closed 2024-09-18 by andig

WolfImBusch, 2024-09-19:
Ich werd verrückt - hab's rausgefunden, warum das Laden abbricht ...😄

Im Fronius muss unter Energiemanagement -> Eigenverbrauchsoptimierung auf "Manuell" umgestellt und konfiguriert werden ... dann bricht er das Laden nicht ab ... 

Habt ihr das bei euch auch so eingestellt?

Viele Grüße!!!!

@andig Nochmal herzlichen Dank für die schnelle Anpassung!!!

JauntyJosef, 2024-09-21:
@WolfImBusch 
- Ich benutze für Wechselrichter und Speicher die `fronius-gen24` Templates.
- Energieverbrauchsoptimierung steht bei auch auf Automatisch.
- "Batterieladung aus dem öffentlichen Netz zulassen" ist ein, allerdings ist das bei mit unter "Gerätekonfiguration" - "Komponenten"
- "Wechselrichter Steuerung über Modbus" ist ein. "Sunspec Model Type" ist "int + SF"

Hast du sonst vielleicht ein Skript laufen, dass den Akku steuert, wie zB prognosenbasiertes Laden?

WolfImBusch, 2024-09-21:
Einen schönen guten Morgen @JauntyJosef,

vielen Dank für Deine Antwort. Bei mir funktioniert es jetzt!!! D.h. das Laden der Hausbatterie beendet sich nicht mehr nach ca. 30s. @andig (nochmal vielen Dank!!) war so nett und hat sich das bei mir nochmal virtuell "vor Ort" angeschaut. 

Ursache war, dass das Register "InOutWRte_RvrtTms" ("Timeout period for charge/discharge rate") von meinem Wechselrichter (warum auch immer) auf "33.00" gesetzt worden war - also die von mir beobachteten 30s. andig hat dies bereits im Code geändert (siehe #16241 ) und die Änderung ist auch bereits im aktuellen Nightly enthalten.

Ich habe es heute morgen kurz getestet und - wie oben beschrieben - sieht es jetzt bei mir sehr gut aus.

Möchtest Du und @Loony2392 das bei euch auch mal testen? Sonst funktioniert es nur bei bei und bei euch nicht ... :-)

Viele Grüße!!

Loony2392, 2024-09-22:
> Einen schönen guten Morgen @JauntyJosef,
>
> vielen Dank für Deine Antwort. Bei mir funktioniert es jetzt!!! D.h. das Laden der Hausbatterie beendet sich nicht mehr nach ca. 30s. @andig (nochmal vielen [...]

Hallöchen, 

leider erst jetzt dazugekommen, Bis jetzt klappt das Netzladen Wunderbar mit der 0.130.12 :) Vielen dank @andig <3