evcc-io/evcc #16012: [OCPP] panic: runtime error: invalid memory address or nil pointer dereference

etecprojekt, 2024-09-10:
### Describe the bug

Wenn ein Ladevorgang aktiv ist und die Wallbox neugestartet wird, 
wird eine Panic ausgelöst und EVCC crasht.

### Steps to reproduce

1. EVCC läuft
2. Wallbox wird neugestartet
3. EVCC crasht und startet neu

### Configuration details

```yaml
#
```

### Log details

```text
[lp-2  ] DEBUG 2024/09/10 10:12:22 charge currents: [0 0 0]A
[lp-3  ] DEBUG 2024/09/10 10:12:22 charge power: 0W
[lp-3  ] DEBUG 2024/09/10 10:12:22 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/09/10 10:12:22 set charge mode: pv
[lp-4  ] ERROR 2024/09/10 10:12:23 charge power: timeout
[lp-4  ] ERROR 2024/09/10 10:12:23 charge currents: timeout
[circuit-wallboxen] DEBUG 2024/09/10 10:12:23 power: 11017W
[circuit-wallboxen] DEBUG 2024/09/10 10:12:23 current: 15.9A
[circuit-main] DEBUG 2024/09/10 10:12:23 power: -4629W
[circuit-main] DEBUG 2024/09/10 10:12:23 current: -6.49A
[site  ] DEBUG 2024/09/10 10:12:23 pv power: 5633W
[site  ] DEBUG 2024/09/10 10:12:23 grid meter: -4629W
[site  ] DEBUG 2024/09/10 10:12:23 grid currents: [-6.49 -6.82 -6.84]A
[site  ] DEBUG 2024/09/10 10:12:23 site power: -4629W
[lp-4  ] ERROR 2024/09/10 10:12:23 charge voltages: timeout
[lp-4  ] DEBUG 2024/09/10 10:12:23 !! session: chargeRater.chargedEnergy=4.9 - chargedAtStartup=0.0
[lp-4  ] ERROR 2024/09/10 10:12:23 charge total import: timeout
[lp-4  ] DEBUG 2024/09/10 10:12:23 charger status: B
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x20 pc=0x1855a80]

goroutine 54 [running]:
github.com/evcc-io/evcc/charger/ocpp.(*CS).GetCompositeScheduleRequest.func1(0x30?, {0x3866ce0, 0xc001176cc0})
	github.com/evcc-io/evcc/charger/ocpp/cs_core.go:80 +0xc0
github.com/lorenzodonini/ocpp-go/ocpp1%2e6.(*centralSystem).GetCompositeSchedule.func1({0x0?, 0x0?}, {0x3866ce0?, 0xc001176cc0?})
	github.com/lorenzodonini/ocpp-go@v0.18.0/ocpp1.6/central_system.go:335 +0x54
github.com/lorenzodonini/ocpp-go/ocpp1%2e6.(*centralSystem).SetChargePointDisconnectedHandler.func1({0x387be90, 0xc000be3040})
	github.com/lorenzodonini/ocpp-go@v0.18.0/ocpp1.6/central_system.go:379 +0xb4
github.com/lorenzodonini/ocpp-go/ocppj.(*Server).onClientDisconnected(0xc00106ea20, {0x387be90, 0xc000be3040})
	github.com/lorenzodonini/ocpp-go@v0.18.0/ocppj/server.go:334 +0x9f
github.com/lorenzodonini/ocpp-go/ws.(*Server).cleanupConnection(0xc0010b20f0, 0xc000be3040)
	github.com/lorenzodonini/ocpp-go@v0.18.0/ws/websocket.go:673 +0x13e
github.com/lorenzodonini/ocpp-go/ws.(*Server).writePump(0xc0010b20f0, 0xc000be3040)
	github.com/lorenzodonini/ocpp-go@v0.18.0/ws/websocket.go:655 +0xce5
created by github.com/lorenzodonini/ocpp-go/ws.(*Server).wsHandler in goroutine 27
	github.com/lorenzodonini/ocpp-go@v0.18.0/ws/websocket.go:556 +0xc9f
[main  ] INFO 2024/09/10 10:12:49 evcc 0.130.7 (9a0d6506)
[main  ] INFO 2024/09/10 10:12:49 using config file: /etc/evcc.yaml
[db    ] INFO 2024/09/10 10:12:49 using sqlite database: /root/.evcc/evcc.db
[main  ] INFO 2024/09/10 10:12:50 listening at :7070
[ABL_3W226305065-1] DEBUG 2024/09/10 10:12:50 waiting for chargepoint: 5m0s
[ABL_3W226305065-2] DEBUG 2024/09/10 10:12:50 failed triggering StatusNotification: cannot send request 3808552360, no client ABL_3W226305065 exists
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc 0.130.7 nightly

andig, 2024-09-10:
> charger/ocpp/cs_core.go:80

I'm sure this line cannot crash anymore. Are you sure you are running the nightly version as you think?

etecprojekt, 2024-09-10:
I re-pull the docker image and the error is gone!

closed 2024-09-10 by etecprojekt

UMLAUTaxl, 2025-03-08:
At 15:22 I think I unplugged the charger:
```
$ evcc -c evcc.yaml --database ~/.evcc/evcc.db
[main  ] INFO 2025/03/08 13:15:47 evcc 0.200.8
[main  ] INFO 2025/03/08 13:15:47 using config file: evcc.yaml
[db    ] INFO 2025/03/08 13:15:47 using sqlite database: /home/alec/.evcc/evcc.db
[main  ] INFO 2025/03/08 13:15:48 listening at :7070
[ocpp  ] INFO 2025/03/08 13:15:48 charge point connected, registering: A22A1445799
[ocpp  ] ERROR 2025/03/08 13:16:04 ocpp message (6604): PropertyConstraintViolation - Field Call.Payload.MeterValue[0].SampledValue must be minimum 1, but was 0 for feature MeterValues ([2,"6604","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2025-03-08T12:16:03Z","sampledValue":[]}],"transactionId":1741418242}])
[lp-1  ] ERROR 2025/03/08 13:16:14 charge total import: not available
[site  ] INFO 2025/03/08 13:16:14 site config:
[site  ] INFO 2025/03/08 13:16:14   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2025/03/08 13:16:14     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2025/03/08 13:16:14     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/03/08 13:16:14   vehicles:
[site  ] INFO 2025/03/08 13:16:14     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2025/03/08 13:16:14   tariffs:
[site  ] INFO 2025/03/08 13:16:14     grid:      ✗
[site  ] INFO 2025/03/08 13:16:14     feed-in:   ✗
[site  ] INFO 2025/03/08 13:16:14     co2:       ✗
[site  ] INFO 2025/03/08 13:16:14     solar:     solar
[lp-1  ] INFO 2025/03/08 13:16:14 loadpoint 1:
[lp-1  ] INFO 2025/03/08 13:16:14   mode:        pv
[lp-1  ] INFO 2025/03/08 13:16:14   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2025/03/08 13:16:14   meters:      charge ✓
[lp-1  ] INFO 2025/03/08 13:16:14     charge:    power ✓ energy ✓ currents ✓
[lp-2  ] INFO 2025/03/08 13:16:14 loadpoint 2:
[lp-2  ] INFO 2025/03/08 13:16:14   mode:        pv
[lp-2  ] INFO 2025/03/08 13:16:14   charger:     power ✓ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-2  ] INFO 2025/03/08 13:16:14   meters:      charge ✓
[lp-2  ] INFO 2025/03/08 13:16:14     charge:    power ✓ energy ✗ currents ✗
[lp-2  ] WARN 2025/03/08 13:16:17 configured phases 3 do not match physical phases 1
[site  ] WARN 2025/03/08 13:16:17 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[lp-1  ] INFO 2025/03/08 13:16:17 car connected
[lp-2  ] INFO 2025/03/08 13:16:28 car connected
[lp-2  ] INFO 2025/03/08 13:16:28 start charging ->
[lp-1  ] INFO 2025/03/08 13:17:51 vehicle updated: unknown -> EUp
[lp-2  ] ERROR 2025/03/08 13:21:35 charge power: request failed: http POST failed: Post "http://192.168.178.46/app/request?seq=-1181135907": EOF
[lp-1  ] ERROR 2025/03/08 13:24:06 charge voltages: timeout
[lp-1  ] ERROR 2025/03/08 13:24:06 charge total import: timeout
[lp-1  ] ERROR 2025/03/08 13:24:07 charge power: timeout
[lp-1  ] ERROR 2025/03/08 13:24:07 charge currents: timeout
[lp-1  ] ERROR 2025/03/08 13:24:08 charge power: timeout
[lp-1  ] ERROR 2025/03/08 13:24:09 charge currents: timeout
[lp-1  ] ERROR 2025/03/08 13:24:09 charge voltages: timeout
[lp-1  ] ERROR 2025/03/08 13:24:09 charge total import: timeout
[lp-1  ] ERROR 2025/03/08 13:24:18 charge power: timeout
[lp-1  ] ERROR 2025/03/08 13:24:19 charge currents: timeout
[ocpp  ] ERROR 2025/03/08 14:10:02 ocpp message (7697): OccurrenceConstraintViolation - Field Call.Payload.MeterValue[0].SampledValue[6].Value required but not found for feature MeterValues ([2,"7697","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2025-03-08T13:09:59Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"16.0","context":"Trigger"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"0.0"},{"measurand":"Current.Import","unit":"A","phase":"L2","value":"0.0","context":"Trigger"},{"measurand":"Current.Import","unit":"A","phase":"L3","value":"0.0","context":"Trigger"},{"measurand":"Voltage","unit":"V","phase":"L1","value":"0.0","context":"Trigger"},{"measurand":"Voltage","unit":"V","phase":"L2","value":"0.0","context":"Trigger"},{"measurand":"Voltage","unit":"riodic"},{"measurand":"Voltage","unit":"V","phase":"L3","value":"0.0","context":"Sample.Periodic"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"365750.0","context":"Sample.Periodic"},{"measurand":"Power.Active.Import","unit":"W","value":"0.0","context":"Sample.Periodic"}]}],"transactionId":1741418242}])
[main  ] ERROR 2025/03/08 14:40:47 cannot save settings: database or disk is full (13); SQL logic error: cannot rollback - no transaction is active (1)
[lp-2  ] INFO 2025/03/08 15:20:18 stop charging <-
[lp-2  ] WARN 2025/03/08 15:20:18 charger out of sync: expected enabled, got disabled
[lp-1  ] INFO 2025/03/08 15:21:48 car disconnected
[lp-1  ] INFO 2025/03/08 15:21:48 vehicle updated: EUp -> unknown
[lp-1  ] ERROR 2025/03/08 15:22:19 charger disable: set charging profile: timeout
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x30 pc=0x1355a95]

goroutine 66 [running]:
github.com/lorenzodonini/ocpp-go/ocppj.(*DefaultServerDispatcher).messagePump(0xc000fd8d00)
	github.com/lorenzodonini/ocpp-go@v0.19.0/ocppj/dispatcher.go:535 +0x375
created by github.com/lorenzodonini/ocpp-go/ocppj.(*DefaultServerDispatcher).Start in goroutine 65
	github.com/lorenzodonini/ocpp-go@v0.19.0/ocppj/dispatcher.go:410 +0x148
```

andig, 2025-03-08:
This is indeed a panic in the Ocpp Library that should not happen. Is there any way to reproduce this?

/cc @lorenzodonini