evcc-io/evcc #16021: ID Charger Connect lässt sich nicht mehr über EVCC steuern

PVMartin, 2024-09-10:
### Describe the bug

Wenn ein Auto an die Wallbox angeschlossen wird, fängt es - unabhängig von der EVCC-Einstellung - mit der konfigurierten Mindestleistung an zu laden.
Im EVCC wird die Ladeleistung als Leistung des Hauses (nicht des Autos) angezeigt.
Die Steuerung der Ladeleistung (AUS, PV, Min+PV, Schnell) reagiert nicht.
Das Auto wird entweder als "verbunden" oder als "Ladebereit. Warte auf Fahrzeug ..." angezeigt.
Der SoC beim Anstecken wird ebenfalls angezeigt, allerdings verändert sich der Wert durch das dann stattfindende Laden nicht.

### Steps to reproduce

1. Update auf 0.130.8
2. Auto mit Wallbox verbinden.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxx
sponsortoken: xxx

interval: 20s # control cycle interval (Tested: shortest possible interval 20s)

meters:
- name: grid1
  type: template
  template: sma-home-manager 
  usage: grid  
  host: 192.168.2.7  
- name: pv2
  type: template
  template: sma-inverter 
  usage: pv  
  host: 192.168.2.17

chargers:
- name: wallbox5
  type: template
  template: elliconnect 
  ski: xxx  
  ip: 192.168.2.16  
- name: wallbox6
  type: template
  template: elliconnect 
  ski: xxx  
  ip: 192.168.2.15

vehicles:
- name: ev3
  type: template
  template: id 
  title: ID.3  
  icon: car  
  user: xxx  
  password: xxx  
  vin: xxx  
  capacity: 45  
  phases: 2     
  minCurrent: 6  
[... 474 lines cut ...]
[lp-1  ] DEBUG 2024/09/10 21:13:05 max charge current: 6A
[eebus ] DEBUG 2024/09/10 21:13:05 Recv: d:_i:47859_Elli-Wallbox-2122A1EC77:LoadControl to LoadControl result 73 27 ResultData 9
[eebus ] DEBUG 2024/09/10 21:13:05 Error Result received 9: Client not bound
[site  ] DEBUG 2024/09/10 21:13:25 ----
[lp-1  ] DEBUG 2024/09/10 21:13:25 charge power: 0W
[lp-2  ] DEBUG 2024/09/10 21:13:25 charge power: 0W
[site  ] DEBUG 2024/09/10 21:13:25 pv power: 0W
[site  ] DEBUG 2024/09/10 21:13:25 grid meter: 4006W
[site  ] DEBUG 2024/09/10 21:13:25 grid powers: [2302 1533 172]W
[site  ] DEBUG 2024/09/10 21:13:25 grid currents: [10.5 7.15 1.19]A
[site  ] DEBUG 2024/09/10 21:13:25 site power: 4006W
[lp-2  ] DEBUG 2024/09/10 21:13:25 charger status: A
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.8 (89fb334)

DerAndereAndi, 2024-09-10:
Kannst du bitte ein Log mit level `trace` für EEBUS bereitstellen?

DerAndereAndi, 2024-09-11:
Ich hab das Problem identifiziert und bin an nem fix dran.

closed 2024-09-11 by andig

TheChatty, 2024-09-11:
Den Fehler habe ich auch. Bin gespannt, ob der Fix funktioniert.

PVMartin, 2024-09-13:
Nach Update auf die Version 0.130.9 trat der Fehler nicht mehr auf.
Großes Lob an das Team für die super-schnelle Bearbeitung! :-)
Toll, dass Ihr Euch so kümmert!!

PVMartin, 2024-09-15:
Nach dem heutigen Update auf 0.130.11 scheint dieser Fehler wieder da zu sein. Anbei das Log:
[evcc-20240915-160723-trace.log.txt](https://github.com/user-attachments/files/17006050/evcc-20240915-160723-trace.log.txt)

andig, 2024-09-15:
Bitte den ganzen SMA kram aus dem Log rauswerfen. Danke.

DerAndereAndi, 2024-09-15:
Nein, das Problem tritt nicht wieder auf. Im Log ist kein Fehlverhalten in EEBUS zu erkennen, auch gibt es in evcc keine Änderungen in EEBUS seit Version 0.130.9.

Das Auto lädt weil es die Ladevorgaben von evcc bekommt, und es kommen auch Messwerte rein.

```
[lp-1  ] DEBUG 2024/09/15 16:05:14 site power -4740W <= 0W enable threshold
[lp-1  ] DEBUG 2024/09/15 16:05:14 pv enable in 0s
[lp-1  ] DEBUG 2024/09/15 16:05:14 pv enable timer elapsed
[eebus ] DEBUG 2024/09/15 16:05:14 Send: d:_i:47859_Elli-Wallbox-2122A1EC77 write 115 LoadControlLimitListData
[lp-1  ] DEBUG 2024/09/15 16:05:14 max charge current: 6A
[eebus ] TRACE 2024/09/15 16:05:14 Send: 03f544c174fd203895fa3e4748919c74cddfb328 {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_n:EVCC_HEMS-EVCC-3463386162356365"},{"entity":[1]},{"feature":7}]},{"addressDestination":[{"device":"d:_i:47859_Elli-Wallbox-2122A1EC77"},{"entity":[1,1]},{"feature":10}]},{"msgCounter":115},{"cmdClassifier":"write"},{"ackRequest":true}]},{"payload":[{"cmd":[[{"function":"loadControlLimitListData"},{"filter":[[{"cmdControl":[{"partial":[]}]}]]},{"loadControlLimitListData":[{"loadControlLimitData":[[{"limitId":0},{"isLimitActive":true},{"value":[{"number":6},{"scale":0}]}],[{"limitId":1},{"isLimitActive":true},{"value":[{"number":6},{"scale":0}]}],[{"limitId":2},{"isLimitActive":true},{"value":[{"number":6},{"scale":0}]}]]}]}]]}]}]}}]}
[eebus ] DEBUG 2024/09/15 16:05:14 Send: d:_i:47859_Elli-Wallbox-2122A1EC77 write 116 LoadControlLimitListData
[lp-1  ] DEBUG 2024/09/15 16:05:14 charger enable
[lp-1  ] DEBUG 2024/09/15 16:05:14 wake-up timer: start
[eebus ] TRACE 2024/09/15 16:05:14 Send: 03f544c174fd203895fa3e4748919c74cddfb328 {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_n:EVCC_HEMS-EVCC-3463386162356365"},{"entity":[1]},{"feature":7}]},{"addressDestination":[{"device":"d:_i:47859_Elli-Wallbox-2122A1EC77"},{"entity":[1,1]},{"feature":10}]},{"msgCounter":116},{"cmdClassifier":"write"},{"ackRequest":true}]},{"payload":[{"cmd":[[{"function":"loadControlLimitListData"},{"filter":[[{"cmdControl":[{"partial":[]}]}]]},{"loadControlLimitListData":[{"loadControlLimitData":[[{"limitId":0},{"isLimitActive":true},{"value":[{"number":6},{"scale":0}]}],[{"limitId":1},{"isLimitActive":true},{"value":[{"number":6},{"scale":0}]}],[{"limitId":2},{"isLimitActive":true},{"value":[{"number":6},{"scale":0}]}]]}]}]]}]}]}}]}
[eebus ] TRACE 2024/09/15 16:05:14 Recv: 03f544c174fd203895fa3e4748919c74cddfb328 {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_i:47859_Elli-Wallbox-2122A1EC77"},{"entity":[1,1]},{"feature":10}]},{"addressDestination":[{"device":"d:_n:EVCC_HEMS-EVCC-3463386162356365"},{"entity":[1]},{"feature":7}]},{"msgCounter":178},{"cmdClassifier":"notify"}]},{"payload":[{"cmd":[[{"function":"loadControlLimitListData"},{"filter":[[{"cmdControl":[{"partial":[]}]}]]},{"loadControlLimitListData":[{"loadControlLimitData":[[{"limitId":0},{"isLimitActive":true},{"value":[{"number":6},{"scale":0}]}],[{"limitId":1},{"isLimitActive":true},{"value":[{"number":6},{"scale":0}]}],[{"limitId":2},{"isLimitActive":true},{"value":[{"number":6},{"scale":0}]}]]}]}]]}]}]}}]}
[eebus ] DEBUG 2024/09/15 16:05:14 Recv: d:_i:47859_Elli-Wallbox-2122A1EC77:LoadControl to LoadControl notify 178 LoadControlLimitListData
```

und 

```
[lp-1  ] DEBUG 2024/09/15 16:07:14 pv charge current: 6.61A = 6A + 0.607A (-419W @ 3p)
[lp-1  ] DEBUG 2024/09/15 16:07:14 pv timer reset
[lp-1  ] DEBUG 2024/09/15 16:07:14 pv timer inactive
[eebus ] DEBUG 2024/09/15 16:07:14 Send: d:_i:47859_Elli-Wallbox-2122A1EC77 write 177 LoadControlLimitListData
[lp-1  ] DEBUG 2024/09/15 16:07:14 max charge current: 6.61A
[eebus ] TRACE 2024/09/15 16:07:14 Send: 03f544c174fd203895fa3e4748919c74cddfb328 {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_n:EVCC_HEMS-EVCC-3463386162356365"},{"entity":[1]},{"feature":7}]},{"addressDestination":[{"device":"d:_i:47859_Elli-Wallbox-2122A1EC77"},{"entity":[1,1]},{"feature":10}]},{"msgCounter":177},{"cmdClassifier":"write"},{"ackRequest":true}]},{"payload":[{"cmd":[[{"function":"loadControlLimitListData"},{"filter":[[{"cmdControl":[{"partial":[]}]}]]},{"loadControlLimitListData":[{"loadControlLimitData":[[{"limitId":0},{"isLimitActive":true},{"value":[{"number":66068},{"scale":-4}]}],[{"limitId":1},{"isLimitActive":true},{"value":[{"number":66068},{"scale":-4}]}],[{"limitId":2},{"isLimitActive":true},{"value":[{"number":66068},{"scale":-4}]}]]}]}]]}]}]}}]}
[eebus ] TRACE 2024/09/15 16:07:14 Recv: 03f544c174fd203895fa3e4748919c74cddfb328 {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_i:47859_Elli-Wallbox-2122A1EC77"},{"entity":[1,1]},{"feature":10}]},{"addressDestination":[{"device":"d:_n:EVCC_HEMS-EVCC-3463386162356365"},{"entity":[1]},{"feature":7}]},{"msgCounter":255},{"cmdClassifier":"notify"}]},{"payload":[{"cmd":[[{"function":"loadControlLimitListData"},{"filter":[[{"cmdControl":[{"partial":[]}]}]]},{"loadControlLimitListData":[{"loadControlLimitData":[[{"limitId":0},{"isLimitActive":true},{"value":[{"number":66068},{ [... cut ...] "},{"filter":[[{"cmdControl":[{"partial":[]}]}]]},{"measurementListData":[{"measurementData":[[{"measurementId":0},{"valueType":"value"},{"value":[{"number":6607},{"scale":-3}]}],[{"measurementId":1},{"valueType":"value"},{"value":[{"number":6607},{"scale":-3}]}],[{"measurementId":2},{"valueType":"value"},{"value":[{"number":6607},{"scale":-3}]}]]}]}]]}]}]}}]}
[eebus ] DEBUG 2024/09/15 16:07:15 Recv: d:_i:47859_Elli-Wallbox-2122A1EC77:Measurement to Measurement notify 257 MeasurementListData
```

PVMartin, 2024-09-15:
Tut mir leid, soweit reicht mein Wissen nicht, dass ich den Inhalt der Log-Dateien nachvollziehen könnte. Ich werde es gerne nochmals ausprobieren, bin aber jetzt erstmal wieder zurück auf 0.130.9.

DerAndereAndi, 2024-09-15:
Musst du auch nicht. Bitte bei Problemen ein NEUES Ticket aufmachen und zum Log auch reinschreiben um welche Uhrzeit welches Verhalten erwartet war, und was man gesehen hat. Hier kann ich nur sagen: Es gibt kein Problem.

Das Problem in diesem Ticket wurde gelöst und damit sollte man hier auch nicht weiter diskutieren.

andig, 2024-09-15:
Dann machen wir mal zu... vielen Dank!