evcc-io/evcc #16038: Renault: Vorklimatisierung erkannt

maatinh, 2024-09-11:
### Describe the bug

evcc erkennt Vorklimatisierung obwohl keine Klima läuft. Das scheint allerdings nicht immer der Fall zu sein. Als ich das Auto angesteckt habe, wurde keine Klima erkannt. Erst während des Ladens irgendwann.

### Steps to reproduce

-

### Configuration details

```yaml
vehicles:
  - name: Zoe
    title: Zoe-PV-Laden
    type: template
    phases: 1
    template: renault
    capacity: 25 # kWh
    user: XXX
    password: XXX
    mode: pv # enable PV-charging when vehicle is identified
    minCurrent: 8
    alternativewakeup: true
```

### Log details

```text
[renault] TRACE 2024/09/11 13:08:30 GET https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/XXXXXXX/kamereon/kca/car-adapter/v2/cars/XXXXXXX/battery-status?country=DE
[renault] TRACE 2024/09/11 13:08:31 {"type":"FUNCTIONAL","messages":[{"code":"err.func.wired.unauthorized","message":"The access is unauthorized"}],"errors":[{"errorCode":"err.func.wired.unauthorized","errorMessage":"The access is unauthorized"}],"error_reference":"FUNCTIONAL"}
[renault] TRACE 2024/09/11 13:08:31 GET https://accounts.eu1.gigya.com/accounts.login?apiKey=XXXXXXX&loginID=***&password=***
[renault] TRACE 2024/09/11 13:08:31 { "callId": "82847644f9c2475da620a08721a95464", "errorCode": 0, "apiVersion": 2, "statusCode": 200, "statusReason": "OK", "time": "2024-09-11T11:08:31.458Z", "registeredTimestamp": 1700745450, "UID": "XXXXXXX", "UIDSignature": "XXXXXXX", "signatureTimestamp": "1726052911", "created": "2023-11-23T13:17:30.843Z", "createdTimestamp": 1700745450, "isActive": true, "isRegistered": true, "isVerified": true, "lastLogin": "2024-09-11T11:08:31.417Z", "lastLoginTimestamp": 1726052911, "lastUpdated": "2024-07-11T12:15:02.467Z", "lastUpdatedTimestamp": 1720700102467, "loginProvider": "site", "oldestDataUpdated": "2023-11-23T13:17:30.843Z", "oldestDataUpdatedTimestamp": 1700745450843, "profile": { "email": "***" }, "registered": "2023-11-23T13:17:30.964Z", "socialProviders": "site", "verified": "2023-11-23T13:17:49.502Z", "verifiedTimestamp": 1700745469502, "newUser": false, "sessionInfo": { "cookieName": "gac_XXXXXXX", "cookieValue": "XXXXXXX" } }
[renault] TRACE 2024/09/11 13:08:31 GET https://accounts.eu1.gigya.com/accounts.getJWT?apiKey=XXXXXXX&expiration=900&fields=data.personId%2Cdata.gigyaDataCenter&login_token=XXXXXXX
[renault] TRACE 2024/09/11 13:08:31 { "callId": "89a93fc6b8e2488ba60c5c7bb7fbfae1", "errorCode": 0, "apiVersion": 2, "statusCode": 200, "statusReason": "OK", "time": "2024-09-11T11:08:31.583Z", "id_token": "XXXXXXX.XXXXXXX.XXXXXXX" }
[renault] TRACE 2024/09/11 13:08:31 GET https://accounts.eu1.gigya.com/accounts.getAccountInfo?apiKey=XXXXXXX&login_token=XXXXXXX
[renault] TRACE 2024/09/11 13:08:31 { "callId": "51c4e2786a2d49e19e43bb159d78b09e", "errorCode": 0, "apiVersion": 2, "statusCode": 200, "statusReason": "OK", "time": "2024-09-11T11:08:31.665Z", "registeredTimestamp": 1700745450000, "UID": "XXXXXXX", "UIDSignature": "XXXXXXX", "signatureTimestamp": "1726052911", "created": "2023-11-23T13:17:30.843Z", "createdTimestamp": 1700745450000, "data": { "personId": "XXXXXXX", "gigyaDataCenter": "eu1.gigya.com" }, "preferences": {}, "emails": { "verified": [ "***" ], "unverified": [] }, "isActive": true, "isRegistered": true, "isVerified": true, "lastLogin": "2024-09-11T11:08:31.417Z", "lastLoginTimestamp": 1726052911000, "lastUpdated": "2024-07-11T12:15:02.467Z", "lastUpdatedTimestamp": 1720700102467, "loginProvider": "site", "oldestDataUpdated": "2023-11-23T13:17:30.843Z", "oldestDataUpdatedTimestamp": 1700745450843, "password": { "created": "2023-11-23T13:17:30.886Z" }, "profile": { "email": "***" }, "registered": "2023-11-23T13:17:30.964Z", "socialProviders": "site", "verified": "2023-11-23T13:17:49.502Z", "verifiedTimestamp": 1700745469502 }
[renault] TRACE 2024/09/11 13:08:31 GET https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/XXXXXXX/kamereon/kca/car-adapter/v2/cars/XXXXXXX/battery-status?country=DE
[renault] TRACE 2024/09/11 13:08:31 {"data":{"id":"XXXXXXX","attributes":{"timestamp":"2024-09-11T10:56:27Z","batteryLevel":62,"batteryAutonomy":73,"plugStatus":1,"chargingStatus":-1.1,"chargingRemainingTime":220}}}
[renault] TRACE 2024/09/11 13:15:09 GET https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/XXXXXXX/kamereon/kca/car-adapter/v1/cars/XXXXXXX/cockpit?country=DE
[renault] TRACE 2024/09/11 13:15:09 GET https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/XXXXXXX/kamereon/kca/car-adapter/v2/cars/XXXXXXX/battery-status?country=DE
[renault] TRACE 2024/09/11 13:15:09 {"type":"FUNCTIONAL","messages":[{"code":"err.func.wired.unauthorized","message":"The access is unauthorized"}],"errors":[{"errorCode":"err.func.wired.unauthorized","errorMessage":"The access is unauthorized"}],"error_reference":"FUNCTIONAL"}
[renault] TRACE 2024/09/11 13:15:09 GET https://accounts.eu1.gigya.com/accounts.login?apiKey=XXXXXXX&loginID=***&password=***
[renault] TRACE 2024/09/11 13:15:10 {"data":{"id":"XXXXXXX","attributes":{"totalMileage":58801}}}
[renault] TRACE 2024/09/11 13:15:10 { "callId": "XXXXXXX", "errorCode": 0, "apiVersion": 2, "statusCode": 200, "statusReason": "OK", "time": "2024-09-11T11:15:10.297Z", "registeredTimestamp": 1700745450, "UID": "XXXXXXX", "UIDSignature": "XXXXXXX=", "signatureTimestamp": "1726053310", "created": "2023-11-23T13:17:30.843Z", "createdTimestamp": 1700745450, "isActive": true, "isRegistered": true, "isVerified": true, "lastLogin": "2024-09-11T11:15:10.255Z", "lastLoginTimestamp": 1726053310, "lastUpdated": "2024-07-11T12:15:02.467Z", "lastUpdatedTimestamp": 1720700102467, "loginProvider": "site", "oldestDataUpdated": "2023-11-23T13:17:30.843Z", "oldestDataUpdatedTimestamp": 1700745450843, "profile": { "email": "***" }, "registered": "2023-11-23T13:17:30.964Z", "socialProviders": "site", "verified": "2023-11-23T13:17:49.502Z", "verifiedTimestamp": 1700745469502, "newUser": false, "sessionInfo": { "cookieName": "gac_XXXXXXX", "cookieValue": "XXXXXXX" } }
[renault] TRACE 2024/09/11 13:15:10 GET https://accounts.eu1.gigya.com/accounts.getJWT?apiKey=XXXXXXX&expiration=900&fields=data.personId%2Cdata.gigyaDataCenter&login_token=XXXXXXX
[renault] TRACE 2024/09/11 13:15:10 { "callId": "3c911c69c207456aa62a750a46a7a334", "errorCode": 0, "apiVersion": 2, "statusCode": 200, "statusReason": "OK", "time": "2024-09-11T11:15:10.378Z", "id_token": "XXXXXXX.XXXXXXX.XXXXXXX-XXXXXXX-XXXXXXX" }
[renault] TRACE 2024/09/11 13:15:10 GET https://accounts.eu1.gigya.com/accounts.getAccountInfo?apiKey=XXXXXXX&login_token=XXXXXXX
[renault] TRACE 2024/09/11 13:15:10 { "callId": "02fd667b5de44797a220782b9e01af98", "errorCode": 0, "apiVersion": 2, "statusCode": 200, "statusReason": "OK", "time": "2024-09-11T11:15:10.461Z", "registeredTimestamp": 1700745450000, "UID": "XXXXXXX", "UIDSignature": "XXXXXXX=", "signatureTimestamp": "1726053310", "created": "2023-11-23T13:17:30.843Z", "createdTimestamp": 1700745450000, "data": { "personId": "XXXXXXX", "gigyaDataCenter": "eu1.gigya.com" }, "preferences": {}, "emails": { "verified": [ "***" ], "unverified": [] }, "isActive": true, "isRegistered": true, "isVerified": true, "lastLogin": "2024-09-11T11:15:10.255Z", "lastLoginTimestamp": 1726053310000, "lastUpdated": "2024-07-11T12:15:02.467Z", "lastUpdatedTimestamp": 1720700102467, "loginProvider": "site", "oldestDataUpdated": "2023-11-23T13:17:30.843Z", "oldestDataUpdatedTimestamp": 1700745450843, "password": { "created": "2023-11-23T13:17:30.886Z" }, "profile": { "email": "***" }, "registered": "2023-11-23T13:17:30.964Z", "socialProviders": "site", "verified": "2023-11-23T13:17:49.502Z", "verifiedTimestamp": 1700745469502 }
[renault] TRACE 2024/09/11 13:15:10 GET https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/XXXXXXX/kamereon/kca/car-adapter/v2/cars/XXXXXXX/battery-status?country=DE
[renault] TRACE 2024/09/11 13:15:11 {"data":{"id":"XXXXXXX","attributes":{"timestamp":"2024-09-11T10:56:27Z","batteryLevel":62,"batteryAutonomy":73,"plugStatus":1,"chargingStatus":-1.1,"chargingRemainingTime":220}}}
[renault] TRACE 2024/09/11 13:15:11 GET https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/XXXXXXX/kamereon/kca/car-adapter/v1/cars/XXXXXXX/cockpit?country=DE
[renault] TRACE 2024/09/11 13:15:11 {"data":{"id":"XXXXXXX","attributes":{"totalMileage":58801}}}
[... cut ...]
[renault] TRACE 2024/09/11 13:43:30 GET https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/XXXXXXX/kamereon/kca/car-adapter/v2/cars/XXXXXXX/battery-status?country=DE
[renault] TRACE 2024/09/11 13:43:31 {"data":{"id":"XXXXXXX","attributes":{"timestamp":"2024-09-11T11:26:26Z","batteryLevel":67,"batteryAutonomy":78,"plugStatus":1,"chargingStatus":-1.1,"chargingRemainingTime":380}}}
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.8

andig, 2024-09-11:
Im Logfile leider nicht erkennbar.

closed 2024-09-11 by andig

maatinh, 2024-09-11:
> Im Logfile leider nicht erkennbar.

[renault] TRACE 2024/09/11 13:39:32 {"data":{"id":"XXXXXXX","attributes":{"hvacStatus":"**unavailable**","lastUpdateTime":"2024-09-11T11:26:26Z"}}} 

??

maatinh, 2024-09-11:
https://github.com/evcc-io/evcc/blob/dbfce599a7c359e8bad87c237143db5ab2e8b855/vehicle/renault/provider.go#L153

Fehlt da "unavailable"?

@andig

maatinh, 2024-09-11:
Danke. Mit der Positiv-ID kann ich leider nicht dienen.
Ich habe gesehen, dass der climater aktualisiert wird, wenn das Auto die eingestellt Ladegrenze erreicht hat. Der Ladestand war 75% im Modus PV. In dem Moment wo ich den Regler auf <75% gezogen habe, wurde die climater API aufgerufen und aufgrund des Rückgabewertes geht er davon aus, dass die Klima läuft. Obwohl ich die Heizung mit der FB aktiviert hatte, kam weiterhin "unavailable" zurück.