evcc-io/evcc #16041: Issue using Hardy Barth eCB1 Wallbox

Christoph-87, 2024-09-11:
### Describe the bug

I try to connect EVCC with two Hardy Barth eCB1 Wallboxes via the template.

```
chargers:
  - name: carport
    type: template
    template: hardybarth-ecb1
    host: 192.168.2.8 # IP-Adresse oder Hostname 
  - name: garage
    type: template
    template: hardybarth-ecb1
    host: 192.168.2.9 # IP-Adresse oder Hostname 
   
loadpoints:
  - title: Carport
    charger: carport
    mode: pv # Standardmodus zum Laden
    vehicle: kona
  - title: Garage
    charger: garage
    mode: pv # Standardmodus zum Laden
```

Unfortunately while loading EVCC, I receive the following error:
> [main  ] INFO 2024/09/11 17:15:22 evcc 0.130.8
> [main  ] INFO 2024/09/11 17:15:22 using config file: /etc/evcc.yaml
> [db    ] INFO 2024/09/11 17:15:22 using sqlite database: /root/.evcc/evcc.db
> [mqtt  ] INFO 2024/09/11 17:15:22 connecting smart at tcp://192.168.2.2:1883
> [mqtt  ] DEBUG 2024/09/11 17:15:22 tcp://192.168.2.2:1883 connected
> [main  ] INFO 2024/09/11 17:15:22 listening at :7070
> [main  ] FATAL 2024/09/11 17:15:35 cannot create charger 'carport': cannot create charger type 'template': cannot create charger type 'hardybarth-ecb1': Post "http://192.168.2.8/api/v1/chargecontrols/1/mode": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
> [main  ] FATAL 2024/09/11 17:15:35 will attempt restart in: 15m0s

Later I also received:
> [main  ] INFO 2024/09/11 19:34:51 evcc 0.130.8
> [main  ] INFO 2024/09/11 19:34:51 using config file: /etc/evcc.yaml
> [db    ] INFO 2024/09/11 19:34:51 using sqlite database: /root/.evcc/evcc.db
> [mqtt  ] INFO 2024/09/11 19:34:51 connecting smart at tcp://192.168.2.2:1883
> [mqtt  ] DEBUG 2024/09/11 19:34:51 tcp://192.168.2.2:1883 connected
> [main  ] INFO 2024/09/11 19:34:51 listening at :7070
> [main  ] FATAL 2024/09/11 19:35:03 cannot create charger 'garage': cannot create charger type 'template': cannot create charger type 'hardybarth-ecb1': invalid status: 404 Not Found
> [main  ] FATAL 2024/09/11 19:35:03 will attempt restart in: 15m0s

I can access the web frontend of the hard barth wallboxes via the IPs from my laptop.

I run EVCC with docker. The compose file is:
version: "3.8"

```
networks:
    net:
        driver: bridge
        name: docker_net

    evcc:
        depends_on: 
            - traefik
        command:
            - evcc
        container_name: evcc
        image: evcc/evcc:latest
        networks:
            - net
        #network_mode: "host"
        expose:
            - 7070
        #ports:
            # - 7070:7070
[... 259 lines cut ...]
[sma   ] TRACE 2024/09/11 19:45:01 send discover package
[sma   ] TRACE 2024/09/11 19:45:01 send discover package
[sma   ] TRACE 2024/09/11 19:45:02 send discover package
[sma   ] TRACE 2024/09/11 19:45:04 login for 192.168.2.81:9522
[sma   ] TRACE 2024/09/11 19:45:04 send 192.168.2.81: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/09/11 19:45:04 recv 192.168.2.81: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/09/11 19:45:04 requestValues for 192.168.2.81:9522: 0x5100 0x263F00 0x263FFF
[sma   ] TRACE 2024/09/11 19:45:04 send 192.168.2.81: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/09/11 19:45:05 recv 192.168.2.81: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/09/11 19:45:05 requestValues for 192.168.2.81:9522: 0x5100 0x295A00 0x295AFF
[sma   ] TRACE 2024/09/11 19:45:05 send 192.168.2.81: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/09/11 19:45:05 recv 192.168.2.81: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

/app # evcc -v evcc version 0.130.8

andig, 2024-09-11:
Sorry, no docker support...

mdkeil, 2024-09-11:
Der Grund des Schließend für mich nicht nachvollziehbar.. Der Ersteller scheint ein Problem mit einer unterstützen Wallbox zu haben.. das evcc unter Docker läuft ist hier nicht das Problem, obgleich die Logauschnitte sehr wirr und am Ende zumindest evcc lauffähig ist..
Wenn nicht bereits geschehen beide Wallboxen vor dem Starten von evcc von Hand in den "manual"-Mode zu stellen. Was für ein Fw-Update läuft denn? Vielleicht ist dies nicht mehr kompatibel zur evcc-Implementierung..

Sind denn die beiden Wallboxen aus dem Docker-Container heraus erreichbar?

andig, 2024-09-11:
> NOTE if you're using HomeAssistant or Docker we ask you to reproduce the problem on plain Linux or Windows first.

Christoph-87, 2024-09-13:
Vielen Dank für die Rückmeldung.

Beide Wallboxen stehen auf "Manuell".

[image]
[image]

Die Einstellungen von Wallbox 1:

> eCB1 |  
> -- | --
> Seriennummer | 75654059
> Firmware | V1.73
> Type | PV
> OS Version | 0.99
> OS Component | 78000001
> MAC-LAN | 00:D0:93:31:14:93
> LAN IP-Adresse | 192.168.2.8
> Netzwerkmaske | 255.255.255.0
> Gateway | 192.168.2.1
> 

Die Einstellungen von Wallbox 2:

> eCB1 |  
> -- | --
> Seriennummer | 75658705
> Firmware | V1.73
> Type | PV
> OS Version | 0.99
> OS Component | 78000001
> MAC-LAN | 00:D0:93:31:19:36
> LAN IP-Adresse | 192.168.2.9
> Netzwerkmaske | 255.255.255.0
> Gateway | 192.168.2.1
 

Aus dem Container heraus kann ich beide Wallboxen anpingen:

> /app # ping 192.168.2.8
> PING 192.168.2.8 (192.168.2.8): 56 data bytes
> 64 bytes from 192.168.2.8: seq=0 ttl=63 time=3.003 ms
> 64 bytes from 192.168.2.8: seq=1 ttl=63 time=1.554 ms
> ^C
> --- 192.168.2.8 ping statistics ---
> 2 packets transmitted, 2 packets received, 0% packet loss
> round-trip min/avg/max = 1.554/2.278/3.003 ms
> /app # ping 192.168.2.9
> PING 192.168.2.9 (192.168.2.9): 56 data bytes
> 64 bytes from 192.168.2.9: seq=0 ttl=63 time=3.481 ms
> 64 bytes from 192.168.2.9: seq=1 ttl=63 time=2.881 ms
> 64 bytes from 192.168.2.9: seq=2 ttl=63 time=6.572 ms
> ^C
> --- 192.168.2.9 ping statistics ---
> 3 packets transmitted, 3 packets received, 0% packet loss
> round-trip min/avg/max = 2.881/4.311/6.572 ms
> /app # 

Gibt es denn sonst Details die hilfreich sind?

Christoph-87, 2024-09-13:
[image]

Inzwischen bekomme ich ausschließlich den 404-Fehler. Gibt es denn eine Möglichkeit herauszufinden, welche URL aufgerufen wird?

mdkeil, 2024-09-13:
> Gibt es denn eine Möglichkeit herauszufinden, welche URL aufgerufen wird?

https://github.com/evcc-io/evcc/blob/ef6ad884c6b723ff7f16cf77c465dec065d3421b/charger/hardybarth-ecb1.go#L79C2-L79C48

also maßgeblich muss die Seite `http://<IP>/api/v1`  Daten lieferb - alle weiteren Abfragen z.B. /chargecontrols, /meters bauen darauf auf.. die kompletten Abfragen finden sich natürlich auch in der hardybarth-ecb1.go wieder, wenn man ein wenig code lesen und verstehen kann.

Ich schau mal, ob ich in der aktuellen Anleitung von deiner WB was finden kann.. ggfs. hat sich ja die API geändert..

Christoph-87, 2024-09-13:
Eventuell ist das das Problem?

[image]

mdkeil, 2024-09-13:
Folgende API wurde implementiert.. V1.3
http://apidoc.ecb1.de/

Mutmaßlich könnte deine Version 1.4 das Problem sein.. du kannst ja mal händisch einige implementierte Abfragen austesten.

Welcher Charge-Controller ist denn in der Konfiguration ausgewählt?

mdkeil, 2024-09-13:
Probiere z.B. mal 
http://<ip>/api/v1/meters/0

Christoph-87, 2024-09-13:
Rückgabe von 
`http://<IP>/api/v1/meters/0`:
> {"meter": {"id": 0, "function": "main", "data": {"lgwb": -1333.5, "1-0:1.4.0": 1333.5, "1-0:61.4.0": 0.0, "1-0:41.4.0": 1542.6, "1-0:22.8.0": 3498.9362, "1-0:2.8.0": 5469.1449, "1-0:1.8.0": 14091.1822, "1-0:32.4.0": 230.524, "1-0:73.4.0": 0.924, "1-0:61.8.0": 5813.6346999999988, "1-0:71.4.0": 0.72400000000000012, "1-0:62.4.0": 141.5, "1-0:41.8.0": 9676.2732, "1-0:72.4.0": 229.22099999999994, "1-0:21.4.0": 0.0, "1-0:22.4.0": 67.6, "1-0:52.4.0": 228.488, "1-0:62.8.0": 6609.9287000000012, "1-0:2.4.0": 0.0, "1-0:21.8.0": 6090.6473, "1-0:31.4.0": 1.1379999999999999, "1-0:51.4.0": 6.9620000000000012, "1-0:53.4.0": 0.988, "1-0:42.8.0": 2849.6529999999994, "1-0:42.4.0": 0.0, "1-0:13.4.0": 0.947, "1-0:33.4.0": 0.412}, "type": "Energymeter", "name": "SMA Energy Meter", "ipaddress": "239.12.255.254", "vendor": "SMA", "serial": 3009885828}, "protocol-version": "1.4"}

Rückgabe von `http://<IP>/api/v1/meters/1`
> {"meter": {"id": 1, "function": "socket", "data": {"lgwb": -4.3, "1-0:1.4.0": 4.3, "1-0:61.4.0": 0.0, "1-0:41.4.0": 0.0, "1-0:22.8.0": 0.0, "1-0:2.8.0": 0.0, "1-0:1.8.0": 0.0914, "1-0:32.4.0": 230.442, "1-0:73.4.0": 0.0, "1-0:61.8.0": 0.0, "1-0:71.4.0": 0.0, "1-0:62.4.0": 0.0, "1-0:41.8.0": 0.0, "1-0:72.4.0": 229.055, "1-0:21.4.0": 4.3, "1-0:22.4.0": 0.0, "1-0:52.4.0": 228.22, "1-0:62.8.0": 0.0, "1-0:2.4.0": 0.0, "1-0:21.8.0": 0.0914, "1-0:31.4.0": 0.017999999999999994, "1-0:51.4.0": 0.0, "1-0:53.4.0": 0.0, "1-0:42.8.0": 0.0, "1-0:42.4.0": 0.0, "1-0:13.4.0": 0.99900000000000012, "1-0:33.4.0": 0.99900000000000012}, "type": "eCB1 intern", "name": "Carport Wallbox", "ipaddress": "127.0.0.1", "vendor": "eCHARGE", "serial": 75654059}, "protocol-version": "1.4"}

Bei `http://<IP>/api/v1/chargecontrols/0`
> {"protocol-version": "1.4", "errors": {"message": "EV-ChargingController Not Found", "id": 0}}

Bei `http://<IP>/api/v1/chargecontrols/1`

> {"protocol-version": "1.4", "errors": {"message": "EV-ChargingController Not Found", "id": 1}}

mdkeil, 2024-09-13:
Das habe ich vermutet, dass kein chargecontroller in der WB-Konfiguration eingestellt ist.. Das bitte mal nachholen. Gerne einen Screenshot machen, was zu Auswahl steht :)

Christoph-87, 2024-09-13:
Da erhalte ich leider die gleiche Fehlermeldung:
[image]

mdkeil, 2024-09-13:
Wenn Du in der WB-Konfiguration noch nichts geändert hast, ist das nicht verwunderlich..
Was ist denn nun in der bei deiner WB in der Konfiguration (nicht in der evcc-config)  unter 
Charge Controller / EVCC eingestellt?

Christoph-87, 2024-09-13:
Aktuell ist das hier die Einstellung? Das muss wie im Screenshot umgestellt werden?
[image]

Christoph-87, 2024-09-13:
Meine Auswahl ist:
[image]

mdkeil, 2024-09-13:
Ich würde beginnend mit Phoenix-RTU testen und wenn es das gleiche Fehlerbild ist auf Phoenix-ETH wechseln.. sollte es dann immer noch nicht funktionieren, gehen mir die Ideen aus.

Christoph-87, 2024-09-13:
Die Fehlermeldung bei Phoenix (RTU) ist nun anders (bei Bus ID habe ich 0 und 1 versucht). 

Es erscheint nicht mehr der 404-Fehler. Stattdessen erhalte ich folgenden Fehler:
> [main  ] FATAL 2024/09/13 15:11:29 cannot create charger 'garage': cannot create charger type 'template': cannot create charger type 'hardybarth-ecb1': Post "http://192.168.2.9/api/v1/chargecontrols/1/mode": context deadline exceeded (Client.Timeout exceeded while awaiting headers)

Bei Phoenix (ETH) muss eine IP angegeben werden. Mir ist nicht klar, welche IP Adresse hier benötigt wird?

mdkeil, 2024-09-13:
ich denke Phoenix-RTU mit ID 1 wird vermutlich korrekt sein. Das Folgeproblem hängt ggfs. wirklich mit dem Repeater zusammen.

Christoph-87, 2024-09-13:
Danke - nach einer Neukonfiguration vom Repeater habe ich es nun geschafft EVCC zumindest mit einem Gerät zu starten - beide gleichzeitig scheint bei mir nicht zu funktionieren. Ich erhalte dann wieder den folgenden Fehler, obwohl das Gerät pingbar ist. Das Gerät wird genau dann nicht mehr ansprechbar, wenn ich EVCC starte. 

> [main ] FATAL 2024/09/13 15:11:29 cannot create charger 'garage': cannot create charger type 'template': cannot create charger type 'hardybarth-ecb1': Post "http://192.168.2.9/api/v1/chargecontrols/1/mode": context deadline exceeded (Client.Timeout exceeded while awaiting headers)

Das Gerät verhält sich nach dem Start von EVCC wie folgt:
```
PS /Users/christoph> ping 192.168.2.62
PING 192.168.2.62 (192.168.2.62): 56 data bytes
64 bytes from 192.168.2.62: icmp_seq=0 ttl=64 time=24.693 ms
64 bytes from 192.168.2.62: icmp_seq=1 ttl=64 time=16.933 ms
^C
--- 192.168.2.62 ping statistics ---
2 packets transmitted, 2 packets received, 0.0% packet loss
round-trip min/avg/max/stddev = 16.933/20.813/24.693/3.880 ms
PS /Users/christoph> nc -zv 192.168.2.62 80                                                                                                                
Connection to 192.168.2.62 port 80 [tcp/http] succeeded!
PS /Users/christoph> curl 192.168.2.62     
```

Bei curl kommt kein Ergebnis. Auch mit dem Browser kann ich die URL nicht mehr aufrufen. Die Wallbox ist erst wieder erreichbar, wenn ich die Sicherung trenne und wieder verbinde. Dann ist die Wallbox so lange erreichbar, bis ich EVCC starte.

mdkeil, 2024-09-13:
hmm.. beide Wallboxen gleich konfiguriert..? also mit Phoenix-RTU und ID 1?

Christoph-87, 2024-09-13:
ja, beide mit Phoenix-RTU und ID 1. Ich habe eine der Wallboxen aufgeschraubt und bei Bus leuchtet es auch grün.

[image]

mdkeil, 2024-09-13:
> Kann das weitere Problem daran liegen, dass beide Wallboxen die Bus ID 1 haben?

nö, sind doch zwei eigenständige Geräte.. oder sind beide Geräte noch untereinander irgendwie verdrahtet?

Christoph-87, 2024-09-17:
Ja, sind zwei eigenständige Geräte und nicht verdrahtet. Ich habe nun gemerkt, dass die Geräte auch ohne Zutun von EVCC nach einer Weile nicht mehr über das Web-Frontend reagieren. Deswegen habe ich den Support von Hardy Barth nun angeschrieben. Falls sich daraus etwas ergibt, melde ich mich hier.

Christoph-87, 2024-09-19:
Der Support von Hardy Barth konnte mir helfen - es lag an der Wallbox oder dem Netzwerk. Ich bin nicht sicher was letztendlich der Grund war, aber nun kann ich EVCC wieder ohne Probleme starten.