evcc-io/evcc #16044: Cheap Grid Charging Price Limit <= 0.0 ct/kWh not shown on LP UX

RenatusRo, 2024-09-11:
### Describe the bug

The limit coins are not shown if limit is set to   <= 0.0 ct/kWh 

### Steps to reproduce

1. set limit to <= 0.0 Ct/kWh
2. enjoy missing visualisation
3. set limit to <= -0.5 Ct/kWh
4. limit is shown correctly

### Configuration details

```yaml
does not matter
```

### Log details

```text
does not matter
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.8

andig, 2024-09-11:
@naltatis evtl. Ist hier `nil` noch nicht umgesetzt und stattdessen ein anderer Vergleichswert aktiv?

naltatis, 2024-09-12:
Ja, das wird mit dem `nil` Umbau zu tun haben. Ich schau da rein.

closed 2024-09-13 by andig