evcc-io/evcc #16070: new sponsortoken dont work

Lucascoco, 2024-09-13:
### Describe the bug

after enter the new token, i geht these error
FATAL 2024/09/13 09:01:44 sponsortoken: rpc error: code = Unknown desc = token has invalid claims: token is expired

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# sponsor token enables optional features (request at https://sponsor.evcc.io)
# trial token, valid until 2024-09-18
sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJldmNjLmlvIiwic3ViIjoidHJpYWwiLCJleHAiOjE3MjY2NTM2MDAsImlhdCI6MTcyNTg3NjAwMCwic3BlIjp0cnVlLCJzcmMiOiJtYSJ9.cR5Gfj9vnHxnGQoho5GIDJlo5tL6exD5yI5vSm2YbVE
```

### Log details

```text
FATAL 2024/09/13 09:01:44 sponsortoken: rpc error: code = Unknown desc = token has invalid claims: token is expired
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.9

naltatis, 2024-09-13:
Did you set the token via `evcc.yaml`? Have you previously used the token form in experimental config ui?

Lucascoco, 2024-09-13:
Hello, i use the evcc.yaml only.

naltatis, 2024-09-13:
This is strange. The token you've posted is still valid and works for me.

Can you remove/rename the `evcc.db` so that evcc creates a fresh one and see if the issue still exists?

Lucascoco, 2024-09-13:
with a new database it will work thx

naltatis, 2024-09-13:
Ok, looks like an old demo token got stuck there.

closed 2024-09-13 by naltatis

mschrotti, 2025-06-17:
Or should I do "evcc migrate"? It is described in the configuration site at the UI

mschrotti, 2025-06-17:
I now found the Sponsorship UI but it does not helps, because I do not have a remove Hyperlink

[image]

naltatis, 2025-06-17:
Can you send me your evcc.yaml and broken db to michael@evcc.io so I can have a look?

mschrotti, 2025-06-17:
You should have it

But it seems OK, because if I rename the evcc.db. and start again it creates a new evcc.db (with empty charging processes :-( ) and it works.

mschrotti, 2025-06-20:
Hi Lucascoco,

one solution was to stop evcc, rename the evcc.db and start evcc again (with the correct token in the evcc.yaml). Evcc will start and generate a new database with the correct token. Because it is a fresh database, all consumption data are lost.

The way we did it. I have some developer in my company and they replace the token direct in the database. Do not ask me how they exactly did it. It worked.