evcc-io/evcc #16077: Easee - Status 403 - Forbidden

puzich, 2024-09-13:
### Describe the bug

Habe zwei Easee Wallboxen. Aktuell wird an einer geladen (LP1). Diese ist auf PV gestellt, lädt aber mit voller Ladeleistung. Ich erhalte den Fehler 403 - Forbidden von Eassee. Habt ihr einen Grund? Ich kann das von heute auf Morgen nicht nachvollziehen.

### Steps to reproduce

Running evcc

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 80

#log: info
#log: debug
log: trace
#levels:
#  bmw: debug
#  cache: error
##  modbus: trace
##log.easee: debug

# unique installation id
plant: *****
sponsortoken: *****
#interval: 30s # control cycle interval
#interval: 10s # control cycle interval
interval: 15s # control cycle interval

site:
  title: Home
  meters:
    grid: my_grid
    pv: my_pv
    battery: my_battery
  ##prioritySoc: 50
#  residualPower: 350
  residualPower: 100
  ##bufferSoc: 80
  ##batteryDischargeControl: true

meters:
  - name: my_grid
    type: template
    template: sofarsolar-g3
    usage: grid
    # modbus: rs485serial
    modbus: tcpip
    id: 1
    # device: /dev/ttyUSB0
    # baudrate: 9600 # Pruefe Geraeteeinstellungen, typische Werte sind 9600, 19200, 38400, 57600, 115200
    # comset: "8N1" # Kommunikationsparameter fuer den Adapter
    # delay: 1s # Optional
    host: 192.168.166.12
    port: 8899
  - name: my_pv
    type: template
    template: sofarsolar-g3
    usage: pv
    # modbus: rs485serial
    modbus: tcpip
    id: 1
    # device: /dev/ttyUSB0
    # baudrate: 9600 # Pruefe Geraeteeinstellungen, typische Werte sind 9600, 19200, 38400, 57600, 115200
    # comset: "8N1" # Kommunikationsparameter fuer den Adapter
    # delay: 1s # Optional
[... 630 lines cut ...]
Sep 13 11:29:06 evcc evcc[94709]: [cache ] TRACE 2024/09/13 11:29:06 lp-2/enabled: false
Sep 13 11:29:06 evcc evcc[94709]: [cache ] TRACE 2024/09/13 11:29:06 lp-2/mode: pv
Sep 13 11:29:06 evcc evcc[94709]: [cache ] TRACE 2024/09/13 11:29:06 lp-2/planProjectedStart: 0001-01-01 00:00:00 +0000 UTC
Sep 13 11:29:06 evcc evcc[94709]: [cache ] TRACE 2024/09/13 11:29:06 lp-2/planProjectedEnd: 0001-01-01 00:00:00 +0000 UTC
Sep 13 11:29:06 evcc evcc[94709]: [cache ] TRACE 2024/09/13 11:29:06 lp-2/planOverrun: 0s
Sep 13 11:29:06 evcc evcc[94709]: [cache ] TRACE 2024/09/13 11:29:06 greenShareHome: 1
Sep 13 11:29:06 evcc evcc[94709]: [cache ] TRACE 2024/09/13 11:29:06 greenShareLoadpoints: 0.07559806925208844
Sep 13 11:29:06 evcc evcc[94709]: [cache ] TRACE 2024/09/13 11:29:06 tariffPriceHome: 0.082
Sep 13 11:29:06 evcc evcc[94709]: [cache ] TRACE 2024/09/13 11:29:06 tariffPriceLoadpoints: 0.2641071803573386
Sep 13 11:29:06 evcc evcc[94709]: [site  ] TRACE 2024/09/13 11:29:06 {"instanceId":"ef3fa0cde6b32068f79ae3ce6a3f0f16d8519711901271e5472a287fb7a17c29","chargePower":10449.999809265137,"greenPower":789.9998092651367,"chargeEnergy":7.915928840637207,"greenEnergy":0.46965305846226557}
Sep 13 11:29:06 evcc evcc[94709]: --
```
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.9

Al3xand3r1987, 2024-09-13:
[easee ] WARN 2024/09/13 11:35:12 smart charging: invalid status: 403

I have the same problem

puzich, 2024-09-13:
I installed version 0.130.3 and now the easee wallbox stopped charging on PV mode (because house batterie is empty). With the older version, it works as it should

Al3xand3r1987, 2024-09-13:
It seems there are currently issues with the Easee portal.

[image]

Al3xand3r1987, 2024-09-13:
Carport: charger logic error: disabled but charging
vor 1 Sekunde
Carport: charger enable: invalid status: 500
vor 13 Sekunden
Carport: charger logic error: disabled but charging
vor 17 Sekunden
Carport: charger enable: invalid status: 500
vor 32 Sekunden
Carport: charger logic error: disabled but charging
vor 36 Sekunden
Carport: charger enable: Post \"https://api.easee.com/api/chargers/EHVG7Y32/commands/resume_charging\": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
vor 56 Sekunden
Carport: charger logic error: disabled but charging
vor 1 Minute
Carport: charger enable: timeout
vor 1 Minute
Carport: charger logic error: disabled but charging
vor 2 Minuten
Carport: charger enable: Post \"https://api.easee.com/api/chargers/EHVG7Y32/settings\": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
vor 2 Minuten
Carport: charger logic error: disabled but charging
vor 2 Minuten
smart charging: Post \"https://api.easee.com/api/chargers/EHVG7Y32/settings\": context deadline exceeded (Client.Timeout exceeded while awaiting headers)