evcc-io/evcc #16080: 0.130.8 und .9 Probleme mit ModBus - evcc startet nicht

Ivanhoe4578, 2024-09-13:
### Describe the bug

Folgender Fehler tritt nach dem Update auf .8 und .9 auf:

### [main ] FATAL 2024/09/12 18:44:22 cannot create meter 'my_battery': cannot create meter type 'my_pv': cannot create meter type 'my_grid': power: connection already registered with different protocol: 192.168.180.107:502

In der Version .7 läuft alles. An der Config wurde nicht geändert. 

### Steps to reproduce

1. Start .8
2. Start .9
3.
4.
...

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local
  port: 7070

interval: 20s 
sponsortoken: ***

log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

modbusproxy:
meters:
  - name: my_grid
    type: template
    template: solaredge-hybrid
    usage: grid
    modbus: rs485tcpip
    id: 1
    host: 192.168.180.107 # Hostname
    port: 502 # Port
  - name: my_pv
    type: template
    template: solaredge-hybrid
    usage: pv
    modbus: rs485tcpip
    id: 1
    host: 192.168.180.107 # Hostname
    port: 502 # Port
  - name: my_battery
    type: template
    template: solaredge-hybrid
    usage: battery
    modbus: rs485tcpip
    id: 1
    host: 192.168.180.107 # Hostname
    port: 502 # Port

  - name: my_pv2
    type: template
    template: eastron-sdm120
    usage: pv
    modbus: tcpip
    id: 1
    host: 192.168.180.116 # Hostname
    port: 8890 # Port 

  - name: my_pv3
    type: template
    template: eastron-sdm120
    usage: pv
    modbus: tcpip
    id: 2
    host: 192.168.180.116 # Hostname
    port: 8890 # Port 

chargers:
  - name: goeGarage
    type: template
    template: go-e-v3
    host: 192.168.180.166

vehicles:
  - name: TeslaM3
    type: custom
    title: TeslaM3 # display name for UI
    capacity: 60 # byttery capacity (kWh)
    soc: # battery soc (%)
      source: mqtt
      topic: teslamate/tesla/cars/1/battery_level
site:
  title: Home # display name for UI
  meters:
    grid: my_grid # grid meter
    pv:
      - my_pv
      - my_pv2
      - my_pv3 # list of pv inverters/ meters
    battery:
      - my_battery # list of battery meters
  residualPower: 0 # additional household usage margin
  maxGridSupplyWhileBatteryCharging: 0 # ignore battery charging if AC consumption is above this value
loadpoints:
  - title: Garage # display name for UI
    charger: goeGarage # charger
    mode: "off" # default charge mode to apply when vehicle is disconnected; use "off" to disable by default if charger is publicly available
    vehicle: TeslaM3
    priority: 0 # relative priority for concurrent charging in PV mode with multiple loadpoints (higher values have higher priority)
    soc:
      poll:
        mode: always
        interval: 60m
      estimate: true # set false to disable interpolating between api updates (not recommended)
    enable: # pv mode enable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 10m # threshold must be exceeded for this long
      threshold: 0 # maximum import power (W)
tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    type: tibber
    token: ***** 
    homeid: *****

mqtt:
  broker: localhost:1883
  topic: evcc # root topic for publishing, set empty to disable
  user: mqtt-user
  password: ******
influx:
eebus:
messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
    soc: # vehicle soc update event
      title: Soc updated
      msg: Battery charged to ${vehicleSoc:%.0f}%
    guest: # vehicle could not be identified
      title: Unknown vehicle
      msg: Unknown vehicle, guest connected?
services:
```

### Log details

```text
[main ] FATAL 2024/09/12 18:44:22 cannot create meter 'my_battery': cannot create meter type 'my_pv': cannot create meter type 'my_grid': power: connection already registered with different protocol: 192.168.180.107:502
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

mdkeil, 2024-09-13:
#16053

andig, 2024-09-13:
@Ivanhoe4578 see linked issue. Still need remote access.

closed 2024-09-13 by andig

andig, 2024-09-13:
The confusing thing here is that all users are ModbusRTU over TCP. I can't really find where the conflict is coming from.