evcc-io/evcc #16081: HOME ASSISTANT ADD ON: veränderbarer Titel = statt evcc z.B. LADER / WALLBOX

typxxi, 2024-09-13:
Hier im UI sollte ein wählbarer TiTEL erscheinen statt fix evcc, denn Älteren kann ich Lader erklären und Wallbox ist auch in deren Sprachgebrauch angekommen, aber evcc nicht.

Nur eine Kleinigkeit, die aber zu Irritationen führte und das schon 2x in 4 Wochen, wenn man am Telefon 85+ Jahre alte Senioren durch das Mysterium Überschuss Laden leiten darf. 

[image]

Größeres / klareres Symbol wäre auch schön wie eine Zapfsäule ohne Stromblitz. Hier sind ältere fix mit der Sehkraft überfordert.

SolarPower2024, 2024-09-13:
Das kannst du selber relativ einfach "beheben".
Erstelle auf dem Dashboard einen großen Button, da kannst du den Titel und das Bild selber wählen. Und als "Tipp-Aktion" wählst du Navigation mit dem Pfad des evcc Addons.

Dann muss man nicht immer links die Leiste öffnen, sondern kann evcc direkt am Dashboard öffnen.

closed 2024-09-13 by naltatis

typxxi, 2024-09-13:
> Das kannst du selber relativ einfach "beheben". Erstelle auf dem Dashboard einen großen Button, da kannst du den Titel und das Bild selber wählen. Und als "Tipp-Aktion" wählst du Navigation mit dem [...]

Das klingt gut, nur steht ich etwas beim Pfad des evcc Addons auf dem Schlauch.

Ich habe das gelbe Icon hier schon in Nutzung, wo ich den Zustand anzeige, ob 
1. einsatzbereit 
2. Fahrzeug verbunden 
3. ladend.
Das wäre dann schon mal der passende Ort und das ICON oder aber ein EXTRA ICON "LADE EINSTELLUNGEN" 
[image]

**Wenn ich dort beim Icon eine Klick Aktion anwähle, dann muss die hinterlegte Aktion für evcc wie aussehen ?**

ANTWORT GEFUNDEN für alle, die so wie ich dann erst einmal im HA file system auf die Suche gehen wollen.

NAVIGIEREN ist eine der TIPP / HALTE Optionen und dann werden sowohl die / alle Elemente / Dashboards der horizontalen Leiste wie auch der Schnellstartleiste am linken Rand angezeigt.
Im Navigationspfad dann "ev" eingeben und man spart sich die Scrollerei, weil zuerst die Dashboardelemente kommen 

[image]

typxxi, 2024-09-13:
> Ist hier auch das falsche Repo dafür. Das Addon lebt hier: https://github.com/evcc-io/hassio-addon

DANKE,
ich lande immer im falschen, wenn ich bei github evcc eingebe, denn da wird mir immer das hier zuerst vorgeschlagen.