evcc-io/evcc #16085: easee invalid Status 403, no matching rate, max charge current, invalid status

phL9r, 2024-09-13:
### Describe the bug

Seit heute Nachmittag bekomme ich plötzlich folgende Fehlermeldungen, ohne etwas anderes konfiguriert zu haben. Heute morgen lief die Ladung noch problemlos:

[image]

### Steps to reproduce

1. Fahrzeug verbinden
2. Überschuss vorhanden
3. Fehlermeldung der easee wallbox
...

### Configuration details

```yaml
Kommt sofort
```

### Log details

```text
[main ] INFO 2024/09/13 14:26:47 evcc 0.130.9
[main ] INFO 2024/09/13 14:26:47 using config file: /etc/evcc.yaml
[db ] INFO 2024/09/13 14:26:47 using sqlite database: /var/lib/evcc/evcc.db
[main ] INFO 2024/09/13 14:26:49 listening at :7070
[lp-1 ] DEBUG 2024/09/13 14:26:52 charge total import: 991.914kWh
[site ] WARN 2024/09/13 14:26:52 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site ] INFO 2024/09/13 14:26:52 site config:
[site ] INFO 2024/09/13 14:26:52 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2024/09/13 14:26:52 grid: power ✓ energy ✓ currents ✓
[site ] INFO 2024/09/13 14:26:52 pv 1: power ✓ energy ✓ currents ✗
[site ] INFO 2024/09/13 14:26:52 pv 2: power ✓ energy ✓ currents ✗
[site ] INFO 2024/09/13 14:26:52 pv 3: power ✓ energy ✓ currents ✗
[site ] INFO 2024/09/13 14:26:52 battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site ] INFO 2024/09/13 14:26:52 vehicles:
[site ] INFO 2024/09/13 14:26:52 vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1 ] INFO 2024/09/13 14:26:52 loadpoint 1:
[lp-1 ] INFO 2024/09/13 14:26:52 mode: pv
[lp-1 ] INFO 2024/09/13 14:26:52 charger: power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1 ] INFO 2024/09/13 14:26:52 meters: charge ✓
[lp-1 ] INFO 2024/09/13 14:26:52 charge: power ✓ energy ✓ currents ✓
[site ] DEBUG 2024/09/13 14:26:52 set buffer soc: 95
[site ] DEBUG 2024/09/13 14:26:52 set buffer start soc: 0
[site ] DEBUG 2024/09/13 14:26:52 set priority soc: 50
[site ] DEBUG 2024/09/13 14:26:52 set battery discharge control: false
[site ] DEBUG 2024/09/13 14:26:52 set residual power: 100
[lp-1 ] DEBUG 2024/09/13 14:26:52 phase timer inactive
[lp-1 ] DEBUG 2024/09/13 14:26:52 pv timer inactive
[lp-1 ] INFO 2024/09/13 14:26:52 vehicle updated: unknown -> Niro
[lp-1 ] DEBUG 2024/09/13 14:26:52 set charge mode: pv
[site ] WARN 2024/09/13 14:26:52 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site ] DEBUG 2024/09/13 14:26:52 ----
[easee ] WARN 2024/09/13 14:26:52 smart charging: invalid status: 403
[lp-1 ] DEBUG 2024/09/13 14:26:52 charge power: 0W
[lp-1 ] DEBUG 2024/09/13 14:26:52 charge currents: [0.747 0.498 0.497]A
[site ] WARN 2024/09/13 14:26:52 planner: no matching rate
[site ] DEBUG 2024/09/13 14:26:54 pv power: 6465W
[site ] DEBUG 2024/09/13 14:26:54 battery soc: 100%
[site ] DEBUG 2024/09/13 14:26:54 battery power: -21W
[site ] DEBUG 2024/09/13 14:26:54 grid meter: -6205W
[site ] DEBUG 2024/09/13 14:26:54 grid currents: [-7.22 -9.68 -9.19]A
[site ] DEBUG 2024/09/13 14:26:54 site power: -6126W
[lp-1 ] DEBUG 2024/09/13 14:26:54 vehicle odometer: 1120km
[lp-1 ] DEBUG 2024/09/13 14:26:54 charge total import: 991.914kWh
[easee ] WARN 2024/09/13 14:26:54 smart charging: invalid status: 403
[lp-1 ] DEBUG 2024/09/13 14:26:54 charger status: B
[lp-1 ] INFO 2024/09/13 14:26:54 car connected
[lp-1 ] DEBUG 2024/09/13 14:26:54 pv timer elapse
[lp-1 ] DEBUG 2024/09/13 14:26:54 pv timer inactive
[... 255 lines cut ...]
[site ] DEBUG 2024/09/13 14:30:08 battery power: -20W
[site ] DEBUG 2024/09/13 14:30:08 grid meter: -6080W
[site ] DEBUG 2024/09/13 14:30:08 grid currents: [-7.13 -9.51 -9.03]A
[site ] DEBUG 2024/09/13 14:30:08 site power: -6000W
[lp-1 ] DEBUG 2024/09/13 14:30:08 charge total import: 991.914kWh
[easee ] WARN 2024/09/13 14:30:08 smart charging: invalid status: 403
[lp-1 ] DEBUG 2024/09/13 14:30:08 charger status: B
[lp-1 ] DEBUG 2024/09/13 14:30:08 pv charge current: 26.1A = 0A + 26.1A (-6000W @ 1p)
[lp-1 ] DEBUG 2024/09/13 14:30:08 site power -6000W <= 0W enable threshold
[lp-1 ] DEBUG 2024/09/13 14:30:08 pv enable in 0s
[lp-1 ] DEBUG 2024/09/13 14:30:08 pv enable timer elapsed
[lp-1 ] ERROR 2024/09/13 14:30:08 max charge current 6A: invalid status: 403
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.9 and 0.130.8 beide Fehlerhaft

andig, 2024-09-13:
Die Fehlermeldung kommt von Easee

closed 2024-09-13 by andig

pmknowles, 2024-09-13:
Easee API appears to be down
https://status.easee.com/

ansysic, 2024-09-14:
Still down, FYI

Schockis, 2024-09-14:
guess we have to wait

magantus, 2024-09-14:
bei mir auch....

es nervt...entweder ist irgendwas mit dem Cupra (VW) Server, dann Datenschutzeinstellungen und jetzt auch noch Easee

rea1ity01, 2024-09-14:
Ja, oder evcc mit dem Lizenz Token. Komisch ist, wenn beides gleichzeitig passiert.

GerritPost, 2024-09-15:
Ich leide auch darunter. Gestern wurden 15 % Solarenergie verbraucht :( bei 40 kW

Iruppi79, 2024-09-15:
ok ich habe das gleich Problem. Die Wallbox lässt sich also über die App überhaupt nicht mehr steuern. Lädt munter vor sich hin. also kann man nur über Easee beenden oder?

phL9r, 2024-09-15:
richtig.

Maexster, 2024-09-15:
.. auch die easee app spinnt, hängt immer wieder.
Zum Teil lädt sie mit 11kw, dann wieder nur 4…

Iruppi79, 2024-09-15:
OK. Dann muss ich das Auto immer abstecken...das blöde bluelink funktioniert nicht immer in der Garage

unstressable, 2024-09-15:
hab das gleiche Problem.
Kann man die Easee nicht lokal einbinden um solche Sachen von vornherein zu verhindern?

Ist doch beschissen, wenn man abhängig von irgendeiner Cloud ist.

pmknowles, 2024-09-15:
> have the same problem. Can't you integrate the Easee locally to prevent such things from the outset?
> 
> It sucks when you're dependent on any cloud.

Easee don't even let OCPP work without using the Easee API.

unstressable, 2024-09-15:
> > have the same problem. Can't you integrate the Easee locally to prevent such things from the outset?
> > It sucks when you're dependent on any cloud.
> 
> Easee don't even let OCPP work without using the Easee API.

So theres no possibility?
Nor Modbus TCP?
Bluetooth?

pmknowles, 2024-09-16:
It's unlikely that Easee Wallboxes are running Modbus. Bluetooth would require  having the evcc server being within Bluetooth range (which sort of defeats the object) plus a lot of work to reverse engineer the app. In theory, it should be possible to spoof the Easee server for local control but, ideally, Easee should allow local OCPP control like other manufacturers.

phL9r, 2024-09-16:
Bei mir läuft es jetzt wieder!

[image]