evcc-io/evcc #16095: sponsortoken "error blocked due to public exposure"

ronajon, 2024-09-13:
### Describe the bug

after restarting evcc, i got an error about the sponsortoken "token blocked due to public exposure, please generate new token or contact info@evcc.io for support"

i'm using the trial token from the evcc site

### Steps to reproduce

1. 
2.
3.
...

### Configuration details

```yaml
sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJldmNjLmlvIiwic3ViIjoidHJpYWwiLCJleHAiOjE3MjY2NTM2MDAsImlhdCI6MTcyNTg3NjAwMCwic3BlIjp0cnVlLCJzcmMiOiJtYSJ9.cR5Gfj9vnHxnGQoho5GIDJlo5tL6exD5yI5vSm2YbVE
```

### Log details

```text
evcc:/opt/evcc# ./evcc -c ./evcc.yaml -l trace
[main  ] INFO 2024/09/13 22:58:01 evcc 0.130.9
[main  ] INFO 2024/09/13 22:58:01 using config file: ./evcc.yaml
[db    ] INFO 2024/09/13 22:58:01 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2024/09/13 22:58:01 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/09/13 22:58:01 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/09/13 22:58:01 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/09/13 22:58:01 SELECT * FROM `settings` 6 <nil>
[main  ] INFO 2024/09/13 22:58:01 listening at :7070
[main  ] FATAL 2024/09/13 22:58:01 sponsortoken: rpc error: code = Unknown desc = token blocked due to public exposure, please generate new token or contact info@evcc.io for support
[main  ] FATAL 2024/09/13 22:58:01 will attempt restart in: 15m0s
[cache ] TRACE 2024/09/13 22:58:01 fatal: <nil>
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.9

ZeroQool007, 2024-09-14:
Same problem

andig, 2024-09-14:
Oh sorry, thats kinda stupid. Will fix asap.

rea1ity01, 2024-09-14:
Same since yesterdays update

doctordsingh, 2024-09-14:
i'm having the same problem. Any idea for a solution?

andig, 2024-09-14:
Fixed

closed 2024-09-14 by andig

rea1ity01, 2024-09-14:
There is no Update available since 0.130.9

ronajon, 2024-09-14:
> There is no Update available since 0.130.9

just reboot your instance

rea1ity01, 2024-09-14:
Works. Fantastic.

ZeroQool007, 2024-09-14:
> > There is no Update available since 0.130.9
> 
> just reboot your instance

same issue, reboot not effected