evcc-io/evcc #16096: Cant read SOC on VW

lordvalium, 2024-09-14:
### Describe the bug

SOC of Passat GTE is not shown in evcc

### Steps to reproduce

1. added config, evcc is working fine with hyundai
2. tried both templates for vw. With "id" and "vw"
3. see error
...

### Configuration details

```yaml
- type: template
  template: vw
  name: ev4
  title: GTE
  user: xxx # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
  password: xxx # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen
  vin: xxxxx # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)
  capacity: 13 # Akkukapazität in kWh (optional)
  timeout: 60s # optional

and also tried with template "id"
```

### Log details

```text
ERROR 2024/09/14 09:15:27 vehicle soc: unknown engine type: electric, got [, ]
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

 0.130.9

TobiasHuber1980, 2024-09-14:
Siehe https://github.com/evcc-io/evcc/issues/16097

andig, 2024-09-14:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2024-09-14 by andig

lordvalium, 2024-09-14:
Hi, 
i have changed the password and also checked the AGB. 

It connects to VW but i still dont get the SOC

[lp-1  ] DEBUG 2024/09/14 14:25:17 vehicle odometer: 54212km
[lp-1  ] DEBUG 2024/09/14 14:25:17 charger status: C
[lp-1  ] ERROR 2024/09/14 14:25:17 vehicle soc: unknown engine type: electric, got [, ]

StefanSchoof, 2024-09-14:
I think you should run a

```
evcc -l trace vehicle 
```

 to produce a log file with enough information.

RaveGun, 2024-12-12:
I have the same issue.
I formatted the output of the trace.

{
  "charging": {
    "batteryStatus": {
      "error": {
        "message": "Bad Gateway",
        "errorTimeStamp": "2024-12-12T18:27:16Z",
        "info": "Upstream service responded with an unexpected status. If the problem persists, please contact our support.",
        "code": 4007,
        "group": 2,
        "retry": true
      }
    },
    "chargingStatus": {
      "error": {
        "message": "Bad Gateway",
        "errorTimeStamp": "2024-12-12T18:27:16Z",
        "info": "Upstream service responded with an unexpected status. If the problem persists, please contact our support.",
        "code": 4007,
        "group": 2,
        "retry": true
      }
    },
    "chargingSettings": {
      "error": {
        "message": "Bad Gateway",
        "errorTimeStamp": "2024-12-12T18:27:14Z",
        "info": "Upstream service responded with an unexpected status. If the problem persists, please contact our support.",
        "code": 4007,
        "group": 2,
        "retry": true
      }
    },
    "plugStatus": {
      "error": {
        "message": "Bad Gateway",
        "errorTimeStamp": "2024-12-12T18:27:16Z",
        "info": "Upstream service responded with an unexpected status. If the problem persists, please contact our support.",
        "code": 4007,
        "group": 2,
        "retry": true
      }
    },
    "chargeMode": {
      "error": {
        "message": "Bad Gateway",
        "errorTimeStamp": "2024-12-12T18:27:14Z",
        "info": "Upstream service responded with an unexpected status. If the problem persists, please contact our support.",
        "code": 4007,
        "group": 2,
        "retry": true
      }
    }
  },
  "climatisation": {
    "climatisationSettings": {
      "error": {
        "message": "Bad Gateway",
        "errorTimeStamp": "2024-12-12T18:27:14Z",
        "info": "Upstream service responded with an unexpected status. If the problem persists, please contact our support.",
        "code": 4007,
        "group": 2,
        "retry": true
      }
    },
    "climatisationStatus": {
      "error": {
        "message": "Bad Gateway",
        "errorTimeStamp": "2024-12-12T18:27:14Z",
        "info": "Upstream service responded with an unexpected status. If the problem persists, please contact our support.",
        "code": 4007,
        "group": 2,
        "retry": true
      }
    },
    "climatisationTemperatureOutside": {
      "error": {
        "message": "Bad Gateway",
        "errorTimeStamp": "2024-12-12T18:27:14Z",
        "info": "Something went wrong. Please try to re-login. If the problem persists, please contact our support.",
        "code": 4004,
        "group": 3,
        "retry": true
      }
    },
    "windowHeatingStatus": {
      "error": {
        "message": "Bad Gateway",
        "errorTimeStamp": "2024-12-12T18:27:14Z",
        "info": "Upstream service responded with an unexpected status. If the problem persists, please contact our support.",
        "code": 4007,
        "group": 2,
        "retry": true
      }
    }
  },
  "fuelStatus": {
    "rangeStatus": {
      "error": {
        "message": "Bad Gateway",
        "errorTimeStamp": "2024-12-12T18:27:14Z",
        "info": "Upstream service responded with an unexpected status. If the problem persists, please contact our support.",
        "code": 4007,
        "group": 2,
        "retry": true
      }
    }
  },
  "measurements": {
    "rangeStatus": {
      "value": {
        "carCapturedTimestamp": "2024-12-12T17:49:12Z",
        "gasolineRange": 510,
        "totalRange_km": 510
      }
    },
    "odometerStatus": {
      "value": {
        "carCapturedTimestamp": "2024-12-12T17:49:12Z",
        "odometer": 51914
      }
    },
    "fuelLevelStatus": {
      "value": {
        "carCapturedTimestamp": "2024-12-12T17:49:12Z",
        "currentFuelLevel_pct": 81,
        "primaryEngineType": "gasoline",
        "secondaryEngineType": "electric",
        "carType": "hybrid"
      }
    }
  }
}

TobiasHuber1980, 2024-12-12:
Heute gab es wieder neue Einwilligungen zu akzeptieren bei VW....

lordvalium, 2024-12-12:
Bringt aber Nix. Geht trotzdem nicht Von meinem iPhone gesendet

RaveGun, 2024-12-12:
Is it the WeConnect Plus needed?

I bought the car a month ago and just got my car back today from the VW as the multimedia device had an old SW and I was not able to become the primary user. Now it has SW version 330 and I was able to setup my account as the primary user. Even if it is now configured, the VW app does not show any charging status or battery status.

Maybe I need to enable some of the services. I will go to check this.

RaveGun, 2024-12-12:
"Laden" is not active without the We Connect Plus.

RaveGun, 2024-12-12:
Bought the "Plus" and now it works:

[id    ] TRACE 2024/12/12 19:59:21 {"charging":{"batteryStatus":{"value":{"carCapturedTimestamp":"2024-12-12T19:58:19Z","currentSOC_pct":97,"cruisingRangeElectric_km":32}},"chargingStatus":{"value":{"carCapturedTimestamp":"2024-12-12T19:58:19Z","chargingState":"notReadyForCharging","chargeMode":"invalid","chargeType":"unsupported"}},"chargingSettings":{"value":{"carCapturedTimestamp":"2024-12-12T19:58:19Z","maxChargeCurrentAC":"maximum"}},"plugStatus":{"value":{"carCapturedTimestamp":"2024-12-12T19:58:19Z","plugConnectionState":"disconnected","plugLockState":"unlocked","externalPower":"unavailable","ledColor":"none"}},"chargeMode":{"error":{"message":"Bad Gateway","errorTimeStamp":"2024-12-12T19:59:21Z","info":"Upstream service responded with an unexpected status. If the problem persists, please contact our support.","code":4111,"group":2,"retry":true}}},"climatisation":{"climatisationSettings":{"value":{"carCapturedTimestamp":"2024-12-12T19:49:45Z","targetTemperature_C":22,"targetTemperature_F":72,"climatisationWithoutExternalPower":false,"heaterSource":"automatic"}},"climatisationStatus":{"value":{"carCapturedTimestamp":"2024-12-12T19:57:49Z","climatisationState":"off"}},"climatisationTemperatureOutside":{"error":{"message":"Bad Gateway","errorTimeStamp":"2024-12-12T19:59:20Z","info":"Something went wrong. Please try to re-login. If the problem persists, please contact our support.","code":4004,"group":3,"retry":true}},"windowHeatingStatus":{"value":{"carCapturedTimestamp":"2024-12-12T19:57:50Z","windowHeatingStatus":[{"windowLocation":"front","windowHeatingState":"off"},{"windowLocation":"rear","windowHeatingState":"off"}]}}},"fuelStatus":{"rangeStatus":{"value":{"carCapturedTimestamp":"2024-12-12T19:58:19Z","carType":"hybrid","primaryEngine":{"type":"gasoline","currentSOC_pct":81,"remainingRange_km":510,"currentFuelLevel_pct":81},"secondaryEngine":{"type":"electric","currentSOC_pct":97,"remainingRange_km":32},"totalRange_km":542}}},"measurements":{"rangeStatus":{"value":{"carCapturedTimestamp":"2024-12-12T19:58:19Z","electricRange":32,"gasolineRange":510,"totalRange_km":542}},"odometerStatus":{"value":{"carCapturedTimestamp":"2024-12-12T19:41:53Z","odometer":51914}},"fuelLevelStatus":{"value":{"carCapturedTimestamp":"2024-12-12T19:58:19Z","currentFuelLevel_pct":81,"currentSOC_pct":97,"primaryEngineType":"gasoline","secondaryEngineType":"electric","carType":"hybrid"}}}}
Soc:            97%
Capacity:       13.0kWh
Charge status:  A
Range:          32km
Odometer:       51914km
Finish time:    2024-12-12 19:58:00 +0000 UTC
Climate active: false
Limit Soc:      not available
OnIdentified:   Mode:now, MinCurrent:6, MaxCurrent:16