evcc-io/evcc #16097: VW Vehicle cannot be created

hradschek, 2024-09-14:
### Describe the bug

Since yesterday, my evcc won't control the charging of my ID.3 anymore, so that the homebattery was drained completly.

### Steps to reproduce

n/a

### Configuration details

```yaml
vehicles:
  - name: my_car
    type: template
    template: vw
    title: ID.3
    user: juliahess4@gmx.de
    password: Holadis!1
    vin: WVWZZZE1ZNP005339
    capacity: 58
    timeout: 10s # optional 
# site describes the EVU connection, PV and home battery

loadpoints:
  - title: Alte Garage - ID.3 # display name for UI
    charger: my_charger # charger
    vehicle: my_car
    mode: pv # default charge mode to apply when vehicle is disconnected; use "off" to disable by default if charger is publicly available
    # remaining settings are experts-only and best left at default values
    priority: 9 # relative priority for concurrent charging in PV mode with multiple loadpoints (higher values have higher priority)
    soc:
      # polling defines usage of the vehicle APIs
      # Modifying the default settings it NOT recommended. It MAY deplete your vehicle's battery
      # or lead to vehicle manufacturer banning you from API use. USE AT YOUR OWN RISK.
      poll:
        # poll mode defines under which condition the vehicle API is called:
        #   charging: update vehicle ONLY when charging (this is the recommended default)
        #   connected: update vehicle when connected (not only charging), interval defines how often
        #   always: always update vehicle regardless of connection state, interval defines how often (only supported for single vehicle)
        mode: charging
        # poll interval defines how often the vehicle API may be polled if NOT charging
        interval: 60m
      estimate: true # set false to disable interpolating between api updates (not recommended)
    enable: # pv mode enable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 3m # threshold must be exceeded for this long
      threshold: 0 # maximum import power (W)
    guardDuration: 5m # switch charger contactor not more often than this (default 5m)
```

### Log details

```text
[main  ] ERROR 2024/09/14 08:59:14 creating vehicle my_car failed: cannot create vehicle type 'template': cannot create vehicle type 'vw': cannot get vehicles: Get "https://emea.bff.cariad.digital/vehicle/v1/vehicles": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[lp-1  ] ERROR 2024/09/14 09:23:44 charger enable: invalid status: 403
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130,9

StefanSchoof, 2024-09-14:
If this is your real password, please change it everywhere where you use it. It is now burned.

My id.3 is working now. Try to log into the my VW web page (not the app) and ensure that there no AGB ore else.

hradschek, 2024-09-14:
Error is caused by EASEE outage

closed 2024-09-14 by hradschek