evcc-io/evcc #16105: KEBA / Charge status:  B, jedoch kein Fahrzeug angeschlossen

TobiasHuber1980, 2024-09-14:
### Describe the bug

Seit 0.130.10 (Ref: https://github.com/evcc-io/evcc/pull/16003) wird Charge status: B gesetzt, obwohl kein Auto angeschlossen ist. (komischerweise nur an zwei (LP-1 und LP-2) von vier Wallboxen)
Ich habe dann die Wallboxen (LP-3 und LP-4) an welchen kein Auto angeschlossen ist remote neu gestartet, und genau in dem Moment (start reboot) steht im Log: 
```
[lp-3 ] DEBUG 2024/09/14 16:46:06 charge total import: 589.699kWh
[lp-3 ] DEBUG 2024/09/14 16:46:06 charger status: B
[lp-3 ] DEBUG 2024/09/14 16:46:06 charge start energy: 589.699kWh
[lp-3 ] INFO 2024/09/14 16:46:06 car connected
[lp-3 ] DEBUG 2024/09/14 16:46:06 vehicle api refresh
[lp-3 ] DEBUG 2024/09/14 16:46:06 pv timer elapse
[lp-3 ] DEBUG 2024/09/14 16:46:06 pv timer inactive
[lp-3 ] DEBUG 2024/09/14 16:46:06 charge total import: 589.699kWh
```

### Steps to reproduce

1. evcc starten
2. an LP-1 und LP-2 werden Autos erkannt, obwohl keine angeschlossen sind.
[image]

Info: 
LP1 = Keba Master; LP2=Keba Slave
LP3 = Keba Master; LP4 = Keba Slave

### Configuration details

```yaml
[main  ] INFO 2024/09/14 16:36:38 evcc 0.130.10
[main  ] INFO 2024/09/14 16:36:38 using config file: /etc/evcc.yaml
[db    ] INFO 2024/09/14 16:36:38 using sqlite database: /home/pi/.evcc/evcc.db

Fehlermeldung:

    failed configuring loadpoints: failed configuring loadpoint: not found: loadpoints

Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug

# unique installation id
plant: *****

#interval: 30s # control cycle interval

sponsortoken: *****

##################################

site:
  - title: name
    meters:
      grid: grid
        
##################################     

meters:
  - name: grid
    type: template
    template: plexlog
    usage: grid
    host: 192.168.1.26
    port: 503

[... 412 lines cut ...]
[keba  ] TRACE 2024/09/14 16:30:31 modbus: send 00 0b 00 00 00 06 ff 03 03 e8 00 02
[keba  ] TRACE 2024/09/14 16:30:31 modbus: recv 00 0b 00 00 00 07 ff 03 04 00 00 00 01
[keba  ] TRACE 2024/09/14 16:30:31 modbus: send 00 0c 00 00 00 06 ff 03 05 dc 00 02
[keba  ] TRACE 2024/09/14 16:30:31 modbus: recv 00 0c 00 00 00 07 ff 03 04 00 00 00 00
Power:          0W
Energy:         589.7kWh
Current L1..L3: 0A 0A 0A
Charge status:  A
Status reason:  unknown
Enabled:        false
Identifier:     <none>
Features:       [WelcomeCharge]
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.10

andig, 2024-09-14:
Fixed in master.

> Ich habe dann die Wallboxen (LP-3 und LP-4) an welchen kein Auto angeschlossen ist remote neu gestartet, und genau in dem Moment (start reboot) steht im Log:

Keine Ahnung, was das bedeuten soll :(

closed 2024-09-14 by andig

TobiasHuber1980, 2024-09-14:
> Fixed in master.
> 
> > Ich habe dann die Wallboxen (LP-3 und LP-4) an welchen kein Auto angeschlossen ist remote neu gestartet, und genau in dem Moment (start reboot) steht im Log:
> 
> Keine Ahnung, was das bedeuten soll :(

1) Es ist doch die Master 0.130.10 installiert.
2) Genau in dem Moment als ich die Wallbox (LP-3) neu gestartet habe, meldet evcc `[lp-3 ] INFO 2024/09/14 16:46:06 car connected` 
3) Ich vermute, dass durch https://github.com/evcc-io/evcc/pull/16003 irgendwie die Status A/B durcheinandergekommen sind.
4) Ich teste nachher nochmal alle Wallboxen stromlos zu machen.

horstvogel, 2024-09-15:
Moin, ich habe das Master auf Home Assistant 0.130.10 installiert, bei mir auch an den Keba´s die nicht belegt sind werden  "Autos" erkannt. Eine Alfen die auch noch im Netzwerk vorhanden ist, die verhält sich weiterhin normal. 
Zurück auf 0.130.9 und alles sieht wieder normal aus.

StefanSchoof, 2024-09-15:
Master ist der aktuelle Code der noch nicht released wurde. Man kann den nur über das nightly oder durch selbst bauen bekommen. (Und nightly für ha gibt es nicht)