evcc-io/evcc #16108: WebUI als iOS App -> Menü nicht auswählbar

sewixp, 2024-09-14:
### Describe the bug

[image]

Wenn man die Weboberfläche als App auf einem iPhone mit Dynamic Island (hier iPhone 15 Pro Max) aufruft, dann erscheint das Menü oben rechts in einem Bereich den man nicht antippen kann. Workaround ist das Telefon ins Querformat zu nehmen, dann zu klicken und dann wieder ins Hochformat zu wechseln. Eventuell kann das Menu einfach unterhalb der iOS UI Elemente platziert werden?

### Steps to reproduce

Webseite als App auf den Homescreen legen und starten

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.9

0x3d13f, 2024-09-14:
Lösung siehe ggf. https://github.com/evcc-io/evcc/discussions/13794

sewixp, 2024-09-14:
wie unangenehm.

Danke und Sorry fürs Ticket

closed 2024-09-14 by andig