evcc-io/evcc #16111: WebUI - SmartPricing UI - Clever Button disapeared - after adding grid meter to site

gramss, 2024-09-14:
### Describe the bug

I had (mis)configured my `evcc.yaml` before and did not add the meter to the site.
[image]

Now I added the tibber pulse grid meter to the "site" property in the `evcc.yaml`

[image]

Sadly, I do not have any PV on that site.. Can I get the "Clever" Button back? SmartPricing functionality still seems to be there over the charge bar (with newest config):
[image]

### Steps to reproduce

added these things to evcc.yaml
1. tibber pulse link in site (before it was just living in the evcc but not connected and therefore not displayed in the WebUI)
2. added co2 grünstrom plugin (unrelated but still mentioned)
3. added currency: EUR (unrelated but still mentioned)

### Configuration details

```yaml
OLD:

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 8a5591514266a12f732e2b3e30d90f130e204e0832f09b360764a3cb53faffe5

interval: 30s # control cycle interval

chargers:
- type: template
  template: pmcc 
  ski: f...fae  
  name: wallbox1

vehicles:
- type: template
  template: iso15118 
  title: X Car  
  capacity: 94  
  name: ev1
  identifiers:
      - XXXXXXXXXX # über `evcc charger` ermittelt

loadpoints:
- title: Garage
  charger: wallbox1
  mode: off

site:
  title: X Home
  meters:

eebus:
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----
          
          -----END EC PRIVATE KEY-----
      public: |
          -----BEGIN CERTIFICATE-----
          
          -----END CERTIFICATE-----

tariffs:
  grid:
    type: template
    template: tibber
    token: XXXXXXXXXXX
    homeid: # Nur erforderlich, wenn du mehrere Häuser in deinem Tibber-Konto hast. (optional)
    charges: # Zusätzlicher fester Aufschlag pro kWh (z.B. 0.05 für 5 Cent) (optional)
    tax: # Zusätzlicher prozentualer Aufschlag (z.B. 0.2 für 20%) (optional)

meters:
  - name: pulse
    type: template
    template: tibber-pulse
    usage: grid
    token: xxxxxxxxxx
    homeid: # optional

New:

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 8a5591514266a12f732e2b3e30d90f130e204e0832f09b360764a3cb53faffe5

interval: 30s # control cycle interval

chargers:
- type: template
  template: pmcc 
  ski: f...fae  
  name: wallbox1

vehicles:
- type: template
  template: iso15118 
  title: X Car  
  capacity: 94  
  name: ev1
  identifiers:
      - XXXXXXXXXX # über `evcc charger` ermittelt

loadpoints:
- title: Garage
  charger: wallbox1
  mode: off

site:
  title: X Home
  meters:
    grid: pulse

eebus:
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----
          
          -----END EC PRIVATE KEY-----
      public: |
          -----BEGIN CERTIFICATE-----
          
          -----END CERTIFICATE-----

tariffs:
  currency: EUR 
  grid:
    type: template
    template: tibber
    token: XXXXXXXXXXX
    homeid: # Nur erforderlich, wenn du mehrere Häuser in deinem Tibber-Konto hast. (optional)
    charges: # Zusätzlicher fester Aufschlag pro kWh (z.B. 0.05 für 5 Cent) (optional)
    tax: # Zusätzlicher prozentualer Aufschlag (z.B. 0.2 für 20%) (optional)
  co2:
    # co2 tariff provides co2 intensity forecast and is for co2-optimized target charging if no variable grid tariff is specified
    type: grünstromindex # GrünStromIndex (Germany only)
    zip: XXXXXX

meters:
  - name: pulse
    type: template
    template: tibber-pulse
    usage: grid
    token: xxxxxxxxxx
    homeid: # optional
```

### Log details

```text
[main ] INFO 2024/09/14 20:00:10 evcc 0.130.10
[main ] INFO 2024/09/14 20:00:10 using config file: /etc/evcc.yaml
[db ] INFO 2024/09/14 20:00:10 using sqlite database: /root/.evcc/evcc.db
[main ] INFO 2024/09/14 20:00:10 listening at :7070
[site ] INFO 2024/09/14 20:00:15 site config:
[site ] INFO 2024/09/14 20:00:15 meters: grid ✓ pv ✗ battery ✗
[site ] INFO 2024/09/14 20:00:15 grid: power ✓ energy ✗ currents ✓
[site ] INFO 2024/09/14 20:00:15 vehicles:
[site ] INFO 2024/09/14 20:00:15 vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1 ] INFO 2024/09/14 20:00:15 loadpoint 1:
[lp-1 ] INFO 2024/09/14 20:00:15 mode: off
[lp-1 ] INFO 2024/09/14 20:00:15 charger: power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1 ] INFO 2024/09/14 20:00:15 meters: charge ✓
[lp-1 ] INFO 2024/09/14 20:00:15 charge: power ✓ energy ✗ currents ✓
[lp-1 ] DEBUG 2024/09/14 20:00:15 set smart cost limit: 0.2
[lp-1 ] DEBUG 2024/09/14 20:00:15 phase timer inactive
[lp-1 ] DEBUG 2024/09/14 20:00:15 pv timer inactive
[site ] DEBUG 2024/09/14 20:00:15 ----
[lp-1 ] DEBUG 2024/09/14 20:00:15 charge power: 0W
[lp-1 ] DEBUG 2024/09/14 20:00:15 charge currents: [0 0 0]A
[site ] DEBUG 2024/09/14 20:00:15 grid meter: 350W
[site ] DEBUG 2024/09/14 20:00:15 grid currents: [0 0 0]A
[site ] DEBUG 2024/09/14 20:00:15 pv power: 0W
[site ] DEBUG 2024/09/14 20:00:15 site power: 350W
[lp-1 ] DEBUG 2024/09/14 20:00:15 charger status: B
[lp-1 ] INFO 2024/09/14 20:00:15 car connected
[lp-1 ] DEBUG 2024/09/14 20:00:15 vehicle api refresh
[lp-1 ] DEBUG 2024/09/14 20:00:15 pv timer elapse
[lp-1 ] DEBUG 2024/09/14 20:00:15 pv timer inactive
[site ] DEBUG 2024/09/14 20:00:45 ----
[lp-1 ] DEBUG 2024/09/14 20:00:45 charge power: 0W
[lp-1 ] ERROR 2024/09/14 20:00:46 charge currents: not available
[site ] DEBUG 2024/09/14 20:00:46 grid meter: 352W
[site ] DEBUG 2024/09/14 20:00:46 grid currents: [0 0 0]A
[site ] DEBUG 2024/09/14 20:00:46 pv power: 0W
[site ] DEBUG 2024/09/14 20:00:46 site power: 352W
[lp-1 ] DEBUG 2024/09/14 20:00:46 charger status: B
[lp-1 ] INFO 2024/09/14 20:01:05 vehicle updated: unknown ->
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.8 -> updated to 0.130.10 in between configs

gramss, 2024-09-14:
might be related to https://github.com/evcc-io/evcc/issues/15056

I have no PV here. 
Seems like, if I want the `clever` button then I should remove any meters from my site.

naltatis, 2024-09-14:
Yes, in cases where we are sure no PV exists we show "off", "smart" and "fast". When a grid meter is present, we assume that a PV can exists and show the PV-related modes.

Technically "smart" and "pv" are identical. We only change the ui label. We know that this is not optimal. We'll find a proper solution once we get to https://github.com/evcc-io/evcc/discussions/3530

closed 2024-09-14 by naltatis