evcc-io/evcc #16131: No Smartcharching with Easee

Simon79-bit, 2024-09-15:
### Describe the bug

Es gibt einen Fehler, mit der easee Box. Pv Überschuss laden geht nicht als Fehler wird angezeigt no Smartcharching
Fehler 403

### Steps to reproduce

Ich kann den Fehler nicht reproduzieren, er trat mit einmal auf und ist nicht mehr abzuschalten

### Configuration details

```yaml
Bis vor einer Woche hat alles gut geklappt. Jetzt plötzlich nicht mehr.
```

### Log details

```text
lp-2  ] ERROR 2024/09/15 13:50:31 charger enable: invalid status: 403
[lp-2  ] ERROR 2024/09/15 13:51:27 charger enable: invalid status: 403
[lp-2  ] ERROR 2024/09/15 13:52:28 charger enable: invalid status: 403
[lp-2  ] ERROR 2024/09/15 13:53:37 charger enable: invalid status: 403
[lp-2  ] ERROR 2024/09/15 13:54:46 charger enable: invalid status: 403
[lp-2  ] ERROR 2024/09/15 13:55:45 charger enable: invalid status: 403
[lp-2  ] ERROR 2024/09/15 13:56:46 charger enable: invalid status: 403
[lp-2  ] ERROR 2024/09/15 13:57:53 charger enable: invalid status: 403
[lp-2  ] ERROR 2024/09/15 13:58:46 charger enable: invalid status: 403
[lp-2  ] ERROR 2024/09/15 13:59:49 charger enable: invalid status: 403
[lp-2  ] ERROR 2024/09/15 14:00:36 charger enable: invalid status: 403
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

130

mike63526, 2024-09-15:
Gleiches Problem bei mir ebenfalls, allerdings habe ich von easee in meinen Mails folgenden hinweis bekommen: 
[image]

Simon79-bit, 2024-09-15:
Hast du das von easee ausprobiert?

andig, 2024-09-15:
https://github.com/evcc-io/evcc/discussions/16079

closed 2024-09-15 by andig