evcc-io/evcc #16136: No surplus charging possible with easee home wallbox

no-banker, 2024-09-15:
### Describe the bug

Presumably since an update of the wallbox, excess charging with evcc is no longer possible. I have restored an older version of evcc 0.130.7 and update to 0.130.11 and the behaviour remains identical. Charging is always carried out with 7.2kWh, there is no adjustment of the charging current.

### Steps to reproduce

1. Update to latest version
2. No other steps 

### Configuration details

```yaml
nothing to to with evcc.yaml or?
```

### Log details

```text
/
Logs
Suchen

DEBUG
Alle Bereiche
[lp-2 ] DEBUG 2024/09/15 14:16:38 phase timer inactive
[lp-2 ] DEBUG 2024/09/15 14:16:38 pv timer inactive
[site ] WARN 2024/09/15 14:16:38 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site ] DEBUG 2024/09/15 14:16:38 ----
[easee ] WARN 2024/09/15 14:16:38 smart charging: invalid status: 403
[lp-1 ] DEBUG 2024/09/15 14:16:38 charge power: 7177W
[lp-1 ] DEBUG 2024/09/15 14:16:38 charge currents: [30.4 0.029 0.011]A
[lp-2 ] DEBUG 2024/09/15 14:16:39 charge power: 0W
[site ] DEBUG 2024/09/15 14:16:39 pv power: 5960W
[site ] DEBUG 2024/09/15 14:16:39 grid meter: 1402W
[site ] DEBUG 2024/09/15 14:16:39 grid powers: [5008 -1714 -1891]W
[site ] DEBUG 2024/09/15 14:16:39 grid currents: [21.7 -7.4 -8.2]A
[site ] DEBUG 2024/09/15 14:16:39 site power: 1302W
[lp-1 ] DEBUG 2024/09/15 14:16:40 vehicle odometer: 23097km
[lp-1 ] DEBUG 2024/09/15 14:16:40 charge total import: 5064.542kWh
[easee ] WARN 2024/09/15 14:16:40 smart charging: invalid status: 403
[lp-1 ] DEBUG 2024/09/15 14:16:40 charger status: C
[lp-1 ] INFO 2024/09/15 14:16:40 car connected
[lp-1 ] DEBUG 2024/09/15 14:16:40 pv timer elapse
[lp-1 ] DEBUG 2024/09/15 14:16:40 pv timer inactive
[lp-1 ] DEBUG 2024/09/15 14:16:40 charge total import: 5064.542kWh
[lp-1 ] INFO 2024/09/15 14:16:40 start charging ->
[lp-1 ] DEBUG 2024/09/15 14:16:40 wake-up timer: stop
[easee ] WARN 2024/09/15 14:16:40 smart charging: invalid status: 403
[lp-1 ] DEBUG 2024/09/15 14:16:40 vehicle status: C
[lp-1 ] DEBUG 2024/09/15 14:16:40 vehicle soc: 52%
[lp-1 ] DEBUG 2024/09/15 14:16:40 vehicle range: 177km
[lp-1 ] WARN 2024/09/15 14:16:40 charger logic error: disabled but charging
[lp-1 ] DEBUG 2024/09/15 14:16:41 charger enable
[lp-1 ] ERROR 2024/09/15 14:16:41 charger enable: invalid status: 403
[site ] DEBUG 2024/09/15 14:16:48 ----
[easee ] WARN 2024/09/15 14:16:48 smart charging: invalid status: 403
[lp-1 ] DEBUG 2024/09/15 14:16:48 charge power: 7177W
[lp-1 ] DEBUG 2024/09/15 14:16:48 charge currents: [30.4 0.029 0.011]A
[lp-2 ] DEBUG 2024/09/15 14:16:49 charge power: 0W
[site ] DEBUG 2024/09/15 14:16:49 pv power: 5891W
[site ] DEBUG 2024/09/15 14:16:49 grid meter: 1438W
[site ] DEBUG 2024/09/15 14:16:49 grid powers: [5036 -1676 -1921]W
[site ] DEBUG 2024/09/15 14:16:49 grid currents: [21.8 -7.3 -8.1]A
[site ] DEBUG 2024/09/15 14:16:50 site power: 1338W
[lp-2 ] DEBUG 2024/09/15 14:16:50 charge total import: 597.286kWh
[lp-2 ] DEBUG 2024/09/15 14:16:51 charger status: B
[lp-2 ] DEBUG 2024/09/15 14:16:51 charge start energy: 597.286kWh
[lp-2 ] INFO 2024/09/15 14:16:51 car connected
[lp-2 ] DEBUG 2024/09/15 14:16:51 vehicle api refresh
[lp-2 ] DEBUG 2024/09/15 14:16:51 pv timer elapse
[... 942 lines cut ...]
[site ] DEBUG 2024/09/15 14:26:39 grid powers: [5198 -1544 -1785]W
[site ] DEBUG 2024/09/15 14:26:39 grid currents: [22.9 -6.6 -7.1]A
[site ] DEBUG 2024/09/15 14:26:39 site power: 1768W
[lp-1 ] DEBUG 2024/09/15 14:26:39 detected active phases: 1p
[lp-1 ] DEBUG 2024/09/15 14:26:39 charge total import: 5066.543kWh
[easee ] WARN 2024/09/15 14:26:39 smart charging: invalid status: 403
[lp-1 ] DEBUG 2024/09/15 14:26:39 charger status: C
[lp-1 ] DEBUG 2024/09/15 14:26:39 soc estimated: 63.56% (vehicle: 52.00%)
[lp-1 ] DEBUG 2024/09/15 14:26:39 vehicle soc: 64%
[lp-1 ] DEBUG 2024/09/15 14:26:39 vehicle range: 178km
[lp-1 ] WARN 2024/09/15 14:26:39 charger logic error: disabled but charging
[lp-1 ] ERROR 2024/09/15 14:26:41 charger enable: invalid status: 403
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.11

andig, 2024-09-15:
Sufu benutzen, Easee API ist kaputt...

closed 2024-09-15 by andig

naltatis, 2024-09-15:
Siehe auch https://status.easee.com/incidents/z00bt9vnjykh

Sobald Easee das behoben hat sollte es wieder klappen.